raspberrypi/linux #6096: i2c controller timed out with 2024-03-15 bookworm-64bit

Tiger0215, 2024-04-07:
### Describe the bug

Device：Raspberry Pi 5 and Waveshare 2.8inch DSI LCD
dmesg errors with 2024-03-15 bookworm 64-bit
i2c_designware 1f00080000.i2c: controller timed out.

### Steps to reproduce the behaviour

When executing the reboot command, there is a probability that you will encounter an i2c timeout problem within about 20 times.

### Device (s)

Raspberry Pi 5

### System

pi@raspberrypi:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2024-03-15
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, f19ee211ddafcae300827f953d143de92a5c6624, stage4

### Logs

pi@raspberrypi:~ $ dmesg |grep i2c
[    0.846566] brcmstb-i2c 107d508200.i2c:  @97500hz registered in interrupt mode
[    0.846762] brcmstb-i2c 107d508280.i2c:  @97500hz registered in interrupt mode
[    1.923798] i2c_designware 1f00080000.i2c: controller timed out
[    2.947791] i2c_designware 1f00080000.i2c: controller timed out
[    3.971790] i2c_designware 1f00080000.i2c: controller timed out
[    4.995801] i2c_designware 1f00080000.i2c: controller timed out
[    6.019791] i2c_designware 1f00080000.i2c: controller timed out
[    7.043791] i2c_designware 1f00080000.i2c: controller timed out
[    8.358758] i2c_dev: i2c /dev entries driver
[   10.883795] i2c_designware 1f00080000.i2c: controller timed out
[   11.907794] i2c_designware 1f00080000.i2c: controller timed out
[   12.931792] i2c_designware 1f00080000.i2c: controller timed out
[   13.955809] i2c_designware 1f00080000.i2c: controller timed out

### Additional context

When i2c timeout problem occurs, the DSI LCD displays a black screen without displaying. From now on, no matter how many restarts are performed, the i2c timeout will always be encountered.

pelwell, 2024-04-07:
Please run `sudo rpi-update` and retest.

Tiger0215, 2024-04-09:
> Please run `sudo rpi-update` and retest.

Thank you, the issue looks to be resolved.

pi@raspberrypi:~ $ uname -a
Linux raspberrypi 6.6.25-v8-16k #1 SMP PREEMPT Mon Apr  8 17:20:29 CST 2024 aarch64 GNU/Linux

closed 2024-04-09 by Tiger0215