raspberrypi/linux #6097: Bluetooth - getting really tired of this...

seamusdemora, 2024-04-08:
### Describe the bug

This is about ongoing issues I have with Bluetooth on my `bullseye` system. Driving me crazy... it works for a while - and then it doesn't. Very frustrating...

Anyway - before I go to the trouble of filing another bug report, I need to ask if `bullseye` is being supported here. I was told last week that it wasn't. If not, guess there's not much point in filing a report. 

Please advise.

### Steps to reproduce the behaviour

use bluetooth

### Device (s)

Raspberry Pi 3 Mod. A+

### System

will comply after I get an answer to my question

JamesH65, 2024-04-08:
Our current release is Bookworm, the previous version, Bullseye (our legacy release) usually gets security updates only. Does the problem exhibit on Bookworm?

seamusdemora, 2024-04-10:
I would expect that it does; it has since at least buster - the first OS that I tried to use BT (I made notes). 

But I haven't actually attempted it yet as I've been too distracted trying to get it to work on bullseye.