raspberrypi/linux #6100: Kernel freeze en 6.6.17 with rpi2 rev B when using usb device

ipoupaille, 2024-04-10:
### Describe the bug

When using usb device 0658:0200  (zwave stick) with zwave-js-ui inside docker with kernel 6.6 (tried 6.6.17 and 6.6.25), kernel freeze and watchdog reboot device.

Work perfectly with kernel 6.1.77.

All others devices work (usb disk, arduino serial device, dvb-t usb device)

### Steps to reproduce the behaviour

Start zwave-js-ui container inside docker with cgroup rule for accessing device.
https://hub.docker.com/r/zwavejs/zwave-js-ui

### Device (s)

Raspberry Pi 2 Mod. B

### System

OS: Alpine 3.19
[    0.120479] raspberrypi-firmware soc:firmware: Attached to firmware from 2024-03-21T14:33:34, variant start
[    0.130506] raspberrypi-firmware soc:firmware: Firmware hash is c14213886ae1e31951022b74ce776ce7ee7023ee
Kernel 6.6.17 (same with 6.6.25). Work with 6.1.77

Usb configuration:
1 zwave stick on port 2
1 powered usb hub on port 3 containing a serial usb device and a dvd-t device
2 powered disks on port 5

lsusb -tv
/:  Bus 001.Port 001: Dev 001, Class=root_hub, Driver=dwc_otg/1p, 480M
    ID 1d6b:0002  
    |__ Port 001: Dev 002, If 0, Class=[unknown], Driver=hub/5p, 480M
        ID 0424:9514  
        |__ Port 001: Dev 003, If 0, Class=[unknown], Driver=smsc95xx, 480M
            ID 0424:ec00  
        |__ Port 002: Dev 009, If 0, Class=[unknown], Driver=cdc_acm, 12M
            ID 0658:0200  
        |__ Port 002: Dev 009, If 1, Class=[unknown], Driver=cdc_acm, 12M
            ID 0658:0200  
        |__ Port 003: Dev 004, If 0, Class=[unknown], Driver=hub/4p, 480M
            ID 214b:7250  
            |__ Port 001: Dev 006, If 0, Class=[unknown], Driver=hub/4p, 480M
                ID 214b:7250  
                |__ Port 001: Dev 007, If 0, Class=[unknown], Driver=cdc_acm, 12M
                    ID 2341:0042  
                |__ Port 001: Dev 007, If 1, Class=[unknown], Driver=cdc_acm, 12M
                    ID 2341:0042  
                |__ Port 002: Dev 008, If 0, Class=[unknown], Driver=em28xx, 480M
                    ID 2040:0265  
        |__ Port 005: Dev 005, If 0, Class=[unknown], Driver=usb-storage, 480M
            ID 067b:2773  

### Logs

[image]

b3nn0, 2024-04-30:
I have the exact same issue with a TTGO T-Motion board. It reports as

ipoupaille, 2024-05-01:
I tried it with 6.6.29. And there is no more freeze anymore.
But my dvb-t stick is not working with tvheadend. I got a poll timeout error. I have not had time yet to investigate.
I did not understand what are the consequences of using the dwc2 driver.