raspberrypi/linux #6102: [feat req] build igb, igc, ixgbe, i40e

delgh1, 2024-04-15:
### Describe the bug

Since CM4 and RPI 5 can have PCIe device attached, more and more people are using Pis as a router etc., but RPI OS doesn't ship with a kernel with those built. Also mentioned in #5797. I myself use a two port Intel I226-V PCIe card which needs igc. I suggest building igb, igc, ixgbe, i40e starting the next release, at least as a module, better as built-in (some people might use NFS root?).

### Steps to reproduce the behaviour

Official RPI OS image

### Device (s)

Raspberry Pi 5, Raspberry Pi CM4, Raspberry Pi CM4 Lite

### System

linux 6.6.20+rpt-rpi-2712

### Logs

`/boot/config-6.6.20+rpt-rpi-2712`

```
CONFIG_NET_VENDOR_AQUANTIA=y
# CONFIG_AQTION is not set
...
CONFIG_NET_VENDOR_INTEL=y
# CONFIG_E100 is not set
# CONFIG_E1000 is not set
# CONFIG_E1000E is not set
# CONFIG_IGB is not set
# CONFIG_IGBVF is not set
# CONFIG_IXGBE is not set
# CONFIG_IXGBEVF is not set
# CONFIG_I40E is not set
# CONFIG_I40EVF is not set
# CONFIG_ICE is not set
# CONFIG_FM10K is not set
# CONFIG_IGC is not set
...
CONFIG_R8169=m
```

pelwell, 2024-04-15:
The modules add up to ~1.2MB for arm and ~1.5MB for arm64. Any objections to adding these to the Pi 4 and Pi 5 kernel defconfigs, @popcornmix?

popcornmix, 2024-04-16:
I'm okay with it if you are happy.

pelwell, 2024-04-16:
Thanks. Resolved by https://github.com/raspberrypi/linux/commit/044dae4ac76ab3677977f432e78b29e423e8c54e.

delgh1, 2024-04-18:
Thanks! I'm gonna close this.

closed 2024-04-18 by delgh1