raspberrypi/linux #6104: Regression with Hifiberry DAC+ Pro since switch to params_width

HiassofT, 2024-04-16:
### Describe the bug

We got a report on our forums that the Hifiberry DAC+ Pro produced choppy audio in Kodi with our latest releases, running kernel 6.6.25. The symptoms looked a lot like the DAC isn't running at the correct rate and Kodi (which outputs 24bit audio) was constantly trying to adjust for it.

https://forum.libreelec.tv/thread/28386-rpi-4-choppy-audio-with-hifiberry-dac2-pro/?postID=190889#post190889

Glancing through the commit history I stumbled upon f0a501b81926 which switched the hifiberry_dacplus driver from `snd_pcm_format_physical_width` to `snd_pcm_format_width`.

With this switch reverted the user reported that kodi worked fine again.

It looks like this driver (and possible others, too), deliberately uses physical_width to ensure the bclk_ratio is either 16 or 32 so the clock rate can be cleanly derived from the oscillators by an integer divider.

### Steps to reproduce the behaviour

Play 24bit audio on a Hifiberry DAC+ Pro (`speaker-test -F S24_LE ...` should do the trick and likely produce slightly off-pitch audio)

### Device (s)

Raspberry Pi 4 Mod. B

### System

LibreELEC 12 beta 2, kernel 6.6.25

### Logs

https://paste.libreelec.tv/logical-cowbird.log

### Additional context

Using physical_width to get either 16 or 32 isn't quite correct as well, as it would fail in case of S24_3LE audio (which, fortunately, isn't being supported on RPis yet), and without additional comments is hard to distinguish from a typo. So it's be better to make the intention clear, eg by using a switch or sth like `width = width <=16 ? 16 : 32`

I suspect other cards may have regressed as well by this change, from a quick glance eg the Allo Boss DAC driver seems to use two fixed frequency oscillators as well and the driver looks like a copycat of the hifiberry one.

ping @pelwell @j-schambacher

j-schambacher, 2024-04-17:
@HiassofT I'll take a look within the next days. Thanks for you proposal. I'll test that, but will take some time.

pelwell, 2024-04-17:
I'm sure @HiassofT is correct. I'll create a PR with the proposed fix, i.e. using snd_pcm_format_width, then forcing either 16 or 32 depending.

j-schambacher, 2024-04-17:
Agreed! Thanks, Phil!

pelwell, 2024-04-17:
#6110 is the suggested fix.

closed 2024-04-18 by HiassofT