raspberrypi/linux #6106: high cpu with 2-CH RS485 HAT Interface-Card and pi5

iiicht, 2024-04-17:
### Describe the bug

I have problems with this card (2-CH RS485 HAT)
https://www.waveshare.com/wiki/2-CH_RS485_HAT

As soon as the module is loaded via /boot/config.txt (dtoverlay=sc16is752-spi1,int_pin=24), the top output shows (33.3 0.0 4:14.01 irq/172-spi1.0 ). So a permanent CPU utilization of over 20-30%.

This problem only exists with the pi5...I tried the same card on a pi4 and it worked perfectly...

The manufacturer has no sutiable solution for it yet.
 
It's also documented in the forum;
https://forums.raspberrypi.com/viewtopic.php?p=2200559&hilit=RS485+HAT#p2200559

### Steps to reproduce the behaviour

install the card on pi 5 and load the module width;
dtoverlay=sc16is752-spi1,int_pin=24

### Device (s)

Raspberry Pi 5

### System

12.5
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 2acf7afcba7d11500313a7b93bb55a2aae20b2d6, stage2
version a482d84b (release) (embedded)
Linux spRasp 6.6.26-v8-16k+ #1754 SMP PREEMPT Thu Apr 11 14:51:20 BST 2024 aarch64 GNU/Linux

### Logs

top:
549 root -51 0 0 0 0 D 22.3 0.0 5:07.35 irq/172-spi1.0

cat /sys/kernel/debug/gpio | grep PIN
gpio-401 (PIN3 )
gpio-402 (PIN5 )
gpio-403 (PIN7 )
gpio-404 (PIN29 )
gpio-405 (PIN31 )
gpio-406 (PIN26 )
gpio-407 (PIN24 )
gpio-408 (PIN21 )
gpio-409 (PIN19 )
gpio-410 (PIN23 )
gpio-411 (PIN32 )
gpio-412 (PIN33 )
gpio-413 (PIN8 )
gpio-414 (PIN10 )
gpio-415 (PIN36 )
gpio-416 (PIN11 )
gpio-417 (PIN12 |spi1 CS0 ) out lo ACTIVE LOW
gpio-418 (PIN35 )
gpio-419 (PIN38 )
gpio-420 (PIN40 )
gpio-421 (PIN15 )
gpio-422 (PIN16 )
gpio-423 (PIN18 )
gpio-424 (PIN22 )
gpio-425 (PIN37 )
gpio-426 (PIN13 )

pinctrl | grep PIN
2: no pu | -- // PIN3/GPIO2 = none
3: no pu | -- // PIN5/GPIO3 = none
4: no pu | -- // PIN7/GPIO4 = none
5: no pu | -- // PIN29/GPIO5 = none
6: no pu | -- // PIN31/GPIO6 = none
7: no pu | -- // PIN26/GPIO7 = none
8: no pu | -- // PIN24/GPIO8 = none
9: no pd | -- // PIN21/GPIO9 = none
10: no pd | -- // PIN19/GPIO10 = none
11: no pd | -- // PIN23/GPIO11 = none
12: no pd | -- // PIN32/GPIO12 = none
13: no pd | -- // PIN33/GPIO13 = none
14: a4 pn | hi // PIN8/GPIO14 = TXD0
15: a4 pu | hi // PIN10/GPIO15 = RXD0
16: no pd | -- // PIN36/GPIO16 = none
17: no pd | -- // PIN11/GPIO17 = none
18: op dl pd | lo // PIN12/GPIO18 = output
19: a0 pd | lo // PIN35/GPIO19 = SPI1_MISO
20: a0 pd | lo // PIN38/GPIO20 = SPI1_MOSI
21: a0 pd | lo // PIN40/GPIO21 = SPI1_SCLK
22: no pd | -- // PIN15/GPIO22 = none
23: no pd | -- // PIN16/GPIO23 = none
24: no pd | -- // PIN18/GPIO24 = none
25: no pd | -- // PIN22/GPIO25 = none
26: no pd | -- // PIN37/GPIO26 = none
27: no pd | -- // PIN13/GPIO27 = none

vmstat 2
procs -----------memory---------- ---swap-- -----io---- -system-- ------cpu-----
r b swpd free buff cache si so bi bo in cs us sy id wa st
2 0 0 3269664 91168 638592 0 0 0 5 49251 42155 0 6 93 0 0
2 0 0 3269664 91168 638592 0 0 0 0 161940 138603 0 7 93 0 0

2 0 0 3269664 91168 638592 0 0 0 4 162138 138771 0 7 93 0 0
2 0 0 3269664 91168 638592 0 0 0 0 162256 138884 0 6 94 0 0
2 0 0 3269664 91168 638592 0 0 0 2 162142 138778 0 7 93 0 0
2 0 0 3269664 91168 638592 0 0 0 0 162047 138699 0 6 94 0 0
2 0 0 3269664 91168 638592 0 0 0 0 162113 138742 0 7 93 0 0
2 0 0 3269664 91168 638592 0 0 0 8 162076 138718 1 7 93 0 0
2 0 0 3269664 91168 638592 0 0 0 0 162207 138828 0 6 94 0 0
3 0 0 3269664 91168 638592 0 0 0 414 162222 138885 0 6 87 7 0
2 0 0 3269664 91168 638592 0 0 0 0 162084 138736 0 6 94 0 0
1 0 0 3269664 91168 638592 0 0 0 0 162222 138855 0 6 94 0 0
2 0 0 3269664 91168 638592 0 0 0 8 162212 138850 0 6 94 0 0
1 0 0 3269664 91168 638592 0 0 0 0 162079 138723 0 6 94 0 0
2 0 0 3269664 91168 638592 0 0 0 0 162036 138692 0 6 94 0 0
2 0 0 3269664 91168 638592 0 0 0 0 162014 138670 0 7 93 0 0
2 0 0 3269664 91168 638592 0 0 0 0 162373 138968 0 6 94 0 0
procs -----------memory---------- ---swap-- -----io---- -system-- ------cpu-----
r b swpd free buff cache si so bi bo in cs us sy id wa st
3 0 0 3269664 91168 638592 0 0 0 0 162208 138842 0 6 94 0 0
1 0 0 3269664 91168 638592 0 0 0 0 162140 138771 0 6 94 0 0
1 0 0 3269664 91168 638592 0 0 0 0 162118 138759 0 6 94 0 0
2 0 0 3269664 91168 638592 0 0 0 0 162142 138789 0 6 94 0 0
2 0 0 3269664 91168 638592 0 0 0 0 162131 138778 0 6 94 0 0

3 0 0 3269664 91168 638592 0 0 0 0 162116 138753 0 6 94 0 0
2 0 0 3269664 91168 638592 0 0 0 0 162178 138824 0 6 94 0 0
2 0 0 3269664 91168 638592 0 0 0 0 162084 138729 0 6 94 0 0
3 0 0 3269664 91168 638592 0 0 0 14 162116 138745 0 7 93 0 0

 cat /proc/interrupts | grep spi
119:          0          0          0          0  rp1_irq_chip  19 Level     1f00050000.spi
120:   69935610          0          0          0  rp1_irq_chip  20 Level     1f00054000.spi
163:          0          0          0          0     GICv2 150 Level     107d004000.spi
186:    8741947          0          0          0  pinctrl-rp1  24 Level     spi1.0

pelwell, 2024-04-17:
69 million is a lot of interrupts for an idle UART - something is not working as it should.

Judging by the names assigned to the GPIOs ("PIN3", etc.), this isn't a fully up-to-date image - it should just be called `GPIO2`. I'd like you to run `sudo BRANCH=stable rpi-update` to ensure everything is in a known state.

I'd also like you to run `pinctrl 24 gp`. What does `pinctrl 24` report now? Has that made any difference to the interrupt load?

pelwell, 2024-04-18:
The problem is as I thought - the sc16is75x overlays don't make the interrupt pins into GPIO inputs. We get away with this on the older Pis because an input GPIO is the default state, but on Pi 5 the default is high-impedance/unconnected/none. Pull request #6111 adds the necessary Device Tree declarations to the affected overlays so that the interrupt works as expected. In about 50 minutes time you should be able to run `sudo rpi-update pulls/6106` to install a new kernel with the updated overlays. Be careful to back-up any important data - this is a beta kernel.

pelwell, 2024-04-18:
[ Edited to correct the install instructions, which should be `sudo rpi-update pulls/6111` ]

iiicht, 2024-04-18:
I tried still with;
dtoverlay=sc16is752-spi1,int_pin=24

pelwell, 2024-04-18:
Oh dear - the spi1 variant differs from the spi0 version in structure, and in making the changes (which I tested on the I2C version of the HAT) I ended up adding something in two different places. It's unfortunate for you that you need the _bad_ version, but I'm happy to have caught the problem now. Of course, I should have waited for your reply before merging.

#6114 is a respin with the fix, but to save you waiting I've attached the updated version.
[sc16is752-spi1.zip](https://github.com/raspberrypi/linux/files/15027674/sc16is752-spi1.zip)

iiicht, 2024-04-18:
hi, thanks for the fix.

now it`s run very well. 

can you estimate when the next stable kernel release with the updated version will appear?

thanks!

pelwell, 2024-04-18:
It shouldn't be long this time - not more than a week or two.

iiicht, 2024-04-18:
OK, thanks!