raspberrypi/linux #6115: Flickering in hyperpixel display

steveiliop56, 2024-04-19:
### Describe the bug

Hello,

I recently got a hyperpixel display (the rectangle one) and I noticed that if I look really close or be in icons that contain a lot of gray like the file explorer in dark mode I can see a flickering issue. I have reported it to the pimoroni GitHub too but since the driver is here I thought it would be a good idea to report it here too.

> Note: It's definitely not a hardware issue since I thought it was and got a replacement from the store and still the same issue.

### Steps to reproduce the behaviour

1. Connect the hyperpixel to the pi
2. Add the dtoverlay to `/boot/firmware/config.txt`
3. Boot the pi
4. Switch to dark mode
5. Open Firefox/File explorer and notice the issue

### Device (s)

Raspberry Pi 5

### System

Os version:

```
Raspberry Pi reference 2024-03-12
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 9a13a81115a8915885e2cc40fed2092e28894d41, stage4
```

Firmware version:

```
2024/04/17 13:54:23
Copyright (c) 2012 Broadcom
version 0b7b6f28 (release) (embedded)
```

Kernel version:

```
6.6.20+rpt-rpi-2712
```

6by9, 2024-04-19:
You say the "hyperpixel display (the rectangle one)". Do you mean the obsolete Hyperpixel (https://github.com/pimoroni/hyperpixel), or [HyperPixel4](https://shop.pimoroni.com/products/hyperpixel-4?variant=12569485443155)? Sorry to be pedantic, but details matter.

You've reported it to Pimoroni? I don't see anything on https://github.com/pimoroni/hyperpixel4/issues

I've just tested a HyperPixel4 (rectangular 480x800) on both a Pi4 and Pi5 and don't see any issues.

I'm afraid this is a 3rd party product, so whilst the driver is part of this repo (indeed we worked with Pimoroni to support the display) and we will offer some support, there is little we can do.

I suspect it will be backlight flicker, which possibly means a power supply noise related issue. There is nothing from the Pi side that it can do, as we only have on/off control for the backlight.

steveiliop56, 2024-04-19:
I have the hyperpixel 4 inch but it has the old special adaptor for i2c instead of the newer Dupont wires. Maybe it's not backlight and it's some kind of interference? I use the official 27w power supply.

steveiliop56, 2024-04-19:
The only difference my screen has is this adaptor: 
[image]

Seems weird since another user in the hyperpixel repo reported the same issue too https://github.com/pimoroni/hyperpixel4/issues/177

steveiliop56, 2024-05-02:
So update, apparently I got the latest version of the screen. I suspect it may be a software issue. I am in touch with the pimoroni support team and we are trying to figure out a solution.