raspberrypi/linux #6129: Compilation errors ("Warning (interrupt_provider)")

by, 2024-04-26:
### Describe the bug

When compiling the kernel, various warnings (e.g., interrupt_provider) pop up.

### Steps to reproduce the behaviour

Comile the kernel as per instructions

### Device (s)

Raspberry Pi 5

### System

Raspberry Pi reference 2023-10-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 962bf483c8f326405794827cce8c0313fd5880a8, stage4

2024/04/20 11:53:30 
Copyright (c) 2012 Broadcom
version d1744d21 (release) (embedded)

Linux (...) 6.6.28+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.28-1+rpt1 (2024-04-22) aarch64 GNU/Linux

### Logs

  DTCO    arch/arm64/boot/dts/overlays/allo-digione.dtbo
./scripts/dtc/include-prefixes/arm/broadcom/bcm2712.dtsi:626.26-637.5: Warning (interrupt_provider): /soc/gpio@7d517c00: '#interrupt-cells' found, but node is not an interrupt provider
  also defined at ./scripts/dtc/include-prefixes/arm/broadcom/bcm2712-rpi-5-b.dts:524.10-529.3
  also defined at ./scripts/dtc/include-prefixes/arm/broadcom/bcm2712-rpi-5-b.dts:641.10-677.3
arch/arm64/boot/dts/broadcom/bcm2712-rpi-5-b.dtb: Warning (interrupt_map): Failed prerequisite 'interrupt_provider'
  DTCO    arch/arm64/boot/dts/overlays/allo-katana-dac-audio.dtbo
arch/arm64/boot/dts/broadcom/../../../../arm/boot/dts/broadcom/bcm2712.dtsi:626.26-637.5: Warning (interrupt_provider): /soc/gpio@7d517c00: '#interrupt-cells' found, but node is not an interrupt provider
  also defined at arch/arm64/boot/dts/broadcom/../../../../arm/boot/dts/broadcom/bcm2712-rpi-5-b.dts:524.10-529.3
  also defined at arch/arm64/boot/dts/broadcom/../../../../arm/boot/dts/broadcom/bcm2712-rpi-5-b.dts:641.10-677.3
  also defined at arch/arm64/boot/dts/broadcom/../../../../arm/boot/dts/broadcom/bcm2712d0-rpi-5-b.dts:46.10-77.3
arch/arm64/boot/dts/broadcom/bcm2712d0-rpi-5-b.dtb: Warning (interrupt_map): Failed prerequisite 'interrupt_provider'
  DTCO    arch/arm64/boot/dts/overlays/allo-piano-dac-pcm512x-audio.dtbo
  DTCO    arch/arm64/boot/dts/overlays/allo-piano-dac-plus-pcm512x-audio.dtbo
  CC      init/main.o
  LDS     scripts/module.lds
  DTC     arch/arm64/boot/dts/broadcom/bcm2712-rpi-cm5-cm5io.dtb
  DTCO    arch/arm64/boot/dts/overlays/anyspi.dtbo
./scripts/dtc/include-prefixes/arm/broadcom/bcm2712.dtsi:626.26-637.5: Warning (interrupt_provider): /soc/gpio@7d517c00: '#interrupt-cells' found, but node is not an interrupt provider
  also defined at ./scripts/dtc/include-prefixes/arm/broadcom/bcm2712-rpi-cm5.dtsi:523.10-528.3
  also defined at ./scripts/dtc/include-prefixes/arm/broadcom/bcm2712-rpi-cm5.dtsi:640.10-692.3
arch/arm64/boot/dts/broadcom/bcm2712-rpi-cm5-cm5io.dtb: Warning (interrupt_map): Failed prerequisite 'interrupt_provider'
  HOSTCC  usr/gen_init_cpio
  DTCO    arch/arm64/boot/dts/overlays/apds9960.dtbo
  DTC     arch/arm64/boot/dts/broadcom/bcm2712-rpi-cm5-cm4io.dtb
  DTCO    arch/arm64/boot/dts/overlays/applepi-dac.dtbo
  DTCO    arch/arm64/boot/dts/overlays/arducam-64mp.dtbo
./scripts/dtc/include-prefixes/arm/broadcom/bcm2712.dtsi:626.26-637.5: Warning (interrupt_provider): /soc/gpio@7d517c00: '#interrupt-cells' found, but node is not an interrupt provider
  also defined at ./scripts/dtc/include-prefixes/arm/broadcom/bcm2712-rpi-cm5.dtsi:523.10-528.3
  also defined at ./scripts/dtc/include-prefixes/arm/broadcom/bcm2712-rpi-cm5.dtsi:640.10-692.3
arch/arm64/boot/dts/broadcom/bcm2712-rpi-cm5-cm4io.dtb: Warning (interrupt_map): Failed prerequisite 'interrupt_provider'

pelwell, 2024-04-26:
> When compiling the kernel, various warnings (e.g., interrupt_provider) pop up.

That looks like one warning, multiple times. Would you agree?

pelwell, 2024-04-26:
I get no such warnings. Which branch are you building, and what is your top-of-tree commit?

by, 2024-04-26:
P.S.: It seems to have appeared at some time during the last <20 days...

by, 2024-04-26:
Dug a bit deeper and it is likely caused by one of the settings to prep for RT kernel patch (RT kernel patch itself is not(!) installed):
-LEDS_TRIGGER_CPU y
 LOCALVERSION "-v8-16k" -> "-v8-16k-NTP"
 PPS_CLIENT_GPIO m -> y
+HZ_1000 y
+HZ_PERIODIC y
+NTP_PPS y
+RTC_INTF_DEV_UIE_EMUL y
+VIRT_CPU_ACCOUNTING_GEN y

by, 2024-04-26:
My bad: probably the conf went bad, so I dod a make menuconfig from the beginning and everything runs well, now!

closed 2024-04-26 by by