raspberrypi/linux #6130: WPA3 broken on Pi 3 with 6.6 (and works with 6.1)

spockfish, 2024-04-29:
### Describe the bug

When I run a 6.1 kernel on a Pi 3, using IWD, WPA3 works as expected. However, simply switching to the 6.6 kernel breaks this: the interface does not come up.

### Steps to reproduce the behaviour

Run a 6.6 kernel, on a Pi 3, accessing a WPA 3 network.

### Device (s)

Raspberry Pi 3 Mod. B

### System

custom built OS (buildroot), with latest 6.1 or 6.6 kernel, IWD for wireless interface mgt.

### Additional context

There's another strange thing going on: I'm using the 'rpi-firmware-nonfree' release (https://github.com/RPi-Distro/firmware-nonfree), but the latest release does not support SAE offload, which is required for WPA3 to function.

So, the latest firmware reports (iw phy) the following:

```
Supported extended features:
		* [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
		* [ 4WAY_HANDSHAKE_STA_PSK ]: 4-way handshake with PSK in station mode
		* [ 4WAY_HANDSHAKE_STA_1X ]: 4-way handshake with 802.1X in station mode
```

And thus WPA3 not functioning, where switching back to the upstream firmware (https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/) reports this:

```
Supported extended features:
		* [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
		* [ 4WAY_HANDSHAKE_STA_PSK ]: 4-way handshake with PSK in station mode
		* [ 4WAY_HANDSHAKE_STA_1X ]: 4-way handshake with 802.1X in station mode
		* [ SAE_OFFLOAD ]: SAE offload support
```

And thus results in a working WPA3 connection, if using 6.1.

peterharperuk, 2024-04-29:
My understanding is you have to use the upstream firmware if you want WPA3 support. Is it just that buildroot is using the wrong version?

spockfish, 2024-04-29:
>  you have to use the upstream firmware if you want WPA3 support

That's what said above ;-) Still does fix the issue that this only works for 6.1, and not for 6.6.

pelwell, 2024-04-29:
The upstream firmware uses the SAE feature, so there is no need to use iwd (in fact it doesn't work) - continue to use wpa_supplicant as before.

spockfish, 2024-04-29:
`so there is no need to use iwd (in fact it doesn't work) - continue to use wpa_supplicant as before.`

Well, it's not about 'need'. I just happen to 'like' IWD, in favour of wpa_supplicant. I've been using it on various Pi's for more than a year now. 

Could you elaborate a bit on the "it doesn't work" part?

spockfish, 2024-04-29:
Hmmm.... I think I know why. IWD does not support CMD_EXTERNAL_AUTH

pelwell, 2024-04-29:
Yes - that's it.

spockfish, 2024-04-29:
> The upstream firmware uses the SAE feature, so there is no need to use iwd (in fact it doesn't work) - continue to use wpa_supplicant as before.

It's still not clear to me why this should be a difference between 6.1 and 6.6. 
Again, with 6.1 I got this working, with 6.6 not.

spockfish, 2024-04-30:
To add to this: the same goes for the Pi 4.

With 6.1 WPA3 is working (upstream firmware), but replacing that with the latest 6.6 (and nothing else) breaks it.

spockfish, 2024-04-30:
Maybe this is bigger than just me: https://bugzilla.redhat.com/show_bug.cgi?id=2275835

EdiDD, 2024-11-02:
@spockfish how did you switch to upstream firmware?. Recompile from scratch or copy pre-compiled binaries to /boot partition ?

spockfish, 2024-11-02:
Hi meant the firmware as packaged in 'linux-firmware' (https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/)

EdiDD, 2024-11-04:
> I meant the firmware as packaged in 'linux-firmware' (https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/)

Did you manually extract this archive to `/lib/firmware` ?

spockfish, 2024-11-05:
> Did you manually extract this archive to `/lib/firmware` ?

I build a custom Linux, and there's a little bit more to it then just copying it over. Most distro's provide this fairly up-to-date.

gearhead, 2024-12-29:
I recently worked with the brcmfmac and the iwd devs and have WPA3 negotiation working with the latest firmware from infineon plus a one line patch to the brcm80211 kernel driver. 
The firmware is here: 
[https://community.infineon.com/t5/AIROC-Wi-Fi-and-Wi-Fi-Bluetooth/Cypress-Linux-WiFi-Driver-Release-FMAC-2024-11-21/td-p/902888](url) 
The specific firmware I tested is attached. There are other firmwares in the package from infineon. The only one I am able to check is the fmac43455 that is used on the RPi3b+,4B,5B. Upon boot, it identifies as:
`Firmware: BCM4345/6 wl0: Oct 28 2024 23:27:00 version 7.45.286 (be70ab3 CY) FWID 01-95efe7fa`

There is still no up to date firmware with the external_auth fixes for the wifi chip on the Zero series and the 3b. The one on the latest Bookworm images appears to be the latest from Infineon:
`Firmware: BCM43430/1 wl0: Jun 14 2023 07:27:45 version 7.45.96.s1 (gf031a129) FWID 01-70bd2af7 es7`

The patch for the kernel driver is a one line patch to the brcm80211 driver developed by Arend Van Spriel and is found in this chain:
[https://lore.kernel.org/linux-wireless/CAG17S_OwtNxetb7XzsxkZcygH_CWkZx15evQZkThb3WjqpiKTQ@mail.gmail.com/](url) 
It was originally a patch for the upstream kernel. I adjusted it to the 6.6.y kernel tree and attach it here. 

I have tried this firmware/kernel patch with wpa_supplicant (that is currently in the Bookwork image/update) on the RPi and it will connect to a WPA3 AP every time and collect a DHCP address and pass traffic.

The current git of iwd works with this kernel patch/firmware most of the time, but sometimes will not grab a DHCP address. A re-try to connect usually will grab a DHCP address. I am certain that this minor bug in iwd will be fixed shortly by the iwd team. The version that has the fix for this firmware/driver that I have tested is: iwd 3.3.r0.g4b535cee.

Please add this to the latest Pi installation and kernel to facilitate WPA3 capability for the 43455 chip. 

[brcm80211_wpa3.zip](https://github.com/user-attachments/files/18270706/brcm80211_wpa3.zip)
[cyfmac43455.zip](https://github.com/user-attachments/files/18270722/cyfmac43455.zip)

Gearhead

pelwell, 2024-12-29:
[I've fixed your URLs]

Does the patch depend on the updated firmware in order to work? If so, what feature or fix is included that makes the difference?

gearhead, 2024-12-29:
> [I've fixed your URLs]
> 
> Does the patch depend on the updated firmware in order to work? If so, what feature or fix is included that makes the difference?

Thanks!
1) the current shipping firmware in Bookworm for the RPis does not work for wpa3 with this patch or without. The driver advertises external_auth for SAE/WPA3 (which is the current direction that Infineon is taking with the fmac cards, apparently) but it does not function. IMO, it is a flawed firmware, somehow. I have never been able to get it to connect to wpa3. It just errors out. It is a kernel message mis-match or something. Neither iwd nor wpa_supplicant will connect with this firmware. Works for wpa2, but not wpa3. To make it work with wpa3, you need both the patch and the firmware. 
2) To get the firmware to work, the driver requires the one line edit to the brcm80211 driver to function. Arend made the patch for upstream kernel which has a refactored brcm80211 tree. I revised it for the simpler tree we have in 6.6.y. If I understand it correctly, it defines a new variable which did not exist in the driver which allows it to communicate properly with the firmware. 
3) I have tried the patched kernel with the older firmware and it does not work with WPA3. 
4) I need to do more testing, but think I saw that, with this kernel patch and the older firmware, that it would not even connect to wpa2 nor wpa3, so both are needed to get it to work.
There is a lot more info in that thread, I am not a programmer, so do not follow all of it.

Apparently, the earlier versions of the firmware had internal wpa3/sae authentication which worked for some users. I never got it to work with internal authentication. I have never been able to connect to wpa3/sae until now with my Pis. 

Now if we can get the same structure set up for the 43430 chip on the Zero2W, this should work there as well. 

also, it would be awesome if nlmon was built as a module in the basic def_configs. 
`CONFIG_NLMON=m`
As it is, to diagnose anything with wlan, you kind of need the nlmon module and to get it you have to build a custom kernel. It does not seem as a huge overhead to have it built as a module that can be dynamically loaded. Is this possible?

gearhead, 2024-12-30:
I just tried this build on a PiZero2W with the 43430 chip. It will connect to a wpa2 network, but not to a wpa3 network (expected behavior).

This build (kernel patch plus 43455 firmware) works identically as the current published Bookworm image with wpa2 APs, but allows Pis with the 43455 chip to connect to wpa3 APs, best I can tell. If ever Infineon updates the 43430 firmware to match the 43455, it 'should' work with it as well.

gearhead, 2025-01-08:
> Are there instructions somewhere on how to apply these patches?
Sure are it is basically this: [RPi Kernel Instructions](https://www.raspberrypi.com/documentation/computers/linux_kernel.html)
but before the 'Build' instruction, you need to apply the patch:
`patch -p1 < brcm80211_wpa3.patch`
then follow the instructions to build and install the patched kernel.

I have been doing more testing and it is iffy, still, with iwd. It works sometimes, but other times, not. With the current shipping version of wpa_supplicant it seems to work every time with NetworkManager or Connman.

gearhead, 2025-01-20:
@pelwell I have been doing quite a bit of testing with this patch+firmware on the Pis I have. It works consistently with the 43455 chip (RPi 3b+,4,5). If I read all this correctly, the previous firmware used an internal firmware command to auth SAE and the later versions use External_Auth (user space?)

```
[    3.911653] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    3.912133] usbcore: registered new interface driver brcmfmac
[    4.205973] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
[    4.207784] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Oct 28 2024 23:27:00 version 7.45.286 (be70ab3 CY) FWID 01-95efe7fa
```

It does not work on the 3b/ZeroW/Zero2W with the 43430 chip as that firmware is not yet patched from cypress to work with SAE. The version of this firmware is from 2022 (and though it says is supports SAE), it does not work with SAE. 
```
[    6.613452] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[    6.615557] usbcore: registered new interface driver brcmfmac
[    6.950102] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
[    6.950986] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Aug 16 2022 03:05:14 version 7.45.98.125 (5b7978c CY) FWID 01-f420b81d
```
There are a lot more brcmfmac firmwares in the package from cypress, but I have no ability to test them.

This setup works with connman and NetworkManager managing the network with wpa_supplicant. Very consistent and reliable. These are the apt, up-to-date versions I have tested of connman, wpa_supplicant and NetworkManager.  
```
connman/stable,now 1.41-3
network-manager/stable 1.42.4-1
wpasupplicant/stable,stable-security 2:2.10-12+deb12u2
```

The current RPiOS version of iwd (2.3) does not work with the brcmfmac card with External_Auth correctly. To work with this you need version 3.3 or later. When version 3.3 is installed, it works consistently with NetworkManager. I can reliably connect and disconnect and it always gets a DHCP Address and I can pass traffic. 
For some reason, iwd with connman is unreliable. I am unsure as to why. It either errors out with 'invalid key' or 'failed connection'. The error in the journal when it fails is pretty coinsistently:
`kernel: ieee80211 phy0: brcmf_cfg80211_mgmt_tx: TX Auth frame operation is failed: status=10)`
I only see this =error with connman/iwd. I never see this error with NetworkManager with either iwd not wpa_supplicant. I also never see this error with connman/wpa_supplicant.

Can this be added to the kernel for RPi? Is there more testing needed? The patch and firmware do not seem to have any negative effects as in I can still connect to wpa2 networks w/o issue.

gearhead, 2025-02-02:
@pelwell 
iwd now has a patch to get iwd to work reliably with WPA3/SAE and the brcmfmac chip:
[[RFC] netdev: avoid PMKSA for fullmac drivers](https://lore.kernel.org/iwd/CAG17S_PQni37Qgbsrqv9FzspCs8D=CVpn+GU3x=iYpJXpkCcFw@mail.gmail.com/T/#m928c551c004c41a033e909604c8f4dae12164a6d)
This patch to iwd 3.3 basically performs the same thing as putting the following in the /etc/iwd/main.conf:
```
[General]
DisablePMKSA=true
```
With this, iwd is now as reliable as wpa_supplicant in connecting to wpa3/SAE APs
I am pretty sure James will get this added to the iwd code base shortly. 

Is there any way of getting this firmware/kernel patch into the RPiOS default kernel? How about also adding iwmon as a module to the kernel so that these incompatibilities are easier to diagnose? Thanks in advance. 
I can provide any testing you may need on RPis. I have a number of Pis here to test with. As for the 'other brcmfmac chip', the 43430, Infineon has not updated it. It appears that they have updated the rest of them to use external_auth for SAE. I have posted on their web site to request that this firmware be updated as well, but this is not the first time that updates to 34355 have dropped and the 43430 was omitted from the release.

jprestwo, 2025-02-12:
I know this issue is specifically for the Pi 3 (which I haven't tested) but we have a similar issue with the Pi 4 which requires this kernel patch:

https://lore.kernel.org/linux-wireless/20241215120401.238320-1-arend.vanspriel@broadcom.com/

It doesn't apply cleanly to 6.6, but its a one liner so its trivial to adapt. With this kernel patch plus the IWD patch (which I'll see if I can get merged today) WPA3/SAE works great on the Pi 4.

gearhead, 2025-02-12:
@jprestwo Thanks for this confirmation.
I am looking at this thread again and notice that it is for the 3B, specifically. The 3B will not work with these fixes, I do not think, as it has the 43430 brcmfmac wifi chip. This chip which was left out of the latest firmware update from cypress. The 3b+,4B,5B do work as they all have the 43455 chip which is in the latest firmware update from fall 2024. The 43430 was last updated in 2022 and does not yet have the external_auth functionality, AFAICT. 
Latest cypress firmware for this chip that is served up with RPiOS:
`Firmware: BCM43430/1 wl0: Jun 14 2023 07:27:45 version 7.45.96.s1 (gf031a129) FWID 01-70bd2af7`
I think there is a newer one floating around somewhere on the infineon web site, but it also does not work with WPA3/SAE

pelwell, 2025-02-13:
Having some time to spare, I took a look at the suggested patch. Although my 43455-equipped systems here can connect to a WPA3 SSID without it, the process is quicker and doesn't produce errors such as:
```
brcmfmac: brcmf_set_channel: set chanspec 0xd022 fail, reason -52
```
Note that I'm using wpa_supplicant rather than iwd - that's the RPiOS default - which may explain why it still works eventually without the patch.

Rather than re-adapt Arend's adapted patch, I've located the original as one of the hundreds of downstream patches that Cypress/Infineon provide. 

See: https://github.com/raspberrypi/linux/commit/009bacab99e1baf79a37563b2f15c394d2e3db8a

jprestwo, 2025-02-13:
> Having some time to spare, I took a look at the suggested patch.

Thank you!

> Rather than re-adapt Arend's adapted patch, I've located the original as one of the hundreds of downstream patches that Cypress/Infineon provide.

Interesting, I did notice on the wireless mailing list that the patch wasn't originally from Arend but I was not aware Cypress/Infineon already had the patch lying around somewhere. Nothing surprises me anymore, but its too bad vendors can't just commit to open source and upstream their fixes.

pelwell, 2025-02-13:
FYI running `sudo rpi-update rpi-6.6.y` will install a trial build including the patch. Note that this is beta software, so back up any important data first.

gearhead, 2025-02-13:
> FYI running `sudo rpi-update rpi-6.6.y` will install a trial build including the patch. Note that this is beta software, so back up any important data first.

@pelwell Sounds good! I will try this tonight. Which firmware is being used with this? Is the firmware for the 43455 version 7.45.286 or version 7.45.265?

pelwell, 2025-02-13:
I've used both of those successfully. The fact that the older, shipping firmware doesn't work for you suggests a difference in our systems - hardware, software or configuration. Are you using iwd or wpa_supplicant?

I'm not against bumping the firmware if the new one has the right feature set. 43455 releases aren't present in every release from Infineon, and I'd not noticed that one - we don't get any notifications.

gearhead, 2025-02-14:
@pelwell @jprestwo I confirm this works with 7.45.265 and 7.45.286 firmware after updating to the latest kernel via `sudo rpi-update rpi-6.6.y`. Did not try wpa_supplicant, but did with iwd-git v3.3 gbf82aff0 with James' PMKSA patch.

adicrescenzo, 2025-03-20:
I've compiled the last kernel version (6.6.78-v8-16k) on my Raspberry Pi 5 and also installed the brcm80211 kernel driver on `/usr/lib/firmware/cypress`:

`sudo find / -type f -name "cyfmac*.bin"`
```
/usr/lib/firmware/cypress/cyfmac4354-sdio.bin
/usr/lib/firmware/cypress/cyfmac4339-sdio.bin
/usr/lib/firmware/cypress/cyfmac43570-pcie.bin
/usr/lib/firmware/cypress/cyfmac54591-pcie.bin
/usr/lib/firmware/cypress/cyfmac43455-sdio-minimal.bin
/usr/lib/firmware/cypress/cyfmac43455-sdio-standard.bin
/usr/lib/firmware/cypress/cyfmac4356-sdio.bin
/usr/lib/firmware/cypress/cyfmac4373-sdio.bin
/usr/lib/firmware/cypress/cyfmac43439-sdio.bin
/usr/lib/firmware/cypress/cyfmac43012-sdio.bin
/usr/lib/firmware/cypress/cyfmac43430-sdio.bin
/usr/lib/firmware/cypress/cyfmac43455-sdio-upstream.bin
/usr/lib/firmware/cypress/cyfmac43340-sdio.bin
/usr/lib/firmware/cypress/cyfmac43362-sdio.bin
/usr/lib/firmware/cypress/cyfmac4356-pcie.bin
/var/lib/dpkg/alternatives/cyfmac43455-sdio.bin
```

. Unfortunately I can't use WPA3 network:

`iw list | grep -A 10 "Supported interface modes"`
```
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * P2P-client
		 * P2P-GO
		 * P2P-device
	Band 1:
		Capabilities: 0x1020
			HT20
			Static SM Power Save
```

Can you please tell what is missing? Thanks

jprestwo, 2025-03-20:
@adicrescenzo first off, the supported interface modes is not going to tell you if WPA3 is supported. Second, even without any of the patches/changes discussed here brcmfmac still reports that it supports SAE, but its just broken without the kernel patch (and IWD changes if you use IWD).

If you've tried connecting to a WPA3 network and it is not working you'll probably have to post some supplicant/kernel logs to debug further. If all your after is to see if brcmfmac "supports" it, you want to look for this line in `iw list`

```
	Device supports SAE with AUTHENTICATE command
```

adicrescenzo, 2025-03-20:
I’ve attempted to connect to a WPA3 network, but it isn’t working. I’ve tried both NetworkManager and wpa_supplicant, but the result is the same.

When I run the following command: `iw list | grep -i SAE` 

I get the output:

`Device supports SAE with AUTHENTICATE command`

Does this mean my device should be capable of connecting to a WPA3 network?

How can I collect logs to troubleshoot this issue?

Thanks!

jprestwo, 2025-03-20:
> Does this mean my device should be capable of connecting to a WPA3 network?

Should be, but as we know its been broken until recently.

Maybe someone else can chime in because I'm not sure if your `6.6.78-v8-16k` kernel would contain the patch needed. I know the patch was merged into the raspi kernel tree, but IIRC it was under some "beta" type build. You probably need to get that specific kernel until its mainlined.

adicrescenzo, 2025-03-20:
@jprestwo thank you for the support. I've tried to run the command:

`sudo rpi-update rpi-6.6.y`

but I got:

```
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
FW_REV:
BOOTLOADER_REV:6c2e2d6833df14134463ea8968151f031950a21e
 *** Invalid hash given
```

gearhead, 2025-03-20:
I got that last night as well. I then did `rpi-update` and ended up with linux-6.12.19! After this and a full apt update, my RPi 3b+/4/5 with  the 43455 chip all will connect to SAE/WPA3 with wpa_supplicant and iwd (built from git HEAD).
The  Zero2W with the 43430 chip will not. I am on an issue thread and have asked if the 43430 firmware can be updated to work like the 43455 and have received input from some knowledgeable people, so maybe it will also soon work with WPA3. 
I did notice a few strange things about the 43430 chip. One is that there are different firwares for the Zero2W and the Zero and the 3B in the RPi repo. I do not know exactly what is going on there. Also, the firmware version currently being used in the RPi repo is flaky even with wpa2. I got it to a state where it would not even connect with wpa2. In order to get it to connect, I had to grab a newer firmware from Ifineon...

gearhead, 2025-03-21:
Well, I am not building wpa_supplicant, but just installing it via 'apt'. I am building iwd-git from 'HEAD' form here:
`https://web.git.kernel.org/pub/scm/network/wireless/iwd.git`
I use makedeb to build it and do have specific RPi patches.
[Rune iwd-git](urlhttps://github.com/gearhead/RuneOS/tree/0.7_OS/rpios_packages/iwd-git)

burny0202, 2025-04-12:
I'm on pi 400. I can't a WPA3 connection to work

Tried wpa_supplicant & network manager.

Tried the cypress firmware fix - Tried updating - Tried rpi-update (6.12)....All nothing.

All I ever get is 

`
wpa_supplicant[3859]: wlan0: WPA: Failed to select authenticated key management type
wpa_supplicant[3859]: wlan0: WPA: Failed to set WPA key management and encryption suites
`

pelwell, 2025-04-12:
Pi 400 uses the Synaptics 43456, not the Infineon 43455 used in the Pi 5, and although the firmware is supposed to support WPA3 it didn't 5 work with our brcmfmac driver.

burny0202, 2025-04-12:
Does that mean I don't stand a chance getting this to work?

So many lost hours!

pelwell, 2025-04-12:
As the software stands now, that's correct.

burny0202, 2025-04-12:
That's really good to know - I don't suppose you know if Pi 5/500 works straight out of the box?

pelwell, 2025-04-12:
Yes - they both use the Infineon 43455.

burny0202, 2025-04-13:
What about the pi4 ?  I just attached my boot drive to an old pi4 I found, and I still can't connect.

Not so much of an error on this one...

`
Apr 13 16:25:28 gympi wpa_supplicant[2182]: wlan0: CTRL-EVENT-ASSOC-REJECT [MACADD] status_code=16
`

pelwell, 2025-04-13:
Try with a fresh image on another SD card. If that doesn't work, put `brcmfmac.debug=0x100000` on the end of cmdline.txt (keep everything in a single line), reboot, and report the output of `dmesg | grep brcmfmac`.

burny0202, 2025-04-14:
This was through network manager - I gave up with the manual wpa_supplicant, as I previously saw network manager was passing the same I'd seen in the manual wpa bits I read elsewhere
`
[    4.052044] brcmfmac: F1 signature read @0x18000000=0x15264345
[    4.056050] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    4.068136] usbcore: registered new interface driver brcmfmac
[    4.364841] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
[    4.365361] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Aug 29 2023 01:47:08 version 7.45.265 (28bca26 CY) FWID 01-b677b91b
[    4.480121] brcmfmac: CONSOLE:  FIXME bt_coex
[    4.480137] brcmfmac: CONSOLE: 000000.267 wl0: wlc_iovar_op: bw_cap BCME -5 (Not down)
[    5.752153] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[    5.776093] brcmfmac: CONSOLE: 000001.539 wl0: unable to find iovar "toe_ol"
[    5.776109] brcmfmac: CONSOLE: 000001.539 wl0: wlc_iovar_op: toe_ol BCME -23 (Unsupported)
[    5.776112] brcmfmac: CONSOLE: 000001.542 wl0: wl_open
[    5.912068] brcmfmac: CONSOLE: 000001.695 wl0: wl_open
[    5.912085] brcmfmac: CONSOLE: 000001.707 wl0: wlc_phy_set_regtbl_on_femctrl: FIXME bt_coex
[    8.668143] brcmfmac: CONSOLE: 000004.418 wl0: link up (wl0)
[    9.688089] brcmfmac: CONSOLE: 000005.424 wl0: unable to find iovar "tdls_sta_info"
[    9.688107] brcmfmac: CONSOLE: 000005.424 wl0: wlc_iovar_op: tdls_sta_info BCME -23 (Unsupported)
[    9.688111] brcmfmac: CONSOLE: 000005.443 wl0: unable to find iovar "nd_hostip_clear"
[    9.688115] brcmfmac: CONSOLE: 000005.443 wl0: wlc_iovar_op: nd_hostip_clear BCME -23 (Unsupported)
[   10.492066] brcmfmac: CONSOLE: 000006.270 wl0.0: wlc_send_bar: seq 0x3 tid 0
[   10.492086] brcmfmac: CONSOLE: 000006.271 wl0.0: wlc_send_bar: seq 0x2 tid 7
[   11.812508] brcmfmac: CONSOLE: 000007.571 wl0: unable to find iovar "tdls_sta_info"
[   11.812532] brcmfmac: CONSOLE: 000007.571 wl0: wlc_iovar_op: tdls_sta_info BCME -23 (Unsupported)
[   15.020114] brcmfmac: CONSOLE: 000010.769 wl0: unable to find iovar "tdls_sta_info"
[   15.020130] brcmfmac: CONSOLE: 000010.769 wl0: wlc_iovar_op: tdls_sta_info BCME -23 (Unsupported)
[   19.976091] brcmfmac: CONSOLE: 000015.698 wl0.0: wlc_rrm_timer: WAIT_STOP_TRAFFIC, entering PS mode for off channel measurement
[   19.976108] brcmfmac: CONSOLE: 000015.699 wlc_rrm_timer: state upd to 5
[   19.976112] brcmfmac: CONSOLE: 000015.699 wl0: wlc_rrm_timer: WAIT_STOP_TRAFFIC, 0 ms until measurement, waiting for PS announcement
[   19.976116] brcmfmac: CONSOLE: 000015.699 wlc_rrm_begin, state upd to 4
[   19.976119] brcmfmac: CONSOLE: 000015.699 wlc_rrm_begin: calling wlc_set_chanspec
[   19.976122] brcmfmac: CONSOLE: 000015.703 wlc_rrm_begin_scan: duration: 100, channel num: 1, bssid: [MACADD]
[   20.015150] brcmfmac: CONSOLE: 000015.760 wlc_rrm_bcnreq_scancb: state: 7, status: 0
[   20.015170] brcmfmac: CONSOLE: 000015.760 wlc_rrm_bcnreq_scancb: scan count: 1
[   20.179089] brcmfmac: CONSOLE: 000015.909 wlc_rrm_meas_end: return to the home channel
[   20.179107] brcmfmac: CONSOLE: 000015.909 wlc_rrm_report_11k: scan type: 1, scan results count = 1, sendout bcnrep
[   21.024142] brcmfmac: CONSOLE: 000016.751 wl0: unable to find iovar "tdls_sta_info"
[   21.024168] brcmfmac: CONSOLE: 000016.752 wl0: wlc_iovar_op: tdls_sta_info BCME -23 (Unsupported)
[   25.056097] brcmfmac: CONSOLE: 000020.773 wlc_rrm_begin: blocked
[   25.176081] brcmfmac: CONSOLE: 000020.869 wl0: unable to find iovar "nd_hostip_clear"
[   25.176098] brcmfmac: CONSOLE: 000020.869 wl0: wlc_iovar_op: nd_hostip_clear BCME -23 (Unsupported)
[   30.904051] brcmfmac: CONSOLE: 000026.612 wlc_phy_rxcal_txrx_gainctrl_acphy_tiny: Too much or too little power? (gain_index=0)
[   31.167984] brcmfmac: CONSOLE: 000026.873 wlc_phy_rxcal_txrx_gainctrl_acphy_tiny: Too much or too little power? (gain_index=0)
[   34.324024] brcmfmac: CONSOLE: 000030.024 wlc_phy_rxcal_txrx_gainctrl_acphy_tiny: Too much or too little power? (gain_index=0)
[   37.995986] brcmfmac: CONSOLE: 000033.680 wlc_phy_rxcal_txrx_gainctrl_acphy_tiny: Too much or too little power? (gain_index=0)
[   43.080030] brcmfmac: CONSOLE: 000038.768 wl0: wlc_phy_set_regtbl_on_femctrl: FIXME bt_coex
[   48.664019] brcmfmac: CONSOLE: 000044.344 wl0: wlc_phy_set_regtbl_on_femctrl: FIXME bt_coex
[   48.684040] brcmfmac: CONSOLE: 000044.364 wlc_phy_rxcal_txrx_gainctrl_acphy_tiny: Too much or too little power? (gain_index=0)
[  164.080074] brcmfmac: CONSOLE: 000159.799 wl0: wlc_phy_set_regtbl_on_femctrl: FIXME bt_coex
[  167.232135] brcmfmac: CONSOLE: 000162.940 wl0: wlc_phy_set_regtbl_on_femctrl: FIXME bt_coex
[  169.816196] brcmfmac: CONSOLE: 000165.514 wlc_phy_rxcal_txrx_gainctrl_acphy_tiny: Too much or too little power? (gain_index=0)
[  179.920171] brcmfmac: CONSOLE: 000175.624 wl0: wlc_phy_set_regtbl_on_femctrl: FIXME bt_coex
[  180.000138] brcmfmac: CONSOLE: 000175.703 wl0: wlc_phy_set_regtbl_on_femctrl: FIXME bt_coex
[  180.020236] brcmfmac: CONSOLE: 000175.722 wlc_phy_rxcal_txrx_gainctrl_acphy_tiny: Too much or too little power? (gain_index=0)
[  190.120121] brcmfmac: CONSOLE: 000185.822 wl0: wlc_phy_set_regtbl_on_femctrl: FIXME bt_coex

`

burny0202, 2025-04-14:
Just tried disabling power save with - Same effect

`iw dev wlan0 set power_save off`

gearhead, 2025-04-14:
My Pi4 connects with a basic headless image. This works with connman and with NetworkManager (nmcli) plus wpa_supplicant. CAn you connect to the WPA3 SSID from your phone?

burny0202, 2025-04-14:
Oh yeah - Got a house full of devices connected to it.

Only came up because I have a few devices that wont (old devices + pis)

burny0202, 2025-04-23:
I thought I would give building the kernel a try - So I've done that twice in the last 2 days. Here's my journalctl 

`Apr 23 17:58:13 gympi NetworkManager[592]: <info>  [1745427493.1010] device (wlan0): supplicant interface state: associating -> completed
Apr 23 17:58:13 gympi NetworkManager[592]: <info>  [1745427493.1011] device (wlan0): Activation: (wifi) Stage 2 of 5 (Device Configure) successful. Connected to wirele>
Apr 23 17:58:13 gympi NetworkManager[592]: <info>  [1745427493.1012] device (p2p-dev-wlan0): supplicant management interface state: associating -> completed
Apr 23 17:58:13 gympi NetworkManager[592]: <info>  [1745427493.1018] device (wlan0): state change: config -> ip-config (reason 'none', sys-iface-state: 'managed')
Apr 23 17:58:13 gympi NetworkManager[592]: <info>  [1745427493.1041] dhcp4 (wlan0): activation: beginning transaction (timeout in 45 seconds)
Apr 23 17:58:13 gympi avahi-daemon[523]: Joining mDNS multicast group on interface wlan0.IPv6 with address MYMAC.
Apr 23 17:58:13 gympi avahi-daemon[523]: New relevant interface wlan0.IPv6 for mDNS.
Apr 23 17:58:13 gympi avahi-daemon[523]: Registering new address record for MYMAC on wlan0.*.
Apr 23 17:58:13 gympi kernel: brcmfmac: CONSOLE: 000303.604 wl0: unable to find iovar "tdls_sta_info"
Apr 23 17:58:13 gympi kernel: brcmfmac: CONSOLE: 000303.605 wl0: wlc_iovar_op: tdls_sta_info BCME -23 (Unsupported)
Apr 23 17:58:13 gympi kernel: brcmfmac: CONSOLE: 000303.611 wl0: wlc_txbf_init_sta: scb aleady has user index 0
Apr 23 17:58:13 gympi kernel: brcmfmac: CONSOLE: 000303.640 wl0: unable to find iovar "nd_hostip_clear"
Apr 23 17:58:13 gympi kernel: brcmfmac: CONSOLE: 000303.640 wl0: wlc_iovar_op: nd_hostip_clear BCME -23 (Unsupported)
Apr 23 17:58:13 gympi kernel: brcmfmac: CONSOLE: 000303.683 wl0: wlc_recvfilter: scb is null, dont resend deauth
Apr 23 17:58:13 gympi kernel: brcmfmac: CONSOLE: 000303.688 wl0.0: wlc_send_bar: seq 0x3 tid 7
Apr 23 17:58:13 gympi kernel: brcmfmac: CONSOLE: 000304.278 wl0.0: wlc_send_bar: seq 0x3 tid 0
`

ValdikSS, 2026-09-07:
Can anyone please test SAE support on the latest Raspberry OS (kernel 6.18) with 43455 (3B/4/5)?

I have Raspberry Pi 400, it has 4345**6** and seems to have never supported SAE.
https://github.com/raspberrypi/linux/issues/7610

Just want to check how `iw phy` / `wpa_cli get_capability key_mgmt` output should look for 43455.

pelwell, 2026-09-10:
On a Pi 5:
```
$ iw phy
Wiphy phy0
        wiphy index: 0
        max # scan SSIDs: 10
        max scan IEs length: 2048 bytes
        max # sched scan SSIDs: 16
        max # match sets: 16
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Supported Ciphers:
                * WEP40 (00-0f-ac:1)
                * WEP104 (00-0f-ac:5)
                * TKIP (00-0f-ac:2)
                * CCMP-128 (00-0f-ac:4)
                * CMAC (00-0f-ac:6)
        Available Antennas: TX 0 RX 0
        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * P2P-client
                 * P2P-GO
                 * P2P-device
        Band 1:
                Capabilities: 0x1020
                        HT20
                        Static SM Power Save
                        RX HT20 SGI
                        No RX STBC
                        Max AMSDU length: 3839 bytes
                        DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: 16 usec (0x07)
                HT TX/RX MCS rate indexes supported: 0-7
                Bitrates (non-HT):
                        * 1.0 Mbps
                        * 2.0 Mbps (short preamble supported)
                        * 5.5 Mbps (short preamble supported)
                        * 11.0 Mbps (short preamble supported)
                        * 6.0 Mbps
                        * 9.0 Mbps
                        * 12.0 Mbps
                        * 18.0 Mbps
                        * 24.0 Mbps
                        * 36.0 Mbps
                        * 48.0 Mbps
                        * 54.0 Mbps
                Frequencies:
                        * 2412.0 MHz [1] (20.0 dBm)
                        * 2417.0 MHz [2] (20.0 dBm)
                        * 2422.0 MHz [3] (20.0 dBm)
                        * 2427.0 MHz [4] (20.0 dBm)
                        * 2432.0 MHz [5] (20.0 dBm)
                        * 2437.0 MHz [6] (20.0 dBm)
                        * 2442.0 MHz [7] (20.0 dBm)
                        * 2447.0 MHz [8] (20.0 dBm)
                        * 2452.0 MHz [9] (20.0 dBm)
                        * 2457.0 MHz [10] (20.0 dBm)
                        * 2462.0 MHz [11] (20.0 dBm)
                        * 2467.0 MHz [12] (20.0 dBm)
                        * 2472.0 MHz [13] (20.0 dBm)
                        * 2484.0 MHz [14] (disabled)
        Band 2:
                Capabilities: 0x1062
                        HT20/HT40
                        Static SM Power Save
                        RX HT20 SGI
[... 98 lines cut ...]
        Device supports SAE with AUTHENTICATE command
        Device supports scan flush.
        Device supports randomizing MAC-addr in sched scans.
        max # scan plans: 1
        max scan plan interval: 508
        max scan plan iterations: 0
        Supported TX frame types:
                 * managed: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
                 * AP: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
                 * P2P-client: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
                 * P2P-GO: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
                 * P2P-device: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
        Supported RX frame types:
                 * managed: 0x40 0xb0 0xd0
                 * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
                 * P2P-client: 0x40 0xd0
                 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
                 * P2P-device: 0x40 0xd0
        Supported extended features:
                * [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
                * [ DFS_OFFLOAD ]: DFS offload
$ wpa_cli get_capability key_mgmt
Selected interface 'wlan0'
NONE IEEE8021X WPA-EAP WPA-PSK WPA-EAP-SUITE-B OWE DPP FT-PSK FT-EAP
```

ValdikSS, 2026-09-10:
@pelwell no mention of SAE anywhere?

There's one caveat with key_mgmt:
wpa_supplicant creates key_mgmt from the chip's cipher list if the chip does not return key_mgmt. It's filled [from the presets](https://git.w1.fi/cgit/hostap/tree/src/drivers/driver_nl80211_capa.c?h=2_12#n1956) and miss for example WPA-EAP-SHA256 needed for WPA3-Enterprise. Most probably it will work even if it's not listed.

To detect whether key_mgmt you see is from chip or filled from the template, run

`wpa_supplicant ... -d | grep 'supported key_mgmt'`

If you don't see this string in the output, this means the key_mgmt is filled from the template.

Could you also provide `wpa_cli get_capability auth_algs` please?

UPD: `Device supports SAE with AUTHENTICATE command`, I see.

pelwell, 2026-09-10:
> @pelwell no mention of SAE anywhere?

Does `Device supports SAE with AUTHENTICATE command` not count?

> Could you also provide wpa_cli get_capability auth_algs please?

```
$ sudo wpa_cli get_capability auth_algs
Selected interface 'wlan0'
FAIL
```

ValdikSS, 2026-09-10:
It's `auth_alg`, sorry.

pelwell, 2026-09-10:
(Whoomp!) There it is:
```
$ sudo wpa_cli get_capability auth_alg
Selected interface 'wlan0'
OPEN SHARED LEAP SAE
```