raspberrypi/linux #6135: Kernel 6.8 and 6.9 builds do not contain bcachefs filesystem support - please activate CONFIG_BCACHEFS_FS=m

bmilde, 2024-04-29:
### Describe the bug

Bcachefs is an exciting new filesystem that was merged into mainline kernel since release 6.7. Please enable support for this FS as a module in 6.8 and 6.9 builds, thanks!

CONFIG_BCACHEFS_FS=m

and possibly also:

CONFIG_BCACHEFS_QUOTA=m
CONFIG_BCACHEFS_ERASURE_CODING=m
CONFIG_BCACHEFS_POSIX_ACL=m
CONFIG_BCACHEFS_TESTS=m
CONFIG_BCACHEFS_LOCK_TIME_STATS=m
CONFIG_BCACHEFS_SIX_OPTIMISTIC_SPIN=m

### Steps to reproduce the behaviour

pi@raspberrypi:~ $ sudo modprobe bcachefs
modprobe: FATAL: Module bcachefs not found in directory /lib/modules/6.8.8-v8-16k+

### Device (s)

Raspberry Pi 5

### System

pi@raspberrypi:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2023-12-05
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 70cd6f2a1e34d07f5cba7047aea5b92457372e05, stage5

pi@raspberrypi:~ $ uname -a
Linux raspberrypi 6.8.8-v8-16k+ #1 SMP PREEMPT Mon Apr 29 14:06:48 UTC 2024 aarch64 GNU/Linux

ticpu, 2024-08-08:
I see a Merged pull request, should this one be closed?
I also see branch 6.10.y has the config enabled in a random config:
https://github.com/raspberrypi/linux/blob/rpi-6.10.y/arch/arm/configs/bcm2711_defconfig#L1483

popcornmix, 2024-08-09:
Yes, request is in default kernel configs now.