raspberrypi/linux #6141: rtl8761_bu usb bluetooth download fw command failed

hkskoglund, 2024-05-01:
### Describe the bug

Hi! I have a Trust usb stick that most of the time fails to download fw during boot on my raspberry pi 5. I have to plug/unplug it until it succeeds. From the btmon log it seems that fw loading stops after the last packet which fails to get HCI Event: Command Complete?

I have no issues with the usb stick on my laptop running fedora 40 with kernel 6.8.7-300.fc40, so maybe its related to the raspberry pi kernel?

### Steps to reproduce the behaviour

dmesg --ctime --follow
Plug/unplug usb stick
Some times download fw fails

### Device (s)

Raspberry Pi 5

### System

Raspberry Pi reference 2024-03-15
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, f19ee211ddafcae300827f953d143de92a5c6624, stage4
2024/04/20 11:53:30 
Copyright (c) 2012 Broadcom
version d1744d21 (release) (embedded)
Linux raspberrypi0 6.6.28+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.28-1+rpt1 (2024-04-22) aarch64 GNU/Linux

### Logs

dmesg --ctime --follow:

[Wed May  1 18:26:14 2024] usb 1-2: new full-speed USB device number 6 using xhci-hcd
[Wed May  1 18:26:14 2024] usb 1-2: New USB device found, idVendor=0bda, idProduct=8771, bcdDevice= 2.00
[Wed May  1 18:26:14 2024] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Wed May  1 18:26:14 2024] usb 1-2: Product: Bluetooth Radio
[Wed May  1 18:26:14 2024] usb 1-2: Manufacturer: Realtek
[Wed May  1 18:26:14 2024] usb 1-2: SerialNumber: 00E04C239987
[Wed May  1 18:26:14 2024] Bluetooth: hci3: RTL: examining hci_ver=0a hci_rev=000b lmp_ver=0a lmp_subver=8761
[Wed May  1 18:26:14 2024] Bluetooth: hci3: RTL: rom_version status=0 version=1
[Wed May  1 18:26:14 2024] Bluetooth: hci3: RTL: loading rtl_bt/rtl8761bu_fw.bin
[Wed May  1 18:26:14 2024] Bluetooth: hci3: RTL: loading rtl_bt/rtl8761bu_config.bin
[Wed May  1 18:26:14 2024] Bluetooth: hci3: RTL: cfg_sz 6, total sz 30210
[Wed May  1 18:26:17 2024] Bluetooth: hci3: command 0xfc20 tx timeout
[Wed May  1 18:26:17 2024] Bluetooth: hci3: RTL: Failed to generate devcoredump
[Wed May  1 18:26:17 2024] Bluetooth: hci3: RTL: download fw command failed (-110)
[Wed May  1 18:26:18 2024] usb 1-2: USB disconnect, device number 6
[Wed May  1 18:26:18 2024] Bluetooth: hci3: RTL: RTL: Read reg16 failed (-19)
[Wed May  1 18:26:29 2024] usb 1-2: new full-speed USB device number 7 using xhci-hcd
[Wed May  1 18:26:29 2024] usb 1-2: New USB device found, idVendor=0bda, idProduct=8771, bcdDevice= 2.00
[Wed May  1 18:26:29 2024] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Wed May  1 18:26:29 2024] usb 1-2: Product: Bluetooth Radio
[Wed May  1 18:26:29 2024] usb 1-2: Manufacturer: Realtek
[Wed May  1 18:26:29 2024] usb 1-2: SerialNumber: 00E04C239987
[Wed May  1 18:26:29 2024] Bluetooth: hci3: RTL: examining hci_ver=0a hci_rev=000b lmp_ver=0a lmp_subver=8761
[Wed May  1 18:26:29 2024] Bluetooth: hci3: RTL: rom_version status=0 version=1
[Wed May  1 18:26:29 2024] Bluetooth: hci3: RTL: loading rtl_bt/rtl8761bu_fw.bin
[Wed May  1 18:26:29 2024] Bluetooth: hci3: RTL: loading rtl_bt/rtl8761bu_config.bin
[Wed May  1 18:26:29 2024] Bluetooth: hci3: RTL: cfg_sz 6, total sz 30210
[Wed May  1 18:26:30 2024] Bluetooth: hci3: RTL: fw version 0xdfc6d922
[Wed May  1 18:26:30 2024] Bluetooth: MGMT ver 1.22

tail btmon-1.log of failed fw download:

< HCI Command: Vendor (0x3f|0x0020) plen 223               #245 [hci1] 6.804458
        f7 68 01 44 51 02 00 40 41 06 00 64 30 08 00 b0  .h.DQ..@A..d0...
        00 66 00 59 40 0a 06 db 50 0c 06 f2 7b 10 06 8c  .f.Y@...P...{...
        55 12 06 0a 28 14 06 27 01 00 f0 60 00 02 90 77  U...(..'...`...w
        f8 64 06 27 00 02 01 00 00 34 00 04 27 38 00 83  .d.'.....4..'8..
        80 00 20 11 50 01 20 00 02 02 20 00 3b 3f 20 28  .. .P. ... .;? (
        03 20 20 2c 90 02 01 20 00 40 00 00 80 42 00 00  .  ,... .@...B..
        01 44 00 00 00 46 00 84 00 48 00 c0 00 4a 00 00  .D...F...H...J..
        00 4c 00 00 00 4e 00 28 00 50 00 40 cc 52 00 00  .L...N.(.P.@.R..
        14 54 00 00 00 56 00 00 20 58 00 11 55 5a 00 00  .T...V.. X..UZ..
        80 5c 00 00 00 5e 00 05 28 0e 01 01 00 02 01 60  .\...^..(......`
        c0 02 01 22 c0 7c e1 64 00 02 01 20 00 30 06 26  ...".|.d... .0.&
        67 34 06 7f c8 3a 06 d3 00 5a 03 45 00 42 02 bf  g4...:...Z.E.B..
        05 2e 01 00 68 02 00 40 41 08 00 b0 00 02 02 47  ....h..@A......G
        99 a4 25 d5 d6 22 d9 c6 df 55 ab 23 87 00 00     ..%.."...U.#... 
= Close Index: 00:00:00:00:00:00                                [hci1] 8.819086

same in fw download success btmon-2.log

< HCI Command: Vendor (0x3f|0x0020) plen 223               #245 [hci3] 7.428504
        f7 68 01 44 51 02 00 40 41 06 00 64 30 08 00 b0  .h.DQ..@A..d0...
        00 66 00 59 40 0a 06 db 50 0c 06 f2 7b 10 06 8c  .f.Y@...P...{...
        55 12 06 0a 28 14 06 27 01 00 f0 60 00 02 90 77  U...(..'...`...w
        f8 64 06 27 00 02 01 00 00 34 00 04 27 38 00 83  .d.'.....4..'8..
        80 00 20 11 50 01 20 00 02 02 20 00 3b 3f 20 28  .. .P. ... .;? (
        03 20 20 2c 90 02 01 20 00 40 00 00 80 42 00 00  .  ,... .@...B..
        01 44 00 00 00 46 00 84 00 48 00 c0 00 4a 00 00  .D...F...H...J..
        00 4c 00 00 00 4e 00 28 00 50 00 40 cc 52 00 00  .L...N.(.P.@.R..
        14 54 00 00 00 56 00 00 20 58 00 11 55 5a 00 00  .T...V.. X..UZ..
        80 5c 00 00 00 5e 00 05 28 0e 01 01 00 02 01 60  .\...^..(......`
        c0 02 01 22 c0 7c e1 64 00 02 01 20 00 30 06 26  ...".|.d... .0.&
        67 34 06 7f c8 3a 06 d3 00 5a 03 45 00 42 02 bf  g4...:...Z.E.B..
        05 2e 01 00 68 02 00 40 41 08 00 b0 00 02 02 47  ....h..@A......G
        99 a4 25 d5 d6 22 d9 c6 df 55 ab 23 87 00 00     ..%.."...U.#... 
> HCI Event: Command Complete (0x0e) plen 5                #246 [hci3] 7.461497
      Vendor (0x3f|0x0020) ncmd 3
        Status: Success (0x00)
        f7                                               .               
< HCI Command: Read Local Version... (0x04|0x0001) plen 0  #247 [hci3] 7.461506
> HCI Event: Command Complete (0x0e) plen 12               #248 [hci3] 7.463502
      Read Local Version Information (0x04|0x0001) ncmd 2
        Status: Success (0x00)
        HCI version: Bluetooth 5.1 (0x0a) - Revision 57286 (0xdfc6)
        LMP version: Bluetooth 5.1 (0x0a) - Subversion 55586 (0xd922)
        Manufacturer: Realtek Semiconductor Corporation (93)

### Additional context

I have dtoverlay=disable-bt in /boot/firmware/config.txt and have disabled hciuart, since I dont need the internal adapter

hkskoglund, 2024-05-02:
I've tried to trace down the source code responsible for downloading firmware, and found it in rtl_download_firmware [https://github.com/torvalds/linux/blob/master/drivers/bluetooth/btrtl.c#L802](url)

The function only tries 1 time to download a fragment before giving up. 

```
/* Send download command */
		skb = __hci_cmd_sync(hdev, 0xfc20, frag_len + 1, dl_cmd,
				     HCI_INIT_TIMEOUT);
		if (IS_ERR(skb)) {
			rtl_dev_err(hdev, "download fw command failed (%ld)",
				    PTR_ERR(skb));
			ret = PTR_ERR(skb);
			goto out;
		}
```

In addition gemini AI/chatgpt gives :

gemini AI

The provided C code appears mostly correct for downloading firmware to an RTL Bluetooth controller. Here's a breakdown with some explanations and potential improvements:

**Functionality:**

* The code downloads firmware in fragments to the Bluetooth controller identified by `hdev`.
* It iterates through the firmware data (`data`) in chunks of `RTL_FRAG_LEN` and sends them using HCI commands. 
* After successful download, it reads the local firmware version using another HCI command.

**Strengths:**

* The code uses kernel memory allocation (`kmalloc`) for the download command structure.
* It checks for errors during command sending and data length verification.
* It logs debugging and informational messages using appropriate functions.

**Potential Improvements:**

* **Error Handling:** Consider adding checks for `dl_cmd` allocation failures beyond just checking the return value. You can use a similar approach for `skb` allocations.
* **Loop Termination:** The loop condition `i < frag_num` might be susceptible to overflow if `frag_num` is very large.  Consider using a loop that iterates until all data is sent (`data + frag_len > data`).
* **Resource Management:** Move the `kfree(dl_cmd);` outside the loop to ensure it's always freed even in error cases.
* **Clarity:** Adding comments within the loop to explain the purpose of setting `dl_cmd->index` and handling the last fragment could improve code readability.

**Uncertainties:**

* The code relies on undefined types like `struct rtl_download_cmd` and `struct rtl_download_response`. Without knowing their definitions, it's impossible to fully verify their usage.
* The functions `rtl_dev_dbg`, `rtl_dev_err`, `rtl_dev_info`, and `btrtl_read_local_version` are assumed to be specific to the RTL Bluetooth driver and are not part of the standard Linux kernel API.

**Overall:**

The code seems like a reasonable implementation for downloading firmware to an RTL Bluetooth controller. However, for a complete review, the missing type definitions and specific driver functionalities would need to be analyzed. The suggested improvements can further enhance the code's robustness and readability.

chatgpt:

The code you provided appears to be a function for downloading firmware to a Realtek Bluetooth device. Here's a quick analysis:

1. **Memory Allocation**: The function allocates memory for a `struct rtl_download_cmd` using `kmalloc`. This allocation is later freed using `kfree`.

2. **Looping through Fragments**: The function loops through the firmware data in fragments of size `RTL_FRAG_LEN`. It constructs a `struct rtl_download_cmd` for each fragment and sends it using `__hci_cmd_sync`. The loop also handles the case when the last fragment might be smaller than `RTL_FRAG_LEN`.

3. **Error Handling**: The function properly handles errors by returning appropriate error codes and freeing allocated resources (`skb`, `dl_cmd`) before returning.

4. **Verification**: It verifies the length of the response received after sending the download command and also reads the local version information after downloading the firmware.

5. **Correctness**: The code appears logically correct, assuming `RTL_FRAG_LEN` is defined and `rtl_dev_dbg`, `rtl_dev_err`, and `rtl_dev_info` are properly implemented for debugging and logging.

However, I noticed a couple of potential issues or areas for improvement:

- **Data Copying**: Inside the loop, `memcpy(dl_cmd->data, data, frag_len);` appears to copy the same data (`data`) into `dl_cmd->data` in each iteration. It seems like it should copy a portion of `data` corresponding to the current fragment. You might need to adjust the starting position in `data` for each fragment.

- **Unused Variables**: The `skb` variable is declared twice inside the loop. You can remove the declaration inside the loop since it's already declared outside.

- **Error Handling in Fragment Loop**: In the fragment loop, if an error occurs during sending the download command (`__hci_cmd_sync`), the loop will break out early with an error code, but it's not clearing resources allocated before the error occurred (e.g., previously allocated `skb`). You should make sure to free resources properly before exiting the function.

Overall, the code seems structurally sound, but you might need to verify the details of the firmware download process and adjust the data copying logic accordingly.

sninaber, 2024-05-04:
hi hkskoglund,

i have the same issue with several aliexpress rtl8761bu dongles on the pi5, plug/unplug until firmware loads and then everything is fine. strange thing is that my pi4's dont have this problem at all.
same raspberry pi os, all rpi-updated to 6.6.30 to be sure. overclocking doesn't seem to matter much.
pi's are in flirc cases so external bluetooth is pretty much essential.

just thought this might be helpful in tracking down the issue.

djholt, 2024-05-23:
Same exact issue here.

sninaber, 2024-05-23:
the latest firmware (https://github.com/raspberrypi/firmware/commit/5c83250276c4fa9d79a93ebc5a739a5ab6a4d6a7) seems to show some improvement on the pi5, sometimes rtl_bt/rtl8761bu now even loads succesfully at boot (about 30-40% succes rate). 
could this be related to this change: https://github.com/raspberrypi/linux/pull/6171 ?

pi4 still loads the driver without problems, also with the same rpi-update.

djholt, 2024-05-23:
@sninaber I'm saw this same improvement just after my comment above. I updated firmware and also enabled usb_max_current_enable=1 at the same time, so I wasn't sure which change helped.

But, still not good reliability. As a workaround, I can repeatedly power cycle USB power using uhubctl until the hci interface comes up successfully, but this isn't ideal. Have you found any way to reload the driver/module or device from software? Or, have you found any alternative Bluetooth chip that plays nicely on Pi 5?

Thanks for the help!

sninaber, 2024-05-23:
i have tried to modprobe, hcitool, rfkill and bluetoothctl (bluez) to no avail. i have some older bluetooth dongles that work with pi5 and some newer that don't have linux drivers yet. but i can't really recommend any for the pi5 as long as we don't know what causes the issue.
let's see if this patch ( https://github.com/raspberrypi/linux/commit/0a19b5256303d2f35be9272832b01a170c9a039b ) for the next firmware release solves the timeout.

P33M, 2024-05-23:
> Same exact issue here.

Thanks for providing a comprehensive bug report. It's very useful.

> i have tried to modprobe, hcitool, rfkill and bluetoothctl (bluez) to no avail. i have some older bluetooth dongles that work with pi5 and some newer that don't have linux drivers yet. but i can't really recommend any for the pi5 as long as we don't know what causes the issue. let's see if this patch ( [0a19b52](https://github.com/raspberrypi/linux/commit/0a19b5256303d2f35be9272832b01a170c9a039b) ) for the next firmware release solves the timeout.

Why would it? You've got a USB device, and the patch there only concerns a SoC:PCIe link.

> the latest firmware ([raspberrypi/firmware@5c83250](https://github.com/raspberrypi/firmware/commit/5c83250276c4fa9d79a93ebc5a739a5ab6a4d6a7)) seems to show some improvement on the pi5, sometimes rtl_bt/rtl8761bu now even loads succesfully at boot (about 30-40% succes rate). could this be related to this change: #6171 ?
> 
> pi4 still loads the driver without problems, also with the same rpi-update.

This is interesting. Can you iterate through rpi-update versions before and after this date and see which lead to the improvement? Commit list here: https://github.com/raspberrypi/firmware/commits/master/

djholt, 2024-05-23:
> Thanks for providing a comprehensive bug report. It's very useful.

Whoa—uncalled for. I had no differentiating information to contribute at the time, but I'm following and interested. Thanks for the support through this issue.

djholt, 2024-05-23:
> Can you iterate through rpi-update versions

Okay. The problem started occurring in e632362—looks like 6.6.16 introduced the issue. I don't experience the issue whatsoever in 6.1.77.

sninaber, 2024-05-23:
ok that was fast i was still typing my reply.. thank for looking in to this!

sninaber, 2024-05-24:
> Okay. The problem started occurring in e632362—looks like 6.6.16 introduced the issue. I don't experience the issue whatsoever in 6.1.77.

i couldn't find this commit, did you mean revert back to https://github.com/raspberrypi/firmware/commit/e7590adb503bd6bc59e598f00e73bbae834d0d81 to check?
> 
> Why would it? You've got a USB device, and the patch there only concerns a SoC:PCIe link.
> 
my guess would be that on the pi5, RP1 is also connected to pcie, which among others holds the usb controller. (see latency considerations: https://datasheets.raspberrypi.com/rp1/rp1-peripherals.pdf ). so any pcie issues would affect RP1 devices.

djholt, 2024-05-24:
I was reverting using rpi-update, so I referenced the commits from rpi-firmware instead: https://github.com/raspberrypi/rpi-firmware/commits/master/

So, for example, the issue didn't manifest after reverting all the way back to 5fc4f64 (kernel 6.1.77):

sudo rpi-update 5fc4f643d2e9c5aa972828705a902d184527ae3f

P33M, 2024-05-24:
That's a clue that it might be an upstream issue (compounded by whatever's different on Pi 5[1]). Does a Pi 4 exhibit the same when updated to either version?

Note that running rpi-update on a Pi 4/5 will also update the bootloader, which has slightly changed XHCI behaviour over time. rpi-update won't downgrade it afterwards, so that should keep the same version for both 6.1 and 6.6.

[1] On Pi 5, the device is directly connected to a root port. On Pi 1-4, all full-speed devices are accessed though at least one hub.

sninaber, 2024-05-24:
> sudo rpi-update 5fc4f643d2e9c5aa972828705a902d184527ae3f

must admit i expected the problem te have existed on the pi5 all along, but can confirm this completely resolves the issue, so my compliments to how quickly you found this. still, looking at the upstream changes to rtl, generic or specifically rtl8761bu, it doesn't really explain why the pi5 sometimes fails to load the firmware while others (including pi4) don't. i have tried different usb hubs on the pi5 (powered and nonpowered) which doesn't make a difference. can't comment on XHCI or the binary files in e632362 where the problem started, so i'm quite useless from here on..

firmware update today https://github.com/raspberrypi/firmware/commit/f31d28d61e048c9d918874be8e7ebbc30db90617 improves a bit further (about 80% succes loading rtl firmware), but still fails sometimes.

sninaber, 2024-06-07:
i thought about what @p33m said earlier about xhci and the bootloader, so i wanted to see if the otg port behaved in the same way. so i added dtoverlay=dwc2,dr_mode=host to config.txt and connected the bluetooth adapter to the otg port. 

[image]

this solves the problem and recognises the adapter without problems, at boot or multiple unplug/plug.  does this help in any way in finding the cause?

is the firmware for RP1 also held in eeprom? if so downgrading to kernel 6.1.77 would not change this.
@p33m: i can send you one of these adapters if that helps.

dbrb2, 2024-06-09:
Thanks for the suggestion @sninaber - I ran into this problem running Home Assistant, and had had to revert the kernel version to keep using the dongles on USBA ports. Using the OSB OTG port in host mode, so far, is working fine on kernel 6.6.28

P33M, 2024-06-17:
I'm insufficiently convinced this isn't a regression between Pi 4 and Pi 5 host controller hardware. @sninaber can you email info < at > raspberrypi.com to start a discussion about getting one of these shipped here?

buganini, 2024-07-04:
https://github.com/raspberrypi/linux/issues/5046#issuecomment-2167137559
may be related

sninaber, 2024-07-24:
interesting!
no problem, thanks for taking time trying to reproduce. so far no issues in linux and firmware seem related, so impact seems low and we may have to close this issue for now. 

i just tried flashing a fresh sdcard for Pi5 with 64bit Pi-OS networkinstall, full-upgrade and rpi-update. offical keyboard, mouse and powersupply. for me with both 4k/16k kernels the problem remains intermittent (20-30% succes) with all rtl8761 dongles directly attached.

so that only leaves the pi5 units which are probably initial revisions. will look into replacing them after summer recess.

/proc/cpuinfo
Revision	: d04170
Serial		: e2fa4c6734b38595
Model		: Raspberry Pi 5 Model B Rev 1.0

uname -a
Linux raspberrypi 6.6.40-v8+ #1784 SMP PREEMPT Tue Jul 16 13:53:31 BST 2024 aarch64 GNU/Linux

last working firmware: af3ab905d42829d4ce860f19a0e049c3bcdbe35f (feb 8 2024).

P33M, 2024-07-24:
Do you have any other USB devices simultaneously plugged in, and if so what are they? (lsusb -v)

hkskoglund, 2024-07-24:
I've tried sudo rpi-update to 6.6.40 and the issue still persist

I then tried the` sudo rpi-update 5fc4f643d2e9c5aa972828705a902d184527ae3f` (6.1.77) and have no issues of downloading fw when plug/unplugging 

Looking at the lsusb output, I take a wild guess that maybe something changed in the root_hub xhci-hcd driver that sometimes interferes with downloading firmware ?

```
lsusb -tvv
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
    /sys/bus/usb/devices/usb4  /dev/bus/usb/004/001
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/2p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    /sys/bus/usb/devices/usb3  /dev/bus/usb/003/001
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
    /sys/bus/usb/devices/usb2  /dev/bus/usb/002/001
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/2p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    /sys/bus/usb/devices/usb1  /dev/bus/usb/001/001
    |__ Port 2: Dev 17, If 0, Class=Wireless, Driver=btusb, 12M
        ID 0bda:8771 Realtek Semiconductor Corp. 
        /sys/bus/usb/devices/1-2  /dev/bus/usb/001/017
    |__ Port 2: Dev 17, If 1, Class=Wireless, Driver=btusb, 12M
        ID 0bda:8771 Realtek Semiconductor Corp. 
        /sys/bus/usb/devices/1-2  /dev/bus/usb/001/017
```

sninaber, 2024-07-24:
just a raspberry pi keyboard and mouse, but as requested: sudo lsusb -v

Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0 
  bDeviceProtocol         3 
  bMaxPacketSize0         9
  idVendor           0x1d6b Linux Foundation
  idProduct          0x0003 3.0 root hub
  bcdDevice            6.06
  iManufacturer           3 Linux 6.6.40-v8-16k+ xhci-hcd
  iProduct                2 xHCI Host Controller
  iSerial                 1 xhci-hcd.1
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x001f
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
        bMaxBurst               0
Hub Descriptor:
  bLength              12
  bDescriptorType      42
  nNbrPorts             1
  wHubCharacteristic 0x0009
    Per-port power switching
    Per-port overcurrent protection
  bPwrOn2PwrGood       50 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  bHubDecLat          0.0 micro seconds
  wHubDelay             0 nano seconds
  DeviceRemovable    0x00
 Hub Port Status:
   Port 1: 0000.02a0 5Gbps power Rx.Detect
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
  wTotalLength       0x000f
  bNumDeviceCaps          1
  SuperSpeed USB Device Capability:
[... 447 lines cut ...]
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             2
  wHubCharacteristic 0x0009
    Per-port power switching
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood       10 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x00
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0100 power
   Port 2: 0000.0503 highspeed power enable connect
Device Status:     0x0001
  Self Powered

P33M, 2024-07-24:
I can reproduce this. There's a dependency between Vbus being connected to the device and the time it takes for initialisation to complete. If the time is too long, the device will not respond to further HCI commands (usually this means failing to load firmware).

If I plug the device in slowly (more likely to occur on a Pi 5, as connectors are quite tight) then I get a high failure rate. If I ram the device in quickly, it almost always works.
I have a hub that switches Vbus on/off at the upstream facing port. If the device is plugged into the first hub port, and enumerated first, then it usually works after being switched on. If I plug it into the last hub port, which is cascaded off the first hub, it never works.
It may also be the case that Pi 5 has more debounce on connection detection, increasing the time taken to enumerate.

My bet is there's some sort of watchdog inside the dongle that's initialised at start of day and is active, and the firmware blob deactivates it.

sninaber, 2024-07-24:
ok, glad you could reproduce after all!. i guess we are slow pluggers. but timeout is indeed the issue.
as hkskoglund mentioned in his/her second post, the routine for loading the firmware is called with a constant HCI_INIT_TIMEOUT which should be 10000 (hci.h) milliseconds. on the pi4 i observed it can take 5 seconds or more before the rtl8761 is initialised. but on the pi5 with kernel > 6.1.77 timeout occurs within 1 or maybe 2 seconds.

hkskoglund, 2024-07-25:
I tried to capture usb traffic on usbmon1 enabling usbmon by `sudo modprobe usbmon`. When fw download fails, tshark does not print "controller → host         HCI_EVT 71 Rcvd Command Complete (Vendor Command 0x0020 [opcode 0xFC20])". 

> henning@raspberrypi0:~ $ uname -a
Linux raspberrypi0 6.6.31+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux
> henning@raspberrypi0:~ $ sudo tshark -i 8

Failed fw download (hci0: command 0xfc20 tx timeout)

> 614   4.495767         host → controller   HCI_CMD 290 Sent Vendor Command 0x0020 (opcode 0xFC20)
  615   4.495786       1.36.0 → host         HCI_USB 64 Rcvd[Malformed Packet]
  616   4.501761       1.36.0 → host         HCI_USB 64 Rcvd[Malformed Packet]
  617   6.506055       1.36.1 → host         USB 64 URB_INTERRUPT in
  618   6.506871       1.36.2 → host         USB 64 URB_BULK in
  619   6.506926       1.36.2 → host         USB 64 URB_BULK in

Successfull fw download:

>  3258 161.271549         host → controller   HCI_CMD 290 Sent Vendor Command 0x0020 (opcode 0xFC20)
 3259 161.271568       1.35.0 → host         HCI_USB 64 Rcvd[Malformed Packet]
 3260 161.277541       1.35.0 → host         HCI_USB 64 Rcvd[Malformed Packet]
 3261 161.304544   controller → host         HCI_EVT 71 Rcvd Command Complete (Vendor Command 0x0020 [opcode 0xFC20])

> $ dmesg --ctime --follow
[Thu Jul 25 09:24:34 2024] usb 1-2: new full-speed USB device number 37 using xhci-hcd
[Thu Jul 25 09:24:34 2024] usb 1-2: New USB device found, idVendor=0bda, idProduct=8771, bcdDevice= 2.00
[Thu Jul 25 09:24:34 2024] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Thu Jul 25 09:24:34 2024] usb 1-2: Product: Bluetooth Radio
[Thu Jul 25 09:24:34 2024] usb 1-2: Manufacturer: Realtek
[Thu Jul 25 09:24:34 2024] usb 1-2: SerialNumber: 00E04C239987
[Thu Jul 25 09:24:34 2024] Bluetooth: hci0: RTL: examining hci_ver=0a hci_rev=000b lmp_ver=0a lmp_subver=8761
[Thu Jul 25 09:24:34 2024] Bluetooth: hci0: RTL: rom_version status=0 version=1
[Thu Jul 25 09:24:34 2024] Bluetooth: hci0: RTL: loading rtl_bt/rtl8761bu_fw.bin
[Thu Jul 25 09:24:34 2024] Bluetooth: hci0: RTL: loading rtl_bt/rtl8761bu_config.bin
[Thu Jul 25 09:24:34 2024] Bluetooth: hci0: RTL: cfg_sz 6, total sz 30210
[Thu Jul 25 09:24:37 2024] Bluetooth: hci0: command 0xfc20 tx timeout
[Thu Jul 25 09:24:37 2024] Bluetooth: hci0: RTL: download fw command failed (-110)

sislakd, 2024-08-02:
I'm using just the single USB device with rtl8761bu and observing the same issue. 

uname -a
Linux 6.6.42-v8+ #1787 SMP PREEMPT Thu Jul 25 17:44:41 BST 2024 aarch64 GNU/Linux

Revision	: d04170
Serial		: 2f2d8041c9185fa0
Model		: Raspberry Pi 5 Model B Rev 1.0

The issue happens also during restart of Rpi. Thus, it is not related to slow plug-in only. Sometimes, I have to un-plug and plug-in several times to get firware loaded successfully.

FraH90, 2024-10-06:
Finally I've discovered why my Asus BT500 BT dongle doesn't work. I'm having the same issue.
I use the dongle for a bit, but if I restart it isn't properly recognized, from what I read here the firmware isn't loaded into the chip at startup.
Has anyone fixed this? Are there any updates since August?

buganini, 2024-10-06:
> Finally I've discovered why my Asus BT500 BT dongle doesn't work. I'm having the same issue.
>
> I use the dongle for a bit, but if I restart it isn't properly recognized, from what I read here the [...]

which rpi release are you using?
there was a release in July which may contain the fix for this issue.

Kaitology, 2024-10-07:
> which rpi release are you using? there was a release in July which may contain the fix for this issue.

Bought it 2 weeks ago.

Linux raspberry 6.6.51+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt2 (2024-10-01) aarch64 GNU/Linux
Raspberry Pi 5 Model B Rev 1.0
Revision        : d04170

mzdrahal, 2024-10-18:
@P33M is there any further information we can supply to tackle this issue? I am experiencing the same problem; device not available on boot, manual re-insertion necessary.

Device works with `sudo rpi-update 5fc4f643d2e9c5aa972828705a902d184527ae3f`.

P33M, 2024-10-21:
The two USB analyser traces I gathered were almost identical. So aside from minor differences in timing, I don't see why 6.1 would work and 6.6 wouldn't. Reversion of the BT commits between 6.1 and 6.6 isn't straightforward - lots of conflicts.

One thing I can think of is to do a device reset right before the driver tries to load firmware - this could make it more robust.

P33M, 2024-10-21:
I've spotted something suspicious. The Pi 5 xHC stops polling the Interrupt endpoint used to report command completions at some point during the firmware download. If control transaction pacing is enabled, it gets through 3 chunks. If pacing is disabled, it gets through 13-14.

So why did it work for the first few commands, then give up?

P33M, 2024-10-23:
I have a crude workaround that improves things for me. Please try `sudo rpi-update pulls/6430` and report back. My theory is that somehow RP1 gets a stale read of a transfer ring with only one TRB outstanding.

This was always theoretically possible because of the race between a write coming across PCIe, DMA waking up and fetching the ring, and the TRB update making it to main memory late, but nothing so far has exposed this as a bug.

mzdrahal, 2024-10-23:
@P33M thanks a lot for your effort.

Just updated and the problem persists (USB device on bus 3; hci1):

```
[    5.591835] Bluetooth: hci1: RTL: examining hci_ver=0a hci_rev=000b lmp_ver=0a lmp_subver=8761
[    5.594852] Bluetooth: hci1: RTL: rom_version status=0 version=1
[    5.594864] Bluetooth: hci1: RTL: loading rtl_bt/rtl8761bu_fw.bin
[    5.595652] Bluetooth: hci1: RTL: loading rtl_bt/rtl8761bu_config.bin
[    5.595763] Bluetooth: hci1: RTL: cfg_sz 6, total sz 30210
[    5.677185] Bluetooth: hci0: BCM: chip id 107
[    5.677408] Bluetooth: hci0: BCM: features 0x2f
[    5.678555] Bluetooth: hci0: BCM4345C0
[    5.678566] Bluetooth: hci0: BCM4345C0 (003.001.025) build 0000
[    5.679387] Bluetooth: hci0: BCM4345C0 'brcm/BCM4345C0.raspberrypi,5-model-b.hcd' Patch
[    6.469338] Bluetooth: hci0: BCM: features 0x2f
[    6.470720] Bluetooth: hci0: BCM43455 37.4MHz Raspberry Pi 3+-0190
[    6.470726] Bluetooth: hci0: BCM4345C0 (003.001.025) build 0382
[    6.471665] Bluetooth: hci0: BCM: Using default device address (43:45:c0:00:1f:ac)
[    8.340653] Bluetooth: hci1: command 0xfc20 tx timeout
[    8.340666] Bluetooth: hci1: RTL: Failed to generate devcoredump
[    8.340669] Bluetooth: hci1: RTL: download fw command failed (-110)
```

Also no dice with USB device on bus 1:

```
[    5.345183] Bluetooth: hci0: RTL: examining hci_ver=0a hci_rev=000b lmp_ver=0a lmp_subver=8761
[    5.348173] Bluetooth: hci0: RTL: rom_version status=0 version=1
[    5.348194] Bluetooth: hci0: RTL: loading rtl_bt/rtl8761bu_fw.bin
[    5.348569] Bluetooth: hci0: RTL: loading rtl_bt/rtl8761bu_config.bin
[    5.348682] Bluetooth: hci0: RTL: cfg_sz 6, total sz 30210
[    5.368761] hci_uart_bcm serial0-0: supply vbat not found, using dummy regulator
[    5.368950] hci_uart_bcm serial0-0: supply vddio not found, using dummy regulator
[    5.745761] Bluetooth: hci1: BCM: chip id 107
[    5.745984] Bluetooth: hci1: BCM: features 0x2f
[    5.747154] Bluetooth: hci1: BCM4345C0
[    5.747157] Bluetooth: hci1: BCM4345C0 (003.001.025) build 0000
[    5.747927] Bluetooth: hci1: BCM4345C0 'brcm/BCM4345C0.raspberrypi,5-model-b.hcd' Patch
[    6.471206] Bluetooth: hci1: BCM: features 0x2f
[    6.472582] Bluetooth: hci1: BCM43455 37.4MHz Raspberry Pi 3+-0190
[    6.472586] Bluetooth: hci1: BCM4345C0 (003.001.025) build 0382
[    6.473011] Bluetooth: hci1: BCM: Using default device address (43:45:c0:00:1f:ac)
[    8.084805] Bluetooth: hci0: command 0xfc20 tx timeout
[    8.084815] Bluetooth: hci0: RTL: Failed to generate devcoredump
[    8.084820] Bluetooth: hci0: RTL: download fw command failed (-110)
```

Unfortunately, the device now also isn't able to retrieve the firmware download when being inserted after the system has booted:

```
[   71.585038] Bluetooth: hci1: RTL: examining hci_ver=0a hci_rev=000b lmp_ver=0a lmp_subver=8761
[   71.588795] Bluetooth: hci1: RTL: rom_version status=0 version=1
[   71.588803] Bluetooth: hci1: RTL: loading rtl_bt/rtl8761bu_fw.bin
[   71.588866] Bluetooth: hci1: RTL: loading rtl_bt/rtl8761bu_config.bin
[   71.588900] Bluetooth: hci1: RTL: cfg_sz 6, total sz 30210
[   72.437787] Bluetooth: hci1: RTL: Failed to generate devcoredump
[   72.516784] Bluetooth: hci1: RTL: Failed to generate devcoredump
[   72.596788] Bluetooth: hci1: RTL: Failed to generate devcoredump
[   72.675790] Bluetooth: hci1: RTL: Failed to generate devcoredump
[   72.754800] Bluetooth: hci1: RTL: Failed to generate devcoredump
[   72.834797] Bluetooth: hci1: RTL: Failed to generate devcoredump
[   72.913800] Bluetooth: hci1: RTL: Failed to generate devcoredump
[   72.992804] Bluetooth: hci1: RTL: Failed to generate devcoredump
[   73.072812] Bluetooth: hci1: RTL: Failed to generate devcoredump
[   73.151810] Bluetooth: hci1: RTL: Failed to generate devcoredump
[   73.230813] Bluetooth: hci1: RTL: Failed to generate devcoredump
[   73.310816] Bluetooth: hci1: RTL: Failed to generate devcoredump
[   73.389820] Bluetooth: hci1: RTL: Failed to generate devcoredump
[   73.468822] Bluetooth: hci1: RTL: Failed to generate devcoredump
[   73.548825] Bluetooth: hci1: RTL: Failed to generate devcoredump
[   73.627842] Bluetooth: hci1: RTL: Failed to generate devcoredump
[   73.706847] Bluetooth: hci1: RTL: Failed to generate devcoredump
[   73.785850] Bluetooth: hci1: RTL: Failed to generate devcoredump
[   73.865852] Bluetooth: hci1: RTL: Failed to generate devcoredump
[   73.944856] Bluetooth: hci1: RTL: Failed to generate devcoredump
[   74.023855] Bluetooth: hci1: RTL: Failed to generate devcoredump
[   74.103862] Bluetooth: hci1: RTL: Failed to generate devcoredump
[   74.182864] Bluetooth: hci1: RTL: Failed to generate devcoredump
[   74.261873] Bluetooth: hci1: RTL: Failed to generate devcoredump
[   74.341871] Bluetooth: hci1: RTL: Failed to generate devcoredump
[   74.363789] Bluetooth: hci1: command 0x1001 tx timeout
[   74.363799] Bluetooth: hci1: RTL: Failed to generate devcoredump
[   74.363805] Bluetooth: hci1: RTL: HCI_OP_READ_LOCAL_VERSION failed (-110)
[   74.363809] Bluetooth: hci1: RTL: read local version failed
```

P33M, 2024-10-30:
Can you re-run `sudo rpi-update pulls/6430` and reboot. With no other USB devices inserted (use SSH or VNC), plug the dongle into the top USB2 slot. Immediately afterwards, run
```
sudo find /sys/kernel/debug/usb/xhci/xhci-hcd.0/devices /sys/kernel/debug/usb/xhci/xhci-hcd.0/event-ring/ -type f -print -exec cat {} \; > dump.txt
```
and attach the contents of dump.txt if the firmware download fails.
I've extended the timeout for the fw download command to 20 seconds.

mzdrahal, 2024-10-31:
Looks promising!

What works:

- plugging in the dongle after boot, firmware gets downloaded, dongle information available at `hciconfig -a`
- re-inserting a dongle that has been inserted during boot now leads to a successful firmware download, with dongle information being available at `hciconfig -a`

What doesn't work:

- booting with the dongle already inserted

The firmware download itself happens within a very short time frame of the dongle being inserted:

`[    7.360068] Bluetooth: RFCOMM TTY layer initialized
[    7.360082] Bluetooth: RFCOMM socket layer initialized
[    7.360090] Bluetooth: RFCOMM ver 1.11
[   27.953915] Bluetooth: hci0: command 0xfc20 tx timeout
[   27.953915] Bluetooth: hci0: RTL: download fw command failed (-110)
[   27.953930] Bluetooth: hci0: RTL: Failed to generate devcoredump
[  183.743135] usb 1-2: USB disconnect, device number 2
[  188.831342] usb 1-2: new full-speed USB device number 3 using xhci-hcd
[  189.001369] usb 1-2: New USB device found, idVendor=7392, idProduct=c611, bcdDevice= 2.00
[  189.001374] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  189.001377] usb 1-2: Product: Edimax Bluetooth Adapter
[  189.001380] usb 1-2: Manufacturer: Realtek
[  189.001382] usb 1-2: SerialNumber: 00E04C239987
[  189.020395] Bluetooth: hci0: RTL: examining hci_ver=0a hci_rev=000b lmp_ver=0a lmp_subver=8761
[  189.023548] Bluetooth: hci0: RTL: rom_version status=0 version=1
[  189.023565] Bluetooth: hci0: RTL: loading rtl_bt/rtl8761bu_fw.bin
[  189.023755] Bluetooth: hci0: RTL: loading rtl_bt/rtl8761bu_config.bin
[  189.023832] Bluetooth: hci0: RTL: cfg_sz 6, total sz 30210
[  189.772415] Bluetooth: hci0: RTL: fw version 0xdfc6d922
[  189.911686] Bluetooth: MGMT ver 1.22`

Interesting enough, in the scenario where the dongle is available during boot, it gets hci0 assigned, instead of the onboard UART adapter.

hkskoglund, 2024-10-31:
Hi, i still have the same issue, it sometimes does not download the firmware and times out after about 20 seconds
[dump-rtl8761bu-fw-download-fail.txt](https://github.com/user-attachments/files/17584598/dump-rtl8761bu-fw-download-fail.txt)

> [Thu Oct 31 08:28:39 2024] Bluetooth: hci2: RTL: examining hci_ver=0a hci_rev=000b lmp_ver=0a lmp_subver=8761
[Thu Oct 31 08:28:40 2024] Bluetooth: hci2: RTL: rom_version status=0 version=1
[Thu Oct 31 08:28:40 2024] Bluetooth: hci2: RTL: loading rtl_bt/rtl8761bu_fw.bin
[Thu Oct 31 08:28:40 2024] Bluetooth: hci2: RTL: loading rtl_bt/rtl8761bu_config.bin
[Thu Oct 31 08:28:40 2024] Bluetooth: hci2: RTL: cfg_sz 6, total sz 30210
[Thu Oct 31 08:28:40 2024] Bluetooth: hci2: RTL: fw version 0xdfc6d922
[Thu Oct 31 08:28:40 2024] Bluetooth: MGMT ver 1.22
[Thu Oct 31 08:28:42 2024] usb 1-2: USB disconnect, device number 18
[Thu Oct 31 08:28:43 2024] usb 1-2: new full-speed USB device number 19 using xhci-hcd
[Thu Oct 31 08:28:43 2024] usb 1-2: New USB device found, idVendor=0bda, idProduct=8771, bcdDevice= 2.00
[Thu Oct 31 08:28:43 2024] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Thu Oct 31 08:28:43 2024] usb 1-2: Product: Bluetooth Radio
[Thu Oct 31 08:28:43 2024] usb 1-2: Manufacturer: Realtek
[Thu Oct 31 08:28:43 2024] usb 1-2: SerialNumber: 00E04C239987
[Thu Oct 31 08:28:43 2024] Bluetooth: hci3: RTL: examining hci_ver=0a hci_rev=000b lmp_ver=0a lmp_subver=8761
[Thu Oct 31 08:28:43 2024] Bluetooth: hci3: RTL: rom_version status=0 version=1
[Thu Oct 31 08:28:43 2024] Bluetooth: hci3: RTL: loading rtl_bt/rtl8761bu_fw.bin
[Thu Oct 31 08:28:43 2024] Bluetooth: hci3: RTL: loading rtl_bt/rtl8761bu_config.bin
[Thu Oct 31 08:28:43 2024] Bluetooth: hci3: RTL: cfg_sz 6, total sz 30210
[Thu Oct 31 08:28:44 2024] Bluetooth: hci3: RTL: fw version 0xdfc6d922
[Thu Oct 31 08:28:44 2024] Bluetooth: MGMT ver 1.22
[Thu Oct 31 08:28:45 2024] usb 1-2: USB disconnect, device number 19
[Thu Oct 31 08:28:48 2024] usb 1-2: new full-speed USB device number 20 using xhci-hcd
[Thu Oct 31 08:28:48 2024] usb 1-2: New USB device found, idVendor=0bda, idProduct=8771, bcdDevice= 2.00
[Thu Oct 31 08:28:48 2024] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Thu Oct 31 08:28:48 2024] usb 1-2: Product: Bluetooth Radio
[Thu Oct 31 08:28:48 2024] usb 1-2: Manufacturer: Realtek
[Thu Oct 31 08:28:48 2024] usb 1-2: SerialNumber: 00E04C239987
[Thu Oct 31 08:28:48 2024] Bluetooth: hci3: RTL: examining hci_ver=0a hci_rev=000b lmp_ver=0a lmp_subver=8761
[Thu Oct 31 08:28:48 2024] Bluetooth: hci3: RTL: rom_version status=0 version=1
[Thu Oct 31 08:28:48 2024] Bluetooth: hci3: RTL: loading rtl_bt/rtl8761bu_fw.bin
[Thu Oct 31 08:28:48 2024] Bluetooth: hci3: RTL: loading rtl_bt/rtl8761bu_config.bin
[Thu Oct 31 08:28:48 2024] Bluetooth: hci3: RTL: cfg_sz 6, total sz 30210
[Thu Oct 31 08:28:49 2024] ieee80211 phy0: brcmf_p2p_send_action_frame: Unknown Frame: category 0xa, action 0x8
[Thu Oct 31 08:29:10 2024] Bluetooth: hci3: RTL: download fw command failed (-110)
[Thu Oct 31 08:29:10 2024] Bluetooth: hci3: command 0xfc20 tx timeout

P33M, 2024-10-31:
It seems there's multiple ways in which the dongle setup can fail. With the extra xhci doorbell read, I don't get broken endpoint polling. The firmware will almost always get fully loaded with the last, smaller chunk never reporting completion via the interrupt endpoint. This results in the same symptom (timeout) as not polling the endpoint.
I've also had a case where the dongle failed to respond to a random command after the firmware download. I'm still looking to see if it's time related from when the device was last reset.

hkskoglund, 2024-10-31:
I am unfamiliar with xhci debugging, but I tried to grep  "ep 3" /sys/kernel/debug/usb/xhci/xhci-hcd.0/event-ring. I was expecting packet length of 253 for all packets except the last which is 223 as reported by sudo btmon.
I also see 'Stopped - Length Invalid' ? Is it something happening with packet length?

> 0x000000100ce5cc70: TRB 000000100dea7b60 status 'Short Packet' len 254 slot 1 ep 3 type 'Transfer Event' flags e:C
> 0x000000100ce5ccc0: TRB 000000100dea7b70 status 'Stopped' len 260 slot 1 ep 3 type 'Transfer Event' flags e:C
> 0x000000100ce5ced0: TRB 000000100ce5f000 status 'Stopped - Length Invalid' len 0 slot 1 ep 3 type 'Transfer Event' flags e:C
> 0x000000100ce5cfb0: TRB 000000100ce5f000 status 'Short Packet' len 252 slot 1 ep 3 type 'Transfer Event' flags e:C
> 0x000000100ce5cfd0: TRB 000000100ce5f010 status 'Short Packet' len 246 slot 1 ep 3 type 'Transfer Event' flags e:C
> 0x000000100ce5cff0: TRB 000000100ce5f020 status 'Short Packet' len 253 slot 1 ep 3 type 'Transfer Event' flags e:C

P33M, 2024-10-31:
That's a consequence of the timeout error cleanup - the endpoint ring was stopped by the kernel but the USB controller has discarded any remaining transfer state as a result.

P33M, 2024-11-04:
The last packet in the firmware download sequence is invariably corrupt. Here's two runs side-by-side done seconds apart with success on the left and failure on the right - on the bad run, there's a chunk of zeroes in the middle of the transfer.
[image]

hkskoglund, 2024-11-04:
If the Buffer 000000100a21db88 is a memory address, I can see it reused each 4 packet in the log. btmon/usbmon writes correct bytes when it fails, so something is happening with the buffer at a later stage
> /sys/kernel/debug/usb/xhci/xhci-hcd.0/devices/01/ep00/trbs
> 0x000000100a3a6870: Buffer **000000100a21db88** length 256 TD size 0 intr 0 type 'Data Stage' flags i:i:c:s:i:e:C
> 0x000000100a3a68a0: Buffer 000000100a21d188 length 256 TD size 0 intr 0 type 'Data Stage' flags i:i:c:s:i:e:C
> 0x000000100a3a68d0: Buffer 000000100a21ef88 length 256 TD size 0 intr 0 type 'Data Stage' flags i:i:c:s:i:e:C
> 0x000000100a3a6900: Buffer **000000100a21db88** length 226 TD size 0 intr 0 type 'Data Stage' flags i:i:c:s:i:e:C

P33M, 2024-11-04:
With no USB devices plugged in (use UART or SSH), what happens if you repeat the plug/unplug after running:
```
sudo busybox devmem 0x1f000c4400 32 0xff
sudo busybox devmem 0x1f000c43b0 32 0xff
```
Note that this won't persist over reboot.

JamesH65, 2024-11-04:
A while back there was a bug in some ethernet drivers where a buffer was being corrupted. It was a shared buffer, and it needed a copy on write to be added to ensure when something wrote to it, grabbed its own copy first. Worked most of the time without but there were some specific circumstances where something else used the buffer and caused weird corruption. No idea if this might be a similar problem. 

https://lore.kernel.org/all/20190218133512.479473907@linuxfoundation.org/

P33M, 2024-11-04:
The failing transfers with bogus data have buffers that span a 4k boundary. Specifically, the last 16B in a 4k page ends up zeroed. There is a feature in RP1 that attempts to split DMA transfers that cross a 4k boundary, but it looks like it misfires on bursts that go up to but do not cross the boundary. The devmem writes disable that.

hkskoglund, 2024-11-05:
I've tested un/plug behaviour and bluetooth scanning in addition to rebooting without any trouble. Thank you all and in particular @P33M for resolving this !

P33M, 2024-11-05:
You're welcome. As an aside, I did notice that there was pretty bad service latency for the Bulk IN endpoint (i.e. radio data) when connected directly to a root port - fiddling with some debug features, I get:
```
NAK_PER_ENH_FS DISABLE_PARKMODE_FS	Bulk IN polls per frame
	0		0		5 / 6.8us period
	1		0		9 / 6.8us period
	0		1		17 / 6.6us period
	1		1		35 / 6.5us period
```
For the top 3, this leaves approximately 85% of a 1ms frame where the endpoint is ignored and 70% for the last option. The same behaviour happens with debugprobe (which has an interrupt endpoint with a much longer interval).

If either FS device is plugged into a hub, this oddness disappears with 84 evenly-spaced IN handshakes, presumably because the HS scheduler operates per-microframe instead of per-frame.

FraH90, 2024-11-10:
> The attached bootloader recovery zip updates the RP1 firmware to disable the DMA interference. Copy the contents of this zipfile to a blank fat32 formatted SD card. Power off the Pi 5, insert the SD card, power on. Wait for rapid green flashes on the LED (or solid green on HDMI) and power off/remove SD card. No kernel changes should be necessary.
> 
> Please test and report back.
> 
> [rpi-eeprom-recovery.zip](https://github.com/user-attachments/files/17632902/rpi-eeprom-recovery.zip)

But could this be a permanent fix, or is it just something to see if we can fix the issue temporary?
How has this firmware been produced? There's code available?

P33M, 2024-11-10:
The required change will be integrated into the next Pi 5 bootloader release.

P33M, 2024-11-10:
@sninaber Do you want your hardware returned? If so please email info@raspberrypi.com referencing this issue number.

closed 2024-11-10 by P33M

sninaber, 2024-11-10:
@p33m Thanks for not giving up and finding the bug!. I still don't fully understand how downgrading to 6.1.77 would also fix this bug, so please keep the dongle just in case.

mzdrahal, 2024-11-12:
Thanks for your work, @P33M - working without a hitch now, across multiple reboots etc.