raspberrypi/linux #6142: Best resolution for monitor is not respected at boot (native is 1366x768, effective is 1920x1080)

token47, 2024-05-01:
### Describe the bug

Booting linux on a Raspberry Pi 4 B 2GB with a monitor that has native resolution of 1366x768 plugged in the HDMI0 port, it does get image but image is very low quality and letters seem very small, a consequence of actual chosen resolution for boot being 1920x1080 instead.

The EDID does list the correct mode and does mention that that is the prerefed one, but it seem not to be respected.

### Steps to reproduce the behaviour

Boot current image of ubuntu in a raspberry pi 4 b and watch chosen resolution at boot time (from firmware to kernel).

More details below.

### Device (s)

Raspberry Pi 4 Mod. B

### System

I don't think the distribution matters here because it seems to happen before that, but in any case I'm testing with the just released Ubuntu noble preinstalled image for arm64+raspi, server edition (not sure if desktop edition would make any difference).

### Logs

The firmware boots:

```
RPi: BOOTSYS release VERSION:0b7b6f28 DATE: 2024/04/17 TIME: 13:51:36
BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1713358296 serial a0940f46 boardrev b03111 stc 452522
PM_RSTS: 0x00001000
(lines suppreseed)

RPi: BOOTLOADER release VERSION:0b7b6f28 DATE: 2024/04/17 TIME: 13:51:36
BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1713358296 serial a0940f46 boardrev b03111 stc 1777411
(lines suppreseed)
Boot mode: USB-MSD (04) order e
HDMI0 edid block 0 offset 0
00ffffffffffff0049f7000000000000
011a0103800000000ad7a5a2594a9624
145054afce0081c00101010101010101
0101010101017f2156aa51001e30468f
33003f432100001ea91a00a050001630
3020370020560000001a000000fd0032
4c1e5010000a202020202020000000fc
004141410a20202020202020200a016f
HDMI0 edid block 1 offset 128
020323744f109f140504131216111503
0702060123090f038301000066030c00
100080011d00bc52d01e20b8285540c4
8e2100001e011d80d0721c1620102c25
80c48e2100009e8c0ad08a20e02d1010
3e9600138e210000188c0ad090204031
200c405500138e210000180000000000
0000000000000000000000000000009d
HDMI0: best-mode 1 (limit 1) 1280x720 60 Hz CEA modes fe007f80000000000000000000000000 extensions 1
(lines suppreseed)
Read config.txt bytes     1235 hnd 0x80fa2
[MSD [03:00] 3.32 000000:03] vl805.bin not found
[MSD [03:00] 3.32 000000:03] pieeprom.upd not found
[MSD [03:00] 3.32 000000:03] recover4.elf not found
[MSD [03:00] 3.32 000000:03] recovery.elf not found
Read start4.elf bytes  2256224 hnd 0x734ef
Read fixup4.dat bytes     5434 hnd 0x79769
0x00b03111 0x00000000 0x00001fff
MEM GPU: 76 ARM: 947 TOTAL: 1023
Firmware: f4e2138c2adc8f3a92a3a65939e458f11d7298ba Feb 29 2024 12:24:53
Starting start4.elf @ 0xfeb00200 partition 0
+
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 6.8.0-1004-raspi (buildd@bos03-arm64-035) (aarch64-linux-gnu-gcc-13 (Ubuntu 13.2.0-23ubuntu4) 13.2.0, GNU ld (GNU Binutils for Ubuntu) 2.42) #4-Ubuntu SMP PREEMPT_DYNAMIC Sat Apr 20 
02:29:55 UTC 2024 (Ubuntu 6.8.0-1004.4-raspi 6.8.1)
[    0.000000] KASLR enabled
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000027c00000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
(lines suppreseed)
```

As you can see, the EDID is correctly received, and it does indicate that best resolution is 1280x720, which does not seem correct but at least would not be too far from it.

As soon as the kernel starts and the four raspberries appear at the top of the screen, the resolution is already wrong at 1920x1080. It does finish booting and when I get a chance to run commands, I can see that:

```
ubuntu@ubuntu:~$ sudo fbset -s

mode "1920x1080"
    geometry 1920 1080 1920 1080 32
    timings 0 0 0 0 0 0 0
    rgba 8/16,8/8,8/0,8/24
endmode
```

This happens both with config.txt overlay for "vc4-kms-v3d", for "vc4-fkms-v3d" or without any of them.

I'll add the decoded EDID next.

token47, 2024-05-01:
```
edid-decode (hex):

00 ff ff ff ff ff ff 00 49 f7 00 00 00 00 00 00
01 1a 01 03 80 00 00 00 0a d7 a5 a2 59 4a 96 24
14 50 54 af ce 00 81 c0 01 01 01 01 01 01 01 01
01 01 01 01 01 01 7f 21 56 aa 51 00 1e 30 46 8f
33 00 3f 43 21 00 00 1e a9 1a 00 a0 50 00 16 30
30 20 37 00 20 56 00 00 00 1a 00 00 00 fd 00 32
4c 1e 50 10 00 0a 20 20 20 20 20 20 00 00 00 fc
00 41 41 41 0a 20 20 20 20 20 20 20 20 0a 01 6f

02 03 23 74 4f 10 9f 14 05 04 13 12 16 11 15 03
07 02 06 01 23 09 0f 03 83 01 00 00 66 03 0c 00
10 00 80 01 1d 00 bc 52 d0 1e 20 b8 28 55 40 c4
8e 21 00 00 1e 01 1d 80 d0 72 1c 16 20 10 2c 25
80 c4 8e 21 00 00 9e 8c 0a d0 8a 20 e0 2d 10 10
3e 96 00 13 8e 21 00 00 18 8c 0a d0 90 20 40 31
20 0c 40 55 00 13 8e 21 00 00 18 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 9d

----------------

Block 0, Base EDID:
  EDID Structure Version & Revision: 1.3
  Vendor & Product Identification:
    Manufacturer: ROW
    Model: 0
    Made in: week 1 of 2016
  Basic Display Parameters & Features:
    Digital display
    Image size is variable
    Gamma: 1.00
    RGB color display
    First detailed timing is the preferred timing
  Color Characteristics:
    Red  : 0.6357, 0.3486
    Green: 0.2900, 0.5888
    Blue : 0.1425, 0.0800
    White: 0.3134, 0.3291
  Established Timings I & II:
    IBM     :   720x400    70.081663 Hz   9:5     31.467 kHz     28.320000 MHz
    DMT 0x04:   640x480    59.940476 Hz   4:3     31.469 kHz     25.175000 MHz
    DMT 0x05:   640x480    72.808802 Hz   4:3     37.861 kHz     31.500000 MHz
    DMT 0x06:   640x480    75.000000 Hz   4:3     37.500 kHz     31.500000 MHz
    DMT 0x08:   800x600    56.250000 Hz   4:3     35.156 kHz     36.000000 MHz
    DMT 0x09:   800x600    60.316541 Hz   4:3     37.879 kHz     40.000000 MHz
    DMT 0x0a:   800x600    72.187572 Hz   4:3     48.077 kHz     50.000000 MHz
    DMT 0x0b:   800x600    75.000000 Hz   4:3     46.875 kHz     49.500000 MHz
    DMT 0x10:  1024x768    60.003840 Hz   4:3     48.363 kHz     65.000000 MHz
    DMT 0x11:  1024x768    70.069359 Hz   4:3     56.476 kHz     75.000000 MHz
    DMT 0x12:  1024x768    75.028582 Hz   4:3     60.023 kHz     78.750000 MHz
  Standard Timings:
    DMT 0x55:  1280x720    60.000000 Hz  16:9     45.000 kHz     74.250000 MHz
  Detailed Timing Descriptors:
    DTD 1:  1366x768    59.964364 Hz 683:384   47.852 kHz     85.750000 MHz (575 mm x 323 mm)
                 Hfront   70 Hsync 143 Hback  213 Hpol P
                 Vfront    3 Vsync   3 Vback   24 Vpol P
    DTD 2:  1280x768    59.994726 Hz   5:3     47.396 kHz     68.250000 MHz (32 mm x 86 mm)
                 Hfront   48 Hsync  32 Hback   80 Hpol P
                 Vfront    3 Vsync   7 Vback   12 Vpol N
    Display Range Limits:
      Monitor ranges (GTF): 50-76 Hz V, 30-80 kHz H, max dotclock 160 MHz
    Display Product Name: 'AAA'
  Extension blocks: 1
Checksum: 0x6f

----------------

Block 1, CTA-861 Extension Block:
  Revision: 3
  Basic audio support
  Supports YCbCr 4:4:4
  Supports YCbCr 4:2:2
  Native detailed modes: 4
  Video Data Block:
    VIC  16:  1920x1080   60.000000 Hz  16:9     67.500 kHz    148.500000 MHz
    VIC  31:  1920x1080   50.000000 Hz  16:9     56.250 kHz    148.500000 MHz (native)
    VIC  20:  1920x1080i  50.000000 Hz  16:9     28.125 kHz     74.250000 MHz
    VIC   5:  1920x1080i  60.000000 Hz  16:9     33.750 kHz     74.250000 MHz
    VIC   4:  1280x720    60.000000 Hz  16:9     45.000 kHz     74.250000 MHz
    VIC  19:  1280x720    50.000000 Hz  16:9     37.500 kHz     74.250000 MHz
    VIC  18:   720x576    50.000000 Hz  16:9     31.250 kHz     27.000000 MHz
    VIC  22:  1440x576i   50.000000 Hz  16:9     15.625 kHz     27.000000 MHz
    VIC  17:   720x576    50.000000 Hz   4:3     31.250 kHz     27.000000 MHz
    VIC  21:  1440x576i   50.000000 Hz   4:3     15.625 kHz     27.000000 MHz
    VIC   3:   720x480    59.940060 Hz  16:9     31.469 kHz     27.000000 MHz
    VIC   7:  1440x480i   59.940060 Hz  16:9     15.734 kHz     27.000000 MHz
    VIC   2:   720x480    59.940060 Hz   4:3     31.469 kHz     27.000000 MHz
    VIC   6:  1440x480i   59.940060 Hz   4:3     15.734 kHz     27.000000 MHz
    VIC   1:   640x480    59.940476 Hz   4:3     31.469 kHz     25.175000 MHz
  Audio Data Block:
    Linear PCM:
      Max channels: 2
      Supported sample rates (kHz): 88.2 48 44.1 32
      Supported sample sizes (bits): 20 16
  Speaker Allocation Data Block:
    FL/FR - Front Left/Right
  Vendor-Specific Data Block (HDMI), OUI 00-0C-03:
    Source physical address: 1.0.0.0
    Supports_AI
  Detailed Timing Descriptors:
    DTD 3:  1280x720    50.000000 Hz  16:9     37.500 kHz     74.250000 MHz (708 mm x 398 mm)
                 Hfront  440 Hsync  40 Hback  220 Hpol P
                 Vfront    5 Vsync   5 Vback   20 Vpol P
    DTD 4:  1920x1080i  50.000000 Hz  16:9     28.125 kHz     74.250000 MHz (708 mm x 398 mm)
                 Hfront  528 Hsync  44 Hback  148 Hpol P
                 Vfront    2 Vsync   5 Vback   15 Vpol P Vfront +0.5 Odd Field
                 Vfront    2 Vsync   5 Vback   15 Vpol P Vback  +0.5 Even Field
    DTD 5:   720x480    59.940060 Hz   3:2     31.469 kHz     27.000000 MHz (531 mm x 398 mm)
                 Hfront   16 Hsync  62 Hback   60 Hpol N
                 Vfront    9 Vsync   6 Vback   30 Vpol N
    DTD 6:   720x576    50.000000 Hz   5:4     31.250 kHz     27.000000 MHz (531 mm x 398 mm)
                 Hfront   12 Hsync  64 Hback   68 Hpol N
                 Vfront    5 Vsync   5 Vback   39 Vpol N
Checksum: 0x9d
```

As can be seen, the "First detailed timing is the preferred timing" and the DTD1 is 1366x768.

popcornmix, 2024-05-01:
I'll need to check the logic the kernel chooses, but:
```
    VIC  31:  1920x1080   50.000000 Hz  16:9     56.250 kHz    148.500000 MHz (native)
```
does show the monitor reports 1920x1080 is native resolution.

token47, 2024-05-01:
Complete boot from firmware to kernel:

```
RPi: BOOTSYS release VERSION:0b7b6f28 DATE: 2024/04/17 TIME: 13:51:36
BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1713358296 serial a0940f46 boardrev b03111 stc 452522
PM_RSTS: 0x00001000
part 00000000 reset_info 00000000
uSD voltage 3.3V
Initialising SDRAM 'Micron' 16Gb x1 total-size: 16 Gbit 3200
DDR 3200 0 0 16 152
OTP boardrev b03111 bootrom 8b0 8b0
Customer key hash 0000000000000000000000000000000000000000000000000000000000000000
VC-JTAG unlocked

RPi: BOOTLOADER release VERSION:0b7b6f28 DATE: 2024/04/17 TIME: 13:51:36
BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1713358296 serial a0940f46 boardrev b03111 stc 1777292
PCI0 init
PCI0 reset
PCIe scan 00001106:00003483
XHCI-STOP
xHC0 ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
USBSTS 11
xHC0 ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
xHC0 ports 5 slots 32 intrs 4
Reset USB port-power 1000 ms
xhci_set_port_power 1 0
xhci_set_port_power 2 0
xhci_set_port_power 3 0
xhci_set_port_power 4 0
xhci_set_port_power 5 0
xhci_set_port_power 1 1
xhci_set_port_power 2 1
xhci_set_port_power 3 1
xhci_set_port_power 4 1
xhci_set_port_power 5 1
XHCI-STOP
xHC0 ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
USBSTS 18
XHCI-STOP
xHC0 ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
USBSTS 19
xHC0 ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
xHC0 ports 5 slots 32 intrs 4
Boot mode: USB-MSD (04) order e
HDMI0 edid block 0 offset 0
00ffffffffffff0049f7000000000000
011a0103800000000ad7a5a2594a9624
145054afce0081c00101010101010101
0101010101017f2156aa51001e30468f
33003f432100001ea91a00a050001630
3020370020560000001a000000fd0032
4c1e5010000a202020202020000000fc
004141410a20202020202020200a016f
HDMI0 edid block 1 offset 128
020323744f109f140504131216111503
0702060123090f038301000066030c00
100080011d00bc52d01e20b8285540c4
8e2100001e011d80d0721c1620102c25
80c48e2100009e8c0ad08a20e02d1010
3e9600138e210000188c0ad090204031
200c405500138e210000180000000000
0000000000000000000000000000009d
HDMI0: best-mode 1 (limit 1) 1280x720 60 Hz CEA modes fe007f80000000000000000000000000 extensions 1
USB2[1] 400202e1 connected
USB2 root HUB port 1 init
DEV [01:00] 2.16 000000:01 class 9 VID 2109 PID 3431
HUB init [01:00] 2.16 000000:01
HUB [01:00] 2.16 000000:01 init port 4 speed 1
DEV [02:01] 2.00 000004:01 class 0 VID 046d PID c52b
HID [02:01] 2.00 000004:01 register HID
[... 412 lines cut ...]
[    5.341811] logitech-hidpp-device 0003:046D:400A.0005: input,hidraw2: USB HID v1.11 Mouse [Logitech M325] on usb-0000:01:00.0-1.4/input2:2
[    5.664211] raid6: neonx8   gen()  3966 MB/s
[    5.685236] raid6: neonx4   gen()  4086 MB/s
[    5.706231] raid6: neonx2   gen()  3348 MB/s
[    5.727218] raid6: neonx1   gen()  2456 MB/s
[    5.748216] raid6: int64x8  gen()  2531 MB/s
[    5.769216] raid6: int64x4  gen()  2519 MB/s
[    5.790213] raid6: int64x2  gen()  2380 MB/s
[    5.811218] raid6: int64x1  gen()  1820 MB/s
[    5.815566] raid6: using algorithm neonx4 gen() 4086 MB/s
[    5.837211] raid6: .... xor() 2742 MB/s, rmw enabled
[    5.842258] raid6: using neon recovery algorithm
[    5.851079] xor: measuring software checksum speed
[    5.857343]    8regs           :  7253 MB/sec
[    5.863040]    32regs          :  7751 MB/sec
[    5.870394]    arm64_neon      :  3365 MB/sec
[    5.874831] xor: using function: 32regs (7751 MB/sec)
[    5.882079] async_tx: api initialized (async)
[    6.098304] Btrfs loaded, zoned=yes, fsverity=no
[    6.424641] EXT4-fs (sda2): mounted filesystem c711ccfa-0386-4932-86cd-bd357d00dd99 ro with ordered data mode. Quota mode: none.
[    7.146088] systemd[1]: Inserted module 'autofs4'
[    7.207502] systemd[1]: systemd 255.4-1ubuntu8 running in system mode (+PAM +AUDIT +SELINUX +APPARMOR +IMA +SMACK +SECCOMP +GCRYPT -GNUTLS +OPENSSL +ACL +BLKID +CURL +ELFUTILS +FIDO2 +IDN2 -IDN +IPTC +KMOD +L
IBCRYPTSETUP +LIBFDISK +PCRE2 -PWQUALITY +P11KIT +QRENCODE +TPM2 +BZIP2 +LZ4 +XZ +ZLIB +ZSTD -BPF_FRAMEWORK -XKBCOMMON +UTMP +SYSVINIT default-hierarchy=unified)

```

token47, 2024-05-01:
> I'll need to check the logic the kernel chooses, but:
> 
> ```
>     VIC  31:  1920x1080   50.000000 Hz  16:9     56.250 kHz    148.500000 MHz (native)
> ```
> 
> does show the monitor reports 1920x1080 is native resolution.

Yeah, that does not seem correct and in fact justifies this being an allowed resolution, but historically the prefered one has been used until very recently (latest images from Noble) it started to do this.

It might be something on Ubuntu's kernel, or it might be the latest firmware that I upgraded recently on Rpi4.

6by9, 2024-05-01:
As documented at https://www.raspberrypi.com/documentation/computers/config_txt.html#hdmi-pipeline-for-raspberry-pi-4-and-5, your 1366x768 isn't supported on Pi4 due to
```
Hfront   70 Hsync 143 Hback  213 Hpol P
```

Are you booting to a Window Manager (ie X11, Wayfire, or similar), or just to the console? Most of the window managers ignore native and preferred modes and go for the highest resolution.

The kernel DRM framebuffer emulation I seem to recall does go for the preferred mode (if available), but otherwise the largest. You can override it with a `video=HDMI-A-1:1280x720@60` type entry on the kernel command line.

token47, 2024-05-01:
Wow, that is very specific:

> Running everything at 2 pixels per clock means that the Raspberry Pi 4 and 5 cannot support a timing where any of the horizontal timings are not divisible by 2. The firmware and Linux kernel filter out any mode that does not fulfil this criteria.
> 
> There is only one incompatible mode in the CEA and DMT standards: DMT mode 81, 1366x768 @ 60Hz. This mode has odd-numbered values for the horizontal sync and back porch timings and a width that indivisible by 8.
> 
> If your monitor has this resolution, Raspberry Pi 4 or 5 automatically drops down to the next mode advertised by the monitor; typically 1280x720.

Indeed this seems to be the reason for the mode to be excluded. Indeed I was using that monitor in a rpi3 where it worked great at native resolution and after moving it to pi4 it started to behave like this (but I also upgraded the firmware and ubuntu versions at the same time so it was not clear where the issue was).

One thing that I would expect then is that, according to the documentation quoted above, it would discard DTD1 and fall back to DTD2 (which is indeed 1280x720), but instead it went for a different one.

Thank you very much for pointing this out. I will test overriding the resolution in kernel cmdline and confirm here.

token47, 2024-05-01:
Answering your previous question, I'm only using the console. No graphical environment on this device.

On the video mode override, it mostly worked. Kernel still gets the screen at 1920x1080p when boot starts but later, almost at the end of the boot (but before init/systemd) it switches to 1280x720p.

I used "video=HDMI-A-1:1280x720@60" at the end of cmdline.txt.

Just as a curiosity I tried forcing the "1366x768@60" mode and the screen flashes but comes back still in 1080p.