raspberrypi/linux #6147: GPIO4 unstable switching because of 1-Wire protocol enabled

ajevticc, 2024-05-05:
### Describe the bug

In this thread (https://forums.raspberrypi.com/viewtopic.php?p=2218276#p2218276) it has been concluded that enabling 1-Wire protocol in the Raspberry Pi Configuration window, affects the usage of the GPIO4 pin as a generic I/O pin. This does not happen with I2C and SPI capable pins. 

### Steps to reproduce the behaviour

Simply running a generic blink.py program when the 1-Wire protocol is enabled shows this behavior of the pin not changing the logic state reliably. 

Here is the test program: 

import RPi.GPIO as GPIO
import time

# Set up GPIO using BCM numbering
GPIO.setmode(GPIO.BCM)
GPIO.setwarnings(False)
# Set GPIO 4 as output
led_pin = 4
GPIO.setup(led_pin, GPIO.OUT)

try:
    while True:
        # Turn on the LED
        GPIO.output(led_pin, GPIO.HIGH)
        print("LED ON")
        time.sleep(1)  # Wait for 1 second

        # Turn off the LED
        GPIO.output(led_pin, GPIO.LOW)
        print("LED OFF")
        time.sleep(1)  # Wait for 1 second

except KeyboardInterrupt:
    # Clean up GPIO
    GPIO.cleanup()

### Device (s)

Raspberry Pi Zero 2 W

### System

Script started on 2024-05-05 20:50:34+02:00 [TERM="xterm-256color" TTY="/dev/pts/0" COLUMNS="80" LINES="24"]
[?2004h]0;sasa@raspberrypi: ~[01;32msasa@raspberrypi[00m:[01;34m~ $[00m raspinfo
[?2004l
System Information
------------------

Raspberry Pi Zero 2 W Rev 1.0
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"

Raspberry Pi reference 2024-03-12
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, b9b581e0a62afb7200d4b0b72ec2c43485fd49cc, stage4

Linux raspberrypi 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux
Revision	: 902120
Serial		: 000000008f8ab336
Model		: Raspberry Pi Zero 2 W Rev 1.0
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0, libcamera interfaces=0

Videocore information
---------------------

Mar 17 2023 10:52:42 
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start)
[... 928 lines cut ...]
[    9.409394] vc4-drm soc:gpu: [drm] Cannot find any crtc or sizes
[   10.283127] uart-pl011 3f201000.serial: no DMA platform data
[   10.415341] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[   10.948437] 8021q: 802.1Q VLAN Support v1.8
[   11.545773] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   15.753182] systemd-journald[128]: File /var/log/journal/c50711eaaddd4d2188798334c8206ef8/user-1000.journal corrupted or uncleanly shut down, renaming and replacing.
[   16.566577] Bluetooth: Core ver 2.22
[   16.566693] NET: Registered PF_BLUETOOTH protocol family
[   16.566700] Bluetooth: HCI device and connection manager initialized
[   16.566725] Bluetooth: HCI socket layer initialized
[   16.566737] Bluetooth: L2CAP socket layer initialized
[   16.566756] Bluetooth: SCO socket layer initialized
[   16.604148] Bluetooth: HCI UART driver ver 2.3
[   16.604175] Bluetooth: HCI UART protocol H4 registered
[   16.604263] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   16.604426] Bluetooth: HCI UART pro

### Additional context

Note in the program itself, there is no call for the 1-Wire protocol. It is just enabled as an option to the Pi's GPIOs. 

Personally, I consider this a bug, because if you are unaware of this behavior, and pin GPIO4 is critical to your application,  a lot of debugging can go in to this. Especially that the hardware I2C ans SPI buses do not cause a similar affect.   
I did not find a similar bug report. 

Thank you for your time and this wonderful platform!

pelwell, 2024-05-05:
If RPi.GPIO is accessing the GPIO registers from user space at the same time as the kernel (in the shape of the onewire driver) then this is the expected outcome. The solution is not to use user space GPIO libraries - try the compatible rpi-lgpio library instead: https://github.com/waveform80/rpi-lgpio/

ajevticc, 2024-05-05:
> If RPi.GPIO is accessing the GPIO registers from user space at the same time as the kernel (in the shape of the onewire driver) then this is the expected outcome. The solution is not to use user [...]

Many thanks for the response, although agreed that this is an easy fix, for most users this is not so obvious. Nor did I find it anywhere convenient as a heads up warning. Hence, wanted to leave a trace somewhere. Will be more aware of this in the future.