raspberrypi/linux #6148: Get sensor link rate through video multiplexer

ArduCAM, 2024-05-06:
### Describe the bug

When using video mux on Raspberry Pi 5, data cannot be obtained normally.

### Steps to reproduce the behaviour

Connect camera using video mux on Raspberry Pi 5.

### Device (s)

Raspberry Pi 5

### System

-

### Additional context

Through inspection, it was found that the cfe driver did not correctly obtain the link rate of the sensor under video-mux.
https://github.com/raspberrypi/linux/blob/dda85fda5b2dda7c4e2ba18770bd2033313006d2/drivers/media/platform/raspberrypi/rp1_cfe/cfe.c#L1070-L1074

Tried adding some code, it solved the problem, but I guess it's not good enough:
```patch
+static struct v4l2_subdev *video_mux_media_entity_to_v4l2_subdev(struct media_entity *entity) {
+	struct media_pad *pad;
+
+	for (int i = 0; i < entity->num_pads; i++) {
+		pad = &entity->pads[i];
+		if (!(pad->flags & MEDIA_PAD_FL_SINK))
+			continue;
+
+		pad = media_pad_remote_pad_first(pad);
+		if (!pad || !is_media_entity_v4l2_subdev(pad->entity))
+			continue;
+
+		return media_entity_to_v4l2_subdev(pad->entity);
+	}
+
+	return NULL;
+}
+
 static u64 sensor_link_rate(struct cfe_device *cfe)
 {
 	struct v4l2_mbus_framefmt *source_fmt;
@@ -1014,6 +1032,14 @@ static u64 sensor_link_rate(struct cfe_device *cfe)
 			goto err;
 
 		entity = pad->entity;
+		if (entity->function == MEDIA_ENT_F_VID_MUX) {
+			subdev = video_mux_media_entity_to_v4l2_subdev(entity);
+			if (!subdev)
+				goto err;
+
+			break;
+		}
+
 		subdev = media_entity_to_v4l2_subdev(entity);
```

6by9, 2024-05-07:
I'm aware that we don't get the configuration through video-mux correctly, but as you say fixing it isn't trivial.
You need to find the active input link into the mux, and I don't believe your changes achieve that.

I think the better solution is in https://patchwork.linuxtv.org/project/linux-media/list/?series=12752 which is being reviewed on mainline, as link frequency then comes via get_mbus_config.
It still needs work on video-mux to derive the link frequency correctly based on the current configuration, but solves the problem of how to find the relevant numbers.

ArduCAM, 2024-05-09:
> I'm aware that we don't get the configuration through video-mux correctly, but as you say fixing it isn't trivial. You need to find the active input link into the mux, and I don't believe your [...]

Thank you very much for the information, I will continue to follow it.

closed 2024-05-09 by ArduCAM