raspberrypi/linux #6154: The dtoverlay=pineboards-hat-ai doesn't work

EnziinSystem, 2024-05-09:
### Describe the bug

The dtoverlay=pineboards-hat-ai doesn't work on Kernel 6.6.30

### Steps to reproduce the behaviour

Here are the steps:

```
$ rpi-eeprom-update -a
$ sudo rpi-update

$ uname -a
Linux smarthome 6.6.30-v8+ #1761 SMP PREEMPT Thu May  2 16:54:52 BST 2024 aarch64 GNU/Linux
```
Configure
```
sudo nano /boot/firmware/config.txt
```

Add lines

```
[all]
# Enable the PCIe External connector.
dtparam=pciex1
dtparam=pciex1_gen=3
kernel=kernel8.img

# Enable Pineboard
dtoverlay=pineboards-hat-ai
```
Turn off PCI ASPM

```
sudo nano /boot/firmware/cmdline.txt

Add **pcie_aspm=off** before **rootwait**

sudo reboot
```

Install the PCIe driver and Edge TPU runtime

```
sudo apt update

echo "deb https://packages.cloud.google.com/apt coral-edgetpu-stable main" | sudo tee /etc/apt/sources.list.d/coral-edgetpu.list

curl https://packages.cloud.google.com/apt/doc/apt-key.gpg | sudo apt-key add -

sudo apt-get update

sudo apt-get install cmake

sudo apt-get install -y libedgetpu1-std

sudo apt install -y devscripts debhelper

sudo apt install -y dkms

sudo apt-get install -y dh-dkms

# Clone the Gasket driver repository

sudo git clone https://github.com/google/gasket-driver.git

cd gasket-driver

sudo debuild -us -uc -tc -b

# Go back to the parent directory
cd ..

# Install the built Gasket driver package

sudo dpkg -i gasket-dkms_1.0-18_all.deb

Selecting previously unselected package gasket-dkms.
(Reading database ... 175909 files and directories currently installed.)
Preparing to unpack gasket-dkms_1.0-18_all.deb ...
Unpacking gasket-dkms (1.0-18) ...
Setting up gasket-dkms (1.0-18) ...
Loading new gasket-1.0 DKMS files...
Deprecated feature: REMAKE_INITRD (/usr/src/gasket-1.0/dkms.conf)
Building for 6.6.30-v8+
Module build for kernel 6.6.30-v8+ was skipped since the
kernel headers for this kernel do not seem to be installed.
```
Note: **Module build for kernel 6.6.30-v8+ was skipped since the kernel headers for this kernel do not seem to be installed**

Set udev rule

```
sudo sh -c "echo 'SUBSYSTEM==\"apex\", MODE=\"0660\", GROUP=\"apex\"' >> /etc/udev/rules.d/65-apex.rules"

sudo groupadd apex

sudo adduser $USER apex

sudo reboot now
```

Result

```
$ lspci
0000:00:00.0 PCI bridge: Broadcom Inc. and subsidiaries Device 2712 (rev 21)
0000:01:00.0 PCI bridge: ASMedia Technology Inc. ASM1182e 2-Port PCIe x1 Gen2 Packet Switch
0000:02:03.0 PCI bridge: ASMedia Technology Inc. ASM1182e 2-Port PCIe x1 Gen2 Packet Switch
0000:02:07.0 PCI bridge: ASMedia Technology Inc. ASM1182e 2-Port PCIe x1 Gen2 Packet Switch
0000:03:00.0 System peripheral: Global Unichip Corp. Coral Edge TPU
0000:04:00.0 Non-Volatile memory controller: Phison Electronics Corporation E8 PCIe3 NVMe Controller (rev 01)
0001:00:00.0 PCI bridge: Broadcom Inc. and subsidiaries Device 2712 (rev 21)
0001:01:00.0 Ethernet controller: Device 1de4:0001
```

List TPU

```
$ lspci -nn | grep 089a
0000:03:00.0 System peripheral [0880]: Global Unichip Corp. Coral Edge TPU [1ac1:089a]

$ ls /dev/apex_0
ls: cannot access '/dev/apex_0': No such file or directory

$ sudo modprobe gasket
modprobe: FATAL: Module gasket not found in directory /lib/modules/6.6.30-v8+

$ sudo modprobe apex
modprobe: FATAL: Module apex not found in directory /lib/modules/6.6.30-v8+
```

Note: I did not make changes to MSI-parent and did not recompile the DTB

### Device (s)

Raspberry Pi 5

### System

```
$ uname -a
Linux smarthome 6.6.30-v8+ #1761 SMP PREEMPT Thu May  2 16:54:52 BST 2024 aarch64 GNU/Linux
```

pelwell, 2024-05-09:
@mikegapinski ?

mikegapinski, 2024-05-09:
@pelwell the overlay works fine - I'll upload a tutorial for the gasket driver today and link it in our docs

mikegapinski, 2024-05-09:
In the case listed above - kernel headers were missing. That's the only issue @EnziinSystem had.

_Module build for kernel 6.6.30-v8+ was skipped since the
kernel headers for this kernel do not seem to be installed._

mikegapinski, 2024-05-09:
@EnziinSystem https://pineboards.io/blogs/tutorials/how-to-configure-the-google-coral-edge-tpu-on-the-raspberry-pi-5

This covers the DMKS drives as well as updating the kernel with rpi-update and installing the headers with rpi-source (since 6.6.30 is not live via apt yet.

@pelwell you can close the issue

x0derp, 2024-05-15:
Hey there,

I have the same Problem. The tutorial doesn’t work for me. I installed step by step (kernel, driver etc.)

Module build for kernel 6.6.30-v8+ was skipped since the kernel headers for this kernel do not seem to be installed.

@pelwell 
Do you solve the Problem on your system?

Thanks

pelwell, 2024-05-15:
> Do you solve the Problem on your system?

I don't have one.

mikegapinski, 2024-05-15:
> > Do you solve the Problem on your system?
> 
> 
> 
> I don't have one.

6.6.30 should roll to stable soon so headers would be available via apt instead of having to deal with rpi-source. Hopefully it won't take long

4tegs, 2025-04-19:
Running Kernel 6.12.22 I'm ending up in not being able to install the gasket drivers. Pineboards documentation has disappeared as the whole website is down. 
Error: 
`Deprecated_ feature: REMAKE_INITRD (/var/lib/dkms/gasket/1.0/source/dkms.conf)`
I have written a small tutorial how to overcome the obstacles: [Tutorial placed on Github](https://y.hs58.de/em8o0)