raspberrypi/linux #6159: SPI transfer fails with [Errno 110] Connection timed out when there are > 4 CS pins configured

itsarune, 2024-05-11:
### Describe the bug

If I configure the device tree with 5 chip selects for SPI0, I am able to observe the chip selects go low (as expected) when I transfer a message to each of them. If I try to issue a transfer to the 5th configured chip select, I observe the chip-select pin going low--but stays low for much longer than expected eventually resulting in a `TimeoutError: [Errno 110] Connection timed out.`

### Steps to reproduce the behaviour

1. Set up device tree for five chip-selects for SPI0 using two default SPI CS pins and three GPIOs
2. error reproducing example. The same issue also occurs with the C++ spidev implementation, but provided is python:
    ```
    import spidev
    spi.open(0, 4)
    to_send = [0x1, 0x2, 0x3]
    spi.xfer(to_send)
    ```
3. Configuring the fifth GPIO pin to another hardware results in the same problem.

### Device (s)

Raspberry Pi 5

### System

system information
------------------

raspberry pi 5 model b rev 1.0
pretty_name="debian gnu/linux 12 (bookworm)"
name="debian gnu/linux"
version_id="12"
version="12 (bookworm)"

raspberry pi reference 2024-03-15
generated using pi-gen, https://github.com/rpi-distro/pi-gen, f19ee211ddafcae300827f953d143de92a5c6624, stage2

linux piedpiper 6.6.30-v8-16k+ #1762 smp preempt thu may  9 15:18:59 bst 2024 aarch64 gnu/linux
revision	: c04170
serial		: 022f76c8c8c62e9a
model		: raspberry pi 5 model b rev 1.0
throttled flag  : throttled=0x0
camera          : vc_gencmd_read_response returned -1 error=1 error_msg="command not registered"

videocore information
---------------------

2024/04/20 11:53:30 
copyright (c) 2012 broadcom
version d1744d21 (release) (embedded)

vc_gencmd_read_response returned -1
error=1 error_msg="command not registered"

filesystem information
----------------------
filesystem     1k-blocks    used available use% mounted on
udev             1899136       0   1899136   0% /dev
tmpfs             414240    5232    409008   2% /run
/dev/mmcblk0p2  60819080 4481464  53229288   8% /
tmpfs            2071136       0   2071136   0% /dev/shm
tmpfs               5120      48      5072   1% /run/lock
/dev/mmcblk0p1    522230   90424    431806  18% /boot/firmware
tmpfs             414224       0    414224   0% /run/user/1000

filename				type		size		used		priority
/var/swap                               file		102368		0		-2

package version information
---------------------------
raspberrypi-ui-mods:
  installed: (none)
raspberrypi-sys-mods:
[... 457 lines cut ...]
                                reg = <2>;
                                #address-cells = <1>;
                                #size-cells = <0>;
                                spi-max-frequency = <125000000>;
                        };

                        spidev0_3: spidev@3 {
                                compatible = "spidev";
                                reg = <3>;
                                #address-cells = <1>;
                                #size-cells = <0>;
                                spi-max-frequency = <125000000>;
                        };

                        spidev0_4: spidev@4 {
                                compatible = "spidev";
                                reg = <4>;
                                #address-cells = <1>;
                                #size-cells = <0>;
                                spi-max-frequency = <125000000>;
                        };
                };
        };
};
```

itsarune, 2024-05-14:
Awesome, updated as you suggested and I can successfully transfer to the 5th chip-select as expected!

pelwell, 2024-05-14:
Thanks for the feedback. The fix has been merged, and will be included in future kernel releases.

closed 2024-05-14 by pelwell