raspberrypi/linux #6160: Unable to load max98357a overlay - no matching platform found

Nate711, 2024-05-13:
### Describe the bug

I'm unable to load the overlay for the max98357a.

```
pi@ubuntu:~$ sudo dtoverlay max98357a
DTOVERLAY[warn]: no matching platform found
```

I'm entirely unfamiliar with how modules and overlays work so not sure how to continue. Thanks!

### Steps to reproduce the behaviour

Install 24.04 lts, run `sudo dtoverlay max98357a`

### Device (s)

Raspberry Pi 5

### System

```
pi@ubuntu:~$ vcgencmd version
2023/10/30 16:45:10
Copyright (c) 2012 Broadcom
version 30de0ba5 (release) (embedded)

pi@ubuntu:~$ uname -r
6.8.0-1004-raspi
```

### Logs

> pi@ubuntu:~$ sudo dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x414fd0b1]
[    0.000000] Linux version 6.8.0-1004-raspi (buildd@bos03-arm64-035) (aarch64-linux-gnu-gcc-13 (Ubuntu 13.2.0-23ubuntu4) 13.2.0, GNU ld (GNU Binutils for Ubuntu) 2.42) #4-Ubuntu SMP PREEMPT_DYNAMIC Sat Apr 20 02:29:55 UTC 2024 (Ubuntu 6.8.0-1004.4-raspi 6.8.1)
[    0.000000] KASLR enabled
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 5 Model B Rev 1.0
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000002000000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x0000000002000000..0x0000000005ffffff (65536 KiB) map reusable linux,cma
[    0.000000] OF: reserved mem: 0x0000000000000000..0x000000000007ffff (512 KiB) nomap non-reusable atf@0
[    0.000000] OF: reserved mem: 0x000000003fd167e0..0x000000003fd16816 (0 KiB) nomap non-reusable nvram@0
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x00000000ffffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000007ffff]
[    0.000000]   node   0: [mem 0x0000000000080000-0x000000003fbfffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000001ffffffff]
[    0.000000] On node 0, zone DMA: 1024 pages in unavailable ranges
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 52 pages/cpu s88232 r8192 d116568 u212992
[    0.000000] pcpu-alloc: s88232 r8192 d116568 u212992 alloc=52*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Virtualization Host Extensions
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] alternatives: applying boot alternatives
[... 515 lines cut ...]
[    7.624917] brcmfmac_wcc: brcmf_wcc_attach: executing
[    7.633593] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
[    7.635030] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Apr 15 2021 03:03:20 version 7.45.234 (4ca95bb CY) FWID 01-996384e2
[    7.740090] Bluetooth: hci0: BCM: chip id 107
[    7.740335] Bluetooth: hci0: BCM: features 0x2f
[    7.741532] Bluetooth: hci0: BCM4345C0
[    7.741540] Bluetooth: hci0: BCM4345C0 (003.001.025) build 0000
[    7.744083] Bluetooth: hci0: BCM4345C0 'brcm/BCM4345C0.raspberrypi,5-model-b.hcd' Patch
[    7.881379] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[    8.419329] Bluetooth: hci0: BCM: features 0x2f
[    8.420711] Bluetooth: hci0: BCM43455 37.4MHz Raspberry Pi 3+-0190
[    8.420715] Bluetooth: hci0: BCM4345C0 (003.001.025) build 0382
[    8.421022] Bluetooth: hci0: BCM: Using default device address (43:45:c0:00:1f:ac)
[   10.816931] systemd-journald[344]: /var/log/journal/50403bee319f43f6bbf3e9439b1db4c5/user-1000.journal: Journal file uses a different sequence number ID, rotating.
[   11.430262] NET: Registered PF_QIPCRTR protocol family
[   11.503337] loop1: detected capacity change from 0 to 8
[  713.134438] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /axi/pcie@120000/rp1/i2s@a0000/status
[  713.145766] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/sound/compatible
[  713.155646] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/sound/simple-audio-card,format
[  713.166744] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/sound/simple-audio-card,name
[  713.177666] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/sound/status

### Additional context

I saw @pelwell fixed an audio issue https://github.com/raspberrypi/linux/pull/5725 and maybe it's related?

pelwell, 2024-05-13:
1. The "no matching platform" warning comes from using a version of the dtoverlay support that pre-dates the Pi 5. You can build and install an update (it's quick and easy) from https://github.com/raspberrypi/utils/tree/master/dtmerge, but I don't think it will change anything except the warning - it is just a warning, unless your overlay name has to be translated to a `-pi5` version.
2. I don't think this is the same cause as #5724 (fixed by #5725) because that bug resulted in errors in the kernel log - your kernel log has none.

Is your user in group `audio` (`groups`)? Does `sudo aplay -l` show anything?

Nate711, 2024-05-15:
Thanks for the helpful tips! What .dtb should I apply the max98357a.dtbo to? 
```
pi@ubuntu:~$ sudo adduser $USER audio
info: The user `pi' is already a member of `audio'.
pi@ubuntu:~$ sudo dtoverlay max98357a
DTOVERLAY[warn]: no matching platform found
pi@ubuntu:~$ sudo aplay -l
aplay: device_list:277: no soundcards found...
```
And side question, does the actual I2S max98357a chip need to be connected for the soundcard to display?

Nate711, 2024-05-21:
@pelwell Sorry to bug you, any ideas given the above information?

pelwell, 2024-05-21:
If you're still getting `DTOVERLAY[warn]: no matching platform found` then you haven't updated the dtmerge utilities.

pelwell, 2024-06-04:
```
$ git clone https://github.com/raspberrypi/utils.git
$ cd utils/dtmerge
```
Then follow these instructions: https://github.com/raspberrypi/utils/tree/master/dtmerge#readme

Nate711, 2024-06-11:
I was able to update my dtmerge utilities which fixed the issue of `DTOVERLAY[warn]: no matching platform found` appearing. As you guessed, it didn't actually fix the problem of no sound carts appearing. 

To try to isolate the issue, I made a Raspberry Pi OS Desktop 64-bit image and tried the max98357a overlay. As expected, if I either apply it with sudo dtoverlay max98357a or add to the config.txt file, I can see the sound card show up when I run `aplay -l`. 

To try to bridge the difference between my ubuntu image and the working pios image, I copied over the max98357a.dtbo overlay from the pios image to the ubuntu image. However, it did _not_ fix the issue -- no sound cards appear with `aplay -l`.

Nate711, 2024-07-04:
Upgrading to 6.8.0-1006.6 https://launchpad.net/ubuntu/+source/linux-raspi/6.8.0-1006.6 made everything work! The max98357a overlay applies without issue and I can hear audio out the speaker. 

I wasn't able to figure out which change made it work but would be happy if someone could figure it out.

Nate711, 2024-07-04:
Actually I still get a bug where sometimes the overlay loads correctly and other times I get an error... But at least it sometimes works.
```
[   18.169159] platform soc:sound: deferred probe pending: asoc-simple-card: parse error 
```

Nate711, 2024-07-04:
Using the hifiberry-dac overlay instead seems to work more reliably than the max98357a overlay! I haven't seen any of the previous `parser error` errors.

My config.txt
```
# dtoverlay=max98357a
dtoverlay=hifiberry-dac
```

closed 2024-07-04 by Nate711