raspberrypi/linux #6167: Kernel 6.6 broken with official 7'' display (at least in the Pi 4)

rafael2k, 2024-05-14:
### Describe the bug

Randomly (after reboots it is pretty consistent) the official 7'' DSI display does not initialize correctly.
I hope to unify bugs #6082 and #6078 in this one.

### Steps to reproduce the behaviour

Just boot Raspberry Pi OS on a Pi 4 with the official 7'' display. Reboot a couple of times (for me it always happens when rebooting) and I get garbled display.
First 6.6 kernels and 6.1 kernels does not present such issue.

### Device (s)

Raspberry Pi 4 Mod. B

### System

```
Raspberry Pi reference 2023-12-11
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 2acf7afcba7d11500313a7b93bb55a2aae20b2d6, stage2
```
```
vcgencmd version
Apr 17 2024 17:27:09 
Copyright (c) 2012 Broadcom
version 86ccc427f35fdc604edc511881cdf579df945fb4 (clean) (release) (start)
```
```
uname -a
Linux station1.hermes.radio 6.6.28+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.28-1+rpt1 (2024-04-22) aarch64 GNU/Linux
```

### Logs

dmesg relevant part:
```
[    2.420137] [drm] Initialized v3d 1.0.0 20180419 for fec00000.v3d on minor 0
[    2.449882] rtc-ds1307 22-0068: registered as rtc0
[    2.452162] rtc-ds1307 22-0068: setting system clock to 2024-05-14T19:03:51 UTC (1715713431)
[    2.516500] Console: switching to colour dummy device 80x25
[    2.554588] vc4-drm gpu: bound fe400000.hvs (ops vc4_hvs_ops [vc4])
[    2.556416] Registered IR keymap rc-cec
[    2.556621] rc rc0: vc4-hdmi-0 as /devices/platform/soc/fef00700.hdmi/rc/rc0
[    2.556816] input: vc4-hdmi-0 as /devices/platform/soc/fef00700.hdmi/rc/rc0/input0
[    2.557963] vc4_hdmi fef00700.hdmi: 'dmas' DT property is missing or empty, no HDMI audio
[    2.558016] vc4-drm gpu: bound fef00700.hdmi (ops vc4_hdmi_ops [vc4])
[    2.562232] Registered IR keymap rc-cec
[    2.562436] rc rc1: vc4-hdmi-1 as /devices/platform/soc/fef05700.hdmi/rc/rc1
[    2.562631] input: vc4-hdmi-1 as /devices/platform/soc/fef05700.hdmi/rc/rc1/input1
[    2.563156] vc4_hdmi fef05700.hdmi: 'dmas' DT property is missing or empty, no HDMI audio
[    2.563201] vc4-drm gpu: bound fef05700.hdmi (ops vc4_hdmi_ops [vc4])
[    2.564326] vc4-drm gpu: bound fe700000.dsi (ops vc4_dsi_ops [vc4])
[    2.564745] vc4-drm gpu: bound fe004000.txp (ops vc4_txp_ops [vc4])
[    2.565093] vc4-drm gpu: bound fe206000.pixelvalve (ops vc4_crtc_ops [vc4])
[    2.567770] vc4-drm gpu: bound fe207000.pixelvalve (ops vc4_crtc_ops [vc4])
[    2.568199] vc4-drm gpu: bound fe20a000.pixelvalve (ops vc4_crtc_ops [vc4])
[    2.568415] vc4-drm gpu: bound fe216000.pixelvalve (ops vc4_crtc_ops [vc4])
[    2.568735] vc4-drm gpu: bound fec12000.pixelvalve (ops vc4_crtc_ops [vc4])
[    2.573399] [drm] Initialized vc4 0.0.0 20140616 for gpu on minor 1
[    3.870689] vc4_dsi fe700000.dsi: transfer interrupt wait timeout
[    3.870705] vc4_dsi fe700000.dsi: instat: 0x00000000
[    3.870715] [drm:vc4_dsi_host_transfer [vc4]] *ERROR* DSI transfer failed, resetting: -110
[    4.086689] tc358762 fe700000.dsi.0: error initializing bridge (-110)
[    4.089361] Console: switching to colour frame buffer device 100x30
[    4.138737] vc4-drm gpu: [drm] fb0: vc4drmfb frame buffer device
```

rafael2k, 2024-05-14:
It seems 6.6.30-v8+ does not present the issue anymore. I wonder what was the regression.

6by9, 2024-05-14:
Why does opening yet another issue unify things?

Duplicates #6082 on which you've also commented, and you've stated it's fixed.

closed 2024-05-14 by 6by9

rafael2k, 2024-05-14:
Very kind from your side @6by9.