raspberrypi/linux #6170: kernel 6.6.30-v8+ strange system freeze

qrp73, 2024-05-15:
### Describe the bug

I catch it several times when switched from 6.6.28 to 6.6.30 (I switch with ```sudo rpi-update rpi-6.6.y```). The mouse cursor freezes, and after some time background sound player is stopped. When it happens I cannot connect with ssh, it asks the password, but then freezes.

Sometimes Alt+SysRq+I allows to restart the Wayfire. But this time it didn't helped. Complete freeze...

I tried to switch back to 6.6.28 and it didn't happens. Then I switched to 6.6.30 and after about half hour it happens again.

### Steps to reproduce the behaviour

I don't know how to reproduce it. It just happens after some runtime. 

### Device (s)

Raspberry Pi 4 Mod. B

### System

```
$ cat /etc/rpi-issue
Raspberry Pi reference 2023-09-22
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 40f37458ae7cadea1aec913ae10b5e7008ebce0a, stage4

$ vcgencmd version
Apr 17 2024 17:27:09 
Copyright (c) 2012 Broadcom
version 86ccc427f35fdc604edc511881cdf579df945fb4 (clean) (release) (start)

$ uname -a
Linux raspi 6.6.30-v8+ #1 SMP PREEMPT Tue May 14 13:53:14 UTC 2024 aarch64 GNU/Linux
```

### Logs

```
May 15 14:18:53 raspi kernel: cma: cma_alloc: linux,cma: alloc failed, req-size: 1024 pages, ret: -12
May 15 14:18:53 raspi kernel: cma: cma_alloc: linux,cma: alloc failed, req-size: 1024 pages, ret: -12
May 15 14:18:53 raspi kernel: cma: cma_alloc: linux,cma: alloc failed, req-size: 1024 pages, ret: -12
May 15 14:18:53 raspi kernel: ThreadPoolSingl: page allocation failure: order:10, mode:0xcc1(GFP_KERNEL|GFP_DMA), nodemask=(null),cpuset=/,mems_allowed=0
May 15 14:18:53 raspi kernel: CPU: 3 PID: 42371 Comm: ThreadPoolSingl Tainted: G         C         6.6.30-v8+ #1
May 15 14:18:53 raspi kernel: Hardware name: Raspberry Pi 4 Model B Rev 1.5 (DT)
May 15 14:18:53 raspi kernel: Call trace:
May 15 14:18:53 raspi kernel:  dump_backtrace+0x9c/0x100
May 15 14:18:53 raspi kernel:  show_stack+0x20/0x38
May 15 14:18:53 raspi kernel:  dump_stack_lvl+0x48/0x60
May 15 14:18:53 raspi kernel:  dump_stack+0x18/0x28
May 15 14:18:53 raspi kernel:  warn_alloc+0x11c/0x1a8
May 15 14:18:53 raspi kernel:  __alloc_pages+0xdbc/0xe28
May 15 14:18:53 raspi kernel:  __dma_direct_alloc_pages.constprop.0+0x16c/0x1d8
May 15 14:18:53 raspi kernel:  dma_direct_alloc+0xd4/0x300
May 15 14:18:53 raspi kernel:  dma_alloc_attrs+0x88/0x108
May 15 14:18:53 raspi kernel:  vb2_dc_alloc+0xa8/0x198 [videobuf2_dma_contig]
May 15 14:18:53 raspi kernel:  __vb2_queue_alloc+0x1a4/0x4a0 [videobuf2_common]
May 15 14:18:53 raspi kernel:  vb2_core_reqbufs+0x258/0x470 [videobuf2_common]
May 15 14:18:53 raspi kernel:  vb2_reqbufs+0x90/0xb8 [videobuf2_v4l2]
May 15 14:18:53 raspi kernel:  v4l2_m2m_ioctl_reqbufs+0x54/0x98 [v4l2_mem2mem]
May 15 14:18:53 raspi kernel:  v4l_reqbufs+0x58/0x78 [videodev]
May 15 14:18:53 raspi kernel:  __video_do_ioctl+0x194/0x400 [videodev]
May 15 14:18:53 raspi kernel:  video_usercopy+0x1e4/0x780 [videodev]
May 15 14:18:53 raspi kernel:  video_ioctl2+0x20/0x40 [videodev]
May 15 14:18:53 raspi kernel:  v4l2_ioctl+0x48/0x70 [videodev]
May 15 14:18:53 raspi kernel:  __arm64_sys_ioctl+0xb0/0x100
May 15 14:18:53 raspi kernel:  invoke_syscall+0x50/0x128
May 15 14:18:53 raspi kernel:  el0_svc_common.constprop.0+0x48/0xf8
May 15 14:18:53 raspi kernel:  do_el0_svc+0x28/0x40
May 15 14:18:53 raspi kernel:  el0_svc+0x40/0xf8
May 15 14:18:53 raspi kernel:  el0t_64_sync_handler+0x13c/0x158
May 15 14:18:53 raspi kernel:  el0t_64_sync+0x190/0x198
May 15 14:18:53 raspi kernel: Mem-Info:
May 15 14:18:53 raspi kernel: active_anon:266632 inactive_anon:822 isolated_anon:0
[... 50 lines cut ...]
May 15 14:20:34 raspi kernel: Call trace:
May 15 14:20:34 raspi kernel:  rpi_firmware_property_list+0x218/0x258
May 15 14:20:34 raspi kernel:  rpi_firmware_property+0x78/0x118
May 15 14:20:34 raspi kernel:  raspberrypi_fw_get_rate+0x4c/0x88
May 15 14:20:34 raspi kernel:  clk_recalc+0x40/0xb8
May 15 14:20:34 raspi kernel:  __clk_recalc_rates+0x44/0xc0
May 15 14:20:34 raspi kernel:  clk_set_rate_range_nolock.part.0+0x1f0/0x2e8
May 15 14:20:34 raspi kernel:  clk_set_min_rate+0x4c/0x108
May 15 14:20:34 raspi kernel:  vc4_atomic_commit_tail+0x23c/0x880 [vc4]
May 15 14:20:34 raspi kernel:  commit_tail+0xac/0x1a0 [drm_kms_helper]
May 15 14:20:34 raspi kernel:  commit_work+0x1c/0x30 [drm_kms_helper]
May 15 14:20:34 raspi kernel:  process_one_work+0x148/0x388
May 15 14:20:34 raspi kernel:  worker_thread+0x338/0x450
May 15 14:20:34 raspi kernel:  kthread+0x120/0x130
May 15 14:20:34 raspi kernel:  ret_from_fork+0x10/0x20
May 15 14:20:34 raspi kernel: ---[ end trace 0000000000000000 ]---
May 15 14:20:36 raspi kernel: hwmon hwmon1: Failed to get throttled (-110)
May 15 14:20:36 raspi kernel: bcm2835_mmal_vchiq: timed out waiting for sync completion
May 15 14:20:37 raspi kernel: raspberrypi-clk soc:firmware:clocks: Failed to change fw-clk-core frequency: -110
May 15 14:21:35 raspi kernel: sysrq: Kill All Tasks
```

### Additional context

3 user processes were running during issue: sdrpp, chromium-browser and vlc. Previously I catched it without sdrpp.

popcornmix, 2024-05-15:
Can you identify the exact update which caused this. See:
https://github.com/raspberrypi/rpi-firmware/commits/master

If you click on each commit the end of the url contains a git hash. Run
```sudo rpi-update <hash>```
to revert back to that version. Report the first version with this error.

You know it's between 6.6.28 and 6.6.30 so there aren't too many commits to check.

qrp73, 2024-05-16:
just catch it again. I will try different commits. But it's not easy, because this issue is floating and I cannot reproduce it with stable steps. 

This time it happens when I clocked on close tab button on chromium-browser:
```
May 16 08:53:41 raspi kernel: cma: cma_alloc: linux,cma: alloc failed, req-size: 1024 pages, ret: -12
May 16 08:53:41 raspi kernel: ThreadPoolSingl: page allocation failure: order:10, mode:0xcc1(GFP_KERNEL|GFP_DMA), nodemask=(null),cpuset=/,mems_allowed=0
May 16 08:53:41 raspi kernel: CPU: 1 PID: 98136 Comm: ThreadPoolSingl Tainted: G         C         6.6.30-v8+ #1
May 16 08:53:41 raspi kernel: Hardware name: Raspberry Pi 4 Model B Rev 1.5 (DT)
May 16 08:53:41 raspi kernel: Call trace:
May 16 08:53:41 raspi kernel:  dump_backtrace+0x9c/0x100
May 16 08:53:41 raspi kernel:  show_stack+0x20/0x38
May 16 08:53:41 raspi kernel:  dump_stack_lvl+0x48/0x60
May 16 08:53:41 raspi kernel:  dump_stack+0x18/0x28
May 16 08:53:41 raspi kernel:  warn_alloc+0x11c/0x1a8
May 16 08:53:41 raspi kernel:  __alloc_pages+0xdbc/0xe28
May 16 08:53:41 raspi kernel:  __dma_direct_alloc_pages.constprop.0+0x16c/0x1d8
May 16 08:53:41 raspi kernel:  dma_direct_alloc+0xd4/0x300
May 16 08:53:41 raspi kernel:  dma_alloc_attrs+0x88/0x108
May 16 08:53:41 raspi kernel:  vb2_dc_alloc+0xa8/0x198 [videobuf2_dma_contig]
May 16 08:53:41 raspi kernel:  __vb2_queue_alloc+0x1a4/0x4a0 [videobuf2_common]
May 16 08:53:41 raspi kernel:  vb2_core_reqbufs+0x258/0x470 [videobuf2_common]
May 16 08:53:41 raspi kernel:  vb2_reqbufs+0x90/0xb8 [videobuf2_v4l2]
May 16 08:53:41 raspi kernel:  v4l2_m2m_ioctl_reqbufs+0x54/0x98 [v4l2_mem2mem]
May 16 08:53:41 raspi kernel:  v4l_reqbufs+0x58/0x78 [videodev]
May 16 08:53:41 raspi kernel:  __video_do_ioctl+0x194/0x400 [videodev]
May 16 08:53:41 raspi kernel:  video_usercopy+0x1e4/0x780 [videodev]
May 16 08:53:41 raspi kernel:  video_ioctl2+0x20/0x40 [videodev]
May 16 08:53:41 raspi kernel:  v4l2_ioctl+0x48/0x70 [videodev]
May 16 08:53:41 raspi kernel:  __arm64_sys_ioctl+0xb0/0x100
May 16 08:53:41 raspi kernel:  invoke_syscall+0x50/0x128
May 16 08:53:41 raspi kernel:  el0_svc_common.constprop.0+0x48/0xf8
May 16 08:53:41 raspi kernel:  do_el0_svc+0x28/0x40
May 16 08:53:41 raspi kernel:  el0_svc+0x40/0xf8
May 16 08:53:41 raspi kernel:  el0t_64_sync_handler+0x13c/0x158
May 16 08:53:41 raspi kernel:  el0t_64_sync+0x190/0x198
May 16 08:53:41 raspi kernel: Mem-Info:
May 16 08:53:41 raspi kernel: active_anon:253404 inactive_anon:8102 isolated_anon:25
                               active_file:280053 inactive_file:168585 isolated_file:0
                               unevictable:84368 dirty:1451 writeback:0
                               slab_reclaimable:13132 slab_unreclaimable:11041
                               mapped:122612 shmem:104701 pagetables:5039
                               sec_pagetables:0 bounce:0
                               kernel_misc_reclaimable:0
                               free:21639 free_pcp:163 free_cma:5817
May 16 08:53:41 raspi kernel: Node 0 active_anon:1013616kB inactive_anon:32408kB active_file:1120212kB inactive_file:674340kB unevictable:337472kB isolated(a>
May 16 08:53:41 raspi kernel: DMA free:65088kB boost:0kB min:3728kB low:4660kB high:5592kB reserved_highatomic:0KB active_anon:193640kB inactive_anon:8kB act>
May 16 08:53:41 raspi kernel: lowmem_reserve[]: 0 2928 2928 2928
May 16 08:53:41 raspi kernel: DMA: 516*4kB (UMEC) 307*8kB (UMEC) 129*16kB (UMEC) 50*32kB (UMEC) 45*64kB (UMEC) 91*128kB (UC) 68*256kB (UEC) 21*512kB (UC) 6*1>
May 16 08:53:41 raspi kernel: 553376 total pagecache pages
May 16 08:53:41 raspi kernel: 0 pages in swap cache
May 16 08:53:41 raspi kernel: Free swap  = 0kB
May 16 08:53:41 raspi kernel: Total swap = 0kB
May 16 08:53:41 raspi kernel: 1012736 pages RAM
May 16 08:53:41 raspi kernel: 0 pages HighMem/MovableOnly
May 16 08:53:41 raspi kernel: 41971 pages reserved
May 16 08:53:41 raspi kernel: 131072 pages cma reserved
May 16 08:53:41 raspi kernel: bcm2835-codec bcm2835-codec: dma alloc of size 4194304 failed
May 16 08:53:49 raspi kernel: cma: cma_alloc: linux,cma: alloc failed, req-size: 1024 pages, ret: -12
May 16 08:53:56 raspi kernel: cma: cma_alloc: linux,cma: alloc failed, req-size: 1024 pages, ret: -12
May 16 08:53:56 raspi kernel: cma: cma_alloc: linux,cma: alloc failed, req-size: 1024 pages, ret: -12
May 16 08:53:56 raspi kernel: cma: cma_alloc: linux,cma: alloc failed, req-size: 1024 pages, ret: -12
May 16 08:53:56 raspi kernel: ThreadPoolSingl: page allocation failure: order:10, mode:0xcc1(GFP_KERNEL|GFP_DMA), nodemask=(null),cpuset=/,mems_allowed=0
May 16 08:53:56 raspi kernel: CPU: 2 PID: 98166 Comm: ThreadPoolSingl Tainted: G         C         6.6.30-v8+ #1
May 16 08:53:56 raspi kernel: Hardware name: Raspberry Pi 4 Model B Rev 1.5 (DT)
May 16 08:53:56 raspi kernel: Call trace:
May 16 08:53:56 raspi kernel:  dump_backtrace+0x9c/0x100
May 16 08:53:56 raspi kernel:  show_stack+0x20/0x38
May 16 08:53:56 raspi kernel:  dump_stack_lvl+0x48/0x60
May 16 08:53:56 raspi kernel:  dump_stack+0x18/0x28
May 16 08:53:56 raspi kernel:  warn_alloc+0x11c/0x1a8
[... 232 lines cut ...]
May 16 08:57:47 raspi kernel: x5 : ffffff80fb76bd48 x4 : 0000000000000000 x3 : 0000000000000027
May 16 08:57:47 raspi kernel: x2 : 0000000000000000 x1 : 0000000000000000 x0 : ffffff8070c5dc40
May 16 08:57:47 raspi kernel: Call trace:
May 16 08:57:47 raspi kernel:  rpi_firmware_property_list+0x218/0x258
[... cut ...]
May 16 08:57:47 raspi kernel:  ret_from_fork+0x10/0x20
May 16 08:57:47 raspi kernel: ---[ end trace 0000000000000000 ]---
May 16 08:57:48 raspi kernel: raspberrypi-clk soc:firmware:clocks: Failed to change fw-clk-core frequency: -110
May 16 08:57:49 raspi kernel: bcm2835_mmal_vchiq: timed out waiting for sync completion
May 16 08:57:49 raspi kernel: hwmon hwmon1: Failed to get throttled (-110)
May 16 08:57:57 raspi kernel: sysrq: Kill All Tasks
```

I see there are a lot of errors, but system freeze happened at about 08:57:45

qrp73, 2024-05-16:
I tried to install ```sudo rpi-update 42312de```, still testing, but found some issues.

I had some short freezes (for about 1-2 sec) and check the log, the last messages are:
```
[ 9962.991318] vc4-drm gpu: swiotlb buffer is full (sz: 524288 bytes), total 32768 (slots), used 76 (slots)
[ 9963.511759] vc4-drm gpu: swiotlb buffer is full (sz: 524288 bytes), total 32768 (slots), used 76 (slots)
[ 9971.347461] bcm2835_mmal_vchiq: timed out waiting for sync completion
[ 9971.347499] bcm2835_mmal_vchiq: vchiq_mmal_component_init: failed to create component -62 (Not enough GPU mem?)
[ 9971.347511] bcm2835-codec bcm2835-codec: bcm2835_codec_create_component: failed to create component ril.video_decode
[ 9991.382612] vc4-drm gpu: swiotlb buffer is full (sz: 376832 bytes), total 32768 (slots), used 1600 (slots)
[ 9991.517494] vc4-drm gpu: swiotlb buffer is full (sz: 376832 bytes), total 32768 (slots), used 1600 (slots)
```

but before that it has a lot of errors, here is some of them:
```
[ 8724.347982] bcm2835_mmal_vchiq: timed out waiting for sync completion
[ 8724.348016] bcm2835-codec bcm2835-codec: bcm2835_codec_stop_streaming: Failed disabling i/p port, ret -62
[ 8726.368001] bcm2835-codec bcm2835-codec: bcm2835_codec_flush_buffers: Timeout waiting for buffers to be returned - 8 outstanding
[ 8726.368146] ------------[ cut here ]------------
[ 8726.368149] WARNING: CPU: 3 PID: 21695 at drivers/media/common/videobuf2/videobuf2-core.c:2024 __vb2_queue_cancel+0x220/0x2a0 [videobuf2_common]
[ 8726.368184] Modules linked in: ntfs3 rfcomm snd_seq_dummy snd_hrtimer snd_seq cmac algif_hash aes_arm64 aes_generic algif_skcipher af_alg bnep brcmfmac_wcc brcmfmac brcmutil joydev binfmt_misc cfg80211 hci_uart btbcm vc4 snd_usb_audio bluetooth raspberrypi_hwmon snd_hwdep snd_soc_hdmi_codec snd_usbmidi_lib bcm2835_v4l2(C) drm_display_helper snd_rawmidi cec bcm2835_isp(C) bcm2835_codec(C) rpivid_hevc(C) drm_dma_helper ecdh_generic snd_seq_device bcm2835_mmal_vchiq(C) sg drm_kms_helper ecc v4l2_mem2mem i2c_brcmstb v3d rfkill snd_soc_core i2c_bcm2835 videobuf2_dma_contig libaes videobuf2_vmalloc snd_bcm2835(C) snd_compress videobuf2_memops snd_pcm_dmaengine videobuf2_v4l2 gpu_sched snd_pcm videodev vc_sm_cma(C) videobuf2_common drm_shmem_helper snd_timer raspberrypi_gpiomem mc snd nvmem_rmem uio_pdrv_genirq uio i2c_dev drm fuse drm_panel_orientation_quirks dm_mod backlight ip_tables x_tables ipv6
[ 8726.368318] CPU: 3 PID: 21695 Comm: ThreadPoolSingl Tainted: G        WC         6.6.29-v8+ #1760
[ 8726.368325] Hardware name: Raspberry Pi 4 Model B Rev 1.5 (DT)
[ 8726.368328] pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[ 8726.368333] pc : __vb2_queue_cancel+0x220/0x2a0 [videobuf2_common]
[ 8726.368347] lr : __vb2_queue_cancel+0x38/0x2a0 [videobuf2_common]
[ 8726.368360] sp : ffffffc083393ae0
[ 8726.368363] x29: ffffffc083393ae0 x28: ffffffdbb742ce18 x27: ffffff80a23d2000
[ 8726.368372] x26: ffffff80442f8530 x25: 0000000000000000 x24: 0000000040045613
[ 8726.368381] x23: 0000000000000000 x22: ffffff8008d9e298 x21: ffffff8008d9e298
[ 8726.368389] x20: 000000000000000a x19: ffffff8008d9e298 x18: ffffffffffffffff
[ 8726.368395] x17: 756f656d6954203a x16: ffffffdc2b2d32a8 x15: 6873756c665f6365
[ 8726.368402] x14: 0000000000000001 x13: ffffff8044240028 x12: 0000000000000000
[ 8726.368411] x11: ffffff8005417260 x10: ffffff8005417230 x9 : ffffffdc2b329e68
[ 8726.368417] x8 : ffffff8005417258 x7 : 0000000000000001 x6 : fffffffffffffffc
[ 8726.368423] x5 : 0000000041001000 x4 : 0000000000150013 x3 : ffffff8045535e40
[ 8726.368429] x2 : 0000000000000000 x1 : 0000000000000000 x0 : 0000000000000008
[ 8726.368437] Call trace:
[ 8726.368443]  __vb2_queue_cancel+0x220/0x2a0 [videobuf2_common]
[ 8726.368457]  vb2_core_streamoff+0x2c/0xe0 [videobuf2_common]
[ 8726.368470]  vb2_streamoff+0x24/0x80 [videobuf2_v4l2]
[ 8726.368499]  v4l2_m2m_streamoff+0x50/0x160 [v4l2_mem2mem]
[ 8726.368534]  v4l2_m2m_ioctl_streamoff+0x20/0x38 [v4l2_mem2mem]
[ 8726.368546]  v4l_streamoff+0x2c/0x40 [videodev]
[ 8726.368665]  __video_do_ioctl+0x194/0x400 [videodev]
[ 8726.368693]  video_usercopy+0x1e4/0x780 [videodev]
[ 8726.368721]  video_ioctl2+0x20/0x40 [videodev]
[ 8726.368750]  v4l2_ioctl+0x48/0x70 [videodev]
[ 8726.368778]  __arm64_sys_ioctl+0xb0/0x100
[ 8726.368803]  invoke_syscall+0x50/0x128
[ 8726.368816]  el0_svc_common.constprop.0+0x48/0xf8
[ 8726.368822]  do_el0_svc+0x28/0x40
[ 8726.368826]  el0_svc+0x40/0xf8
[ 8726.368840]  el0t_64_sync_handler+0x13c/0x158
[ 8726.368847]  el0t_64_sync+0x190/0x198
[ 8726.368854] ---[ end trace 0000000000000000 ]---
[ 8726.368863] videobuf2_common: driver bug: stop_streaming operation is leaving buf 00000000cdf0c294 in active state
[ 8726.368878] videobuf2_common: driver bug: stop_streaming operation is leaving buf 00000000266bf4a1 in active state
[ 8726.368881] videobuf2_common: driver bug: stop_streaming operation is leaving buf 00000000dfd94176 in active state
[ 8726.368887] videobuf2_common: driver bug: stop_streaming operation is leaving buf 0000000084ff8e03 in active state
[ 8726.368891] videobuf2_common: driver bug: stop_streaming operation is leaving buf 0000000003024712 in active state
[ 8726.368894] videobuf2_common: driver bug: stop_streaming operation is leaving buf 0000000018a71992 in active state
[ 8726.368900] videobuf2_common: driver bug: stop_streaming operation is leaving buf 0000000038cfb820 in active state
[... cut ...]
1869178.725: e=7) failed�Fqm
1818324.594: 291328, in_low_region=0, mode=7) failed@�wm
1919905.824: uffer header callback not defined)��m
1986355.060: : event lost on port 1,0 (buffer header callback not defined)�"�m
539766.845: failed@�m
1818324.333: ent_send: event lost on port 1,0 (buffer header callback not defined)Ԩm
1918859.119:  mode=7) failed0K�m
1953656.688: ffer header callback not defined)�w�m
1763716.152: gion=0, mode=7) failed >�m
544108.320:  (buffer header callback not defined)�\�m
```

qrp73, 2024-05-16:
I found that there is huge memory leak in firefox, so out of memory may be the reason for this issue. 
But I'm not sure why it cannot kill all processes and restart wayfire.
Needs more testing.

qrp73, 2024-05-25:
catch the issue several times with ```Linux raspi 6.6.31-v8+ #1764 SMP PREEMPT Mon May 20 15:26:28 BST 2024 aarch64 GNU/Linux```

The system is freezed and don't respond even on ssh connections. Alt+SysRq+E don't works. It happens when I click on close tab in chromium-browser.

```
May 25 22:40:57 raspi kernel: cma_alloc: 105 callbacks suppressed
May 25 22:40:57 raspi kernel: cma: cma_alloc: linux,cma: alloc failed, req-size: 1024 pages, ret: -12
May 25 22:40:57 raspi kernel: cma: cma_alloc: linux,cma: alloc failed, req-size: 1024 pages, ret: -12
May 25 22:40:57 raspi kernel: cma: cma_alloc: linux,cma: alloc failed, req-size: 1024 pages, ret: -12
May 25 22:40:57 raspi kernel: cma: cma_alloc: linux,cma: alloc failed, req-size: 1024 pages, ret: -12
May 25 22:41:02 raspi kernel: cma: cma_alloc: linux,cma: alloc failed, req-size: 1024 pages, ret: -12
May 25 22:41:02 raspi kernel: cma: cma_alloc: linux,cma: alloc failed, req-size: 1024 pages, ret: -12
May 25 22:41:02 raspi kernel: cma: cma_alloc: linux,cma: alloc failed, req-size: 1024 pages, ret: -12
May 25 22:41:02 raspi kernel: cma: cma_alloc: linux,cma: alloc failed, req-size: 1024 pages, ret: -12
May 25 22:41:02 raspi kernel: cma: cma_alloc: linux,cma: alloc failed, req-size: 1024 pages, ret: -12
May 25 22:41:02 raspi kernel: cma: cma_alloc: linux,cma: alloc failed, req-size: 1024 pages, ret: -12
May 25 22:41:02 raspi kernel: warn_alloc: 4 callbacks suppressed
May 25 22:41:02 raspi kernel: ThreadPoolSingl: page allocation failure: order:10, mode:0xcc1(GFP_KERNEL|GFP_DMA), nodemask=(null),cpuset=/,mems_allowed=0
May 25 22:41:02 raspi kernel: CPU: 2 PID: 138261 Comm: ThreadPoolSingl Tainted: G        WC         6.6.31-v8+ #1764
May 25 22:41:02 raspi kernel: Hardware name: Raspberry Pi 4 Model B Rev 1.5 (DT)
May 25 22:41:02 raspi kernel: Call trace:
May 25 22:41:02 raspi kernel:  dump_backtrace+0x9c/0x100
May 25 22:41:02 raspi kernel:  show_stack+0x20/0x38
May 25 22:41:02 raspi kernel:  dump_stack_lvl+0x48/0x60
May 25 22:41:02 raspi kernel:  dump_stack+0x18/0x28
May 25 22:41:02 raspi kernel:  warn_alloc+0x11c/0x1a8
May 25 22:41:02 raspi kernel:  __alloc_pages+0xdbc/0xe28
May 25 22:41:02 raspi kernel:  __dma_direct_alloc_pages.constprop.0+0x16c/0x1d8
May 25 22:41:02 raspi kernel:  dma_direct_alloc+0xd4/0x300
May 25 22:41:02 raspi kernel:  dma_alloc_attrs+0x88/0x108
May 25 22:41:02 raspi kernel:  vb2_dc_alloc+0xa8/0x198 [videobuf2_dma_contig]
May 25 22:41:02 raspi kernel:  __vb2_queue_alloc+0x1a4/0x4a0 [videobuf2_common]
May 25 22:41:02 raspi kernel:  vb2_core_reqbufs+0x258/0x470 [videobuf2_common]
May 25 22:41:02 raspi kernel:  vb2_reqbufs+0x90/0xb8 [videobuf2_v4l2]
May 25 22:41:02 raspi kernel:  v4l2_m2m_ioctl_reqbufs+0x54/0x98 [v4l2_mem2mem]
May 25 22:41:02 raspi kernel:  v4l_reqbufs+0x58/0x78 [videodev]
May 25 22:41:02 raspi kernel:  __video_do_ioctl+0x194/0x400 [videodev]
May 25 22:41:02 raspi kernel:  video_usercopy+0x1e4/0x780 [videodev]
May 25 22:41:02 raspi kernel:  video_ioctl2+0x20/0x40 [videodev]
May 25 22:41:02 raspi kernel:  v4l2_ioctl+0x48/0x70 [videodev]
May 25 22:41:02 raspi kernel:  __arm64_sys_ioctl+0xb0/0x100
May 25 22:41:02 raspi kernel:  invoke_syscall+0x50/0x128
May 25 22:41:02 raspi kernel:  el0_svc_common.constprop.0+0x48/0xf8
May 25 22:41:02 raspi kernel:  do_el0_svc+0x28/0x40
May 25 22:41:02 raspi kernel:  el0_svc+0x40/0xf8
May 25 22:41:02 raspi kernel:  el0t_64_sync_handler+0x13c/0x158
May 25 22:41:02 raspi kernel:  el0t_64_sync+0x190/0x198
May 25 22:41:02 raspi kernel: Mem-Info:
May 25 22:41:02 raspi kernel: active_anon:132946 inactive_anon:102507 isolated_anon:0
                               active_file:145804 inactive_file:208771 isolated_file:0
                               unevictable:69566 dirty:1382 writeback:0
                               slab_reclaimable:14608 slab_unreclaimable:11834
                               mapped:110352 shmem:84325 pagetables:4766
                               sec_pagetables:0 bounce:0
                               kernel_misc_reclaimable:0
                               free:137639 free_pcp:131 free_cma:9970
May 25 22:41:02 raspi kernel: Node 0 active_anon:531784kB inactive_anon:410028kB active_file:583216kB inactive_file:835084kB unevictable:278264kB isolated(an>
May 25 22:41:02 raspi kernel: DMA free:206968kB boost:0kB min:3660kB low:4572kB high:5484kB reserved_highatomic:0KB active_anon:2268kB inactive_anon:70560kB >
May 25 22:41:02 raspi kernel: lowmem_reserve[]: 0 2928 2928 2928
May 25 22:41:02 raspi kernel: DMA: 918*4kB (UMEC) 795*8kB (UMEC) 728*16kB (UMEC) 610*32kB (UMEC) 410*64kB (UMC) 276*128kB (UMEC) 148*256kB (UMEC) 70*512kB (U>
May 25 22:41:02 raspi kernel: 438863 total pagecache pages
May 25 22:41:02 raspi kernel: 0 pages in swap cache
May 25 22:41:02 raspi kernel: Free swap  = 0kB
May 25 22:41:02 raspi kernel: Total swap = 0kB
May 25 22:41:02 raspi kernel: 1007616 pages RAM
May 25 22:41:02 raspi kernel: 0 pages HighMem/MovableOnly
May 25 22:41:02 raspi kernel: 41962 pages reserved
May 25 22:41:02 raspi kernel: 131072 pages cma reserved
May 25 22:41:02 raspi kernel: bcm2835-codec bcm2835-codec: dma alloc of size 4194304 failed
May 25 22:41:04 raspi kernel: cma: cma_alloc: linux,cma: alloc failed, req-size: 1024 pages, ret: -12
[... 24 lines cut ...]
May 25 22:41:10 raspi kernel:  rpi_firmware_property_list+0x218/0x258
May 25 22:41:10 raspi kernel:  rpi_firmware_property+0x78/0x118
May 25 22:41:10 raspi kernel:  raspberrypi_fw_get_rate+0x4c/0x88
[... cut ...]
May 25 22:41:12 raspi kernel: bcm2835_mmal_vchiq: timed out waiting for sync completion
May 25 22:41:12 raspi kernel: hwmon hwmon1: Failed to get throttled (-110)
May 25 22:41:18 raspi kernel: sysrq: Terminate All Tasks
May 25 22:41:18 raspi systemd-journald[276]: Received SIGTERM.
May 25 22:41:18 raspi avahi-daemon[669]: Got SIGTERM, quitting.
May 25 22:41:18 raspi systemd[1]: cron.service: Deactivated successfully.
May 25 22:41:18 raspi bluetoothd[768]: Terminating
May 25 22:41:18 raspi org.freedesktop.secrets[1463]: discover_other_daemon: 1
```

qrp73, 2024-05-25:
just catch it again, but this time I was able to connect via ssh and see the logs before the system stopped responding completely.

This time I reproduced it with reboot and opening chromium-browser with one tab (tg.me page), I read page for some time and then opened new tab and clicked on close button for old tab, at this moment Wayland desktop was freezes, but I connected via ssh and take logs:

sudo vclog -m:
```
1476722.338: alloc_compact_internal(space=3291328, in_low_region=0, mode=7) failed
1476722.378: mmal: mmal_port_event_send: event lost on port 1,0 (buffer header callback not defined)
1476722.621: alloc_compact_internal(space=3291328, in_low_region=0, mode=7) failed
1476722.664: mmal: mmal_port_event_send: event lost on port 1,0 (buffer header callback not defined)
1476722.874: alloc_compact_internal(space=3291328, in_low_region=0, mode=7) failed
1476722.909: mmal: mmal_port_event_send: event lost on port 1,0 (buffer header callback not defined)
1476723.115: alloc_compact_internal(space=3291328, in_low_region=0, mode=7) failed
1476723.152: mmal: mmal_port_event_send: event lost on port 1,0 (buffer header callback not defined)
1476727.195: alloc_compact_internal(space=3291328, in_low_region=0, mode=7) failed
1476727.233: mmal: mmal_port_event_send: event lost on port 1,0 (buffer header callback not defined)
1476731.550: alloc_compact_internal(space=3291328, in_low_region=0, mode=7) failed
<...skip thousands of the same alloc_compact error messages...>
1483624.136: alloc_compact_internal(space=3291328, in_low_region=0, mode=7) failed
1483624.167: mmal: mmal_port_event_send: event lost on port 1,0 (buffer header callback not defined)
1483641.663: alloc_compact_internal(space=3291328, in_low_region=0, mode=7) failed
1483641.701: mmal: mmal_port_event_send: event lost on port 1,0 (buffer header callback not defined)
1483641.901: alloc_compact_internal(space=3291328, in_low_region=0, mode=7) failed
1483641.932: mmal: mmal_port_event_send: event lost on port 1,0 (buffer header callback not defined)
1483691.042: smservice: sm_remove_resource: failed to find resource (0) in resource list
1483691.063: smservice: sm_remove_resource: failed to find resource (0) in resource list
1483691.081: smservice: sm_remove_resource: failed to find resource (0) in resource list
1483691.102: smservice: sm_remove_resource: failed to find resource (0) in resource list
1483691.119: smservice: sm_remove_resource: failed to find resource (0) in resource list
1483691.141: smservice: sm_remove_resource: failed to find resource (0) in resource list
1483691.161: smservice: sm_remove_resource: failed to find resource (0) in resource list
1483691.182: smservice: sm_remove_resource: failed to find resource (0) in resource list
1483691.202: smservice: sm_remove_resource: failed to find resource (0) in resource list
1483691.224: smservice: sm_remove_resource: failed to find resource (0) in resource list
1483691.248: smservice: sm_remove_resource: failed to find resource (0) in resource list
1483691.269: smservice: sm_remove_resource: failed to find resource (0) in resource list
1483691.288: smservice: sm_remove_resource: failed to find resource (0) in resource list
1483691.309: smservice: sm_remove_resource: failed to find resource (0) in resource list
1483691.328: smservice: sm_remove_resource: failed to find resource (0) in resource list
1483691.349: smservice: sm_remove_resource: failed to find resource (0) in resource list
1483691.368: smservice: sm_remove_resource: failed to find resource (0) in resource list
1483691.405: smservice: sm_remove_resource: failed to find resource (0) in resource list
1483691.440: smservice: sm_remove_resource: failed to find resource (0) in resource list
1483691.460: smservice: sm_remove_resource: failed to find resource (0) in resource list
1483691.478: smservice: sm_remove_resource: failed to find resource (0) in resource list
1483691.499: smservice: sm_remove_resource: failed to find resource (0) in resource list
1483691.518: smservice: sm_remove_resource: failed to find resource (0) in resource list
1483691.538: smservice: sm_remove_resource: failed to find resource (0) in resource list
1483691.555: smservice: sm_remove_resource: failed to find resource (0) in resource list
1483691.576: smservice: sm_remove_resource: failed to find resource (0) in resource list
```

dmesg:
```
[ 1090.500290] vc4-drm gpu: swiotlb buffer is full (sz: 581632 bytes), total 32768 (slots), used 0 (slots)
[ 1090.504107] vc4-drm gpu: swiotlb buffer is full (sz: 843776 bytes), total 32768 (slots), used 0 (slots)
[ 1090.508205] vc4-drm gpu: swiotlb buffer is full (sz: 1048576 bytes), total 32768 (slots), used 28 (slots)
[ 1090.513071] vc4-drm gpu: swiotlb buffer is full (sz: 581632 bytes), total 32768 (slots), used 0 (slots)
[ 1091.079446] vc4-drm gpu: swiotlb buffer is full (sz: 843776 bytes), total 32768 (slots), used 0 (slots)
[ 1091.089192] vc4-drm gpu: swiotlb buffer is full (sz: 1048576 bytes), total 32768 (slots), used 28 (slots)
[ 1113.275010] vc4-drm gpu: swiotlb buffer is full (sz: 524288 bytes), total 32768 (slots), used 1596 (slots)
[ 1113.968446] vc4-drm gpu: swiotlb buffer is full (sz: 524288 bytes), total 32768 (slots), used 1596 (slots)
[ 1446.670829] cma: cma_alloc: linux,cma: alloc failed, req-size: 1024 pages, ret: -12
[ 1446.672935] cma: cma_alloc: linux,cma: alloc failed, req-size: 1024 pages, ret: -12
[ 1451.478444] cma: cma_alloc: linux,cma: alloc failed, req-size: 1024 pages, ret: -12
[... cut ...]
[ 1692.663878]  wait_for_completion_timeout+0x7c/0x158
[ 1692.663882]  mbox_send_message+0xc0/0x158
[ 1692.663886]  rpi_firmware_property_list+0xf4/0x258
[ 1692.663889]  vcio_user_property_list+0xd4/0x210
[ 1692.663894]  vcio_device_ioctl+0x34/0x58
[ 1692.663898]  __arm64_sys_ioctl+0xb0/0x100
[ 1692.663902]  invoke_syscall+0x50/0x128
[ 1692.663907]  el0_svc_common.constprop.0+0x48/0xf8
[ 1692.663912]  do_el0_svc+0x28/0x40
[ 1692.663917]  el0_svc+0x40/0xf8
[ 1692.663922]  el0t_64_sync_handler+0x13c/0x158
[ 1692.663924]  el0t_64_sync+0x190/0x198
```

qrp73, 2024-06-03:
after latest update to 6.6.31 it happens again:
```
Jun 04 00:35:54 raspi kernel: cma_alloc: 11 callbacks suppressed
Jun 04 00:35:54 raspi kernel: cma: cma_alloc: linux,cma: alloc failed, req-size: 1024 pages, ret: -16
Jun 04 00:35:54 raspi kernel: ThreadPoolSingl: page allocation failure: order:10, mode:0xcc1(GFP_KERNEL|GFP_DMA), nodemask=(null),cpuset=/,mems_allowed=0
Jun 04 00:35:54 raspi kernel: CPU: 1 PID: 352523 Comm: ThreadPoolSingl Tainted: G        WC         6.6.31+rpt-rpi-v8 #1  Debian 1:6.6.31-1+rpt1
Jun 04 00:35:54 raspi kernel: Hardware name: Raspberry Pi 4 Model B Rev 1.5 (DT)
Jun 04 00:35:54 raspi kernel: Call trace:
Jun 04 00:35:54 raspi kernel:  dump_backtrace+0xa0/0x100
Jun 04 00:35:54 raspi kernel:  show_stack+0x20/0x38
Jun 04 00:35:54 raspi kernel:  dump_stack_lvl+0x48/0x60
Jun 04 00:35:54 raspi kernel:  dump_stack+0x18/0x28
Jun 04 00:35:54 raspi kernel:  warn_alloc+0x128/0x1b8
Jun 04 00:35:54 raspi kernel:  __alloc_pages+0xdc0/0xe68
Jun 04 00:35:54 raspi kernel:  __dma_direct_alloc_pages.constprop.0+0x15c/0x238
Jun 04 00:35:54 raspi kernel:  dma_direct_alloc+0x78/0x310
Jun 04 00:35:54 raspi kernel:  dma_alloc_attrs+0x8c/0x108
Jun 04 00:35:54 raspi kernel:  vb2_dc_alloc+0xa8/0x188 [videobuf2_dma_contig]
Jun 04 00:35:54 raspi kernel:  __vb2_queue_alloc+0x1cc/0x4a8 [videobuf2_common]
Jun 04 00:35:54 raspi kernel:  vb2_core_reqbufs+0x244/0x470 [videobuf2_common]
Jun 04 00:35:54 raspi kernel:  vb2_reqbufs+0x90/0xb8 [videobuf2_v4l2]
Jun 04 00:35:54 raspi kernel:  v4l2_m2m_ioctl_reqbufs+0x54/0x98 [v4l2_mem2mem]
Jun 04 00:35:54 raspi kernel:  v4l_reqbufs+0x58/0x78 [videodev]
Jun 04 00:35:54 raspi kernel:  __video_do_ioctl+0x170/0x3e0 [videodev]
Jun 04 00:35:54 raspi kernel:  video_usercopy+0x208/0x770 [videodev]
Jun 04 00:35:54 raspi kernel:  video_ioctl2+0x20/0x40 [videodev]
Jun 04 00:35:54 raspi kernel:  v4l2_ioctl+0x48/0x70 [videodev]
Jun 04 00:35:54 raspi kernel:  __arm64_sys_ioctl+0xb4/0x100
Jun 04 00:35:54 raspi kernel:  invoke_syscall+0x50/0x128
Jun 04 00:35:54 raspi kernel:  el0_svc_common.constprop.0+0x48/0xf0
Jun 04 00:35:54 raspi kernel:  do_el0_svc+0x24/0x38
Jun 04 00:35:54 raspi kernel:  el0_svc+0x40/0xe8
Jun 04 00:35:54 raspi kernel:  el0t_64_sync_handler+0x100/0x130
Jun 04 00:35:54 raspi kernel:  el0t_64_sync+0x190/0x198
Jun 04 00:35:54 raspi kernel: Mem-Info:
Jun 04 00:35:54 raspi kernel: active_anon:341433 inactive_anon:58266 isolated_anon:0
                               active_file:204163 inactive_file:92138 isolated_file:0
                               unevictable:90421 dirty:2384 writeback:0
                               slab_reclaimable:15666 slab_unreclaimable:13238
                               mapped:134485 shmem:110979 pagetables:6680
                               sec_pagetables:0 bounce:0
                               kernel_misc_reclaimable:0
                               free:17601 free_pcp:0 free_cma:1471
Jun 04 00:35:54 raspi kernel: Node 0 active_anon:1365732kB inactive_anon:233064kB active_file:816652kB inactive_file:368552kB unevictable:361684kB isolated(a>
Jun 04 00:35:54 raspi kernel: DMA free:28456kB boost:0kB min:3660kB low:4572kB high:5484kB reserved_highatomic:4096KB active_anon:215092kB inactive_anon:3332>
Jun 04 00:35:54 raspi kernel: lowmem_reserve[]: 0 2928 2928 2928
Jun 04 00:35:54 raspi kernel: DMA: 313*4kB (UMEHC) 179*8kB (UMEHC) 46*16kB (UMEHC) 54*32kB (UMEHC) 39*64kB (UMEHC) 10*128kB (HC) 11*256kB (UHC) 9*512kB (UC) >
Jun 04 00:35:54 raspi kernel: 406540 total pagecache pages
Jun 04 00:35:54 raspi kernel: 0 pages in swap cache
Jun 04 00:35:54 raspi kernel: Free swap  = 0kB
Jun 04 00:35:54 raspi kernel: Total swap = 0kB
Jun 04 00:35:54 raspi kernel: 1007616 pages RAM
Jun 04 00:35:54 raspi kernel: 0 pages HighMem/MovableOnly
Jun 04 00:35:54 raspi kernel: 41961 pages reserved
Jun 04 00:35:54 raspi kernel: 131072 pages cma reserved
Jun 04 00:35:54 raspi kernel: bcm2835-codec bcm2835-codec: dma alloc of size 4194304 failed
Jun 04 00:36:00 raspi kernel: ------------[ cut here ]------------
Jun 04 00:36:00 raspi kernel: Firmware transaction timeout
Jun 04 00:36:00 raspi kernel: WARNING: CPU: 3 PID: 329121 at drivers/firmware/raspberrypi.c:68 rpi_firmware_property_list+0x254/0x278
Jun 04 00:36:00 raspi kernel: Modules linked in: ntfs3 rfcomm snd_seq_dummy snd_hrtimer snd_seq cmac algif_hash aes_arm64 aes_generic algif_skcipher af_alg b>
Jun 04 00:36:00 raspi kernel: CPU: 3 PID: 329121 Comm: kworker/u12:2 Tainted: G        WC         6.6.31+rpt-rpi-v8 #1  Debian 1:6.6.31-1+rpt1
Jun 04 00:36:00 raspi kernel: Hardware name: Raspberry Pi 4 Model B Rev 1.5 (DT)
Jun 04 00:36:00 raspi kernel: Workqueue: events_unbound commit_work [drm_kms_helper]
Jun 04 00:36:00 raspi kernel: pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Jun 04 00:36:00 raspi kernel: pc : rpi_firmware_property_list+0x254/0x278
Jun 04 00:36:00 raspi kernel: lr : rpi_firmware_property_list+0x254/0x278
Jun 04 00:36:00 raspi kernel: sp : ffffffc08c7e3b40
Jun 04 00:36:00 raspi kernel: x29: ffffffc08c7e3b40 x28: 0000000000000000 x27: ffffff804092bcc0
Jun 04 00:36:00 raspi kernel: x26: 000000000000000b x25: ffffffc08049e008 x24: 0000000000001000
Jun 04 00:36:00 raspi kernel: x23: ffffff806d225100 x22: ffffffe0d51caea0 x21: 0000000000000018
[... 147 lines cut ...]
Jun 04 00:37:17 raspi kernel:  v4l2_m2m_streamoff+0x50/0x160 [v4l2_mem2mem]
Jun 04 00:37:17 raspi kernel:  v4l2_m2m_ioctl_streamoff+0x20/0x38 [v4l2_mem2mem]
[... cut ...]
Jun 04 00:37:17 raspi kernel: videobuf2_common: driver bug: stop_streaming operation is leaving buf 000000007a822fab in active state
Jun 04 00:37:17 raspi kernel: videobuf2_common: driver bug: stop_streaming operation is leaving buf 000000004e3005f5 in active state
Jun 04 00:37:17 raspi kernel: videobuf2_common: driver bug: stop_streaming operation is leaving buf 000000007408ecf8 in active state
Jun 04 00:37:17 raspi kernel: videobuf2_common: driver bug: stop_streaming operation is leaving buf 00000000f91afb97 in active state
Jun 04 00:37:20 raspi kernel: bcm2835_mmal_vchiq: timed out waiting for sync completion
```

popcornmix, 2024-06-04:
Are you viewing a web page with multiple videos on? What is the url?

qrp73, 2024-06-08:
usually this is telegram site. Almost any channel has many videos on the page.
For example: https://t.me/s/wildlifen

The issue happens again and again on a regular basis. But note, it don't happens immediately, it needs to spend some long time (at minimum several hours) to catch it. But in some rare cases I catch it immediately after reboot.

The log consists a bunch of errors in drivers/media/common/videobuf2/videobuf2-core.c
vclog -m shows thousands of the same errors like 
```alloc_compact_internal(space=3291328, in_low_region=0, mode=7) failed
mmal: mmal_port_event_send: event lost on port 1,0 (buffer header callback not defined)```

Usually these error messages appears after visiting web sites with video, but the system still works normally. 

But after some time something happens and some kernel code stops to respond. First it leads to a frozen chormium-browser or firefox. I notice that at this time web sites which uses graphics (web games like slither.io and agar.io) can't be loaded and attempt to open it leads to a browser freeze with no way to close it in usual way. Force kill from system-monitor kills it, but attempt to start it again and open such page again leads to the same app freeze issue. Only system reboot can solve that state.

Then after some time Wayfire apparently freezes (current picture stops and don't update anymore), but the system may still respond on ssh connections. Attempt to kill and restart Wayfire don't have effect, it still shows the frozen picture with no changes.

At this point it is noticeable that ssh connection has some strange random long duration freezes even when I type command it may stop to give echo for several seconds.

And in short period after Wayfire freeze, entire system freezes and stops to respond even on ssh connections. At early stage of this state it just refuse ssh connections, but later just stops to respond at all.

Only manual reboot with power off can fix this state.

qrp73, 2024-06-09:
just catch it again (complete freeze, but ssh still alive), here is the log:
dmesg:
```
[20187.913299] sd 0:0:0:0: [sda] Attached SCSI removable disk
[20188.395468] ntfs3: Max link count 4000
[22338.297672] vc4-drm gpu: swiotlb buffer is full (sz: 524288 bytes), total 32768 (slots), used 2354 (slots)
[22338.308075] vc4-drm gpu: swiotlb buffer is full (sz: 524288 bytes), total 32768 (slots), used 1020 (slots)
[22338.318831] vc4-drm gpu: swiotlb buffer is full (sz: 499712 bytes), total 32768 (slots), used 192 (slots)
[22338.753802] vc4-drm gpu: swiotlb buffer is full (sz: 524288 bytes), total 32768 (slots), used 2354 (slots)
[22338.896181] vc4-drm gpu: swiotlb buffer is full (sz: 524288 bytes), total 32768 (slots), used 1020 (slots)
[22338.945986] vc4-drm gpu: swiotlb buffer is full (sz: 499712 bytes), total 32768 (slots), used 192 (slots)
[22340.128838] vc4-drm gpu: swiotlb buffer is full (sz: 524288 bytes), total 32768 (slots), used 1462 (slots)
[22340.321078] vc4-drm gpu: swiotlb buffer is full (sz: 524288 bytes), total 32768 (slots), used 1462 (slots)
[24595.826238] cma: cma_alloc: linux,cma: alloc failed, req-size: 900 pages, ret: -16
[24595.826389] ThreadPoolSingl: page allocation failure: order:10, mode:0xcc1(GFP_KERNEL|GFP_DMA), nodemask=(null),cpuset=/,mems_allowed=0
[24595.826413] CPU: 0 PID: 59062 Comm: ThreadPoolSingl Tainted: G         C         6.6.31+rpt-rpi-v8 #1  Debian 1:6.6.31-1+rpt1
[24595.826419] Hardware name: Raspberry Pi 4 Model B Rev 1.5 (DT)
[24595.826422] Call trace:
[24595.826425]  dump_backtrace+0xa0/0x100
[24595.826435]  show_stack+0x20/0x38
[24595.826439]  dump_stack_lvl+0x48/0x60
[24595.826445]  dump_stack+0x18/0x28
[24595.826449]  warn_alloc+0x128/0x1b8
[24595.826454]  __alloc_pages+0xdc0/0xe68
[24595.826457]  __dma_direct_alloc_pages.constprop.0+0x15c/0x238
[24595.826462]  dma_direct_alloc+0x78/0x310
[24595.826465]  dma_alloc_attrs+0x8c/0x108
[24595.826471]  vb2_dc_alloc+0xa8/0x188 [videobuf2_dma_contig]
[24595.826485]  __vb2_queue_alloc+0x1cc/0x4a8 [videobuf2_common]
[24595.826505]  vb2_core_reqbufs+0x244/0x470 [videobuf2_common]
[24595.826518]  vb2_reqbufs+0x90/0xb8 [videobuf2_v4l2]
[24595.826532]  v4l2_m2m_ioctl_reqbufs+0x54/0x98 [v4l2_mem2mem]
[24595.826548]  v4l_reqbufs+0x58/0x78 [videodev]
[24595.826600]  __video_do_ioctl+0x170/0x3e0 [videodev]
[24595.826632]  video_usercopy+0x208/0x770 [videodev]
[24595.826663]  video_ioctl2+0x20/0x40 [videodev]
[24595.826693]  v4l2_ioctl+0x48/0x70 [videodev]
[24595.826724]  __arm64_sys_ioctl+0xb4/0x100
[24595.826733]  invoke_syscall+0x50/0x128
[24595.826738]  el0_svc_common.constprop.0+0x48/0xf0
[24595.826743]  do_el0_svc+0x24/0x38
[24595.826748]  el0_svc+0x40/0xe8
[24595.826754]  el0t_64_sync_handler+0x100/0x130
[24595.826759]  el0t_64_sync+0x190/0x198
[24595.826781] Mem-Info:
[24595.826784] active_anon:389361 inactive_anon:34871 isolated_anon:0
                active_file:181521 inactive_file:107326 isolated_file:0
                unevictable:68597 dirty:2652 writeback:0
                slab_reclaimable:18728 slab_unreclaimable:12868
                mapped:122242 shmem:94840 pagetables:6419
                sec_pagetables:0 bounce:0
                kernel_misc_reclaimable:0
                free:26355 free_pcp:195 free_cma:8178
[24595.826793] Node 0 active_anon:1557444kB inactive_anon:139484kB active_file:726084kB inactive_file:429304kB unevictable:274388kB isolated(anon):0kB isolated(file):0kB mapped:488968kB dirty:10608kB writeback:0kB shmem:379360kB writeback_tmp:0kB kernel_stack:11552kB pagetables:25676kB sec_pagetables:0kB all_unreclaimable? no
[24595.826800] DMA free:50260kB boost:0kB min:3660kB low:4572kB high:5484kB reserved_highatomic:0KB active_anon:217316kB inactive_anon:2480kB active_file:21984kB inactive_file:46672kB unevictable:57396kB writepending:7840kB present:950272kB managed:863628kB mlocked:72kB bounce:0kB free_pcp:780kB local_pcp:0kB free_cma:32712kB
[24595.826808] lowmem_reserve[]: 0 2928 2928 2928
[24595.826817] DMA: 305*4kB (UMEC) 137*8kB (UMEC) 32*16kB (UMEC) 39*32kB (UEC) 19*64kB (UEC) 16*128kB (UC) 28*256kB (UC) 16*512kB (C) 15*1024kB (UC) 6*2048kB (UC) 0*4096kB = 50348kB
[24595.826851] 383688 total pagecache pages
[24595.826853] 0 pages in swap cache
[24595.826855] Free swap  = 0kB
[24595.826856] Total swap = 0kB
[24595.826858] 1007616 pages RAM
[24595.826895] 0 pages HighMem/MovableOnly
[24595.826899] 41961 pages reserved
[24595.826901] 131072 pages cma reserved
[24595.826907] bcm2835-codec bcm2835-codec: dma alloc of size 3686400 failed
[24597.226913] ------------[ cut here ]------------
[24597.226925] Firmware transaction timeout
[24597.226959] WARNING: CPU: 2 PID: 58403 at drivers/firmware/raspberrypi.c:68 rpi_firmware_property_list+0x254/0x278
[24597.226977] Modules linked in: ntfs3 rfcomm snd_seq_dummy snd_hrtimer snd_seq cmac algif_hash aes_arm64 aes_generic algif_skcipher af_alg bnep brcmfmac_wcc hci_uart btbcm brcmfmac bluetooth brcmutil raspberrypi_hwmon binfmt_misc sg snd_usb_audio cfg80211 bcm2835_codec(C) rpivid_hevc(C) joydev bcm2835_isp(C) v4l2_mem2mem ecdh_generic ecc bcm2835_v4l2(C) bcm2835_mmal_vchiq(C) videobuf2_vmalloc videobuf2_dma_contig snd_hwdep videobuf2_memops rfkill libaes snd_usbmidi_lib videobuf2_v4l2 snd_rawmidi videodev raspberrypi_gpiomem snd_seq_device snd_bcm2835(C) videobuf2_common vc_sm_cma(C) mc nvmem_rmem uio_pdrv_genirq uio i2c_dev fuse dm_mod ip_tables x_tables ipv6 vc4 snd_soc_hdmi_codec drm_display_helper cec drm_dma_helper v3d gpu_sched drm_kms_helper drm [... cut ...]
3136805.206: smservice: sm_remove_resource: failed to find resource (0) in resource list
3136805.225: smservice: sm_remove_resource: failed to find resource (0) in resource list
3136805.247: smservice: sm_remove_resource: failed to find resource (0) in resource list
3136805.264: smservice: sm_remove_resource: failed to find resource (0) in resource list
3136805.286: smservice: sm_remove_resource: failed to find resource (0) in resource list
3136805.306: smservice: sm_remove_resource: failed to find resource (0) in resource list
```

qrp73, 2024-06-10:
The same report from other user: https://forums.raspberrypi.com/viewtopic.php?t=372078

qrp73, 2024-07-23:
Since there is kernel 6.6.32-v8+ and issue is not rerpoduced for a long time, the issue is not relevant anymore and I will close it.

closed 2024-07-23 by qrp73