raspberrypi/linux #6173: HifiBerry DAC+ADC audio glitches on Pi 5 at 192 kHz when stressing memory

hansvanzutphen, 2024-05-16:
### Describe the bug

On a Pi 5, when memory is stressed (see below), the HifiBerry DAC+ADC has audio glitches when playing audio at high sample rates. Placing the exact same SD card in a Pi 4 works fine.

### Steps to reproduce the behaviour

- Create a 192 kHz file
- Play it with aplay _filename_
- Run `stress -vm 2` (or higher)  (`sudo apt-get install stress` if needed)

### Device (s)

Raspberry Pi 5

### System

Raspberry Pi reference 2022-04-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 27a8050c3c06e567c794620394a8c2d74262a516, stage2

2023/10/30 16:45:10
Copyright (c) 2012 Broadcom
version 30de0ba5 (release) (embedded)

Linux thimeo-pi 6.6.30-v8+ #1762 SMP PREEMPT Thu May  9 15:17:51 BST 2024 aarch64 GNU/Linux

This Linux version is the one that you get with rpi-upgrade as suggested in https://www.hifiberry.com/blog/changes-in-hifiberry-drivers/ . But I experienced the exact same behavior with the currently released version (6.1.0 if I recall correctly), when adding ",slave" to the dtoverlay to make the HifiBerry card work on that system. There, also, the same image worked fine on a Pi 4.

### Additional context

I noticed this problem with our own software, and with "stress" I stressed several things separately. Only memory (-vm option) causes issues; I/O, HDD and CPU do not. Glitches clearly get worse if the stressing is increased; with -vm 2 I'm getting dozens of cracks (it almost sounds like a tone) per second. Talking to ALSA with bigger block sizes does not help; we've tried sizes up to 69 ms with no effect at all on the resulting audio. Lowering the priority of the threads that stress the memory, even to idle, does not help.

pelwell, 2024-05-16:
@P33M Do you think a modified QoS map on pcie2 would help here?

P33M, 2024-05-28:
QoS for PCIE2 is already set high. I think it's a manifestation of https://github.com/raspberrypi/linux/commit/0a19b5256303d2f35be9272832b01a170c9a039b

What happens if you run iperf (or ping flood) on the Pi 5 at the same time?

hansvanzutphen, 2024-10-09:
The issue no longer reproduced after doing an apt-update and apt-upgrade to the latest Raspbian version (6.6 if I recall correctly). So I'll close this issue.

closed 2024-10-09 by hansvanzutphen