raspberrypi/linux #6175: hdmi_max_pixel_freq support with KMS driver

nicoske, 2024-05-17:
### Describe the bug

It seems hdmi_max_pixel_freq is not supported anymore using the KMS driver (no effect).

### Steps to reproduce the behaviour

Set hdmi_max_pixel_freq=148500000 to avoid any resolution above 1920x1080p
Reboot
Detected resolution is still 4k

### Device (s)

Raspberry Pi 4 Mod. B

### System

Raspberry Pi reference 2024-03-15
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, f19ee211ddafcae300827f953d143de92a5c6624, stage2

Apr 17 2024 17:27:09 
Copyright (c) 2012 Broadcom
version 86ccc427f35fdc604edc511881cdf579df945fb4 (clean) (release) (start)

Linux raspberrypi 6.6.28+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.28-1+rpt1 (2024-04-22) aarch64 GNU/Linux

popcornmix, 2024-05-17:
I don't believe `hdmi_max_pixel_freq` has ever been a setting.
`hdmi_pixel_freq_limit` is used to limit the pixel clock on the firmware driver.

But this is a firmware driver thing.
Linux kernel doesn't support that as a user setting on any platforms.

What is the actual actual issue you want to resolve?

nicoske, 2024-05-17:
I used this configuration in the past before the KMS driver. I've found out that it has been moved to legacy configuration and u sed it exactly as in the example: https://www.raspberrypi.com/documentation/computers/legacy_config_txt.html

My issue is that I experience several issues with Pi4 and 4k resolution as it defaults to 30Hz and enabling 4kp60 overheat the device.
I want to ensure this resolution is not picked up. I ended up adding video=HDMI-A-1:1920x1080@60 as kernel parameters, but since I have around 500 devices with a lot of screen types this is not very handy.

nicoske, 2024-05-17:
Could we consider adding support for the hdmi_max_pixel_freq parameter in the VC4 device tree overlay to manage maximum HDMI pixel frequencies across various devices?

6by9, 2024-05-17:
> Could we consider adding support for the hdmi_max_pixel_freq parameter in the VC4 device tree overlay to manage maximum HDMI pixel frequencies across various devices?

How is having `vc4.hdmi_max_pixel_freq=XXX` in cmdline.txt any easier than `video=HDMI-A-1:1920x1080@60`?

Each platform has a max pixel clock defined in the driver
Pi0-3 https://github.com/raspberrypi/linux/blob/rpi-6.6.y/drivers/gpu/drm/vc4/vc4_hdmi.c#L3884
Pi4 HDMI-0 https://github.com/raspberrypi/linux/blob/rpi-6.6.y/drivers/gpu/drm/vc4/vc4_hdmi.c#L3904
Pi4 HDMI-1 https://github.com/raspberrypi/linux/blob/rpi-6.6.y/drivers/gpu/drm/vc4/vc4_hdmi.c#L3933
Pi5 HDMI-0 https://github.com/raspberrypi/linux/blob/rpi-6.6.y/drivers/gpu/drm/vc4/vc4_hdmi.c#L3962
Pi5 HDMI-1 https://github.com/raspberrypi/linux/blob/rpi-6.6.y/drivers/gpu/drm/vc4/vc4_hdmi.c#L3989

If you've got one module parameter, how does that reflect on devices with 2 HDMI ports?

Easiest solution is actually to use HDMI-1 (further away from the power connector), as that is limited to 1080p60 on Pi4/CM4 anyway.

nicoske, 2024-05-17:
In my project I release one version for both the Pi4 and CM4, we use a diverse range of display screens.  I prefer relying on each screen's EDID to automatically determine the best resolution, rather than setting a fixed resolution like 1920x1080, which could lead to compatibility issues with some older monitors and TVs. Also I sometimes have no control on where/who connects the HDMI plug (1 or 2).

Anyway thanks a lot for pointing me to the max pixel clock line in the driver. I can produce a patch in my projec that will enforce a lower value !