raspberrypi/linux #6178: broken sudo vclog -m output

qrp73, 2024-05-18:
### Describe the bug

The output of the command ```sudo vclog -m``` very often is broken. See log for example

### Steps to reproduce the behaviour

1) Open console
2) run command ```sudo vclog -m``` many times

Expected result: all attempt should give correct output

Actual result: in most cases the output is broken (see log for example)

### Device (s)

Raspberry Pi 4 Mod. B

### System

```
$ cat /etc/rpi-issue
Raspberry Pi reference 2023-09-22
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 40f37458ae7cadea1aec913ae10b5e7008ebce0a, stage4

$ vcgencmd version
Apr 26 2024 18:18:11 
Copyright (c) 2012 Broadcom
version f1a80576d5322dde60a2e7e9e3f4b56fd5fbe18d (clean) (release) (start)

$ uname -a
Linux raspi 6.6.30-v8+ #1 SMP PREEMPT Wed May 15 15:36:33 UTC 2024 aarch64 GNU/Linux
```

### Logs

```
$ sudo vclog -m
1830825.008: ailed+�
1768383.858: de=7) failedn���
1869488.235: d)i鴡
1601463.137: (space=3291328, in_low_region=0, mode=7) failedr���
1767990.816: 
691486.053: .֡
1763716.152: gion=0, mode=7) failed�Uߡ
544108.320:  (buffer header callback not defined)��
809332.335: ) failed�*�
1701209.717:  callback not defined)����
000000.081: compact_internal(space=3291328, in_low_region=0, mode=7) failed��
678191.470: 91328, in_low_region=0, mode=7) failed8��
1634561.280: port_event_send: event lost on port 1,0 (buffer header callback not defined)ʣ�
544502.639: 1,0 (buffer header callback not defined)r!�

$ sudo vclog -m
1830825.008: ailed(�i�
1768383.858: de=7) failed�4�
1869488.235: d)�x�
1601463.137: (space=3291328, in_low_region=0, mode=7) failed(1��
1767990.816: 
pi@raspi:~ $ sudo vclog -m
1830825.008: ailed�훢
1768383.858: de=7) failed
                         �!�
1869488.235: d)��%�
1601463.137: (space=3291328, in_low_region=0, mode=7) failed��.�
1767990.816: 

$ sudo vclog -m
1830825.008: ailed*���
1768383.858: de=7) failed��?�
1869488.235: d)��C�
1601463.137: (space=3291328, in_low_region=0, mode=7) failed��L�
1767990.816: 
539768.882: egion=0, mode=7) failed��f�
926770.532: �pp�
1801675.106: ined)1�x�
1828716.800: l_port_event_send: event lost on port 1,0 (buffer header callback not defined)���
1700754.546: d: event lost on port 1,0 (buffer header callback not defined)���
1932029.025: 1328, in_low_region=0, mode=7) failed@+��
��01080.941: edf
1667330.157: al(space=3291328, in_low_region=0, mode=7) failednÝ�
1953392.991: ace=3291328, in_low_region=0, mode=7) failed�_��
1953722.220:  1,0 (buffer header callback not defined)&ů�

$ sudo vclog -m
1830825.008: ailed��Ӣ
1768383.858: de=7) failedj�Y�
1869488.235: d))�]�
1601463.137: (space=3291328, in_low_region=0, mode=7) failedT�f�
1767990.816: 
1767990.816: 
1701322.866: lback not defined)䛂�
540699.745: t_event_send: event lost on port 1,0 (buffer header callback not defined)����
1702257.952: on port 1,0 (buffer header callback not defined)o���
2672462.180: 
942814.001: _region=0, mode=7) failed����
1601071.392: on=0, mode=7) failed7���
000065.536: mpact_internal(space=3291328, in_low_region=0, mode=7) failed<׹�
1634561.280: port_event_send: event lost on port 1,0 (buffer header callback not defined)S�â

$ sudo vclog -m
2726388.521: alloc_compact_internal(space=3291328, in_low_region=0, mode=7) failed
2726388.563: mmal: mmal_port_event_send: event lost on port 1,0 (buffer header callback not defined)
2726397.969: alloc_compact_internal(space=3291328, in_low_region=0, mode=7) failed
2726398.010: mmal: mmal_port_event_send: event lost on port 1,0 (buffer header callback not defined)
2726398.219: alloc_compact_internal(space=3291328, in_low_region=0, mode=7) failed
2726398.259: mmal: mmal_port_event_send: event lost on port 1,0 (buffer header callback not defined)
2726407.994: alloc_compact_internal(space=3291328, in_low_region=0, mode=7) failed
2726408.035: mmal: mmal_port_event_send: event lost on port 1,0 (buffer header callback not defined)
2726408.291: alloc_compact_internal(space=3291328, in_low_region=0, mode=7) failed
<...>
```

popcornmix, 2024-05-18:
The messages suggest there have been gpu allocation failures.
What are you running prior to getting corrupt output?

Do you ever see the corrupt output soon after a reboot, without running vc side code (like video encode/decode)?

qrp73, 2024-05-18:
I running chromium-browser and firefox. Since some update it always full of such messages. It usually happens when playing videos on youtube and telegram.

But this issue about corrupted messages in console output

pelwell, 2024-05-18:
The nature of this corruption makes me think that either we are reading from the wrong start point or that new messages are overwriting the old ones before we have a chance to read them, making use lose track of the message boundaries.

Looking at the last group of messages above, that's quite a high volume of data, and I suspect it was higher before, making the above scenarios more likely.

qrp73, 2024-05-18:
I think there is some issue to read log buffer, because when it shows log properly it shows all messages with no issue, even these which previously shown as broken text

pelwell, 2024-05-18:
> I think there is some issue to read log buffer, because when it shows log properly it shows all messages with no issue, even these which previously shown as broken text

Your opinion is noted.

qrp73, 2024-05-18:
Is there any chance to fix it?

alessandromrc, 2024-05-18:
> ### Describe the bug
> The output of the command `sudo vclog -m` very often is broken. See log for example
>
> ### Steps to reproduce the behaviour
> 1. Open console
> 2. run command `sudo vclog -m` many [...]

Not able to reproduce at all... All the time I get this output (SSH and also OS Terminal):
```
006805.064: Initial voltage 800000 temp 33433
007002.495: avs_2712: AVS pred 8649 864900 temp 33433
007002.506: vpred 864 mV +0
007624.873: FB framebuffer_swap 1
007653.822: Select resolution HDMI0/2 hotplug 0 max_mode 2
007653.838: Select resolution HDMI1/2 hotplug 0 max_mode 2
008301.465: dtb_file 'bcm2712-rpi-5-b.dtb'
011614.305: Starting OS 11614 ms
011616.941: 00000040: -> 00000480
011616.958: 00000030: -> 00100080
011616.973: 00000034: -> 00100080
011616.990: 00000038: -> 00100080
011617.006: 0000003c: -> 00100080
```

alessandromrc, 2024-05-18:
I've just tried on Pi4 watching the Telegram videos and can confirm one doesn't work but that seems just to be related to the video format not being supported... Other than that, I am having no such errors in the log or any corruption.

qrp73, 2024-05-19:
> Other than that, I am having no such errors in the log or any corruption.
> Running on: `Linux raspberrypi4 6.6.28+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.28-1+rpt1 (2024-04-22) aarch64 GNU/Linux`

You're using old kernel 6.6.28 which don't support some video codecs and don't have imrovements and fixes which was done in 6.6.29 and 6.6.30.

If you read carefully by bug report, it is tested on kernel 6.6.30.
You can update it with ```rpi-update rpi-6.6.y```

alessandromrc, 2024-05-19:
> > Other than that, I am having no such errors in the log or any corruption.
> 
> > Running on: `Linux raspberrypi4 6.6.28+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.28-1+rpt1 (2024-04-22) aarch64 GNU/Linux`
> 
> 
> 
> You're using old kernel 6.6.28 which don't support some video codecs and don't have imrovements and fixes for video codec hw acceleration which was done in 6.6.29 and 6.6.30.
> 
> 
> 
> If you read carefully my bug report, it is tested on kernel 6.6.30.
> 
> You can update your old 6.6.28 to 6.6.30 with ```sudo rpi-update rpi-6.6.y```
> 
> 
> 
> 

Oh, my bad, I misread the actual Kernel version. I will try later with the newer kernels and see what happens. Thanks!

qrp73, 2024-05-19:
I opened separate issue https://github.com/raspberrypi/linux/issues/6179 about video codec errors. 
This issue is about corrupted text in vclog output.

alessandromrc, 2024-05-19:
> vcgencmd version

Can confirm it doesn't happen even on this kernel:

`Linux raspberrypi4 6.6.30-v8+ #1763 SMP PREEMPT Mon May 13 17:50:31 BST 2024 aarch64 GNU/Linux
`

vclog output:
```
037901.354: vchiq_core: vchiq_init_state: slot_zero = 0xce080000, is_master = 1
147026.672: video_decode:8:RIL: resolution changing
154141.536: video_decode:9:RIL: resolution changing
154145.273: mmal: mmal_port_event_send: event lost on port 1,0 (buffer header callback not defined)
154826.838: video_decode:10:RIL: resolution changing
157482.271: video_decode:11:RIL: resolution changing
```

qrp73, 2024-05-21:
I notice that corrupted vclog output happens when there is opened web page with video, probably it happens when there is some calls to codec.

popcornmix, 2024-05-21:
We are reading the log buffer without synchronisation(*). 
If you try to read while a message is being written you may get corruption. 
Typically log messages on VC side are infrequent so the chance of corruption is low.

But it seem the bug where the browser tries to hw decode too many streams causes many error messages.
It's possible once that issue is resolved the corruption issue will be much less common.

(*) adding synchronisation between arm and VC would slow things down significantly so will not be done for a debug feature.

qrp73, 2024-05-21:
well, I agree that adding sync on VC side just to read log is not good, because it will slow down the main work of VC, but maybe there is some way to sync it on arm side with no changes on VC?

qrp73, 2024-07-19:
In case if someone want to fix this bug, it can be fixed with using counter variable with atomic type to track every log update on kernel side. On vclog side this variable can be read before and after log read and check - if counter is changed, then log re-read is required. In that way it allows to not use thread locking/synchronization at all on kernel side and allows vclog to check if aquired log buffer is correct or not. It helps to avoid read garbage instead of log.