raspberrypi/linux #6179: alloc_compact_internal(space=3291328, in_low_region=0, mode=7) failed

qrp73, 2024-05-18:
### Describe the bug

When playing video on web pages with several videos there is strange error appears in the log ```sudo vclog -m```:
```
1061192.736: alloc_compact_internal(space=3291328, in_low_region=0, mode=7) failed
1061192.780: mmal: mmal_port_event_send: event lost on port 1,0 (buffer header callback not defined)
```

### Steps to reproduce the behaviour

1) reboot
2) start chromium-browser
3) open page https://t.me/s/wildlifen
4) play video from last message (about red bird)
5) try to play video from previous message (about elephant),it will not be played due to some issue
6) try to play video from previous message (about wolf)
7) start console and execute sudo vclog -m

### Device (s)

Raspberry Pi 4 Mod. B

### System

```
$ cat /etc/rpi-issue
Raspberry Pi reference 2023-09-22
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 40f37458ae7cadea1aec913ae10b5e7008ebce0a, stage4

$ vcgencmd version
Apr 26 2024 18:18:11 
Copyright (c) 2012 Broadcom
version f1a80576d5322dde60a2e7e9e3f4b56fd5fbe18d (clean) (release) (start)

$ uname -a
Linux raspi 6.6.30-v8+ #1 SMP PREEMPT Wed May 15 15:36:33 UTC 2024 aarch64 GNU/Linux
```

alessandromrc, 2024-05-20:
I personally tried to play that video (Elephant one) on my Debian machine and it still wouldn't play within Brave / Chrome. It plays on the Telegram Desktop APP.

I guess this has more to do with chromium itself than anything else.

6by9, 2024-05-20:
You're running out of gpu_mem as it's trying to decode all those video clips.
Try `gpu_mem=128` or `gpu_mem=192` in config.txt.

I thought Chromium with our patches included limiting the number of decodes that it attempted at one time. @jc-kynesim am I totally misremembering that?

popcornmix, 2024-05-20:
I was about to post the same - my understanding was chromium was changed to only hw decode the first video, and if there were additional ones they fell back to software. Perhaps that patch isn't behaving.

jc-kynesim, 2024-05-20:
The issue is that that the decoder that did that has been superseded - I'll see if I can find the equivalent in the new one. But I think the current default of 64M? may be a bit low, given that I've now had two reports of the same BBB test clip not playing due to not enough GPU mem with the default setting.

6by9, 2024-05-20:
Default is 76MB on Pi4.
I think it's only 64MB on Pi1, but we're not considering Chromium supported on Pi0 or 1.

popcornmix, 2024-05-20:
The default gpu_mem should be enough to play one BluRay (High profile, Level 4.1) stream.
If we are failing, it sounds like we are either attempting to play more than one stream at once, or it's an unsupported (by default) profile.

I don't think we want to increase the default gpu_mem without understanding why this is failing here (and is not in, say, kodi) as that memory becomes unavailable for anyone, and the whole gpu_mem usage is being removed as much as possible.

alessandromrc, 2024-05-20:
> You're running out of gpu_mem as it's trying to decode all those video clips. Try `gpu_mem=128` or `gpu_mem=192` in config.txt.
> 
> I thought Chromium with our patches included limiting the number of decodes that it attempted at one time. @jc-kynesim am I totally misremembering that?

can confirm that stops having errors in vclog but the video still won't play, I guess that's just the encoding not being supported.

qrp73, 2024-05-20:
> can confirm that stops having errors in vclog but the video still won't play, I guess that's just the encoding not being supported.

I think these errors are not related with that video in the middle which is not played at all.

I can reproduce these errors just by opening these two videos on separate tabs in chromium-browser:
1) https://t.me/wildlifen/2713
2) https://t.me/wildlifen/2711
No needs to open more video, these two is enough to get a lot of error messages in vclog.

qrp73, 2024-05-20:
It seems that ```gpu_mem=96``` also solves issue with a bunch of error messages.