raspberrypi/linux #6185: Module nvme-fabrics not selected in cofnog but service running

by, 2024-05-24:
### Describe the bug

In kernel configs, module nvme-fabrics is not selected but at startup, the respective service is launched, leading to the error `modprobe: FATAL: Module nvme-fabrics not found in directory /lib/modules/(...)`

### Steps to reproduce the behaviour

Restart with NVME drive attached, check log

### Device (s)

Raspberry Pi 5

### System

cat /etc/rpi-issue
Raspberry Pi reference 2023-10-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 962bf483c8f326405794827cce8c0313fd5880a8, stage4

vcgencmd version
2024/05/17 12:29:43 
Copyright (c) 2012 Broadcom
version 72caf667 (release) (embedded)

uname -a
Linux aion 6.9.1-v8-16k-NTP+ #1 SMP Fri May 24 14:01:51 CEST 2024 aarch64 GNU/Linux

### Logs

May 24 14:32:35 aion systemd[1]: Starting nvmf-autoconnect.service - Connect NVMe-oF subsystems automatically during boot...
May 24 14:32:35 aion modprobe[1420]: modprobe: FATAL: Module nvme-fabrics not found in directory /lib/modules/6.9.1-v8-16k-NTP+

pelwell, 2024-05-24:
It's not clear what your complaint is really about:
* Do you have an NVMe-oF system?
* Did you install optional component nvme-cli?
* Does failure to start the nvmf-autoconnect service cause problems, other than the messages in the kernel log?

pelwell, 2024-05-24:
I suspect the answer to question 1 above is "no", otherwise your issue title would have been rather different, in which case what you probably want is:
```
$  sudo systemctl disable nvmf-autoconnect
```
Or complain to the nvme-cli maintainers: https://github.com/linux-nvme/nvme-cli/issues