raspberrypi/linux #6192: unable to connect an arduino portenta lite via usb

davidt322, 2024-05-28:
### Describe the bug

when I plug a portenta lite h7 into raspberry pi 2b or 4b no ttyxxx device gets created

tried:
   tried 2 different cables (ineffective)
   powered portenta with +5V to the Vin pins on J1 (ineffective)
   upgraded my pi2b to latest install (ineffective)
   tried it on my pi4 (this computer ineffective)
   plugged same cable to my windows 11 computer and the portenta 
          showed up in the device manager as com10 with device id 
          USB\VID_2341&PID_025B&REV_0101&MI_00 (this worked but not
          what I want to do)

### Steps to reproduce the behaviour

   I bought a usb a to c cable from amazon p/n B09FYVD6B5
   I plugged usb cable from both raspberry pi 2b and 4 between pi and portenta lite 
(https://store-usa.arduino.cc/products/portenta-h7-lite?queryID=undefined&selectedStore=us)

### Device (s)

Raspberry Pi 2 Mod. B, Raspberry Pi 4 Mod. B

### System

from raspberry pi 4
https://pastebin.com/ZJ0NdK1x

pi@ALLEN-ACP:/tmp $ cat /etc/rpi-issue
Raspberry Pi reference 2022-09-22
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 005a8c73b05a2cab394073150208bf4f069e861a, stage4

pi@ALLEN-ACP:/tmp $ vcgencmd version
Jan  5 2023 10:46:54 
Copyright (c) 2012 Broadcom
version 8ba17717fbcedd4c3b6d4bce7e50c7af4155cba9 (clean) (release) (start)

pi@ALLEN-ACP:/tmp $ uname -a
Linux ALLEN-ACP 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64 GNU/Linux

### Logs

complete dmesg log in pastebin link above
[  139.251221] usb 1-1-port1: Cannot enable. Maybe the USB cable is bad?
[  140.103638] usb 1-1-port1: attempt power cycle
[  142.123606] usb 1-1-port1: unable to enumerate USB device

### Additional context

reported bug to https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1071986
they said:
This does not seem to run a (native) Debian system, so closing.
Please reopen if an issue with a plain Debian system.

davidt322, 2024-05-29:
I tried the blue and the black usb ports and got the same message
I also tried plugging it into a 4 port hub and it did enumerate

blue usb port
[518036.470297] usb 1-1-port2: Cannot enable. Maybe the USB cable is bad?
[518037.322532] usb 1-1-port2: Cannot enable. Maybe the USB cable is bad?
[518037.323311] usb 1-1-port2: attempt power cycle
[518038.490246] usb 1-1-port2: Cannot enable. Maybe the USB cable is bad?
[518039.342307] usb 1-1-port2: Cannot enable. Maybe the USB cable is bad?
[518039.343087] usb 1-1-port2: unable to enumerate USB device

black usb port
[518051.322024] usb 1-1-port4: Cannot enable. Maybe the USB cable is bad?
[518052.174049] usb 1-1-port4: Cannot enable. Maybe the USB cable is bad?
[518052.174858] usb 1-1-port4: attempt power cycle
[518053.342009] usb 1-1-port4: Cannot enable. Maybe the USB cable is bad?
[518054.193972] usb 1-1-port4: Cannot enable. Maybe the USB cable is bad?
[518054.194635] usb 1-1-port4: unable to enumerate USB device

port on my Ativa 4 port hub, which is plugged into a black usb port on my pi 4
[518075.357104] usb 1-1.3.4: new high-speed USB device number 27 using xhci_hcd
[518075.558166] usb 1-1.3.4: New USB device found, idVendor=2341, idProduct=025b, bcdDevice= 1.01
[518075.558201] usb 1-1.3.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[518075.558219] usb 1-1.3.4: Product: Envie M7
[518075.558233] usb 1-1.3.4: Manufacturer: Arduino
[518075.558246] usb 1-1.3.4: SerialNumber: 003D00243132510F39383539
[518075.623562] cdc_acm 1-1.3.4:1.0: ttyACM0: USB ACM device
[518075.623727] usbcore: registered new interface driver cdc_acm
[518075.623739] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters

so I can debug my new code from a pi if I add a $3-$10 hub.  Perhaps the developers can fix this software bug so I don't need an extra hub.

davidt322, 2024-05-29:
it even works same way on my pi 2b
[   88.893226] usb 1-1.3.4: new high-speed USB device number 12 using dwc_otg
[   89.123998] usb 1-1.3.4: New USB device found, idVendor=2341, idProduct=025b, bcdDevice= 1.01
[   89.124038] usb 1-1.3.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   89.124054] usb 1-1.3.4: Product: Envie M7
[   89.124066] usb 1-1.3.4: Manufacturer: Arduino
[   89.124077] usb 1-1.3.4: SerialNumber: 003D00243132510F39383539
[   89.403952] cdc_acm 1-1.3.4:1.0: ttyACM0: USB ACM device
[   89.404206] usbcore: registered new interface driver cdc_acm
[   89.404218] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters

pelwell, 2024-05-29:
I'm not the resident USB expert, but I think this is unlikely to be a software bug. The fact that both a Pi 2 and Pi 4 - different USB hardware, different drivers - think there's something bad about the device+cable combination suggests it is not standards-compliant in some way.

pelwell, 2024-05-29:
> it even works same way on my pi 2b

You mean with the hub?

davidt322, 2024-05-29:
I don't see how it can be a cable issue when I used same cable working/not working
[image]

pelwell, 2024-05-29:
You could say the same about the Portenta.

By acting as an electrical break between the device and host, hubs can make bad things less bad.

davidt322, 2024-05-29:
This is the first time that I am aware of the adding a usb hub fixes a usb issue.

So you are saying 1 of 2 things
1) you don't think you should be able to plug an arduino portenta into a raspberry pi
2) you think the problem is in the boot loader of the portenta and arduino should fix the problem

I am currently buying about 100 raspberry pi's a year
Someone has to fix the issue, ether raspberry pi are arduino.
Can you tell me how to find out which open source project is responsible to fix issue?
I can plug a portenta into a windows computer no issue.
At this point I think it is probably the raspberry pi software.

Is there a test version of the OS are some other log I can get.

pelwell, 2024-05-30:
No, I'm suggesting:
3. there is some hardware incompatibility between the device+cable combination you are using and the Pis you have. That fact that they are two different Pis with very different USB hardware does point towards it being a device problem.

That some hardware is more tolerant than others doesn't mean the problem is with the less tolerant hardware, rather it's with the bad behaviour that requires tolerance.

Any thoughts, @P33M? Could this be a timing/protocoal problem caused by the Arduino firmware?

P33M, 2024-05-30:
If it is a protocol failure, it's happening right at the start - before enumeration is even attempted. The device might be trying to do some sort of OTG session negotiation when it shouldn't, and a plausible reason for the working setup is that the hub just ignores this state.

There's not going to be a fix here - both the Pi 4 and Pi 2 integrated hubs have no software settings to change.

davidt322, 2024-05-30:
What can you tell the arduino team to try and fix the issue.  Can you suggest some debug changes to hub.c or other core usb files that will tell what is going on. I can compile a new version of the os so we can determine the cause of the failure?

davidt322, 2024-05-30:
I could loan/give you a portenta if that will help.  Perhaps just set some breakpoints in the os and look at some variables.

pelwell, 2024-05-30:
> Perhaps just set some breakpoints in the os and look at some variables.

The kernel is already reporting all the information that is available. A proper analysis is going to require at least a USB analyser and some motivation. Given that, as @P33M says, "There's not going to be a fix here", why would we?

It might be worth trying another USB cable though - after all, "Maybe the USB cable is bad?"

closed 2024-05-30 by pelwell

davidt322, 2024-06-05:
what would be considered motivation?  Already tried 2 usb cables.

davidt322, 2024-07-24:
this problem was solved by supplying 5.1v at 2 amps to raspberry pi

pelwell, 2024-07-24:
Wow - what were you providing before? For a Pi 4 we say 51.V at 3A as a minimum.

And beware, 5.1V at the supply is not going to be 5.1V at the Pi - cable losses can be significant.

davidt322, 2024-07-24:
It did not work when I plugged it into a powered usb hub.  Lightning bolt was on, on the pi 2b.  Lightning bolt comes on at 4.6V.  Also Works when plugged into a 1A 5V usb charger too (no lightning bolt).