raspberrypi/linux #6195: M.2 HAT+ - incompatible NVMe (Sandisk Corp WD Blue SN550 NVMe SSD/SanDisk Ultra 3D NVMe 500GB)

aBUGSworstnightmare-rpi, 2024-05-29:
### Describe the bug

The NVMe is working on CM4 but not even detected on the official M.2 HAT+. Drive is capable of gen3 --> test conducted on gen1/gen2/gen3.

### Steps to reproduce the behaviour

Test the NVMe in CM4 (see logs) - NVMe linked on the bus, fully working.

Connect the NVMe to the official M.2 HAT+ --> device is not even detected on the PCIe bus.

### Device (s)

Raspberry Pi 5, Raspberry Pi CM4

### System

for CM4: Linux raspberrypi 6.6.31-v8+ #1 SMP PREEMPT Fri May 24 14:09:25 CEST 2024 aarch64 GNU/Linux

for Pi5: Linux m2hattest 6.6.31-v8-16k+ #1766 SMP PREEMPT Fri May 24 12:15:34 BST 2024 aarch64 GNU/Linux

### Logs

Relevant info from the CM4 (CM4 on CM4IO)
On CM4

pi@raspberrypi:~ $ lspci -n
00:00.0 0604: 14e4:2711 (rev 20)
01:00.0 0108: 15b7:5009 (rev 01)

pi@raspberrypi:~ $ sudo lspci -q
00:00.0 PCI bridge: Broadcom Inc. and subsidiaries BCM2711 PCIe Bridge (rev 20)
01:00.0 Non-Volatile memory controller: Sandisk Corp WD Blue SN550 NVMe SSD (rev 01)

pi@raspberrypi:~ $ sudo lspci -vvxxx
00:00.0 PCI bridge: Broadcom Inc. and subsidiaries BCM2711 PCIe Bridge (rev 20) (prog-if 00 [Normal decode])
	Device tree node: /sys/firmware/devicetree/base/scb/pcie@7d500000/pci@0,0
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0
	Interrupt: pin A routed to IRQ 27
	Bus: primary=00, secondary=01, subordinate=01, sec-latency=0
	Memory behind bridge: 00000000-000fffff [size=1M] [32-bit]
	Prefetchable memory behind bridge: [disabled] [64-bit]
	Secondary status: 66MHz- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- <SERR- <PERR-
	BridgeCtl: Parity- SERR- NoISA- VGA- VGA16- MAbort- >Reset- FastB2B-
		PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
	Capabilities: [48] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0+,D1-,D2-,D3hot+,D3cold-)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=1 PME-
	Capabilities: [ac] Express (v2) Root Port (Slot-), MSI 00
		DevCap:	MaxPayload 512 bytes, PhantFunc 0
			ExtTag- RBE+
		DevCtl:	CorrErr- NonFatalErr- FatalErr- UnsupReq-
			RlxdOrd+ ExtTag- PhantFunc- AuxPwr+ NoSnoop+
			MaxPayload 128 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr- TransPend-
		LnkCap:	Port #0, Speed 5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <1us, L1 <2us
			ClockPM+ Surprise- LLActRep- BwNot+ ASPMOptComp+
		LnkCtl:	ASPM L1 Enabled; RCB 64 bytes, Disabled- CommClk+
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 5GT/s, Width x1
			TrErr- Train- SlotClk+ DLActive- BWMgmt+ ABWMgmt+
		RootCap: CRSVisible+
		RootCtl: ErrCorrectable- ErrNon-Fatal- ErrFatal- PMEIntEna+ CRSVisible+
		RootSta: PME ReqID 0000, PMEStatus- PMEPending-
		DevCap2: Completion Timeout: Range ABCD, TimeoutDis+ NROPrPrP- LTR+
			 10BitTagComp- 10BitTagReq- OBFF Via WAKE#, ExtFmt- EETLPPrefix-
			 EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
			 FRS- LN System CLS Not Supported, TPHComp- ExtTPHComp- ARIFwd-
			 AtomicOpsCap: Routing- 32bit- 64bit- 128bitCAS-
[... 587 lines cut ...]
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete- EqualizationPhase1-
pi@m2hattest:~ $ sudo nvme id-ctrl -H /dev/nvme0
/dev/nvme0: No such file or directory
Usage: nvme id-ctrl <device> [OPTIONS]

Send an Identify Controller command to the given device and report
information about the specified controller in human-readable or binary
format. May also return vendor-specific controller attributes in hex-dump if
requested.

Options:
  [  --vendor-specific, -v ]            --- dump binary vendor field
  [  --output-format=<FMT>, -o <FMT> ]  --- Output format: normal|json|binary
  [  --raw-binary, -b ]                 --- show identify in binary format
  [  --human-readable, -H ]             --- show identify in readable format
pi@m2hattest:~ $ 

Tested with PCIe_gen forced to 1 (param=pciex1_gen=1)
raspinfo output: https://paste.debian.net/hidden/d5bb727e/

Tested with PCIe_gen forced to 2 param=pciex1_gen=2)
https://paste.debian.net/hidden/950b4ce8/

as the drive is capable tested at gen3 as well --> all non working
https://paste.debian.net/hidden/d19c0467/

pelwell, 2024-05-29:
For testing purposes, does `dtparam=nvme` in config.txt make any difference?

aBUGSworstnightmare-rpi, 2024-05-30:
> For testing purposes, does `dtparam=nvme` in config.txt make any difference?

Adding the parameter shows no effect; external PCIe is still down.
https://paste.debian.net/hidden/44e00785/

to my current understanding 'PCIE_PROBE=1' in EEPROM config is the same. Tested as well, but link remains down.

timg236, 2024-05-30:
The boot UART log will indicate the status the DET WAKE signal and PCIE_PROBE will force PWR_EN.

It should be possible to see the state of those pins via pinctrl

aBUGSworstnightmare-rpi, 2024-05-30:
> The boot UART log will indicate the status the DET WAKE signal and PCIE_PROBE will force PWR_EN.
> 
> It should be possible to see the state of those pins via pinctrl

Find the log below. I've used a Pi Debug probe and 'minicom' (on another Pi5) for caputring the boot loader output.

> 3.18 PCIEx1: PWR 0 DET_WAKE 1

Question: why does the boot loader provides info of 4GB RAM while the Pi is an 8GB version
pi@m2hattest:~ $ vcgencmd get_config total_mem
total_mem=8192
or how to read
>   1.35 Initialising SDRAM 'Micron' 32Gb x2 total-size: 64 Gbit 4267
>   1.38 DDR 4267 1 0 64 152

>   0.82 RPi: BOOTSYS release VERSION:72caf667 DATE: 2024/05/17 TIME: 12:29:43
>   0.86 BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1715945383 serial 9a8412f7 boardrev d04170 stc 886147
>   0.96 AON_RESET: 00000003 PM_RSTS 00001000
>   0.03 RP1_BOOT chip ID: 0x20001927
>   0.06 PM_RSTS: 0x00001000
>   0.06 part 00000000 reset_info 00000000
>   0.10 PMIC reset-event 00000000 rtc 6658738f alarm 00000000 enabled 0
>   0.16 uSD voltage 3.3V
>   1.35 Initialising SDRAM 'Micron' 32Gb x2 total-size: 64 Gbit 4267
>   1.38 DDR 4267 1 0 64 152
>   2.55 OTP boardrev d04170 bootrom a a
>   2.56 Customer key hash 0000000000000000000000000000000000000000000000000000000000000000
>   2.63 VC-JTAG unlocked
>   2.86 RP1_BOOT chip ID: 0x20001927
> 
>   3.93 RP1_BOOT chip ID: 0x20001927
>   3.94 RP1_BOOT: fw size 25992
>   3.49 PCI2 init
>   3.49 PCI2 reset
>   3.93 PCIe scan 00001de4:00000001
>   3.94 RP1_CHIP_INFO 20001927
> 
>   3.97 RPi: BOOTLOADER release VERSION:72caf667 DATE: 2024/05/17 TIME: 12:29:43
>   3.04 BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1715945383 serial 9a8412f7 boardrev d04170 stc 3904199
>   3.14 AON_RESET: 00000003 PM_RSTS 00001000
>   3.18 PCIEx1: PWR 0 DET_WAKE 1
>   3.45 usb_pd_init status 5
>   3.51 XHCI-STOP
>   3.51 xHC0 ver: 272 HCS: 03000440 140000f1 07ff000a HCC: 0240fe6d
>   3.55 USBSTS 1
>   3.58 xHC0 ver: 272 HCS: 03000440 140000f1 07ff000a HCC: 0240fe6d
>   3.63 xHC0 ports 3 slots 64 intrs 4
>   3.75 XHCI-STOP
>   3.75 xHC1 ver: 272 HCS: 03000440 140000f1 07ff000a HCC: 0240fe6d
>   3.79 USBSTS 1
>   3.82 xHC1 ver: 272 HCS: 03000440 140000f1 07ff000a HCC: 0240fe6d
>   3.87 xHC1 ports 3 slots 64 intrs 4
>   3.94 Boot mode: SD (01) order f46
>   4.09 SD HOST: 200000000 CTL0: 0x00800000 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
>   4.20 SD HOST: 200000000 CTL0: 0x00800f00 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
>   4.50 OCR c0ff8000 [190]
> CID: 000353445343333247809318a100014b
> CSD: 400e00325b590000edc87f800a404000
>   4.59 SD: bus-width: 4 spec: 2 SCR: 0x02358043 0x00000000
>   4.66 SD HOST: 200000000 CTL0: 0x00800f04 BUS: 50000000 Hz actual: 50000000 HZ div: 4 (2) status: 0x1fff0000 delay: 2
>   4.78 MBR: 0x00002000, 1048576 type: 0x0c
>   4.79 MBR: 0x00102000,61277184 type: 0x83
>   4.83 MBR: 0x00000000,       0 type: 0x00
>   4.87 MBR: 0x00000000,       0 type: 0x00
>   4.91 Trying partition: 0
>   4.96 type: 32 lba: 8192 'mkfs.fat' ' bootfs     ' clusters 261116 (4)
>   4.00 rsc 32 fat-sectors 2040 root dir cluster 2 sectors 0 entries 0
>   4.08 FAT32 clusters 261116
>   4.14 [sdcard] autoboot.txt not found
>   4.15 Select partition rsts 0 C(boot_partition) 0 EEPROM config 0 result 0
>   4.22 Trying partition: 0
>   4.26 type: 32 lba: 8192 'mkfs.fat' ' bootfs     ' clusters 261115 (4)
>   4.30 rsc 32 fat-sectors 2040 root dir cluster 2 sectors 0 entries 0
>   4.39 FAT32 clusters 261115
>   4.43 Read config.txt bytes     2717 hnd 0x2808e
>   4.47 [sdcard] pieeprom.upd not found
>   4.48 usb_max_current_enable default 0 max-current 5000
>   4.60 Read bcm2712-rpi-5-b.dtb bytes    80218 hnd 0x1defd
>   4.62 dt-match: compatible: raspberrypi,5-model-b match: brcm,bcm2712
>   4.69 dt-match: compatible: brcm,bcm2712 match: brcm,bcm2712
> 
> NOTICE:  BL31: v2.6(release):v2.6-239-g2a9ede0bd
> NOTICE:  BL31: Built : 14:26:57, Jun 22 2023
> [    0.839738] brcm-pcie 1000110000.pcie: link down
> 
> Debian GNU/Linux 12 m2hattest ttyAMA10
> 
> m2hattest login:

timg236, 2024-05-30:
It doesn't say 4GB. Your log and the HDMI diagnostic screen will say it's a 64 GBit so 8 gigabyte SDRAM
"1.35 Initialising SDRAM 'Micron' 32Gb x2 total-size: 64 Gbit 4267"

timg236, 2024-05-30:
Regarding the original question - 
"3.18 PCIEx1: PWR 0 DET_WAKE 1"

So the M2.HAT is detected and the bootloader will have enabled power to the NVMe and enabled the PCIe as a bootable device. This particular boot had SD has a higher priority so it's not possible to determine whether  bootloader could see the NVMe device, but probably not if the device is not visible under Linux.

Do you see the amber power indicator on the M2.HAT ?

aBUGSworstnightmare-rpi, 2024-05-30:
> Regarding the original question - "3.18 PCIEx1: PWR 0 DET_WAKE 1"
>
> So the M2.HAT is detected and the bootloader will have enabled power to the NVMe and enabled the PCIe as a bootable device. This [...]

the amber PWR LED is on.

changed the boot order to 0f416

  0.81 RPi: BOOTSYS release VERSION:72caf667 DATE: 2024/05/17 TIME: 12:29:43
  0.85 BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1715945383 serial 9a8412f7 boardrev d04170 stc 885883
  0.95 AON_RESET: 00000003 PM_RSTS 00001000
  0.03 RP1_BOOT chip ID: 0x20001927
  0.05 PM_RSTS: 0x00001000
  0.06 part 00000000 reset_info 00000000
  0.09 PMIC reset-event 00000000 rtc 6658b261 alarm 00000000 enabled 0
  0.16 uSD voltage 3.3V
  1.35 Initialising SDRAM 'Micron' 32Gb x2 total-size: 64 Gbit 4267
  1.38 DDR 4267 1 0 64 152
  2.57 OTP boardrev d04170 bootrom a a
  2.59 Customer key hash 0000000000000000000000000000000000000000000000000000000000000000
  2.66 VC-JTAG unlocked
  2.89 RP1_BOOT chip ID: 0x20001927

  3.96 RP1_BOOT chip ID: 0x20001927
  3.96 RP1_BOOT: fw size 25992
  3.51 PCI2 init
  3.51 PCI2 reset
  3.96 PCIe scan 00001de4:00000001
  3.96 RP1_CHIP_INFO 20001927

  3.99 RPi: BOOTLOADER release VERSION:72caf667 DATE: 2024/05/17 TIME: 12:29:43
  3.06 BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1715945383 serial 9a8412f7 boardrev d04170 stc 3906607
  3.16 AON_RESET: 00000003 PM_RSTS 00001000
  3.20 PCIEx1: PWR 0 DET_WAKE 1
  3.49 usb_pd_init status 5
  3.53 XHCI-STOP
  3.53 xHC0 ver: 272 HCS: 03000440 140000f1 07ff000a HCC: 0240fe6d
  3.58 USBSTS 11
  3.60 xHC0 ver: 272 HCS: 03000440 140000f1 07ff000a HCC: 0240fe6d
  3.65 xHC0 ports 3 slots 64 intrs 4
  3.77 XHCI-STOP
  3.77 xHC1 ver: 272 HCS: 03000440 140000f1 07ff000a HCC: 0240fe6d
  3.82 USBSTS 11
  3.84 xHC1 ver: 272 HCS: 03000440 140000f1 07ff000a HCC: 0240fe6d
  3.89 xHC1 ports 3 slots 64 intrs 4
  3.97 Boot mode: NVME (06) order f41
  4.98 USB2[2] 000206e1 connected
  4.13 USB3[3] 002a1203 connected enabled
  4.14 USB3 root HUB port 3 init
  4.18 DEV [01:00] 3.00 000000:03 class 0 VID 05dc PID b051
  4.22 MSD device [01:00] 3.00 000000:03 conf 0 iface 0 ep 81#1024 02#1024
  4.29 MSD [01:00] 3.00 000000:03 register MSD
  4.48 USB2[2] 00200e03 connected enabled
  4.49 USB2 root HUB port 2 init
  4.54 DEV [01:00] 2.00 000000:02 class 9 VID 05e3 PID 0610
  4.57 HUB init [01:00] 2.00 000000:02
  4.70 HUB [01:00] 2.00 000000:02 init port 1 speed 2
  4.99 DEV [02:01] 2.00 000001:02 class 0 VID 04d9 PID 0006
  4.06 HID [02:01] 2.00 000001:02 register HID
  4.76 HUB [01:00] 2.00 000000:02 init port 3 speed 2
  4.82 DEV [03:01] 1.16 000003:02 class 0 VID 093a PID 2510
  4.87 HID [03:01] 1.16 000003:02 register HID
  4.17 PCI1 init
  4.17 PCI1 reset
  5.38 PCIe timeout: 0x0001e08f
  5.38 Failed to open device: 'nvme'
  5.41 Retry NVME 1
  5.43 PCI1 init
  5.44 PCI1 reset
  5.66 PCIe timeout: 0x0001e08f
  5.67 Failed to open device: 'nvme'
  5.70 Boot mode: SD (01) order f4
  5.73 HUB [01:00] 2.00 000000:02 init port 4 speed 3
  5.92 SD HOST: 200000000 CTL0: 0x00800000 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
  5.03 SD HOST: 200000000 CTL0: 0x00800f00 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
  5.19 DEV [04:01] 2.00 000004:02 class 0 VID 148f PID 5572
  5.34 OCR c0ff8000 [188]
CID: 000353445343333247809318a100014b
CSD: 400e00325b590000edc87f800a404000
  5.44 SD: bus-width: 4 spec: 2 SCR: 0x02358043 0x00000000
  5.51 SD HOST: 200000000 CTL0: 0x00800f04 BUS: 50000000 Hz actual: 50000000 HZ div: 4 (2) status: 0x1fff0000 delay: 2
  5.63 MBR: 0x00002000, 1048576 type: 0x0c
  5.64 MBR: 0x00102000,61277184 type: 0x83
  5.67 MBR: 0x00000000,       0 type: 0x00
  5.71 MBR: 0x00000000,       0 type: 0x00
  5.75 Trying partition: 0
  5.80 type: 32 lba: 8192 'mkfs.fat' ' bootfs     ' clusters 261116 (4)
  5.84 rsc 32 fat-sectors 2040 root dir cluster 2 sectors 0 entries 0
  5.92 FAT32 clusters 261116
  5.99 [sdcard] autoboot.txt not found
  5.00 Select partition rsts 0 C(boot_partition) 0 EEPROM config 0 result 0
  5.06 Trying partition: 0
  5.11 type: 32 lba: 8192 'mkfs.fat' ' bootfs     ' clusters 261115 (4)
  5.15 rsc 32 fat-sectors 2040 root dir cluster 2 sectors 0 entries 0
  5.24 FAT32 clusters 261115
  5.28 Read config.txt bytes     2717 hnd 0x2808e
  5.32 [sdcard] pieeprom.upd not found
  5.33 usb_max_current_enable default 0 max-current 5000
  5.45 Read bcm2712-rpi-5-b.dtb bytes    80218 hnd 0x1defd
  5.47 dt-match: compatible: raspberrypi,5-model-b match: brcm,bcm2712
  5.53 dt-match: compatible: brcm,bcm2712 match: brcm,bcm2712

NOTICE:  BL31: v2.6(release):v2.6-239-g2a9ede0bd
NOTICE:  BL31: Built : 14:26:57, Jun 22 2023
[    0.839764] brcm-pcie 1000110000.pcie: link down

aBUGSworstnightmare-rpi, 2024-06-03:
Any idea what is causing the link to time out? Any parameter which I can add to prevent this from happening?

What is different when comparing PCIe on CM4 to PCIe on Pi5?

aBUGSworstnightmare-rpi, 2024-06-16:
As no-one seems to be able to answer what is the difference between PCIe on CM4 (working SN550) compared to Pi5 (SN550 non-working) let me add some **CM4 bootlaoder logs**

### booted with no PCIe NVMe drive attached
  0.10 RPi: BOOTSYS release VERSION:72caf667 DATE: 2024/05/17 TIME: 12:26:58
  0.14 BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1715945218 serial 6906841c boardrev c03140 stc 514243
  0.30 PM_RSTS: 0x00001020
  0.30 part 00000000 reset_info 00000000
  0.34 uSD voltage 3.3V
  0.59 Initialising SDRAM 'Samsung' 16Gb x2 total-size: 32 Gbit 3200
  0.63 DDR 3200 1 0 32 152
  2.98 OTP boardrev c03140 bootrom 48b0 48b0
  2.00 Customer key hash 0000000000000000000000000000000000000000000000000000000000000000
  2.07 VC-JTAG unlocked

  2.83 RPi: BOOTLOADER release VERSION:72caf667 DATE: 2024/05/17 TIME: 12:26:58
  2.87 BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1715945218 serial 6906841c boardrev c03140 stc 2387769
  2.23 PCI0 init
  2.23 PCI0 reset
  2.43 PCIe timeout: 0x0000008f
  2.43 USB xHC init failed
  2.46 Boot mode: SD (01) order f2546
  2.63 SD HOST: 200000000 CTL0: 0x00800000 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
  2.74 SD HOST: 200000000 CTL0: 0x00800f00 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
  3.90 OCR c0ff8000 [261]
CID: 00035344534c363447804a6e0d1a0114
CSD: 400e00325b590001dbd37f800a404000
  3.01 SD: bus-width: 4 spec: 2 SCR: 0x02458003 0x00000000
  3.09 SD HOST: 200000000 CTL0: 0x00800f04 BUS: 50000000 Hz actual: 50000000 HZ div: 4 (2) status: 0x1fff0000 delay: 2
  3.22 MBR: 0x00002000, 1048576 type: 0x0c
  3.23 MBR: 0x00102000,123678720 type: 0x83
  3.27 MBR: 0x00000000,       0 type: 0x00
  3.31 MBR: 0x00000000,       0 type: 0x00
  3.34 Trying partition: 0
  3.40 type: 32 lba: 8192 'mkfs.fat' ' bootfs     ' clusters 130812 (8)
  3.43 rsc 32 fat-sectors 1024 root dir cluster 2 sectors 0 entries 0
  3.52 FAT32 clusters 130812
  3.59 [sdcard] autoboot.txt not found
  3.60 Select partition rsts 0 C(boot_partition) 0 EEPROM config 0 result 0
  3.67 Trying partition: 0
  3.70 type: 32 lba: 8192 'mkfs.fat' ' bootfs     ' clusters 130812 (8)
  3.75 rsc 32 fat-sectors 1024 root dir cluster 2 sectors 0 entries 0
  3.85 FAT32 clusters 130812
  3.87 Read config.txt bytes     3431 hnd 0xcf85
  3.91 [sdcard] pieeprom.upd not found
  3.93 [sdcard] recover4.elf not found
  3.97 [sdcard] recovery.elf not found
  3.20 Read start4.elf bytes  2257216 hnd 0xaa26
  3.23 Read fixup4.dat bytes     5436 hnd 0xbaf5
  3.26 0x00c03140 0x00000000 0x00001fff
  3.30 MEM GPU: 76 ARM: 948 TOTAL: 1024
  3.08 Firmware: 4942b7633c0ff1af1ee95a51a33b56a9dae47529 May 24 2024 15:30:04
  3.41 Starting start4.elf @ 0xfec00200 partition 0
  3.43 PCI0 reset
  3.55 +

  0.10 RPi: BOOTSYS release VERSION:72caf667 DATE: 2024/05/17 TIME: 12:26:58
  0.14 BOOTMODE: 0x06 partition 63 build-ts BUILD_TIMESTAMP=1715945218 serial 6906841c boardrev c03140 stc 514243
  0.30 Halt: wake: 1 power_off: 0
--> powered down

--> attaching the SN550 to PCIe### 
  0.11 RPi: BOOTSYS release VERSION:72caf667 DATE: 2024/05/17 TIME: 12:26:58
  0.15 BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1715945218 serial 6906841c boardrev c03140 stc 515918
  0.32 PM_RSTS: 0x00001000
  0.32 part 00000000 reset_info 00000000
  0.35 uSD voltage 3.3V
  0.61 Initialising SDRAM 'Samsung' 16Gb x2 total-size: 32 Gbit 3200
  0.64 DDR 3200 1 0 32 152
  2.01 OTP boardrev c03140 bootrom 48b0 48b0
  2.03 Customer key hash 0000000000000000000000000000000000000000000000000000000000000000
  2.10 VC-JTAG unlocked

  2.86 RPi: BOOTLOADER release VERSION:72caf667 DATE: 2024/05/17 TIME: 12:26:58
  2.90 BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1715945218 serial 6906841c boardrev c03140 stc 2390617
  2.26 PCI0 init
  2.26 PCI0 reset
  2.29 PCIe scan 000015b7:00005009
  2.30 USB xHC init failed
  2.32 Boot mode: SD (01) order f2546
  2.49 SD HOST: 200000000 CTL0: 0x00800000 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
  2.60 SD HOST: 200000000 CTL0: 0x00800f00 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
  2.83 OCR c0ff8000 [269]
CID: 00035344534c363447804a6e0d1a0114
CSD: 400e00325b590001dbd37f800a404000
  2.93 SD: bus-width: 4 spec: 2 SCR: 0x02458003 0x00000000
  2.01 SD HOST: 200000000 CTL0: 0x00800f04 BUS: 50000000 Hz actual: 50000000 HZ div: 4 (2) status: 0x1fff0000 delay: 2
  2.14 MBR: 0x00002000, 1048576 type: 0x0c
  2.15 MBR: 0x00102000,123678720 type: 0x83
  2.19 MBR: 0x00000000,       0 type: 0x00
  2.23 MBR: 0x00000000,       0 type: 0x00
  2.27 Trying partition: 0
  2.32 type: 32 lba: 8192 'mkfs.fat' ' bootfs     ' clusters 130812 (8)
  2.36 rsc 32 fat-sectors 1024 root dir cluster 2 sectors 0 entries 0
  2.45 FAT32 clusters 130812
  2.52 [sdcard] autoboot.txt not found
  2.52 Select partition rsts 0 C(boot_partition) 0 EEPROM config 0 result 0
  2.59 Trying partition: 0
  2.62 type: 32 lba: 8192 'mkfs.fat' ' bootfs     ' clusters 130812 (8)
  2.68 rsc 32 fat-sectors 1024 root dir cluster 2 sectors 0 entries 0
  2.77 FAT32 clusters 130812
  2.79 Read config.txt bytes     3431 hnd 0xcf85
  2.83 [sdcard] pieeprom.upd not found
  2.86 [sdcard] recover4.elf not found
  2.89 [sdcard] recovery.elf not found
  3.13 Read start4.elf bytes  2257216 hnd 0xaa26
  3.16 Read fixup4.dat bytes     5436 hnd 0xbaf5
  3.18 0x00c03140 0x00000000 0x00001fff
  3.22 MEM GPU: 76 ARM: 948 TOTAL: 1024
  3.01 Firmware: 4942b7633c0ff1af1ee95a51a33b56a9dae47529 May 24 2024 15:30:04
  3.34 Starting start4.elf @ 0xfec00200 partition 0
  3.36 PCI0 reset
  3.47 +
 
```
pi@raspberrypi:~ $ lspci
00:00.0 PCI bridge: Broadcom Inc. and subsidiaries BCM2711 PCIe Bridge (rev 20)
01:00.0 Non-Volatile memory controller: Sandisk Corp WD Blue SN550 NVMe SSD (rev 01)
```

timg236, 2024-06-17:
Until it works under Linux there's little point in trying to boot from this. 
There are some assorted PCIe compatibility levers that you could try here dtoverlay=pciex1-compat-pi5,<param>=<val>
https://github.com/raspberrypi/firmware/issues/1833#issuecomment-2168427105

timg236, 2024-06-17:
It working on a CM4 obviously won’t help you with a pi5

aBUGSworstnightmare-rpi, 2024-06-17:
> It working on a CM4 obviously won’t help you with a pi5

That's why I'm asking here what makes PCIe on Pi5 differ from PCIe on CM4, but either nobody is able or not willing wo answer.

Whst are these 'compatibility flags' good for and why wasn't there none for CM4 do far?

pelwell, 2024-06-17:
Because it's a different version of the PCIe RC IP, with 2 lanes, Gen-3-capable, etc.

timg236, 2024-06-17:
The question is really, what's different about this drive vs all the other drives that work on Pi5. I've already walked you through the steps to verify that the M2.HAT is working and powering on the drive etc.

The assorted Linux PCIe compatiblity quirks do disable features e.g. ASPM are an obvious place to start when trying to isolate the problem.

aBUGSworstnightmare-rpi, 2024-06-18:
> The question is really, what's different about this drive vs all the other drives that work on Pi5. I've already walked you through the steps to verify that the M2.HAT is working and powering on the drive etc.
> 
> The assorted Linux PCIe compatiblity quirks to disable features e.g. ASPM are an obvious place to start when trying to isolate the problem.

Went through all this permutations and always end up with a down link 
(yes, I know that first two parameters were superfluous, but as I've previously been requested to check if `dtparam=nvme` makes a difference I've kept them in)

>   4.02 Boot mode: NVME (06) order f41
>   4.02 PCI1 init
>   4.02 PCI1 reset
>   5.23 PCIe timeout: 0x0001e08f
>   5.23 Failed to open device: 'nvme'
>   5.26 Retry NVME 1
>   5.28 PCI1 init
>   5.30 PCI1 reset
>   5.52 PCIe timeout: 0x0001e08f
>   5.52 Failed to open device: 'nvme'
> ..
> NOTICE:  BL31: v2.6(release):v2.6-239-g2a9ede0bd
> NOTICE:  BL31: Built : 14:26:57, Jun 22 2023
> [    0.815752] brcm-pcie 1000110000.pcie: link down

```
# PCIe related parameters
dtparam=pciex1
dtparam=pciex1_gen=2
#dtoverlay=pciex1-compat-pi5,l1ss
#dtoverlay=pciex1-compat-pi5,no-l0s
#dtoverlay=pciex1-compat-pi5,l1ss,no-l0s
#dtoverlay=pciex1-compat-pi5,no-mip
#dtoverlay=pciex1-compat-pi5,l1ss,no-mip
#dtoverlay=pciex1-compat-pi5,no-l0s,no-mip
#dtoverlay=pciex1-compat-pi5,l1ss,no-l0s,no-mip
#dtoverlay=pciex1-compat-pi5,mmio-hi
#dtoverlay=pciex1-compat-pi5,l1ss,mmio-hi
#dtoverlay=pciex1-compat-pi5,l1ss,no-l0s,mmio-hi
dtoverlay=pciex1-compat-pi5,l1ss,no-l0s,no-mip,mmio-hi
```

> Because it's a different version of the PCIe RC IP, with 2 lanes, Gen-3-capable, etc.

sorry, but do I read/understand that correctly here? External PCIe is 2-lanes, but Pi5 exposes one lane only? Or what is the background of this comment?
Under the context of having an external 2-lane PCIe interface the naming of the overlay would make some sense at least (for dealing with quirks when running it single lane only).

pelwell, 2024-06-18:
> External PCIe is 2-lanes, but Pi5 exposes one lane only?

No it's not - I had a feeling that one of the non-RP1 PCIe RCs was 2-lane capable, but I was remembering wrong and should have checked my facts first. The comment has been corrected to avoid further confusion.

iamursky, 2024-06-19:
Couldn't install both Ubuntu and Raspberry Pi OS on WD_BLACK SN770M 1TB.

Hardware:
- [Raspberry Pi 5 8GB](https://www.raspberrypi.com/products/raspberry-pi-5/)
- [Raspberry Pi M.2 HAT+](https://www.raspberrypi.com/products/m2-hat-plus/)
- [Raspberry Pi 27W USB-C Power Supply](https://www.raspberrypi.com/products/27w-power-supply/)
- [WD_BLACK SN770M M.2 NVMe 1TB SSD](https://www.westerndigital.com/products/internal-drives/wd-black-sn770m-nvme-ssd?sku=WDBDNH0010BBK-WRSN)

Here are captures made with the [Debug Probe](https://www.raspberrypi.com/products/debug-probe/):
- Ubuntu: https://gist.github.com/iamursky/1c7c25c751f97fce8f46a6ab755203d4
- Raspberry Pi OS: https://gist.github.com/iamursky/84f150bd435b08d26d457f4c8465f7fb

Raspberry Pi Forum post: https://forums.raspberrypi.com/viewtopic.php?t=372325

aBUGSworstnightmare-rpi, 2024-06-19:
@iamursky This is not related to the topic discussed in this issue most likely. Please open a new issue.

aBUGSworstnightmare-rpi, 2024-06-23:
as a result of my findings here https://github.com/raspberrypi/linux/issues/6228 I've ended up with a working NVMe on Pi5 now!

As I already was under the impression that this issue might be related to a firmware issue as well I've started to investigate further down that road. Testbed for what is shown next is the CM4 as the drive is running there!

started checking in detail again what I have in front of me
```
pi@raspberrypi:~ $ cat /sys/class/nvme/nvme0/model && cat /sys/class/nvme/nvme0/firmware_rev
SanDisk Ultra 3D NVMe                   
211070WD
```

checking based on https://wddashboarddownloads.wdc.com/wdDashboard/config/devices/lista_devices.xml reveals

```
<!-- Calypso - WD Blue and SanDisk Ultra -->
<lista_device model="SanDisk Ultra 3D NVMe">
<url>
 wdDashboard/firmware/SanDisk_Ultra_3D_NVMe/211210WD/device_properties.xml
</url>
```

**NOTE: WHAT IS SHOWN NEXT IS NOT OFFICIALLY SUPPORTED BY WD --> USE AT OWN RISK!!!**

checking the decency of that firmware shows it should be compatible (as it depends on 211070WD which is what my drive is using) 
```
pi@raspberrypi:~ $ curl https://wddashboarddownloads.wdc.com/wdDashboard/firmware/SanDisk_Ultra_3D_NVMe/211210WD/device_properties.xml

<?xml version="1.0"?>
<ffu version="1.0">
    <model>SanDisk Ultra 3D NVMe</model>
    <fwmode>mode3</fwmode>
    <fwversion>211210WD</fwversion>
    <fwfile>211210WD.fluf</fwfile>
    <dependency model="SanDisk Ultra 3D NVMe">211070WD</dependency>
</ffu>
```

the new firmware is '211210WD.fluf' so let's download it
```
pi@raspberrypi:~ $ curl --remote-name https://wddashboarddownloads.wdc.com/wdDashboard/firmware/SanDisk_Ultra_3D_NVMe_500GB/211210WD/211210WD.fluf
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   342    0   342    0     0    428      0 --:--:-- --:--:-- --:--:--   428
```

hmm .. only 342kb? looks strange but let's give it a try.

as firmware can be managed by name-cli I've started the process. Refer to nvmi-cli documentation for details on the commands in use.

```
pi@raspberrypi:~ $ sudo nvme fw-log /dev/nvme0
Firmware Log for device:nvme0
afi  : 0x1
frs1 : 0x4457303730313132 (211070WD)
pi@raspberrypi:~ $ sudo nvme id-ctrl /dev/nvme0 -H | grep Firmware
  [9:9] : 0x1	Firmware Activation Notices Supported
  [4:4] : 0x1	Firmware Activate Without Reset Supported
  [3:1] : 0x2	Number of Firmware Slots
  [0:0] : 0	Firmware Slot 1 Read/Write

pi@raspberrypi:~ $ sudo nvme fw-download -f 211210WD.fluf /dev/nvme0
Invalid size:342 for f/w image
```

BAM! That firmware file seems to be corrupted as it is reported as too small. **DEAD END?**

No! As I also have a SN550 in my possession (and lspci reports it as such anyhow - see https://github.com/raspberrypi/linux/issues/6190) I took a closer look to that one again as well:

```
pi@raspberrypi:~ $ cat /sys/class/nvme/nvme0/model && cat /sys/class/nvme/nvme0/firmware_rev
[... 271 lines cut ...]
host_read_commands			: 11,392,058
host_write_commands			: 5,927,484
controller_busy_time			: 72
power_cycles				: 559
power_on_hours				: 230
unsafe_shutdowns			: 146
media_errors				: 0
num_err_log_entries			: 1
Warning Temperature Time		: 0
Critical Composite Temperature Time	: 0
Thermal Management T1 Trans Count	: 0
Thermal Management T2 Trans Count	: 0
Thermal Management T1 Total Time	: 0
Thermal Management T2 Total Time	: 0
pi@m2hattest:~ $ sudo lspci -vvv | grep -E 'nvme|LnkSta'
		LnkSta:	Speed 5GT/s, Width x1
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete- EqualizationPhase1-
		LnkSta:	Speed 5GT/s (downgraded), Width x1 (downgraded)
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete- EqualizationPhase1-
	Kernel driver in use: nvme
		LnkSta:	Speed 5GT/s, Width x4
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete- EqualizationPhase1-
		LnkSta:	Speed 5GT/s, Width x4
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete- EqualizationPhase1-
```

closed 2024-06-23 by aBUGSworstnightmare-rpi