raspberrypi/linux #6196: /soc/nvmem_* nodes in RPI5 device tree violate DTS specification

lorc, 2024-05-29:
### Describe the bug

`bcm2712-rpi.dtsi` file provide `/soc/nvmem_*` nodes with `reg` values like `<0 192>` which violates DTS specification as those nodes reside inside `/soc/` node which has `ranges` defined as follows:

```
		ranges     = <0x7c000000  0x10 0x7c000000  0x04000000>;
```

So it is impossible to calculate, say, `/soc/nvmem_otp` address. Moreover, `reg` property inside those nodes does not provide memory address at all, which also violates dts, because `/soc/` node is compatible with "simple-bus".

This causes issues for non-Linux kernel. For example Xen is not happy about this.

### Steps to reproduce the behaviour

Just look at the source code (`arch/arm/boot/dts/broadcom/bcm2712-rpi.dtsi`). And compare this with Device Tree Specification (https://readthedocs.org/projects/devicetree-specification/downloads/pdf/latest/) sections `2.3.8` and `4.5`.

### Device (s)

Raspberry Pi 5

### System

Linux version rpi-6.6.y 

### Logs

This is how Xen complains about those entries:
```
I/TC: Added guest 1
(XEN) *** LOADING DOMAIN 0 ***
(XEN) Loading d0 kernel from boot module @ 0000000000e00000
(XEN) Allocating 1:1 mappings totalling 128MB for dom0:
(XEN) BANK[0] 0x00000010000000-0x00000018000000 (128MB)
(XEN) Grant table range: 0x00000002000000-0x00000002040000
(XEN) Unable to retrieve address 0 for /soc/nvmem_otp
(XEN) Device tree generation failed (-22).
(XEN) 
(XEN) ****************************************
(XEN) Panic on CPU 0:
(XEN) Could not set up DOM0 guest OS (rc = -22)
(XEN) ****************************************
(XEN) 
(XEN) Reboot in five seconds...
```

### Additional context

Proper solution is to provide own device with own addressing mode, like this:

```
	nvmem {
		compatible = "raspberrypi,rpi-otp";
		#address-cells = <1>;
		#size-cells = <1>;
		firmware = <&firmware>;
		nvmem_otp: nvmem_otp {
			reg = <0 192>;
		};

		nvmem_cust: nvmem_cust {
			reg = <1 8>;
		};

		nvmem_mac: nvmem_mac {
			reg = <2 6>;
		};

		nvmem_priv: nvmem_priv {
			reg = <3 16>;
		};
	};
```

This is how other devices like SPI or I2C controllers are working.

pelwell, 2024-05-29:
How is this different from an SPI or I2C controller, both of which have their own kind of addressing? There is no memory address for an I2C device or an SPI CS line.

Section 2.3.8 of the specification you link to where it talks about the `ranges` property says:
> If the property is not present in a bus node, it is assumed that no mapping exists between children of the node
and the parent address space.

`nvmem` above does not include a `ranges` property, because _no mapping exists between children of the node and the parent address space_.

pelwell, 2024-05-29:
I suspect that Xen is seeing the size values and _assuming_ it is dealing with something which is CPU addressable. I2C and SPI chilren have no sizes, so this may have seemed like the right thing to do, but the truth is less simple.

pelwell, 2024-05-29:
Show me an example of an I2C controller with an I2C device, with the controller being a child of a simple bus. Then explain how our nvmem node is not like the I2C controller.

pelwell, 2024-05-29:
Also look at the cpus node in figure 2.1.

lorc, 2024-05-29:
> Show me an example of an I2C controller with an I2C device, with the controller being a child of a simple bus.

I hope you don't mind decompiled example, because it is easier than providing links to bunch of dtsi files:

```
/ {
[...]
	soc {
		compatible = "simple-bus";
		interrupt-parent = <0x11>;
		#address-cells = <0x02>;
		#size-cells = <0x02>;
		ranges;
[...]

		i2c@e66d8000 {
			compatible = "renesas,i2c-r8a779f0\0renesas,rcar-gen4-i2c";
			reg = <0x00 0xe66d8000 0x00 0x40>;
			interrupts = <0x00 0xf2 0x04>;
			clocks = <0x0b 0x01 0x20a>;
			power-domains = <0x09 0x20>;
			resets = <0x0b 0x20a>;
			dmas = <0x16 0x99 0x16 0x98 0x17 0x99 0x17 0x98>;
			dma-names = "tx\0rx\0tx\0rx";
			i2c-scl-internal-delay-ns = <0x6e>;
			#address-cells = <0x01>;
			#size-cells = <0x00>;
			status = "okay";
			pinctrl-0 = <0x19>;
			pinctrl-names = "default";
			clock-frequency = <0x61a80>;
			xen,passthrough;

			eeprom@50 {
				compatible = "rohm,br24g01\0atmel,24c01";
				label = "cpu-board";
				reg = <0x50>;
				pagesize = <0x08>;
			};

			eeprom@52 {
				compatible = "rohm,br24g01\0atmel,24c01";
				label = "ethernet-sub-board";
				reg = <0x52>;
				pagesize = <0x08>;
			};

			eeprom@51 {
				compatible = "rohm,br24g01\0atmel,24c01";
				label = "breakout-board";
				reg = <0x51>;
				pagesize = <0x08>;
			};
		};
[...]
};
```

>  Then explain how our nvmem node is not like the I2C controller.

There is no `nvmem` node in the `bcm2712-rpi.dtsi`. You are referring to my example with proposed fix to the issue.
In `bcm2712-rpi.dtsi` there are bunch of  `/soc/nvmem_*` nodes.

> Also look at the cpus node in figure 2.1.

`cpus` is the separate node with own address space. This is completely legal. `nvmem` should be done in the exact same way.

I am telling that there is a problem because there are bunch of `/soc/nvmem_*` entries, each with own `reg` property. As they are direct childs of `/soc`, they must obey `/soc` bus addressing rules. But they are not.

What I am proposing is to create `/soc/nvmem` node without `reg` property, but with `#address_cells` and `#size_cells`.  This will start new address space as intended by Device Tree specification.  Then put those `nvmem_*` entries inside it /soc/nvmem` and voila, you have  completely correct DTS.

GrygiriiS, 2024-05-30:
@pelwell You brutally abusing DT with this nvmemX nodes (and not only), and how you are referring to I2C above regarding this issue - sorry, but it is absolute incorrect.
The  I2C is a **BUS** (and as other *busses*  like MDIO, etc) It defines it's own `#address-cells/#size-cells`.
  For example: ` #address-cells = <0x01>; #size-cells = <0x00>;`

The  `soc` is a generic bus intended to contain:

1. IO mapped SoC HW modules definitions
2. other buses
Nothing else.

These nvmemX nodes (which actually not describing HW at all, but rather firmware features (SW) ) should be defined as child nodes of `firmware` node  (seems like) or other MFD/syscon like nodes, and corresponding devices populated by it if required.

Even "firmaware" node should not be part of the `soc `bus - it's not HW. It should be in `root`.

pelwell, 2024-05-30:
Easy - I've realised my mistake (skim reading), and will come up with a response later.

lorc, 2024-05-30:
No problem. Thanks for the quick fix. I verified it on my setup and it is working as intended.

lorc, 2024-05-30:
Fixed via #6197

closed 2024-05-30 by lorc