raspberrypi/linux #6198: kernel 6.6.x compilation crash when CONFIG_BACKLIGHT_CLASS_DEVICE is not y

jlsalvador, 2024-06-01:
### Describe the bug

When CONFIG_BACKLIGHT_CLASS_DEVICE is not y; `backlight_set_display_name` will not be defined, so compiling `drivers/gpu/drm/bridge/panel.o` will crash with an `backlight_set_display_name undefined`

This issue was added by commit 573f8fd0abf1d63ef719672a3c26e7abc0169620

Tested on the 6.6.x branch.

### Steps to reproduce the behaviour

Generate a kernel config with CONFIG_BACKLIGHT_CLASS_DEVICE not set.

Try to compile the kernel.

### Device (s)

Raspberry Pi 4 Mod. B, Raspberry Pi 5, Other

### System

Cross compiling kernel by Buildroot.

6by9, 2024-06-01:
It should be defined as an empty function via the #else at https://github.com/raspberrypi/linux/blob/rpi-6.6.y/include/linux/backlight.h#L496-L500

I'll test on Monday.

pelwell, 2024-06-01:
Please provide an example configuration showing the build failure.

jlsalvador, 2024-06-01:
> Please provide an example configuration showing the build failure.

https://gist.github.com/jlsalvador/37316ff616a0428adbac7cc2cecdddab

Please, take a look at the bottom, the `output.txt` file.

6by9, 2024-06-01:
Your original description said CONFIG_BACKLIGHT_CLASS_DEVICE not set. I believe that should be fine.

The awkwardness is actually `CONFIG_DRM_PANEL=y` at line 4404 with `CONFIG_BACKLIGHT_CLASS_DEVICE=m`. Having that built in when backlight is a module will lead to a link error. There are magic Kconfig runes around forcing modules to be built in to match other built ins, but it becomes ugly - there's no obligation for selecting BACKLIGHT from DRM_PANEL, and having BACKLIGHT declare all the modules that could use it and change structure accordingly is equally ugly.

I suspect I need to resort to `#ifdef` in the code. It looks like drm_panel already has 
`#if [IS_REACHABLE](https://elixir.bootlin.com/linux/latest/C/ident/IS_REACHABLE)([CONFIG_BACKLIGHT_CLASS_DEVICE](https://elixir.bootlin.com/linux/latest/K/ident/CONFIG_BACKLIGHT_CLASS_DEVICE))`
for accessing `devm_of_find_backlight` which is similarly stubbed.

adminy, 2024-06-18:
I got:
```
QUESTION: Lowlevel Backlight controls, NAME: BACKLIGHT_CLASS_DEVICE, ALTS: M/y/?, ANSWER: n
GOT: n
GOT: 
GOT: CONFIG_BACKLIGHT_CLASS_DEVICE:
GOT: 
GOT: This framework adds support for low-level control of the LCD
GOT: backlight. This includes support for brightness and power.
GOT: 
GOT: To have support for your specific LCD panel you will have to
GOT: select the proper drivers which depend on this option.
GOT: 
GOT: Symbol: BACKLIGHT_CLASS_DEVICE [=m]
GOT: Type  : tristate
GOT: Defined at drivers/video/backlight/Kconfig:136
GOT:   Prompt: Lowlevel Backlight controls
GOT:   Depends on: HAS_IOMEM [=y]
GOT:   Location:
GOT:     -> Device Drivers
GOT:       -> Graphics support
GOT:         -> Backlight & LCD device support
GOT:           -> Lowlevel Backlight controls (BACKLIGHT_CLASS_DEVICE [=m])
GOT: Selected by [m]:
GOT:   - DRM_RADEON [=m] && HAS_IOMEM [=y] && DRM [=y] && PCI [=y] && MMU [=y] && (AGP [=n] || !AGP [=n])
GOT:   - DRM_AMDGPU [=m] && HAS_IOMEM [=y] && DRM [=y] && PCI [=y] && MMU [=y] && !UML
GOT:   - DRM_NOUVEAU [=m] && HAS_IOMEM [=y] && DRM [=y] && PCI [=y] && MMU [=y] && DRM_NOUVEAU_BACKLIGHT [=y]
GOT:   - DRM_PANEL_MIPI_DBI [=m] && HAS_IOMEM [=y] && DRM [=y] && SPI [=y]
GOT:   - TINYDRM_HX8357D [=m] && HAS_IOMEM [=y] && DRM [=y] && SPI [=y]
GOT:   - TINYDRM_ILI9163 [=m] && HAS_IOMEM [=y] && DRM [=y] && SPI [=y]
GOT:   - TINYDRM_ILI9341 [=m] && HAS_IOMEM [=y] && DRM [=y] && SPI [=y]
GOT:   - TINYDRM_ILI9486 [=m] && HAS_IOMEM [=y] && DRM [=y] && SPI [=y]
GOT:   - TINYDRM_MI0283QT [=m] && HAS_IOMEM [=y] && DRM [=y] && SPI [=y]
GOT:   - TINYDRM_ST7735R [=m] && HAS_IOMEM [=y] && DRM [=y] && SPI [=y]
GOT:   - DRM_GUD [=m] && HAS_IOMEM [=y] && DRM [=y] && USB [=y] && MMU [=y]
GOT:   - DRM_SSD130X [=m] && HAS_IOMEM [=y] && DRM [=y] && MMU [=y]
GOT:   - DRM_PARADE_PS8622 [=m] && HAS_IOMEM [=y] && DRM [=y] && DRM_BRIDGE [=y] && OF [=y]
GOT:   - FB_ARMCLCD [=m] && (ARM || ARM64 [=y] || COMPILE_TEST [=n]) && FB [=y] && ARM_AMBA [=y] && HAS_IOMEM [=y] && OF [=y]
GOT:   - FB_BACKLIGHT [=m] && HAS_IOMEM [=y] && FB [=y]
GOT:   - USB_APPLEDISPLAY [=m] && USB_SUPPORT [=y] && USB [=y]
GOT: Selected by [n]:
GOT:   - DRM_NOUVEAU [=m] && HAS_IOMEM [=y] && DRM [=y] && PCI [=y] && MMU [=y] && ACPI [=n] && X86
GOT:   - DRM_I915 [=n] && HAS_IOMEM [=y] && DRM [=y] && X86 && PCI [=y] && !PREEMPT_RT [=n] && ACPI [=n]
GOT:   - DRM_GMA500 [=n] && HAS_IOMEM [=y] && DRM [=y] && PCI [=y] && X86 && MMU [=y] && ACPI [=n]
GOT:   - DRM_SHMOBILE [=n] && HAS_IOMEM [=y] && DRM [=y] && (ARCH_RENESAS [=n] || ARCH_SHMOBILE || COMPILE_TEST [=n])
GOT:   - DRM_TILCDC [=n] && HAS_IOMEM [=y] && DRM [=y] && OF [=y] && ARM
GOT:   - DRM_FSL_DCU [=n] && HAS_IOMEM [=y] && DRM [=y] && OF [=y] && ARM && COMMON_CLK [=y]
GOT:   - FB_OLPC_DCON [=n] && STAGING [=y] && OLPC && FB [=y] && BROKEN [=n] && I2C [=y] && GPIO_CS5535 [=n] && ACPI [=n]
GOT:   - SAMSUNG_Q10 [=n] && X86_PLATFORM_DEVICES [=n] && ACPI [=n]
GOT:   - ACPI_CMPC [=n] && X86_PLATFORM_DEVICES [=n] && ACPI [=n] && INPUT [=y] && (RFKILL [=m] || RFKILL [=m]=n [=n])
GOT: 
GOT: 
GOT: 
QUESTION: Lowlevel Backlight controls, NAME: BACKLIGHT_CLASS_DEVICE, ALTS: M/y/?, ANSWER: n
repeated question: Lowlevel Backlight controls at /nix/store/k5nz4dqvifjqgr2m3ya4n1012jnn9zjb-generate-config.pl line 88.

x999 of:
Error in reading or end of file.
```
And even set `CONFIG_DRM_PANEL=n`

6by9, 2024-06-18:
I'm struggling to get a configuration that fails, but #6230 should work around it anyway.

pelwell, 2024-06-18:
That's now merged. Please test again.

adminy, 2024-06-18:
okay, thank you

jlsalvador, 2024-06-20:
Fixed by da87f91ad8450ccc5274cd7b6ba8d823b396c96f
Thank you all!

closed 2024-06-20 by jlsalvador