raspberrypi/linux #6200: Patch for CVE-2024-1086 not available in Bullseye stable (6.1.21) yet, kernel still vulnerable

bcutter, 2024-06-02:
### Describe the bug

Cause: CVE-2024-1086

Patch https://github.com/raspberrypi/linux/commit/8e34430e33b8a80bc014f3efe29cac76bc30a4b4 seems to not be merged to any stable release. Bullseye 6.1.21 dates back to 2023-04-03 and therefore is vulnerable. It has been patched already in original Debian release (see https://security-tracker.debian.org/tracker/CVE-2024-1086), Raspberry Pi OS seems to stay behind.

Background: https://www-heise-de.translate.goog/news/Linux-Luecke-Angreifer-verschaffen-sich-root-Rechte-9742699.html?_x_tr_sl=auto&_x_tr_tl=en&_x_tr_hl=de&_x_tr_pto=wapp

Not sure if Raspberry Pi OS Bookworm release channel kernels are also still affected

### Steps to reproduce the behaviour

Use Raspberry Pi OS with latest kernel.

### Device (s)

Raspberry Pi 4 Mod. B

### System

Raspberry Pi OS v11 (Bullseye)

### Additional context

https://github.com/Notselwyn/CVE-2024-1086

popcornmix, 2024-06-04:
I've bumped the 6.1 tree to latest upstream stable, which does include the referenced commit.
I've built a candidate kernel which can be got with:
`sudo rpi-update oldstable`

If you could test this, it would be helpful.

We'll update the apt kernel if no regressions are reported in this test kernel.

bcutter, 2024-06-04:
Great, thanks.

Unfortunately as a passionate user I don‘t do rpi-update anymore - strictly on my productive systems. I only have one test system (Pi 2 B, 32 bit OS) left. I can test it there but as it is not used productively my feedback would likely be limited to „boots, runs, no issues discovered so far“.

Let me know if that’s helpful at all or sufficient.

pelwell, 2024-06-04:
> as a passionate user I don‘t do rpi-update anymore

Passion usually lends to a lack of caution...

popcornmix, 2024-06-04:
> Let me know if that’s helpful at all or sufficient.

Any testing is better than no testing, so go ahead.

bcutter, 2024-06-04:
> > as a passionate user I don‘t do rpi-update anymore
> 
> 
> 
> Passion usually leads to a lack of caution...

😁

Maybe a bit lost in translation. The focus is on "user" not the adjective in front of it. Spent 2 whole weekends recently with Pi system mgmt stuff (OS maintenance and kernel things) so indeed the lack of free time equals the lack of passion currently.

In general I was a bit surprised this has not been fixed (shipped) yet while all security magazines are talking bout this CVE and it has been fixed in most distributions.

How to downgrade / revert the rpi-update just in case? If I remind correctly that never was possible (bleeding edge/beta back to stable).

bcutter, 2024-06-14:
`sudo rpi-update oldstable` on a test system:
- Before: `Linux Test 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux`
- After: `______`

---

Update not performed yet, because: Is this correct? "bumps to rpi-5.10.y linux tree" is a bit confusing:

[image]

pelwell, 2024-06-14:
That's a human-written comment that, quite reasonably, hasn't been updated for this trial build.

popcornmix, 2024-06-14:
I have now updated the notice file (so the message will change), but you'll actually get the 6.1 kernel either way.

bcutter, 2024-06-14:
Looking better now, including some more helpful details on firmware/bootloader versions:

[image]

...even the information bout the eeprom firmware also confuses a bit. Is it "too old" because it is Bullseye or because it's not a Pi 4 upwards hardware and therefore doesn't even have an eeprom?

---

`sudo rpi-update oldstable` on a test system (Pi 2 B hardware):

- Before: `Linux Test 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux`
- After: `Linux Test 6.1.77-v7+ #1768 SMP Tue Jun  4 14:45:11 BST 2024 armv7l GNU/Linux`

Well, it booted fine so... nothing to complain I guess. As mentioned: test system.
I only noticed by the way that the former kernel files (6.1.21*) in `/lib/modules` remained (expected them to get removed, but maybe that's only done with apt packages and rpi-update behaves different).

bcutter, 2024-07-14:
Still running/working.

Status/progress? Enough testing/confidence to release it or...?