raspberrypi/linux #6202: Why is there no DT overlay for the ofdicial M.2 HAT+?

aBUGSworstnightmare-rpi, 2024-06-03:
### Describe the bug

From the HAT+ specification:

The ID EEPROM content provides: 
• A product UUID (required) 
• A product ID (optional – zero if not used) 
• A product version (optional – zero if not used) 
• A vendor name string (required), e.g. “ACME Technology Company” 
• A product description string (required), e.g. “Special Sensor Board” 
**--> • A Device Tree overlay name string (required) <--**
• Other user defined data (optional)

Why is there no overlay for the official M.2 HAT+ although it is specified as mandatory?

Having the EEPROM on the HAT+ has stoped my DPI display from working -> why was an EEPROM added when it just holds 'informative (production traceability) data'?

> # ---------- Dump generated by eepdump handling format version 0x02 ----------
> #
> # --Header--
> # signature=0x69502d52
> # version=0x02
> # reserved=0
> # numatoms=1
> # eeplen=70
> # ----------
> 
> 
> # Start of atom #0 of type 0x0001 and length 50
> # Vendor info
> product_uuid 833ad001-a8e0-4779-8e6f-3c4ee6676e9c
> product_id 0x0000
> product_ver 0x0003
> vendor "Raspberry Pi"   # length=12
> product "M.2 HAT+ M Key"   # length=14
> # End of atom. CRC16=0x1a90

### Steps to reproduce the behaviour

Check the kernel source, no such overlay existant.

### Device (s)

Raspberry Pi 5

### System

Linux hatplus 6.6.31-v8-16k+ https://github.com/raspberrypi/linux/issues/1766 SMP PREEMPT Fri May 24 12:15:34 BST 2024 aarch64 GNU/Linux

pelwell, 2024-06-03:
Because the documentation is wrong. That should be fixed once the documentaiton logjam clears.

aBUGSworstnightmare-rpi, 2024-06-03:
> Because the documentation is wrong. That should be fixed once the documentation logjam clears.

That said will allow me to remove the EEPROM incl I2C pull-ups to get my DPI screen back into working condition. Thanks.

closed 2024-06-03 by aBUGSworstnightmare-rpi