raspberrypi/linux #6207: RPi5 I2C HDMI interface (/dev/i2c-2) not working from userspace

cagnulein, 2024-06-05:
### Describe the bug

Hi,

I'm in the process of migrating a project from RPI4 to RPI5 (4GB).

One of the functions on RPI4 was turning on/off and changing the brightness of HDMI connected LCD via ddcutil command (details see https://www.ddcutil.com/raspberry/).

Unfortunately, on RPI5 I'm struggling with accessing /dev/i2c-2 HDMI1 I2C from userspace:

"ddcutil detect" command takes long, then outputs "No displays found".

root@raspberrypi:/home/pi# ls -la /dev/i2c*
crw-rw---- 1 root i2c 89, 1 Jun  5 09:33 /dev/i2c-1

root@raspberrypi:/home/pi# uname -a
Linux raspberrypi 6.6.31-v8+ #1766 SMP PREEMPT Fri May 24 12:14:10 BST 2024 aarch64 GNU/Linux

ddcutil --version
ddcutil 0.9.5

Any idea how to fix the problem? Thanks!

### Steps to reproduce the behaviour

# ddcutil detect

### Device (s)

Raspberry Pi 5

### System

Raspberry Pi reference 2020-05-27
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 825107f04027269db77426046f5085475b1ea22f, stage4

root@raspberrypi:/home/pi# vcgencmd version
VCHI initialization failed

root@raspberrypi:/home/pi# uname -a
Linux raspberrypi 6.6.31-v8+ #1766 SMP PREEMPT Fri May 24 12:14:10 BST 2024 aarch64 GNU/Linux

6by9, 2024-06-05:
What distribution are you using?

Raspberry Pi OS Bookworm 64bit
```
pi@bookworm64:~ $ i2cdetect -l
i2c-11	i2c       	107d508200.i2c                  	I2C adapter
i2c-12	i2c       	107d508280.i2c                  	I2C adapter
```
i2c-11 is HDMI-1 (closest to the USB-C port), and i2c-12 is HDMI-2.
```
pi@bookworm64:~ $ ddcutil --version
ddcutil 1.4.1
```
```
pi@bookworm64:~ $ ddcutil detect
Display 1
   I2C bus:  /dev/i2c-11
   DRM connector:           card1-HDMI-A-1
   EDID synopsis:
      Mfg id:               DEL - Dell Inc.
      Model:                DELL P2214H
      Product code:         41112  (0xa098)
      Serial number:        29C2953U9R5S
      Binary serial number: 961688915 (0x39523553)
      Manufacture year:     2015,  Week: 14
   VCP version:         2.1
```

cagnulein, 2024-06-05:
sorry fixed, it was a xorg issue after all

closed 2024-06-05 by cagnulein