raspberrypi/linux #6224: RBD module should be added

rbartl, 2024-06-15:
### Describe the bug

RBD module is not available in compiled Raspberry linux kernel images but is needed for CEPH

in #2916 CEPH module was added - additionally RBD would also be very useful to use e.g. CEPH ROOK on raspberry systems

i tried looking into various branches in /arch/arm64/config/*defconfig* and found no mention of RBD.
having it as a module should not have many downsides apart from a bit more compliation time and a bit bigger deb package.

### Steps to reproduce the behaviour

using raspbian (debian on raspberry) 
:/boot# grep CONFIG_BLK_DEV_RBD /boot/config*
/boot/config-6.6.28+rpt-rpi-2712:# CONFIG_BLK_DEV_RBD is not set

or

/boot# modprobe rbd
modprobe: FATAL: Module rbd not found in directory /lib/modules/6.6.31+rpt-rpi-v8

### Device (s)

Raspberry Pi 4 Mod. B

### System

/boot# cat /etc/rpi-issue
Raspberry Pi reference 2023-11-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 8b2feb9380ee45e03cc4c5753dbf6ca7ea70bc6f, stage6

vcgencmd version
May 24 2024 15:30:04
Copyright (c) 2012 Broadcom
version 4942b7633c0ff1af1ee95a51a33b56a9dae47529 (clean) (release) (start)

uname -a
Linux k3sraspi0 6.6.31+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux

rbartl, 2024-06-15:
it was already merged in PR: https://github.com/dwrobel/kernel/pull/3/commits and discussed in https://github.com/raspberrypi/linux/pull/4461 
but seems to be not enabled in current 6' versions

pelwell, 2024-06-17:
New PR here: #6226

snooops, 2024-06-26:
Is there a way to test the kernel with the this merge?

pelwell, 2024-06-26:
`sudo rpi-update pulls/6226` is the way to test a pull request before (and after) it has been merged. However, once it has been merged, all future kernel builds will include it, so you could run `sudo rpi-update rpi-6.6.y` to get the current 6.6 build, or after a few days, just `sudo rpi-update`.

There's a list of all rpi-update firmwares here: https://github.com/raspberrypi/rpi-firmware/commits/master
Anything after (chronologically, before when reading down the page) the 6.6.34 release contains the RBD module.

snooops, 2024-06-27:
Thank you for the explaination pelwell, i can confirm that the rbd module works. 
My Proxmox is now able to work with ceph disks and to use the rbd kernel module.

xorinzor, 2025-02-15:
Has this been included in future versions now? Since the issue is still open.

pelwell, 2025-02-15:
We normally leave it to the opener to close issues, but that doesn't always happen.