raspberrypi/linux #6228: Incompatible NVMe: 250GB Western Digital PC SN530 (Pi 5, 8GB, X1001)

twoplan, 2024-06-17:
### Describe the bug

The **250GB WD PC SN530** NVMe is not detected with an Geekworm X1001 Adapter on my 8GB Pi 5. 
Tested with dtparam=pciex1_gen=2 and 3.

With a different 500GB NVMe **Sandisk SN740**, the X1001 NVMe adapter works perfect (dtparam=pciex1_gen=3).

### Steps to reproduce the behaviour

You need a SN530 NVMe from Western Digital.

### Device (s)

Raspberry Pi 5

### System

cat /etc/rpi-issue
```
Raspberry Pi reference 2024-03-15
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, f19ee211ddafcae300827f953d143de92a5c6624, stage4
```
vcgencmd version
```
2024/06/05 16:41:49 
Copyright (c) 2012 Broadcom
version 6fe0b091 (release) (embedded)
```
uname -a
`Linux raspi5 6.6.31+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux`

### Logs

**250GB Western Digital PC SN530**
sudo lspci
```
00:00.0 PCI bridge: Broadcom Inc. and subsidiaries Device 2712 (rev 21)
01:00.0 Ethernet controller: Device 1de4:0001
```
With 

**500GB Sandisk SN740**
sudo lspci
```
0000:00:00.0 PCI bridge: Broadcom Inc. and subsidiaries Device 2712 (rev 21)
0000:01:00.0 Non-Volatile memory controller: Sandisk Corp WD PC SN740 NVMe SSD 512GB (DRAM-less) (rev 01)
0001:00:00.0 PCI bridge: Broadcom Inc. and subsidiaries Device 2712 (rev 21)
0001:01:00.0 Ethernet controller: Device 1de4:0001
```

timg236, 2024-06-17:
Possibly a duplicate of https://github.com/raspberrypi/linux/issues/6195

aBUGSworstnightmare-rpi, 2024-06-17:
> Possibly a duplicate of #6195
Not a duplicate! PCIe device Id's of the drives are different.
Unknown if they were using the same firmware (so far at least). I will test two more SN350 labeled devices later this week in order to check what @twoplan is seeing.

Btw: I have another SN350 labeled device (again different ID) which is working fine but most likely a re-labeled SN550.

aBUGSworstnightmare-rpi, 2024-06-18:
Tests concluded!
2242 sized SN530 is working here for me
it shows expected ID 
`-+-[0000:00]---00.0-[01]----00.0  Sandisk Corp PC SN530 NVMe SSD (DRAM-less) [15b7:5008]`
[image]
[image]
[image]

> pi@m2hattest:~ $ sudo nvme --smart-log /dev/nvme0n1 -H
> Smart Log for NVME device:nvme0n1 namespace-id:ffffffff
> critical_warning			: 0
>       Available Spare[0]             : 0
>       Temp. Threshold[1]             : 0
>       NVM subsystem Reliability[2]   : 0
>       Read-only[3]                   : 0
>       Volatile mem. backup failed[4] : 0
>       Persistent Mem. RO[5]          : 0
> temperature				: 26°C (299 Kelvin)
> available_spare				: 100%
> available_spare_threshold		: 10%
> percentage_used				: 0%
> endurance group critical warning summary: 0
> Data Units Read				: 58,670 (30.04 GB)
> Data Units Written			: 56,463 (28.91 GB)
> host_read_commands			: 829,446
> host_write_commands			: 255,675
> controller_busy_time			: 3
> power_cycles				: 14
> power_on_hours				: 2
> unsafe_shutdowns			: 2
> media_errors				: 0
> num_err_log_entries			: 1
> Warning Temperature Time		: 0
> Critical Composite Temperature Time	: 0
> Thermal Management T1 Trans Count	: 0
> Thermal Management T2 Trans Count	: 0
> Thermal Management T1 Total Time	: 0
> Thermal Management T2 Total Time	: 0
> pi@m2hattest:~ $ sudo nvme id-ctrl -H /dev/nvme0
> NVME Identify Controller:
> vid       : 0x15b7
> ssvid     : 0x15b7
> sn        : 22293C452220        
> mn        : WDC PC SN530 SDBPMPZ-256G-1101          
> fr        : 21160001
> rab       : 4
> ieee      : 001b44
> cmic      : 0
>   [3:3] : 0	ANA not supported
>   [2:2] : 0	PCI
>   [1:1] : 0	Single Controller
>   [0:0] : 0	Single Port
> 
> mdts      : 7
> cntlid    : 0x1
> ver       : 0x10400
> rtd3r     : 0x7a120
> rtd3e     : 0xf4240
> oaes      : 0x200
>   [31:31] : 0	Discovery Log Change Notice Not Supported
>   [27:27] : 0	Zone Descriptor Changed Notices Not Supported
>   [15:15] : 0	Normal NSS Shutdown Event Not Supported
>   [14:14] : 0	Endurance Group Event Aggregate Log Page Change Notice Not Supported
>   [13:13] : 0	LBA Status Information Notices Not Supported
>   [12:12] : 0	Predictable Latency Event Aggregate Log Change Notices Not Supported
>   [11:11] : 0	Asymmetric Namespace Access Change Notices Not Supported
>   [9:9] : 0x1	Firmware Activation Notices Supported
>   [8:8] : 0	Namespace Attribute Changed Event Not Supported
> 
> ctratt    : 0x2
[... 199 lines cut ...]
> ioccsz    : 0
> iorcsz    : 0
> icdoff    : 0
> fcatt     : 0
>   [0:0] : 0	Dynamic Controller Model
> 
> msdbd     : 0
> ofcs      : 0
>   [0:0] : 0	Disconnect command Not Supported
> 
> ps      0 : mp:3.50W operational enlat:0 exlat:0 rrt:0 rrl:0
>             rwt:0 rwl:0 idle_power:0.6300W active_power:1.80W
>             active_power_workload:80K 128KiB SW
> ps      1 : mp:2.40W operational enlat:0 exlat:0 rrt:0 rrl:0
>             rwt:0 rwl:0 idle_power:0.6300W active_power:1.60W
>             active_power_workload:80K 128KiB SW
> ps      2 : mp:1.90W operational enlat:0 exlat:0 rrt:0 rrl:0
>             rwt:0 rwl:0 idle_power:0.6300W active_power:1.50W
>             active_power_workload:80K 128KiB SW
> ps      3 : mp:0.0250W non-operational enlat:3900 exlat:11000 rrt:3 rrl:3
>             rwt:3 rwl:3 idle_power:0.0250W active_power:-
>             active_power_workload:-
> ps      4 : mp:0.0050W non-operational enlat:5000 exlat:39000 rrt:4 rrl:4
>             rwt:4 rwl:4 idle_power:0.0050W active_power:-
>             active_power_workload:-

aBUGSworstnightmare-rpi, 2024-06-18:
2280 sized SN530 is working here for me
**it shows an unexpected ID**
`pi@m2hattest:~ $ lspci -tvnn
-+-[0000:00]---00.0-[01]----00.0  Sandisk Corp WD Blue SN550 NVMe SSD [15b7:5009]`
which is in fact the same as the drive in https://github.com/raspberrypi/linux/issues/6195
so this one seems to be a relabeled SN550 in 256GB and/or the data base is incorrect.
[image]
[image]
[image]

> pi@m2hattest:~ $ sudo nvme --smart-log /dev/nvme0n1 -H
> Smart Log for NVME device:nvme0n1 namespace-id:ffffffff
> critical_warning			: 0
>       Available Spare[0]             : 0
>       Temp. Threshold[1]             : 0
>       NVM subsystem Reliability[2]   : 0
>       Read-only[3]                   : 0
>       Volatile mem. backup failed[4] : 0
>       Persistent Mem. RO[5]          : 0
> temperature				: 27°C (300 Kelvin)
> available_spare				: 100%
> available_spare_threshold		: 10%
> percentage_used				: 0%
> endurance group critical warning summary: 0
> Data Units Read				: 206,111 (105.53 GB)
> Data Units Written			: 246,671 (126.30 GB)
> host_read_commands			: 1,846,942
> host_write_commands			: 1,113,758
> controller_busy_time			: 10
> power_cycles				: 30
> power_on_hours				: 4
> unsafe_shutdowns			: 16
> media_errors				: 0
> num_err_log_entries			: 1
> Warning Temperature Time		: 0
> Critical Composite Temperature Time	: 0
> Thermal Management T1 Trans Count	: 0
> Thermal Management T2 Trans Count	: 0
> Thermal Management T1 Total Time	: 0
> Thermal Management T2 Total Time	: 0
> pi@m2hattest:~ $ sudo nvme id-ctrl -H /dev/nvme0
> NVME Identify Controller:
> vid       : 0x15b7
> ssvid     : 0x15b7
> sn        : 2210D2494001        
> mn        : WDC PC SN530 SDBPNPZ-256G-1114          
> fr        : 21113000
> rab       : 4
> ieee      : 001b44
> cmic      : 0
>   [3:3] : 0	ANA not supported
>   [2:2] : 0	PCI
>   [1:1] : 0	Single Controller
>   [0:0] : 0	Single Port
> 
> mdts      : 7
> cntlid    : 0x1
> ver       : 0x10400
> rtd3r     : 0x7a120
> rtd3e     : 0xf4240
> oaes      : 0x200
>   [31:31] : 0	Discovery Log Change Notice Not Supported
>   [27:27] : 0	Zone Descriptor Changed Notices Not Supported
>   [15:15] : 0	Normal NSS Shutdown Event Not Supported
>   [14:14] : 0	Endurance Group Event Aggregate Log Page Change Notice Not Supported
>   [13:13] : 0	LBA Status Information Notices Not Supported
>   [12:12] : 0	Predictable Latency Event Aggregate Log Change Notices Not Supported
>   [11:11] : 0	Asymmetric Namespace Access Change Notices Not Supported
>   [9:9] : 0x1	Firmware Activation Notices Supported
>   [8:8] : 0	Namespace Attribute Changed Event Not Supported
[... 201 lines cut ...]
> ioccsz    : 0
> iorcsz    : 0
> icdoff    : 0
> fcatt     : 0
>   [0:0] : 0	Dynamic Controller Model
> 
> msdbd     : 0
> ofcs      : 0
>   [0:0] : 0	Disconnect command Not Supported
> 
> ps      0 : mp:3.50W operational enlat:0 exlat:0 rrt:0 rrl:0
>             rwt:0 rwl:0 idle_power:0.6300W active_power:1.80W
>             active_power_workload:80K 128KiB SW
> ps      1 : mp:2.40W operational enlat:0 exlat:0 rrt:0 rrl:0
>             rwt:0 rwl:0 idle_power:0.6300W active_power:1.60W
>             active_power_workload:80K 128KiB SW
> ps      2 : mp:1.90W operational enlat:0 exlat:0 rrt:0 rrl:0
>             rwt:0 rwl:0 idle_power:0.6300W active_power:1.50W
>             active_power_workload:80K 128KiB SW
> ps      3 : mp:0.0250W non-operational enlat:3900 exlat:11000 rrt:3 rrl:3
>             rwt:3 rwl:3 idle_power:0.0250W active_power:-
>             active_power_workload:-
> ps      4 : mp:0.0050W non-operational enlat:5000 exlat:39000 rrt:4 rrl:4
>             rwt:4 rwl:4 idle_power:0.0050W active_power:-
>             active_power_workload:-

aBUGSworstnightmare-rpi, 2024-06-18:
Note: `pciex1-compat-pi5` wasn't used here!

twoplan, 2024-06-18:
@aBUGSworstnightmare-rpi
The **firmware is different**. Yours is **21113000** from 04MAR 2022. 
And mine is **21103900** from 24JAN 2022.

[image]

aBUGSworstnightmare-rpi, 2024-06-18:
@twoplan noted! Will test on Pimoroni NVMe base later. Official M.2 HAT+ has the 32kHz sub clock mounted which isn't there on your HW and the NVMe base (at least the revision which I have).

Will also test on CM4 to see if the PCIe IP on CM4 has any issues with these drives.

aBUGSworstnightmare-rpi, 2024-06-20:
Both SN530 drives (2245 and 2280size) were check with Pimoroni NVMe base as well as on CM4: working on both platforms.

as @twoplan 's isn't working it might be a firmware issue. I'm now investigating possibilities of doing a FW update (on both SN530 as well as the still non-working SN550).

Will check by using Western Digital SSD Dashboard (https://support-en.wd.com/app/answers/detailweb/a_id/31759#subject3 or https://support-en.wd.com/app/products/downloads/softwaredownloads) but need to wait until a USB NVMe adapter got delivered.

twoplan, 2024-06-21:
Installed latest WDC Dashboard on my Windows laptop and connected the NVMe via an USB-C adapter. Unfortunately the software reports, that there is **no firmware update available** for this SN530. 
Which obviously is wrong, as @aBUGSworstnightmare-rpi 's firmware is newer.

[image]

So I opened a ticket at WDC support ... let's see where this leads to.

aBUGSworstnightmare-rpi, 2024-06-22:
seems like SN530 is not listed in there https://wddashboarddownloads.wdc.com/wdDashboard/config/devices/lista_devices.xml

WD Green SN530 are different. Hopefully WD has some interrest in supporting usage of their products on Raspberry Pi platforms.

aBUGSworstnightmare-rpi, 2024-06-23:
should be solvable by writing a new firmware (see last commit in https://github.com/raspberrypi/linux/issues/6195).

Question now is how to read the NVMe firmware by using name-cli.

twoplan, 2024-06-24:
This is todays feedback from Western Digitals Customers Support:

> Thank you for contacting Western Digital Customer Support. My name is Matt R.
> 
> As verified, your drive is an Original Equipment Manufacturer (OEM) drive. Western Digital supplies hardware to manufacturers that can be used as OEM products.
> 
> **Western Digital® Global Customer Support does not provide technical support or warranty replacement service for Original Equipment Manufacturer (OEM) products.**
> 
> Please contact the OEM manufacturer or the place of purchase of the OEM product.

The 250GB SN530 was left over from a storage upgrade of my Acer laptop two years ago. As WD is not willing to help, geting the firmware from Acer seems very unlikely too. There is no firmware update on the laptops support website.

Maybe there exists a newer firmware from a different OEM manufacturer? Found only this **3 years old** [reddit](https://www.reddit.com/r/techsupport/comments/lzni1u/latest_wd_sn530_firmware_version/).

Dumping it from a working SN530 NVMe looks like the way to go ... but how?

aBUGSworstnightmare-rpi, 2024-06-24:
honestly speaking that's what I've expected!

There are two packages on the internet which include Firmware for the SN530 devices - one is Dell, the other one Lenovo.
Both are fully packaged executable (.exe) where there seems to be no possibility of getting access to the binary file only.

https://download.lenovo.com/pccbbs/mobiles/fwnva68.exe
https://dl.dell.com/FOLDER07599485M/1/Western-Digital-Corp-SN530-Solid-State-Drive-Firmware_XTHG0_WIN64_2111.3012_A00.EXE

Although the firmware is marked as read/write I have no idea how to dump it from the drive. nvme-cli seems to not support such action as well.

aBUGSworstnightmare-rpi, 2024-06-24:
Which USB-C adapter did you use when testing with WD Dashboard? I only have a Realtek RTL9210B based USB 3.0 connectorized board which isn't supported.

twoplan, 2024-06-24:
An [Icy Box IB-1817M-C31](https://raidsonic-static-content.s3.eu-central-1.amazonaws.com/IcyBox/Files/manual_IB-1817M-C31_Web.pdf) with JMicron controller.

```
lsusb
Bus 004 Device 002: ID 152d:0580 JMicron Technology Corp. / JMicron USA Technology Corp. USB 3.1 Storage Device
```

twoplan, 2024-06-24:
In the unpacked Lenovo folder are 4 firmware files for WD SN520, SN530, SN720 and SN730 NVMe. 
But all have dates from 2019 or 2020 and I don't kwow which is for the SN530 (21160001.fluf ?)

```
2277376 Dec 18  2019 10190101.fluf
2588672 Sep  1  2020 11170101.fluf
2121728 Aug 28  2020 20370001.fluf
2588672 Jul  6  2020 21160001.fluf 
```

aBUGSworstnightmare-rpi, 2024-06-24:
21160001 is the firmware of my WDC PC SN530 SDBPMPZ-256G-1101 (the 2242 sized SN530)

You can try it, if it doesn't match you should get some error message.

BUT: USE AT OWN RISK!

P.S. can you share the file here on git?

twoplan, 2024-06-24:
On windows you can download and run fwnva68.exe from the Lenovo site ... just choose "**unpack**" in the dialog instead of "install".

[21160001.zip](https://github.com/user-attachments/files/15958774/21160001.zip)

Tomorrow I will try 21160001.fluf in WD Dashboard.

aBUGSworstnightmare-rpi, 2024-06-24:
doesn't look promising:
```
pi@Pi5-SN530-250:~ $ cd /home/pi/Downloads
pi@Pi5-SN530-250:~/Downloads $ cat /sys/class/nvme/nvme0/model && cat /sys/class/nvme/nvme0/firmware_rev
WDC PC SN530 SDBPNPZ-256G-1114          
pi@Pi5-SN530-250:~/Downloads $ sudo nvme fw-download -f 21160001.fluf /dev/nvme0
Firmware download success
pi@Pi5-SN530-250:~/Downloads $ sudo nvme fw-commit -s 2 -a 0 /dev/nvme0
NVMe status: Invalid Firmware Image: The firmware image specified for activation is invalid and not loaded by the controller(0x4107)
Multiple Update Detected (MUD) Value: 100665094
Detected an overlapping firmware/boot partition image update command sequence due to processing a command from an Admin SQ on a controller
```

twoplan, 2024-06-24:
Perhaps you could try the cab-file from [ArchWiki](https://fwupd.org/lvfs/devices/com.lenovo.PCSN530.firmware) and [fwupd](https://wiki.archlinux.org/title/Fwupd)?

Instructions from the readme:
```
You can install this archive on Linux using fwupd >= 1.0.8 on supported
hardware. Use this command to install:

    fwupdmgr install Lenovo-WD-PCSN530-NVMe-21160001.cab

You can check the version of fwupd using:

    fwupdmgr --version
```

twoplan, 2024-06-25:
Just "updated" the SN530 with firmware 21160001.fluf in WD Dashboard. 
Nice progress bar and success message! 

[image]

**But the firmware remains at version 21103900.**

aBUGSworstnightmare-rpi, 2024-06-25:
> Perhaps you could try the cab-file from [ArchWiki](https://fwupd.org/lvfs/devices/com.lenovo.PCSN530.firmware) and [fwupd](https://wiki.archlinux.org/title/Fwupd)?
>
> Instructions from the readme: [...]
that is not working!
```
pi@Pi5-SN530-250:~/Downloads $ fwupdmgr install Lenovo-WD-PCSN530-NVMe-21160001.cab
WARNING: UEFI capsule updates not available or enabled in firmware setup
See https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported for more information.
Decompressingâ€¦           [    \                                  ]
No supported devices found
```

Trying to flash the firmware binary from the .cab file fails --> the firmware has wrong dependencies
```
pi@Pi5-SN530-250:~/Downloads $ sudo nvme fw-download -f firmware.bin /dev/nvme0
Firmware download success
pi@Pi5-SN530-250:~/Downloads $ sudo nvme fw-commit -s 2 -a 0 /dev/nvme0
NVMe status: Invalid Firmware Image: The firmware image specified for activation is invalid and not loaded by the controller(0x4107)
Multiple Update Detected (MUD) Value: 100665094
Detected an overlapping firmware/boot partition image update command sequence due to processing a command from an Admin SQ on a controller
```

As there is no support from WD to be expected I would suggest to buy another NVMe. You can use your SN530 as external storage still.

squealteam6, 2024-06-26:
I'm not sure if this helps but I've got an IX SN530 NVMe 256GB, FW:2412T000 (july 27, 2023), m.2 2230. 

I'm able to get it to boot, Raspbian using the Pineboard Hat from a complete power down, but if I reset the the boards with the power still connected it will not boot until I power cycle the board.

If there is any data i can provide to might be useful to this topic let me know.

mikegapinski, 2024-06-28:
> I'm not sure if this helps but I've got an IX SN530 NVMe 256GB, FW:2412T000 (july 27, 2023), m.2 2230.
>
> I'm able to get it to boot, Raspbian using the Pineboard Hat from a complete power down, but [...]

Pineboards HATs made after Jan 2023 (for HatDrive Bottom) and March (for HatDrive! Top) have HAT+ power management that solves the problem you mentioned. The official M.2 HAT+ also works in this case.

The Pi 5 simply disables the buck converter on the HAT when the pi reboots and it restarts the drive. It was needed for Western Digital drives

P33M, 2025-04-04:
I have a WD PC SN530 256GB w/ firmware 21160001 and it works in the official M2 HAT+. The HAT power cycles the drive across a reboot, this is required or it will fail to enumerate. There are no newer firmwares available.

closed 2025-04-04 by P33M

ThomasLevering, 2025-06-23:
We found 3 different Firmware in Siemens Industrial PanelPC
2410T000  (SSD from MAR2021)
2412T000
2414T000
on Intel Atom x6413E
after same days, weeks there is no ssd in BIOS. Powercycle resolve the issue.
we are observe, if the newer have the same problem
[image]