raspberrypi/linux #6236: two finger touchscreen gestures do not work in gtk3-demo or gtk4-demo on x11 raspberry bookworm arm64

mgrouch, 2024-06-21:
### Describe the bug

two finger touchscreen gestures do not work in gtk3-demo or gtk4-demo on x11 raspberry bookworm arm64

### Steps to reproduce the behaviour

I've installed

````
sudo apt install gtk-3-examples gtk-4-examples
````

and tested with 

````
gtk3-demo
````

picked 'Gestures' and clicked 'Run'. Swipe gesture on touchscreen draws a line but two finger gestures (rotate, zoom) do not work.

same issue is with gtk4-demo.

This is on x11 (without wayland)

### Device (s)

Raspberry Pi 5

### System

Bookworm

pelwell, 2024-06-21:
If you're seeing events (`sudo apt install evtest`) then this is unlikely to be a kernel issue.

6by9, 2024-06-21:
If you want a quick test of whether multi-touch is working from your touch screen, then https://naqtn.github.io/WBBMTT/ is one of the easiest ways.

mgrouch, 2024-06-21:
Yes I forgot to mention it's working in Chromium. So issue is somewhere between X11 and gtk3/4
on raspberry bookworm

mgrouch, 2024-06-21:
Correction:

Test works
https://naqtn.github.io/WBBMTT/  

Works:
````
gtk4-demo --run=gestures
````

Gtk3 test doesn't work with two finger gestures:
````
gtk3-demo --run=gestures
````

pelwell, 2024-06-21:
So GTK3 deserves a one-finger gesture.

mgrouch, 2024-06-22:
Doc on gtk3 events for touch and gestures 

https://docs.gtk.org/gtk3/input-handling.html

popcornmix, 2024-06-26:
Closing as this issue is not linux kernel related.

closed 2024-06-26 by popcornmix

mgrouch, 2024-06-26:
I’ve opened it as Debian bug related to gtk3

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1074283

thanks