raspberrypi/linux #6237: Wifi on RPi 5 seems to be broken after commit 5c0f94088e06

iucoen, 2024-06-22:
### Describe the bug

Wifi on RPi5 stopped working since this commit (5c0f94088e06). Looks like scanning is timing out with this kernel error message:
```
ieee80211 phy0: brcmf_escan_timeout: timer expired
````

### Steps to reproduce the behaviour

Boot, try to connect to wifi with iwd

### Device (s)

Raspberry Pi 5

### System

OS is Gentoo

P33M, 2024-06-22:
Are you sure that's the exact commit that causes the issue? I.e. have you reverted it and tested?

iucoen, 2024-06-22:
> Are you sure that's the exact commit that causes the issue? I.e. have you reverted it and tested?

Yes. Wifi works after reverting that commit.

P33M, 2024-06-22:
The first chunk is a no-op for SDIO cards, but the second will have the side effect of re-enabling clock gating. I wonder if you previously got one timeout, then with this commit the controller got fully reset (and CG got turned off).

https://github.com/raspberrypi/linux/blob/5c0f94088e0694220a2f0d8ad6e8216b50a80f2e/drivers/mmc/host/sdhci-brcmstb.c#L432

Does commenting out that line fix it?

iucoen, 2024-06-23:
> https://github.com/raspberrypi/linux/blob/5c0f94088e0694220a2f0d8ad6e8216b50a80f2e/drivers/mmc/host/sdhci-brcmstb.c#L432
> 
> Does commenting out that line fix it?

Yep. fixed.

spockfish, 2024-06-23:
I can confirm the issue.

Dark-Sky, 2024-06-24:
Built today's 6.6. 6.9 and 6.10-rc kernels today and none of them have wifi on the pi5. I was hoping this was fixed but guess not....

spockfish, 2024-06-24:
Yes I had exactly the same experience.

However, I did not comment out the line mentioned, but reverted the complete patch. 
The result is working wifi again.

Thanks

pelwell, 2024-06-24:
@P33M Describe the correct fix and I'll get it merged.

spockfish, 2024-06-24:
If it helps: I reverse applied this patch: https://github.com/openwrt/openwrt/pull/15792/files

P33M, 2024-06-24:
> @P33M Describe the correct fix and I'll get it merged.
drivers/mmc/host/sdhci-brcmstb.c:432 - struct brcmstb_match_priv match_priv_2712 should not have no flags. Declaring auto-clockgate support for a host that can interface with SDIO cards is a bug. 
(Why this is a property of the host and not of the platform is a discussion for another time.)

pelwell, 2024-06-24:
#6238 is the resulting patch, now merged and ported to rpi-6.9.y and rpi-6.10.y.

Dark-Sky, 2024-06-25:
> #6238 is the resulting patch, now merged and ported to rpi-6.9.y and rpi-6.10.y.

Built and tested all 3 kernels and wifi seems to be ok now in all of them. Thank you!!

closed 2024-06-25 by iucoen