raspberrypi/linux #6240: Failed to build due to README file.

adminy, 2024-06-27:
### Describe the bug

Trying to compile the latest 6.10.y kernel gives:
```
make[3]: *** No rule to make target 'arch/arm64/boot/dts/overlays/README', needed by 'arch/arm64/boot/dts/overlays/dtbs-list'.  Stop.
make[3]: *** Waiting for unfinished jobs....
make[1]: *** [/tmp/nix-build-linux-rpi-6.10.0-rc5.drv-0/source/Makefile:1405: dtbs] Error 2
make: *** [../Makefile:240: __sub-make] Error 2
```

### Steps to reproduce the behaviour

I'm just trying to normally compile the kernel. no special steps.
```Makefile
cmd_gen_order = { $(foreach m, $(real-prereqs), \
	$(if $(filter %/$(notdir $@), $m), cat $m, echo $m);) :; } \
	> $@
```
I suspect its this piece of code that filters directories but not README files?

### Device (s)

Raspberry Pi 5

### System

N/a

adminy, 2024-06-27:
https://github.com/raspberrypi/linux/pull/6241

pelwell, 2024-06-27:
> I'm just trying to normally compile the kernel. no special steps.

List them - my builds are working.

adminy, 2024-06-27:
I'm using the nixpkgs `buildLinux` derivation with the following flags:
```nix
modDirVersion = "6.10.0-rc5";
version = "6.10.0-rc5";
pname = "linux-rpi";
src = rpi-linux-6_10_0-rc5-src;
defconfig = "bcm2712_defconfig";
structuredExtraConfig = with lib.kernel; {
  BACKLIGHT_CLASS_DEVICE = yes;
  KUNIT = no;
};
features.efiBootStub = false;
kernelPatches = [];
```

with a postconfigure like so:
```nix
postConfigure = ''
  sed -i $buildRoot/.config -e 's/^CONFIG_LOCALVERSION=.*/CONFIG_LOCALVERSION=""/'
  sed -i $buildRoot/include/config/auto.conf -e 's/^CONFIG_LOCALVERSION=.*/CONFIG_LOCALVERSION=""/'
'';
```
The flake builds fine for 6.6.y branch, if you look carefully, that branch does not have the `README` file.

pelwell, 2024-06-28:
That's not a "normal" Linux build. I want a list of "make" commands that demonstrate the problem.

adminy, 2024-06-28:
This in turn calls the make commands in:

https://github.com/NixOS/nixpkgs/blob/5f24895313965092dbd4a622b85d1986f01e2138/pkgs/os-specific/linux/kernel/generic.nix#L167-L189

Normal is relative. To me its more normal not to make assumptions about which environment your machine is using. Because that leads to "works on my machine" problems. This is why I use the normal nix way of building the kernel, so that other people can replicate it. Anyways it seems like https://github.com/raspberrypi/linux/pull/6241 does actually fix the problem. You're probably not building the dtbs with make.

by, 2024-06-28:
Can you build https://github.com/torvalds/linux ? -- That's the linux source.

adminy, 2024-06-28:
yeah, I can try. Probably won't have the defconfig file though.

by, 2024-06-29:
Alternatively and/or additionally, you will likely need to create a manual config with your environment (i.e., https://github.com/NixOS/nixpkgs/blob/d77bda728d5041c1294a68fb25c79e2d161f62b9/pkgs/os-specific/linux/kernel/manual-config.nix); if this fails, you might want to move your issue to https://github.com/NixOS/nixpkgs, as https://github.com/raspberrypi/linux compiles well according to the official way @ https://www.raspberrypi.com/documentation/computers/linux_kernel.html#build-the-kernel

pelwell, 2024-06-29:
To be clear: I'm not saying that there isn't a problem somewhere in the overlay build rules, but unless I can reproduce the failure I won't feel comfortable that I have understood the issue, and therefore that the fix is a good one.

sanderhollaar, 2025-02-22:
https://github.com/raspberrypi/linux/pull/6681 - `fix: remove README from Makefile`

sanderhollaar, 2025-03-07:
Fixed with https://github.com/raspberrypi/linux/commit/f9d8731cc63800cd85dfa653099bdc6c8147f474

(I do not have permissions to `close with comment`)

closed 2025-03-07 by pelwell

adminy, 2025-03-07:
That's great but still an issue in 6.12.y and 6.13.y branches. I believe your changes only made it to 6.14.y

sanderhollaar, 2025-03-07:
I can create additional pull requests for `6.12.y` and `6.13.y` if these kernel versions are still relevant.

adminy, 2025-03-07:
currently raspberrypi/firmware only ships 6.12.y files.

pelwell, 2025-03-07:
It's in progress