raspberrypi/linux #6245: Attempts to write file `/sys/class/rtc/rtc0/wakealarm` produce `write error: Device or resource busy`

seamusdemora, 2024-07-01:
### Describe the bug

When using the overlay `dtoverlay=i2c-rtc,ds3231,wakeup-source`, the file at `/sys/class/rtc/rtc0/wakealarm` gives errors for write attempts: `write error: Device or resource busy`

### Steps to reproduce the behaviour

1. Set up the overlay: `dtoverlay=i2c-rtc,ds3231,wakeup-source` 
2. Write a value to the file: `/sys/class/rtc/rtc0/wakealarm` 
3. Attempt to revise the file's contents by writing a new value to the file: `/sys/class/rtc/rtc0/wakealarm` 
4. Note the error: `write error: Device or resource busy`

### Device (s)

Raspberry Pi Zero, Raspberry Pi 3 Mod. A+

### System

cat /etc/rpi-issue:
```
Raspberry Pi reference 2024-03-15
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 11096428148f0f2be3985ef3126ee71f99c7f1c2, stage2
```

vcgencmd version:
```
May 24 2024 15:31:28
Copyright (c) 2012 Broadcom
version 4942b7633c0ff1af1ee95a51a33b56a9dae47529 (clean) (release) (start)
```

uname -a:
```
Linux rpi3a 6.6.31+rpt-rpi-v7 #1 SMP Raspbian 1:6.6.31-1+rpt1 (2024-05-29) armv7l GNU/Linux
```

### Additional context

If given enough time (~ 15-30 minutes), whatever it is that has the `wakealarm` file locked up - releases it. But this seems nonsensical. I'm not good with 'C', but I did attempt to read the [driver source code??](https://github.com/raspberrypi/linux/blob/rpi-6.6.y/drivers/rtc/rtc-ds1307.c). And I've read through the DS3231 data sheet on multiple occasions, but neither of these sources suggest any explanation for the file being "locked up"

Hopefully, someone here can explain what the issue is. FWIW, here's the script I'm using to attempt to write to `wakealarm`: 

```bash
#!/usr/bin/bash

if [ ! -f /sys/class/rtc/rtc0/wakealarm ]; then
    echo "file '/sys/class/rtc/rtc0/wakealarm' does not exist; exiting...";
    exit 1;
fi

if [ -s /sys/class/rtc/rtc0/wakealarm ]; then
    printf "overwrite 'wakealarm' value: %s with '/dev/null'\n" "$(< /sys/class/rtc/rtc0/wakealarm)"
    /usr/bin/cp /dev/null /sys/class/rtc/rtc0/wakealarm    # overwrite wakealarm w/ /dev/null
    /usr/bin/sleep 3
fi

off_time=30
# /usr/bin/date '+%s' -d "+ $off_time minutes" | /usr/bin/tee /sys/class/rtc/rtc0/wakealarm    # <== also produces error
alarm=$(/usr/bin/date '+%s' -d "+ $off_time minutes")
printf "%s" "$alarm" > "/sys/class/rtc/rtc0/wakealarm"     # <== source of error
# echo "'wakealarm' time set to: $(/usr/bin/cat /sys/class/rtc/rtc0/wakealarm)"
```

pelwell, 2024-07-01:
wakealarm is handled by the common RTC framework - drivers just have to say they are potential wake sources. The relevant code is here: https://github.com/raspberrypi/linux/blob/rpi-6.6.y/drivers/rtc/sysfs.c#L152

The "Device or resource busy" message corresponds to a return value of `-EBUSY`, which suggests that something has already enabled the alarm. You can confirm the diagnosis by using the `+=<interval>` syntax - to delay that alarm - that should succeed. The question is then what is setting the alaram, but that isn't a kernel issue.

seamusdemora, 2024-07-01:
> by using the `+=<interval>` syntax - to delay that alarm

I'm not quite clear on how to do that. Would it be:  `printf "%s" "$alarm+=1" > "/sys/class/rtc/rtc0/wakealarm"`

pelwell, 2024-07-01:
To defer the alarm by one second, do the equivalent of:
```
echo '+=1' > /sys/class/rtc/rtc0/wakealarm
```

You could also try writing 0 to see if that disables the alarm.

seamusdemora, 2024-07-02:
Completed my tests using the script below for evaluation. 

Conclusions:
1. Copying `/dev/null` to `wakealarm` does nothing useful; it doesn't overwrite an existing value! (no clue why)
2. Writing a new value to `wakealarm` requires writing a '0' beforehand; otherwise the Device busy error is thrown
3. Writing an incremental increase works fine; use either `echo '+=N' > "wakealarm"`, or `printf '%s' "+=N" >...` 
4. The "Device or resource busy" error seems *misleading*; something like "Illegal write attempt" makes more sense (to me)
5. This may be "much ado about nothing", but I do feel it deserves to be documented. 

```bash
#!/usr/bin/bash

while :
do
#    /usr/bin/cp "/dev/null" "/sys/class/rtc/rtc0/wakealarm"
#    printf "%s" "0" > "/sys/class/rtc/rtc0/wakealarm"
    /usr/bin/sleep 10
    alarm=$(/usr/bin/date '+%s' -d "+ 1 day")
#    printf "%s" "$alarm" > "/sys/class/rtc/rtc0/wakealarm"
#    echo '+=1' > "/sys/class/rtc/rtc0/wakealarm"
    printf '%s' "+=1" > "/sys/class/rtc/rtc0/wakealarm"
    echo "Current time: $(date); aka: $(date '+%s');  Contents of 'wakealarm': $(cat "/sys/class/rtc/rtc0/wakealarm")"
done
``` 

Thanks for your help with this!

closed 2024-07-02 by seamusdemora