raspberrypi/linux #6246: USB OTG Initialization Issue with dwc2 Driver and g_mass_storage Gadget on Raspberry Pi CM3

techniker, 2024-07-01:
### Describe the bug

I’m experiencing issues with the USB OTG functionality on my Raspberry Pi Compute Module 3 (CM3). The problem seems to be related to the dwc2 driver and the g_mass_storage gadget. Upon boot, I see the following errors in the kernel log:

[    3.546076] dwc2 3f980000.usb: 256 invalid for host_nperio_tx_fifo_size. Check HW configuration.
[    3.557306] dwc2 3f980000.usb: 512 invalid for host_perio_tx_fifo_size. Check HW configuration.
[    3.568522] dwc2 3f980000.usb: EPs: 8, dedicated fifos, 4080 entries in SPRAM
...
[    3.609244] g_mass_storage gadget: Mass Storage Gadget, version: 2009/09/11
[    3.617616] g_mass_storage gadget: userspace failed to provide iSerialNumber
[    3.626124] g_mass_storage gadget: g_mass_storage ready
[    3.632785] dwc2 3f980000.usb: bound driver g_mass_storage
[    3.817578] dwc2 3f980000.usb: new device is high-speed
[    3.846967] dwc2 3f980000.usb: new address 46
[    3.869133] g_mass_storage gadget: high-speed config #1: Linux File-Backed Storage
...
[   63.435425] WARNING: CPU: 0 PID: 0 at /home/qpc/builds/resinio/amber-etcher-kernel/build/tmp/work-shared/raspberrypi-cm3/kernel-source/drivers/usb/dwc2/gadget.c:195 dwc2_hsotg_init_fifo+0x1b0/0x1b8 [dwc2]
[   63.456926] Modules linked in: g_mass_storage usb_f_mass_storage libcomposite dwc2

### Steps to reproduce the behaviour

1.	Set up a Raspberry Pi Compute Module 3 with the following configuration in /boot/config.txt: 
dtoverlay=dwc2
2.     Connect the PI via USB OTG to a Computer to mount the eMMC flash drive.
4.	Observe the kernel log for errors related to dwc2 and g_mass_storage.

### Device (s)

Raspberry Pi CM3

### System

2024-03-15-raspios-bookworm-arm64-lite.img

Others not possible due to no boot.

### Logs

[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 989008K/1021952K available (7168K kernel code, 488K rwdata, 2116K rodata, 3072K init, 778K bss, 24752K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xbe800000 - 0xff800000   (1040 MB)
[    0.000000]     lowmem  : 0x80000000 - 0xbe600000   ( 998 MB)
[    0.000000]     modules : 0x7f000000 - 0x80000000   (  16 MB)
[    0.000000]       .text : 0x80008000 - 0x80800000   (8160 kB)
[    0.000000]       .init : 0x80b00000 - 0x80e00000   (3072 kB)
[    0.000000]       .data : 0x80e00000 - 0x80e7a1c4   ( 489 kB)
[    0.000000]        .bss : 0x80e7c000 - 0x80f3e8d4   ( 779 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000]  Build-time adjustment of leaf fanout to 32.
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] Failed to get local register map. FIQ is disabled for cpus > 1
[    0.000000] arm_arch_timer: Architected cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000006] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000022] Switching to timer-based delay loop, resolution 52ns
[    0.000269] Console: colour dummy device 80x30
[    0.001161] console [tty1] enabled
[    0.001203] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.001269] pid_max: default: 32768 minimum: 301
[    0.001595] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001636] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002523] Disabling cpuset control group subsystem
[    0.002684] CPU: Testing write buffer coherency: ok
[    0.002748] ftrace: allocating 22526 entries in 67 pages
[... 111 lines cut ...]
[    3.604902] Number of LUNs=1
[    3.609244] g_mass_storage gadget: Mass Storage Gadget, version: 2009/09/11
[    3.617616] g_mass_storage gadget: userspace failed to provide iSerialNumber
[    3.626124] g_mass_storage gadget: g_mass_storage ready
[    3.632785] dwc2 3f980000.usb: bound driver g_mass_storage
[    3.817578] dwc2 3f980000.usb: new device is high-speed
[    3.846967] dwc2 3f980000.usb: new address 46
[    3.869133] g_mass_storage gadget: high-speed config #1: Linux File-Backed Storage
[    5.028074] dwc2 3f980000.usb: dwc2_hsotg_ep_sethalt(ep bbce0090 ep1in, 1)
[    5.153293] dwc2 3f980000.usb: dwc2_hsotg_ep_sethalt(ep bbce0090 ep1in, 1)
[    5.161692] dwc2 3f980000.usb: dwc2_hsotg_start_req: ep1 is stalled
[    5.169780] dwc2 3f980000.usb: dwc2_hsotg_ep_sethalt(ep bbce0090 ep1in, 0)
[    5.178666] dwc2 3f980000.usb: dwc2_hsotg_ep_sethalt(ep bbce0090 ep1in, 1)
[    5.303275] dwc2 3f980000.usb: dwc2_hsotg_ep_sethalt(ep bbce0090 ep1in, 1)
[    5.311408] dwc2 3f980000.usb: dwc2_hsotg_start_req: ep1 is stalled
[    5.319192] dwc2 3f980000.usb: dwc2_hsotg_ep_sethalt(ep bbce0090 ep1in, 0)
[    5.327930] dwc2 3f980000.usb: dwc2_hsotg_ep_sethalt(ep bbce0090 ep1in, 1)
[    5.453274] dwc2 3f980000.usb: dwc2_hsotg_ep_sethalt(ep bbce0090 ep1in, 1)
[    5.461303] dwc2 3f980000.usb: dwc2_hsotg_start_req: ep1 is stalled
[    5.468960] dwc2 3f980000.usb: dwc2_hsotg_ep_sethalt(ep bbce0090 ep1in, 0)
[    5.641760] random: fast init done
[   15.656524] random: crng init done
[   63.429663] ------------[ cut here ]------------
[   63.435425] WARNING: CPU: 0 PID: 0 at /home/qpc/builds/resinio/amber-etcher-kernel/build/tmp/work-shared/raspberrypi-cm3/kernel-source/drivers/usb/dwc2/gadget.c:195 dwc2_hsotg_init_fifo+0x1b0/0x1b8 [dwc2]
[   63.456926] Modules linked in: g_mass_storage usb_f_mass_storage libcomposite dwc2

stefaanv, 2025-02-01:
I have a similar issue on a raspberry pi 5
`dwc2_hsotg_ep_start_req: ep1 is stalled`
An duploading the boot file is very slow

christhomas, 2025-09-16:
I have a raspberry pi zero and the composite gadget to create a usb disk that the other computer mounts when it's plugged in. With one device it works and mounts as a drive and data can be written to it. But when I plug it into another computer, the same device, will have these exact errors in the dmesg output. 

Same raspberry pi, but different host computer, so it's like some host computers react differently, or behave differently. 

```
[   31.151871] dwc2 20980000.usb: dwc2_hsotg_start_req: ep3 is stalled
[   31.152272] dwc2 20980000.usb: dwc2_hsotg_ep_sethalt(ep 0c8394b5 ep3in, 0)
[   31.154471] dwc2 20980000.usb: dwc2_hsotg_ep_sethalt(ep 0c8394b5 ep3in, 1)
[   31.270882] dwc2 20980000.usb: dwc2_hsotg_ep_sethalt(ep 0c8394b5 ep3in, 1)
[   31.270954] dwc2 20980000.usb: dwc2_hsotg_start_req: ep3 is stalled
[   31.271326] dwc2 20980000.usb: dwc2_hsotg_ep_sethalt(ep 0c8394b5 ep3in, 0)
[   31.290948] dwc2 20980000.usb: dwc2_hsotg_ep_sethalt(ep 0c8394b5 ep3in, 1)
[   31.410835] dwc2 20980000.usb: dwc2_hsotg_ep_sethalt(ep 0c8394b5 ep3in, 1)
```

macmpi, 2025-10-06:
@christhomas you may want to check if https://github.com/macmpi/xg_multi solves your issue.
Tested OK on PiZeroW with Trixie.
`Linux raspberrypi 6.12.47+rpt-rpi-v6 #1 Raspbian 1:6.12.47-1+rpt1 (2025-09-16) armv6l GNU/Linux`

christhomas, 2025-10-06:
> [@christhomas](https://github.com/christhomas) you may want to check if https://github.com/macmpi/xg_multi solves your issue. Tested OK on PiZeroW with Trixie. `Linux raspberrypi 6.12.47+rpt-rpi-v6 #1 Raspbian 1:6.12.47-1+rpt1 (2025-09-16) armv6l GNU/Linux`

Thanks! I'll take a look