raspberrypi/linux #6247: [rpi zero 2w] Disconnect/connect USB devices on the fly leads to unrecoverable error in USB driver

qrp73, 2024-07-01:
Steps to reproduce:
1) Use Raspberry Pi Zero 2W device with latest raspi os bookworm 64 ```Linux rpi2w 6.6.31+rpt-rpi-v8 raspberrypi/bookworm-feedback#1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux```
2) Run ```dmesg -w``` to monitor kernel errors
3) Disconnect all devices from USB port
4) Connect USB keyboard
5) Disconnect USB keyboard
6) Repeat steps 4 & 5 in a loop until this error in dmesg output:
```
[  431.506128] usb 1-1: device descriptor read/64, error -71
```

After that the system is unable to detect USB device connections anymore. Only reboot can fix it. But it still can accept ssh connection through WiFi.

It may take 10-30 connect/disconnect loops to catch the issue.

Device: Raspberry Pi Zero 2W Rev 1.0
Revision	: 902120

```
$ cat /etc/rpi-issue
Raspberry Pi reference 2024-03-15
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, f19ee211ddafcae300827f953d143de92a5c6624, stage2

$ vcgencmd version
May 24 2024 15:31:28 
Copyright (c) 2012 Broadcom
version 4942b7633c0ff1af1ee95a51a33b56a9dae47529 (clean) (release) (start)

$ uname -a
Linux rpi2w 6.6.31+rpt-rpi-v8 raspberrypi/bookworm-feedback#1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux
```

The log:
```
[  427.362039] Indeed it is in host mode hprt0 = 00041501
[  428.178199] Indeed it is in host mode hprt0 = 00021d01
[  428.506226] Indeed it is in host mode hprt0 = 00021d01
[  428.694165] usb 1-1: new full-speed USB device number 14 using dwc_otg
[  430.358619] Indeed it is in host mode hprt0 = 00021501
[  431.178152] Indeed it is in host mode hprt0 = 00021d01
[  431.506128] usb 1-1: device descriptor read/64, error -71
[  433.680703] Indeed it is in host mode hprt0 = 00021501
[  434.498169] Indeed it is in host mode hprt0 = 00021d01
[  434.826084] usb 1-1: device descriptor read/64, error -71
[  434.934131] usb usb1-port1: attempt power cycle
```

pelwell, 2024-07-01:
And does adding `dtoverlay=dwc2,dr_mode=host` make it more reliable? Note that I would expect a loss of performance with that configuration.

qrp73, 2024-07-02:
> And does adding `dtoverlay=dwc2,dr_mode=host` make it more reliable? Note that I would expect a loss of performance with that configuration.

No, it don't helps. It still happens with this line in config.txt:
```
[   16.746657] systemd[691]: memfd_create() called without MFD_EXEC or MFD_NOEXEC_SEAL set
[   49.530105] usb 1-1: new full-speed USB device number 2 using dwc2
[   52.386155] usb 1-1: new full-speed USB device number 3 using dwc2
[   54.986123] usb 1-1: new full-speed USB device number 4 using dwc2
[   57.414126] usb 1-1: new full-speed USB device number 5 using dwc2
[   59.862129] usb 1-1: new full-speed USB device number 6 using dwc2
[   62.214135] usb 1-1: new full-speed USB device number 7 using dwc2
[   64.410126] usb 1-1: new full-speed USB device number 8 using dwc2
[   66.618129] usb 1-1: new full-speed USB device number 9 using dwc2
[   69.058149] usb 1-1: new full-speed USB device number 10 using dwc2
[   71.198127] usb 1-1: new full-speed USB device number 11 using dwc2
[   73.294130] usb 1-1: new full-speed USB device number 12 using dwc2
[   75.358170] usb 1-1: new full-speed USB device number 13 using dwc2
[   77.546156] usb 1-1: new full-speed USB device number 14 using dwc2
[   79.766176] usb 1-1: new full-speed USB device number 15 using dwc2
[   81.786202] usb 1-1: new full-speed USB device number 16 using dwc2
[   83.986221] usb 1-1: new full-speed USB device number 17 using dwc2
[   85.970239] usb 1-1: new full-speed USB device number 18 using dwc2
[   86.994249] usb 1-1: device descriptor read/64, error -71
[   87.102293] dwc2 3f980000.usb: Not connected
[   87.102336] dwc2 3f980000.usb: Not connected
[   87.102356] dwc2 3f980000.usb: Not connected
[   88.838260] usb 1-1: new full-speed USB device number 19 using dwc2
[   90.882299] usb 1-1: new full-speed USB device number 20 using dwc2
[   93.346285] usb 1-1: new full-speed USB device number 21 using dwc2
[   95.762293] usb 1-1: new full-speed USB device number 22 using dwc2
[  155.657248] usb 1-1: new full-speed USB device number 23 using dwc2
[  171.160539] usb 1-1: device descriptor read/64, error -110
[  186.776034] usb 1-1: device descriptor read/64, error -110
[  187.072011] usb 1-1: new full-speed USB device number 24 using dwc2
[  195.303805] usb 1-1: new full-speed USB device number 25 using dwc2
[  200.411900] dwc2 3f980000.usb: dwc2_hc_chhltd_intr_dma: Channel 3 - ChHltd set, but reason is unknown
[  200.411932] dwc2 3f980000.usb: hcint 0x00000002, intsts 0x04000029
[  200.411973] dwc2 3f980000.usb: dwc2_hc_chhltd_intr_dma: Channel 4 - ChHltd set, but reason is unknown
[  200.411994] dwc2 3f980000.usb: hcint 0x00000002, intsts 0x04000029
[  200.412029] dwc2 3f980000.usb: dwc2_hc_chhltd_intr_dma: Channel 5 - ChHltd set, but reason is unknown
[  200.412049] dwc2 3f980000.usb: hcint 0x00000002, intsts 0x04000029
[  201.059703] usb 1-1: new low-speed USB device number 26 using dwc2
[  203.024764] dwc2 3f980000.usb: dwc2_hc_chhltd_intr_dma: Channel 7 - ChHltd set, but reason is unknown
[  203.024795] dwc2 3f980000.usb: hcint 0x00000002, intsts 0x04000029
[  203.024834] dwc2 3f980000.usb: dwc2_hc_chhltd_intr_dma: Channel 0 - ChHltd set, but reason is unknown
[  203.024855] dwc2 3f980000.usb: hcint 0x00000002, intsts 0x04000029
[  203.027749] dwc2 3f980000.usb: Not connected
[  203.027794] dwc2 3f980000.usb: Not connected
[  204.115703] usb usb1-port1: attempt power cycle
```
But your line in config.txt eliminates other errors, like this one (this is the log taken without ```dtoverlay=dwc2,dr_mode=host```):
```
[   17.326440] systemd[692]: memfd_create() called without MFD_EXEC or MFD_NOEXEC_SEAL set
[  107.927526] Indeed it is in host mode hprt0 = 00041d01
[  108.115436] usb 1-1: new low-speed USB device number 2 using dwc_otg

[  108.802692] ERROR::dwc_otg_hcd_urb_enqueue:501: Not connected

[  108.802714] ERROR::dwc_otg_hcd_urb_enqueue:501: Not connected

[  111.647392] Indeed it is in host mode hprt0 = 00021501
[  111.775384] Indeed it is in host mode hprt0 = 00021501
[  112.043387] Indeed it is in host mode hprt0 = 00021501
[  112.311382] Indeed it is in host mode hprt0 = 00021501
[  112.579378] Indeed it is in host mode hprt0 = 00021d01
[  115.119385] Indeed it is in host mode hprt0 = 00021d01
[  115.307322] usb 1-1: new full-speed USB device number 4 using dwc_otg

[  117.201496] ERROR::dwc_otg_hcd_urb_enqueue:501: Not connected

[... 29 lines cut ...]
[  260.104890] usb 1-1: Manufacturer: SIGMACHIP
[  260.111813] input: SIGMACHIP USB Keyboard as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1:1.0/0003:1C4F:0026.0003/input/input5
[  260.171326] hid-generic 0003:1C4F:0026.0003: input,hidraw0: USB HID v1.10 Keyboard [SIGMACHIP USB Keyboard] on usb-3f980000.usb-1/input0
[  260.176030] input: SIGMACHIP USB Keyboard Consumer Control as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1:1.1/0003:1C4F:0026.0004/input/input6
[  260.235083] input: SIGMACHIP USB Keyboard System Control as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1:1.1/0003:1C4F:0026.0004/input/input7
[  260.235360] hid-generic 0003:1C4F:0026.0004: input,hidraw1: USB HID v1.10 Device [SIGMACHIP USB Keyboard] on usb-3f980000.usb-1/input1
[  261.046730] usb 1-1: USB disconnect, device number 3
[  263.174719] Indeed it is in host mode hprt0 = 00041901
[  263.362642] usb 1-1: new low-speed USB device number 4 using dwc_otg
[  263.363536] Indeed it is in host mode hprt0 = 00041901
[... cut ...]
As you can see, with USB OTG cable which has not reliable contacts, it detects USB connection, but does not have time to read information about the device, because it immediately disconnected in very short time. And when this happens many times, the driver is going into unrecoverable state... Once it happens, any further attempt to connect device with a good USB OTG cable don't helps - the driver remains in unrecoverable error state and stops to handle USB devices.

So, I think there is some issue in the driver which happens when it detects a new connection, but does not have time to read data about the new device because it quickly disconnects.

qrp73, 2024-07-02:
Some new finding. I'm able to reproduce the issue on a good USB OTG cable, but its' much more hard. It requires to connect/disconnect USB very shortly to not allow read information about device (you can visially check it with ```dmesg -w```. 

With a good USB OTG cable it take about 117 connect/disconnect cycles to catch the issue:
```
[ 1469.821421] Indeed it is in host mode hprt0 = 00041901
[ 1471.176546] Indeed it is in host mode hprt0 = 00041901
[ 1471.996533] Indeed it is in host mode hprt0 = 00041901
[ 1472.324507] usb 1-1: new low-speed USB device number 112 using dwc_otg

[ 1472.324567] ERROR::dwc_otg_hcd_urb_enqueue:501: Not connected

[ 1472.324598] ERROR::dwc_otg_hcd_urb_enqueue:501: Not connected

[ 1472.324623] ERROR::dwc_otg_hcd_urb_enqueue:501: Not connected

[ 1473.656537] Indeed it is in host mode hprt0 = 00041901
[ 1475.144531] Indeed it is in host mode hprt0 = 00041901
[ 1476.144486] Indeed it is in host mode hprt0 = 00041901
[ 1476.332453] usb 1-1: new low-speed USB device number 115 using dwc_otg
[ 1476.333268] Indeed it is in host mode hprt0 = 00041901
[ 1477.152471] Indeed it is in host mode hprt0 = 00041901

[ 1477.480470] ERROR::dwc_otg_hcd_urb_enqueue:501: Not connected

[ 1477.688478] ERROR::dwc_otg_hcd_urb_enqueue:501: Not connected

[ 1478.240472] Indeed it is in host mode hprt0 = 00041901
[ 1479.132470] Indeed it is in host mode hprt0 = 00041901
[ 1479.460441] usb 1-1: new low-speed USB device number 116 using dwc_otg
[ 1479.461250] Indeed it is in host mode hprt0 = 00041901
[ 1480.280503] Indeed it is in host mode hprt0 = 00041901

[ 1480.628473] ERROR::dwc_otg_hcd_urb_enqueue:501: Not connected

[ 1480.628506] ERROR::dwc_otg_hcd_urb_enqueue:501: Not connected

[ 1480.628531] ERROR::dwc_otg_hcd_urb_enqueue:501: Not connected

[ 1480.628589] usb usb1-port1: attempt power cycle
```

after that USB driver going into unrecoverable error and stops to detect any further USB device connections. It just don't react on USB connections after that (device is not detected and no new log messages at all).

qrp73, 2024-07-18:
Can you please fix it? This is important issue, because it leads to random unrecoverable error in USB drivers which makes any USB device unavailable until system reboot.

HectorPulido, 2024-11-21:
I have the exact same issue, also `usb 1-1: device descriptor read/64, error -71`

lategoodbye, 2024-11-21:
Hi, in order to investigate further this needs more specific information:
- Is this specific to the Raspberry Pi 2 W or is this reproducible with a different Pi?
- Is it relevant that the keyboard is directly connected to the Pi 2 W or placing a USB hub between avoid the issue?

lategoodbye, 2024-11-24:
I was able to reproduce this issue with Mainline 6.12 on a Raspberry Pi 3A+ and the real issue seems the power cycle for the root port. It seems that hub control isn't tested very often because most of the Pis have a builtin hub.

The following branch contains the relevant fix ( usb: dwc2: hcd: Fix GetPortStatus & SetPortFeature ):
https://github.com/lategoodbye/linux-dev/commits/v6.12-dwc_hcd_fixes/

The dwc_otg driver seems to suffer from the same pattern.

I'm still a newbie to the dwc2 driver, so a review or at least test feedback is welcome.

popcornmix, 2024-11-25:
@qrp73 are you able to build a kernel with @lategoodbye's patch?
@P33M does the patch make sense to you?

P33M, 2024-11-26:
I guess so. Root port behaviours with the wacky OTG session negotiation aren't really tested as Pi board tend to force modes one way or the other - and usually the Zero boards are used with just the data socket providing power.

lategoodbye, 2024-11-26:
One of the reasons why I'm publishing these patches here is that the Mainline DWC2 maintainer from Synopsys wasn't active on linux-usb since July.

popcornmix, 2024-11-26:
@qrp73 (and others) if you are willing to switch to a newer kernel (6.12) with lategoodbye's proposed patches you can run:

```sudo rpi--update pulls/6496```

Wait about 45mins for the build to complete.

lategoodbye, 2024-12-09:
The 3 fixes has been applied to gregkh/usb-linus:
https://git.kernel.org/pub/scm/linux/kernel/git/gregkh/usb.git/log/?h=usb-linus

Hopefully this find its way to Linux 6.13 soon and then stable later ...

gabrielkrell, 2026-10-21:
> The dwc_otg driver seems to suffer from the same pattern.

I think I have encountered this.  I'm running Mainsail on a raspi zero 2w with an Ender 3 connected. The Ender 3 board is not very well designed and requires a USB "isolator" (a disconnected 5V pin) to avoid backfeeding power to the printer when it's off or attempting to power the Pi by its data port when it's on. When the printer is on, everything works as expected. When the printer is off, dwc_otg makes a few reconnect attempts, tries to power cycle, and then never comes back up - even if I power up the ender 3 or plug in a normal peripheral like a flash drive.

```
[  227.449785] Indeed it is in host mode hprt0 = 00021501
[  227.629716] usb 1-1: new full-speed USB device number 4 using dwc_otg
[  227.630174] Indeed it is in host mode hprt0 = 00021501
[  227.809712] usb 1-1: device descriptor read/64, error -71
[  227.914116] Indeed it is in host mode hprt0 = 00021501
[  228.093694] usb 1-1: device descriptor read/64, error -71
[  228.197779] Indeed it is in host mode hprt0 = 00021501
[  228.377698] usb 1-1: new full-speed USB device number 5 using dwc_otg
[  228.378063] Indeed it is in host mode hprt0 = 00021501
[  228.557693] usb 1-1: device descriptor read/64, error -71
[  228.662002] Indeed it is in host mode hprt0 = 00021501
[  228.841691] usb 1-1: device descriptor read/64, error -71
[  228.945753] usb usb1-port1: attempt power cycle
```

The only thing that gets the port back online is a reboot. Re-binding from the driver results in a segfault:

```bash
pi@mainsailos:~ $ ls -l /sys/bus/platform/drivers/dwc_otg/
total 0
lrwxrwxrwx 1 root root    0 Oct 21 19:38 3f980000.usb -> ../../../../devices/platform/soc/3f980000.usb
--w------- 1 root root 4096 Oct 21 19:08 bind
--w------- 1 root root 4096 Oct 21 19:38 uevent
--w------- 1 root root 4096 Oct 21 19:08 unbind
-r--r--r-- 1 root root 4096 Oct 21 19:38 version
pi@mainsailos:~ $ echo 3f980000.usb | sudo tee /sys/bus/platform/drivers/dwc_otg/unbind
3f980000.usb

[  130.530324] dwc_otg 3f980000.usb: remove, state 1
[  130.530369] usb usb1: USB disconnect, device number 1
[  130.531273] PortPower off
[  130.532313] dwc_otg 3f980000.usb: USB bus 1 deregistered

pi@mainsailos:~ $ echo 3f980000.usb | sudo tee /sys/bus/platform/drivers/dwc_otg/bind
3f980000.usb
Segmentation fault

[  164.177686] Unable to handle kernel NULL pointer dereference at virtual address 0000000000000040
[  164.177740] Mem abort info:
[  164.177751]   ESR = 0x0000000096000005
[  164.177764]   EC = 0x25: DABT (current EL), IL = 32 bits
[  164.177780]   SET = 0, FnV = 0
[  164.177792]   EA = 0, S1PTW = 0
[  164.177803]   FSC = 0x05: level 1 translation fault
[  164.177817] Data abort info:
[  164.177827]   ISV = 0, ISS = 0x00000005, ISS2 = 0x00000000
[  164.177842]   CM = 0, WnR = 0, TnD = 0, TagAccess = 0
[  164.177856]   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0
[  164.177872] user pgtable: 4k pages, 39-bit VAs, pgdp=00000000060c6000
[  164.177890] [0000000000000040] pgd=0000000000000000, p4d=0000000000000000, pud=0000000000000000
[  164.177924] Internal error: Oops: 0000000096000005 [#1]  SMP
[  164.177943] Modules linked in: binfmt_misc brcmfmac_cyw spidev vc4 brcmfmac snd_soc_hdmi_codec drm_exec drm_display_helper brcmutil cec drm_dma_helper drm_client_lib drm_kms_helper bcm2835_codec(C) bcm2835_isp(C) bcm2835_v4l2(C) cfg80211 bcm2835_mmal_vchiq(C) vc_sm_cma(C) videobuf2_vmalloc v4l2_mem2mem snd_soc_core videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videodev snd_compress snd_pcm_dmaengine rfkill videobuf2_common mc raspberrypi_hwmon snd_bcm2835(C) snd_pcm snd_timer spi_bcm2835 snd raspberrypi_gpiomem sg sch_fq_codel i2c_dev zram lz4_compress drm fuse drm_panel_orientation_quirks backlight nfnetlink ipv6 libsha1 i2c_bcm2835
[  164.178248] CPU: 1 UID: 0 PID: 1206 Comm: tee Tainted: G         C          6.18.50+rpt-rpi-v8 #1 PREEMPT  Debian 1:6.18.50-1+rpt1 
[  164.178283] Tainted: [C]=CRAP
[  164.178296] Hardware name: Raspberry Pi Zero 2 W Rev 1.0 (DT)
[  164.178314] pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  164.178336] pc : dwc_otg_driver_remove+0x28/0xa8
[  164.178370] lr : dwc_otg_driver_probe+0x3a8/0xb48
[  164.178392] sp : ffffffc08158bb40
[  164.178405] x29: ffffffc08158bb40 x28: ffffff80046d0000 x27: 0000000000000000
[  164.178436] x26: 0000000000000000 x25: ffffffd662ed8d40 x24: ffffff80027a7400
[  164.178467] x23: ffffff8003552280 x22: 0000000000000000 x21: ffffff80027a7410
[  164.178497] x20: ffffff80027a7400 x19: 0000000000000000 x18: 0000000000000000
[  164.178527] x17: ffffffa9a4217000 x16: ffffffd662aa6348 x15: 00000030ca558cbf
[  164.178558] x14: 00000000000000f2 x13: 0000000100000003 x12: 8000000800000008
[... 18 lines cut ...]
[  164.179020]  invoke_syscall+0x4c/0x100
[  164.179043]  el0_svc_common.constprop.0+0x48/0xf0
[  164.179065]  do_el0_svc+0x24/0x38
[  164.179086]  el0_svc+0x38/0x120
[  164.179107]  el0t_64_sync_handler+0xa0/0xe8
[  164.179128]  el0t_64_sync+0x198/0x1a0
[  164.179155] Code: 91004015 a90153f3 aa0003f4 f9403eb3 (f9402261) 
[  164.179176] ---[ end trace 0000000000000000 ]---
```

Using `dwc2` by changing

```conf
[cm5]
dtoverlay=dwc2,dr_mode=host
```

to
 
```conf
[all]
dtoverlay=dwc2,dr_mode=host
```

in `/boot/firmware/config.txt` seems to resolve the issue.