raspberrypi/linux #6251: null pointer exception on RPi 0W "Oops - undefined instruction: 0 [#2] ARM"

xsolinsx, 2024-07-05:
### Describe the bug

getting this when trying to upload a tar.xz file using a telegram bot written in python
https://github.com/xsolinsx/RPiCamBot/blob/master/WebcamBot/utils.py#L89
it backups the folder of the telegram bot + some additional files specified in the configuration
this archive is then sent to my telegram user from my bot, this is the step where it gets stuck, last time it broke at this situation
```
I am sending the automatic backup.[███████████████░░░░░] 78.38%
18.00 MB/22.96 MB
1.12 MB/s 16s/20s
```
the tmux window then is blocked on the logs reported below and I cannot kill the python process in any way

### Steps to reproduce the behaviour

it does not happen consistently, but I experienced this multiple times in the last months, every week or so
I don't have a clue of what the steps to reproduce could be
I would say get a telegram bot api key, install the required libraries and try to upload a file of some MB, but I'm not sure this would definitely cause the issue to occur
these lines should be enough to create the client with the library I'm using https://github.com/xsolinsx/RPiCamBot/blob/master/WebcamBot/timelapse_bot.py#L25-L36
while these lines should be enough to actually send a document https://github.com/xsolinsx/RPiCamBot/blob/master/WebcamBot/utils.py#L98-L104 (the progress function could be useful to debug but it's not mandatory)

### Device (s)

Raspberry Pi Zero W / WH

### System

Raspberry Pi reference 2023-12-05
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 446f3e34c8ceb33dfe7bd38578f1cd6b6afe19da, stage2

Mar 17 2023 10:53:39
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start_x)

Linux raspberrypi 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux

### Logs

tmux console where python was being executed
```
Message from syslogd@raspberrypi at Jul  5 04:42:54 ...
 kernel:[145517.794414] Internal error: Oops - undefined instruction: 0 [#2] ARM

Message from syslogd@raspberrypi at Jul  5 04:42:54 ...
 kernel:[145517.935834] Process python (pid: 2297, stack limit = 0x3bc5047c)

Message from syslogd@raspberrypi at Jul  5 04:42:54 ...
 kernel:[145517.937678] Stack: (0xd8cc5ec8 to 0xd8cc6000)

Message from syslogd@raspberrypi at Jul  5 04:42:54 ...
 kernel:[145517.939519] 5ec0:                   c2637380 00000000 d8cc5f2c c08c6378 00000000 0a6ac34f

Message from syslogd@raspberrypi at Jul  5 04:42:54 ...
 kernel:[145517.943279] 5ee0: b289d000 d7de8030 ffffffff c3b435e8 00000cc0 000b289d b289d000 b289d000

Message from syslogd@raspberrypi at Jul  5 04:42:54 ...
 kernel:[145517.949780] 5f00: 00000a55 d2a86ca0 d2a86ca0 00000000 00000000 00000000 c7b4e274 c26373b0

Message from syslogd@raspberrypi at Jul  5 04:42:54 ...
 kernel:[145517.954258] 5f20: 00000000 5f3e4581 00000817 d8cc5fb0 c2637380 b289d000 00000817 00000255

Message from syslogd@raspberrypi at Jul  5 04:42:54 ...
 kernel:[145517.959145] 5f40: c3b435e8 00000002 c26373b0 c08cd358 d8cc5f74 c01b2f4c 00000022 000b2837

Message from syslogd@raspberrypi at Jul  5 04:42:54 ...
 kernel:[145517.963781] 5f60: d8cc5f70 c0c34bfc 00000817 c08cd220 b289d000 d8cc5fb0 f82fb0b1 00489fd8

Message from syslogd@raspberrypi at Jul  5 04:42:54 ...
 kernel:[145517.969023] 5f80: 00489fd8 c00189fc 00000000 b6f16b44 20000010 ffffffff b6f16b44 20000010

[... 64 lines cut ...]
Jul  5 04:42:54 raspberrypi kernel: [145517.949780] 5f00: 00000a55 d2a86ca0 d2a86ca0 00000000 00000000 00000000 c7b4e274 c26373b0
Jul  5 04:42:54 raspberrypi kernel: [145517.954258] 5f20: 00000000 5f3e4581 00000817 d8cc5fb0 c2637380 b289d000 00000817 00000255
Jul  5 04:42:54 raspberrypi kernel: [145517.959145] 5f40: c3b435e8 00000002 c26373b0 c08cd358 d8cc5f74 c01b2f4c 00000022 000b2837
Jul  5 04:42:54 raspberrypi kernel: [145517.963781] 5f60: d8cc5f70 c0c34bfc 00000817 c08cd220 b289d000 d8cc5fb0 f82fb0b1 00489fd8
Jul  5 04:42:54 raspberrypi kernel: [145517.969023] 5f80: 00489fd8 c00189fc 00000000 b6f16b44 20000010 ffffffff b6f16b44 20000010
Jul  5 04:42:54 raspberrypi kernel: [145517.975412] 5fa0: ffffffff c445a080 00c5387d c0008f94 b289d000 b2699000 00019fb0 b9b05160
Jul  5 04:42:54 raspberrypi kernel: [145517.980850] 5fc0: 3c140794 64b612e9 6701a3af 5c12f40c e114d2ca f82fb0b1 00489fd8 00489fd8
Jul  5 04:42:54 raspberrypi kernel: [145517.989004] 5fe0: 779c4ddf b457e4f4 00000060 b6f16b44 20000010 ffffffff 00000000 00000000
Jul  5 04:42:54 raspberrypi kernel: [145517.995158]  handle_mm_fault from do_page_fault+0x138/0x3a8
Jul  5 04:42:54 raspberrypi kernel: [145517.998317]  do_page_fault from do_DataAbort+0x40/0xb4
Jul  5 04:42:54 raspberrypi kernel: [145518.001806]  do_DataAbort from __dabt_usr+0x54/0x60
Jul  5 04:42:54 raspberrypi kernel: [145518.004925] Exception stack(0xd8cc5fb0 to 0xd8cc5ff8)
Jul  5 04:42:54 raspberrypi kernel: [145518.008030] 5fa0:                                     b289d000 b2699000 00019fb0 b9b05160
Jul  5 04:42:54 raspberrypi kernel: [145518.015772] 5fc0: 3c140794 64b612e9 6701a3af 5c12f40c e114d2ca f82fb0b1 00489fd8 00489fd8
Jul  5 04:42:54 raspberrypi kernel: [145518.021711] 5fe0: 779c4ddf b457e4f4 00000060 b6f16b44 20000010 ffffffff
Jul  5 04:42:54 raspberrypi kernel: [145518.026121] Code: 73c5cb76 68737213 70f86db3 5fe469a4 (f6c517fa)
Jul  5 04:42:54 raspberrypi kernel: [145518.030452] ---[ end trace 0000000000000000 ]---
Jul  5 04:42:54 raspberrypi kernel: [145518.033432] note: python[2297] exited with irqs disabled
```

### Additional context

Python 3.9.2

docs of the library https://docs.pyrogram.org/

popcornmix, 2024-07-05:
Can you try adding `over_voltage=2` to config.txt and see if that helps?

xsolinsx, 2024-09-17:
the issue have never happened again, thanks for the help
can you maybe give a little detail on what was happening before the change of over_voltage?

closed 2024-09-17 by xsolinsx

popcornmix, 2024-09-17:
A small percentage of Pi0 need a slightly higher core voltage to run reliably.
On later pi models we run AVS at boot to determine what voltage the specific silicon used needs,
but that isn't available on Pi0.

You should be fine with manually setting it. You could try lowering it (to 1),
but it's unlikely to make much difference to observed power used or temperature.