raspberrypi/linux #6256: aplay maximum period_size issue for 'HIFIBERRY DAC+ ADC PRO' and RPi5 running Raspberry Pi OS bookworm

anchung-chen, 2024-07-06:
### Describe the bug

On the Raspberry Pi 5: aplay reports a 'period_size' of 512, which is very small and could not change to a large value.
On the Raspberry Pi 4: aplay reports a 'period_size' of 16384, which is normal and change to could change to a large value

p.s. Pi5 and Pi4 boot the same uSD Card with lately updated Raspberry Pi OS bookworm with kernel 6.6.31+rpt-rpi-2712.

### Steps to reproduce the behaviour

sox -V -r 48000 -n -b 32 -c 2 sin_1K_32bit.wav synth 30 sin 1000 vol -10dB
aplay -D hw:CARD=sndrpihifiberry,DEV=0  -v sin_1K_32bit.wav 
aplay -D hw:CARD=sndrpihifiberry,DEV=0 --period-size=32768  -v sin_1K_32bit.wav

### Device (s)

Raspberry Pi 5

### System

$ cat /etc/rpi-issue
Raspberry Pi reference 2023-10-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 962bf483c8f326405794827cce8c0313fd5880a8, stage4
$ vcgencmd version
2024/06/05 16:41:49 
Copyright (c) 2012 Broadcom
version 6fe0b091 (release) (embedded)
$ uname -a
Linux raspberrypi 6.6.31+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux

### Logs

--------------------------------RPi5-------------------------------------------
$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: sndrpihifiberry [snd_rpi_hifiberry_dacplusadcpro], device 0: HiFiBerry DAC+ADC Pro HiFi multicodec-0 [HiFiBerry DAC+ADC Pro HiFi multicodec-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

$ sox -V -r 48000 -n -b 32 -c 2 sin_1K_32bit.wav synth 30 sin 1000 vol -10dB

$ aplay -D hw:CARD=sndrpihifiberry,DEV=0  -v sin_1K_32bit.wav 
Playing WAVE 'sin_1K_32bit.wav' : Signed 32 bit Little Endian, Rate 48000 Hz, Stereo
Hardware PCM card 2 'snd_rpi_hifiberry_dacplusadcpro' device 0 subdevice 0
Its setup is:
  stream       : PLAYBACK
  access       : RW_INTERLEAVED
  format       : S32_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (384000/8)
  msbits       : 32
  buffer_size  : 24064
  period_size  : 512
  period_time  : 10666
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 512
  period_event : 0
  start_threshold  : 24064
  stop_threshold   : 24064
  silence_threshold: 0
  silence_size : 0
  boundary     : 6773413839565225984
  appl_ptr     : 0
  hw_ptr       : 0

$ aplay -D hw:CARD=sndrpihifiberry,DEV=0 --period-size=32768  -v sin_1K_32bit.wav
Playing WAVE 'sin_1K_32bit.wav' : Signed 32 bit Little Endian, Rate 48000 Hz, Stereo
Hardware PCM card 2 'snd_rpi_hifiberry_dacplusadcpro' device 0 subdevice 0
Its setup is:
  stream       : PLAYBACK
  access       : RW_INTERLEAVED
  format       : S32_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (384000/8)
  msbits       : 32
  buffer_size  : 24064
  period_size  : 512
  period_time  : 10666
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 512
  period_event : 0
  start_threshold  : 24064
  stop_threshold   : 24064
  silence_threshold: 0
  silence_size : 0
  boundary     : 6773413839565225984
  appl_ptr     : 0
  hw_ptr       : 0

--------------------------------RPi4-------------------------------------------
$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 3: sndrpihifiberry [snd_rpi_hifiberry_dacplusadcpro], device 0: HiFiBerry DAC+ADC Pro HiFi multicodec-0 [HiFiBerry DAC+ADC Pro HiFi multicodec-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

$ aplay -D hw:CARD=sndrpihifiberry,DEV=0  -v sin_1K_32bit.wav
Playing WAVE 'sin_1K_32bit.wav' : Signed 32 bit Little Endian, Rate 48000 Hz, Stereo
Hardware PCM card 3 'snd_rpi_hifiberry_dacplusadcpro' device 0 subdevice 0
Its setup is:
  stream       : PLAYBACK
  access       : RW_INTERLEAVED
  format       : S32_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (384000/8)
  msbits       : 32
  buffer_size  : 24000
  period_size  : 6000
  period_time  : 125000
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 6000
  period_event : 0
  start_threshold  : 24000
  stop_threshold   : 24000
  silence_threshold: 0
  silence_size : 0
  boundary     : 6755399441055744000
  appl_ptr     : 0
  hw_ptr       : 0

$ aplay -D hw:CARD=sndrpihifiberry,DEV=0 --period-size=32768  -v sin_1K_32bit.wav
Playing WAVE 'sin_1K_32bit.wav' : Signed 32 bit Little Endian, Rate 48000 Hz, Stereo
Hardware PCM card 3 'snd_rpi_hifiberry_dacplusadcpro' device 0 subdevice 0
Its setup is:
  stream       : PLAYBACK
  access       : RW_INTERLEAVED
  format       : S32_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (384000/8)
  msbits       : 32
  buffer_size  : 65536
  period_size  : 32768
  period_time  : 682666
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 32768
  period_event : 0
  start_threshold  : 65536
  stop_threshold   : 65536
  silence_threshold: 0
  silence_size : 0
  boundary     : 4611686018427387904
  appl_ptr     : 0
  hw_ptr       : 0

 

### Additional context

My board is the HIFIBERRY DAC+ ADC PRO, running the lately updated Raspberry Pi OS bookworm with default config.txt settings.

On the Raspberry Pi 5: aplay reports a 'period_size' of 512, which is very small and could not change to a large value.

On the Raspberry Pi 4: aplay reports a 'period_size' of 16384, which is normal.

According to the description in pyalsaaudio (https://larsimmisch.github.io/pyalsaaudio/terminology.html), the period_size in ALSA (Advanced Linux Sound Architecture) affects the interrupt interval and determines how often the user's program needs to read or write data to the ALSA data buffer.

I have several Python scripts using pyalsaaudio that set the period_size to a large value like 8192, and they work fine on Raspberry Pi 4 with Pi Bookworm OS. However, these Python scripts do not function properly on Raspberry Pi 5 with the same Pi Bookworm OS. After several hours of debugging, I discovered that the maximum value for period_size is limited to 512.

My hypothesis is that the Raspberry Pi 5 kernel or the RP1 chip restricts the maximum value of period_size.

Also try the Raspberry Pi OS Lite, results are the same as GUI version.

pelwell, 2024-07-10:
This was a timely Issue, coinciding as it did with a Pi 5 SPI investigation, during which I discovered that the DMA driver was declaring the maximum block size to be 4KB, even though the Device Tree configuration says that all channels are capable of using 256KB blocks.

Pull request #6263 makes use of the DT values, setting the maximum block size to the expected 256KB. With that patch applied, aplay -v returns the following:
```
Playing WAVE 'piano2.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
Hardware PCM card 0 'IQaudIOCODEC' device 0 subdevice 0
Its setup is:
  stream       : PLAYBACK
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16
  buffer_size  : 24000
  period_size  : 6000
  period_time  : 125000
```
In about 40 minutes time the autobuilds will have completed, at which point you will be able to install a trial kernel using `sudo rpi-update pulls/6263`. Be sure to back up any valuable data - this is beta software.

anchung-chen, 2024-07-11:
Yes, using sudo rpi-update pulls/6263 this issue had been fixed. 

Thank you pelwell.

closed 2024-07-11 by anchung-chen