raspberrypi/linux #6258: Pi 5, I2C does not detect any device

manidmt, 2024-07-08:
### Describe the bug

It seems like my Raspberry Pi 5 does not detect any device via I2C. I have tried with multiples devices (TPM or SSD1306 e.g) and the result is always the same:
nel@nel:~ $ sudo i2cdetect -y 11
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --   

nel@nel:~ $ sudo i2cdetect -y 0
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --   

These are the I2C ports that I have:
nel@nel:~ $ i2cdetect -l
i2c-0	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-3	unknown   	300000002.i2c                   	N/A
i2c-4	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-11	i2c            i2c-gpio@1                      	I2C adapter
i2c-12	i2c       	107d508200.i2c                  	I2C adapter
i2c-13	i2c       	107d508280.i2c                  	I2C adapter

Originally, there only were 3 ports, 11, 12 and 13, the other 3 were added by editing the `boot/firmware/config.txt` file:
```
 dtparam=i2c_arm=on
#dtparam=i2s=on
dtparam=i2c0=on
dtparam=i2c_csi_dsi=on
dtparam=spi=on
dtoverlay=tpm-slb9670
dtoverlay=tpm-slb9673
dtoverlay=i2c-gpio,bus=3
#dtparam=i2c_baudrate=10000
#dtoverlay=tpm-stm
# Enable audio (loads snd_bcm2835)
dtparam=audio=on
```
The lines that are commented have also been proved, but the result has been always the same. 

### Steps to reproduce the behaviour

What have I tried?

- Updating the OS, `sudo rpi-update`

- Adding to the config.txt file (I have tried this both 1 by 1 and several at a time.):

1.  dtparam=i2c_arm=on
2.  dtparam=i2c0=on
3.  dtparam=i2c_csi_dsi=on
4.  dtparam=i2c_baudrate=10000
5.  dtoverlay=tpm-slb9670
6.  dtoverlay=tpm-slb9673
7.  dtoverlay=i2c-gpio,bus=3 (in order to 

- Adding in` /etc/modules i2c_bcm2712 `and `i2c_dev`, and in `/etc/modprobe.d/raspi-blacklist.conf` the lines `blacklist spi_bcm2712` and` blacklist i2c_bcm2712`.
- Installing the Linux Kernel (https://www.raspberrypi.com/documentation/computers/linux_kernel.html) and then
1. In linux directory `make menuconfig`  and: ```
Device Drivers --->
Character devices --->
-*- TPM Hardware Support --->
<*>  TPM Interface Specification 1.3 Interface / TPM 2.0 FIFO Interface - (I2C - generic)   
)
I2C support  --->
{*} I2C support                                                                                          
[*]   Enable compatibility bits for old user-space                                                           
<*>   I2C device interface                                                                                  
{*}   I2C bus multiplexing support 
```
2. [In arch/arm/boot/dts/](https://github.com/raspberrypi/linux/commit/0727b80aac499e376a5bfbe33157dbf912490068)

### Device (s)

Raspberry Pi 5

### System

nel@nel:~ $ cat /etc/rpi-issue 
Raspberry Pi reference 2024-03-15
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, f19ee211ddafcae300827f953d143de92a5c6624, stage4

nel@nel:~ $ vcgencmd version
2024/06/05 16:41:49 
Copyright (c) 2012 Broadcom
version 6fe0b091 (release) (embedded)

nel@nel:~ $ uname -a
Linux nel 6.6.36-v8-16k+ #2 SMP PREEMPT Fri Jul  5 13:14:07 BST 2024 aarch64 GNU/Linux

### Logs

nel@nel:~ $ dmesg | grep i2c
[    0.478400] i2c_dev: i2c /dev entries driver
[    2.852788] i2c-gpio 300000002.i2c: using lines 594 (SDA) and 595 (SCL)
[    2.921180] brcmstb-i2c 107d508200.i2c:  @97500hz registered in interrupt mode
[    2.931864] brcmstb-i2c 107d508280.i2c:  @97500hz registered in interrupt mode
[  225.404231] i2c-gpio i2c-gpio@1: using lines 573 (SDA) and 574 (SCL)
[  355.290211] i2c_designware 1f00070000.i2c: controller timed out
[  356.314206] i2c_designware 1f00070000.i2c: controller timed out
[  357.338194] i2c_designware 1f00070000.i2c: controller timed out
[  358.362182] i2c_designware 1f00070000.i2c: controller timed out
[  359.386172] i2c_designware 1f00070000.i2c: controller timed out
[  360.410162] i2c_designware 1f00070000.i2c: controller timed out
[  361.434148] i2c_designware 1f00070000.i2c: controller timed out

pelwell, 2024-07-08:
You've omitted the most important piece of information - which header pins are your I2C devices connected to? I2C1 is the most common I2C interface, and it appears on GPIOs 2 & 3 (pins 3 & 5).

manidmt, 2024-07-08:
Yes, I have used those PINS and also 16 & 18 with software I2C, however, none of that worked.

pelwell, 2024-07-08:
Please give an example of at least one configuration you have tried. I want to see the wiring (preferably a photo), and the config.txt changes.

pelwell, 2024-07-08:
Hint: this seems to work for everyone else, so we're trying to narrow down what you are doing wrong.

Note that unless you are using I2C1 on GPIOs 2 & 3 (pins 3 & 5) you will need to use an external pull-up resistor.

manidmt, 2024-07-08:
[image]
[config.txt](https://github.com/user-attachments/files/16125648/config.txt)

pelwell, 2024-07-08:
The wiring looks OK, but you need to enable I2C1:
```
dtparam=i2c_arm=on
```
Check if that has worked with `pinctrl 2-3` (or `raspi-gpio get 2-3` on an older image).

manidmt, 2024-07-08:
Im sorry the config.txt file is not correct, 
[config.txt](https://github.com/user-attachments/files/16126185/config.txt)
This one is the good one

pelwell, 2024-07-08:
And what does `i2cdetect -y 1` report?

6by9, 2024-07-08:
You may need to do `i2cdetect -y -q 1`. For some reason the Pi5 is giving odd results with the SMBus read probe with i2cdetect on some devices.
https://github.com/raspberrypi/bookworm-feedback/issues/263

manidmt, 2024-07-08:
nel@nel:~ $ i2cdetect -y 1
Error: Could not open file `/dev/i2c-1' or `/dev/i2c/1': No such file or directory
nel@nel:~ $ i2cdetect -y 0
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
nel@nel:~ $ i2cdetect -y 11
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
nel@nel:~ $ i2cdetect -y -q 11
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
nel@nel:~ $ i2cdetect -y -q 0
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
nel@nel:~ $ i2cdetect -y -q 1
Error: Could not open file `/dev/i2c-1' or `/dev/i2c/1': No such file or directory

6by9, 2024-07-08:
`dtparam=i2c_arm=on` will enable i2c-1. You've got a space at the start of the line when you've quoted config.txt in your first post which may be causing the issue. You can enable it dynamically with `sudo dtparam i2c_arm`.
It should be listed by `i2cdetect -l` as "Synopsys DesignWare I2C adapter"

i2c-0 is on GPIO 0&1 (pins 27&28) 
i2c-3 is a bit bashed i2c-gpio bus from `dtoverlay=i2c-gpio,bus=3`
i2c-11 is a bit bashed i2c-gpio bus, probably from `dtoverlay=tpm-slb9673`
i2c-12 and i2c-13 are the HDMI DDC connections.

manidmt, 2024-07-08:
nel@nel:~ $ sudo dtparam i2c_arm
nel@nel:~ $ i2cdetect -l
i2c-0	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-1	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-11	i2c       	i2c-gpio@1                      	I2C adapter
i2c-12	i2c       	107d508200.i2c                  	I2C adapter
i2c-13	i2c       	107d508280.i2c                  	I2C adapter
nel@nel:~ $ i2cdetect -y -q 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --

manidmt, 2024-07-08:
After testing with another RPi5, it seems to be a hardware issue of this particular one because I have moved the SD to another RPi5 and now it is working correctly. 
nel@nel:~ $ i2cdetect -y -q 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- 2e -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --

Thanks :)

6by9, 2024-07-08:
Link pin 3 (GPIO2 / SDA) to ground, and rerun `i2cdetect`. All addresses should be reported.

Remove that link, link pin 5 (GPIO3 / SCL) to ground, and rerun `i2cdetect`. It should report no addresses detected but run very slowly (around 1second per address).

If either of those doesn't happen, then you've killed the GPIO.
If both of those does happen, then your display is probably dead. Try with another device.

closed 2024-07-08 by manidmt

v0za, 2025-07-27:
> Link pin 3 (GPIO2 / SDA) to ground, and rerun `i2cdetect`. All addresses should be reported.
>
> Remove that link, link pin 5 (GPIO3 / SCL) to ground, and rerun `i2cdetect`. It should report no [...]

When I do this it just slowly reports addresses on both scl snd sda all addresses don't get reported, i had this same issue of i2c device not getting detected it magically fixed itself the next morning it's now no longer working again @6by9

javiervillarreal2002, 2025-07-28:
I am experiencing the same issue. I am trying to use a PCA9685 GPIO Extender module with my Raspberry Pi 5 and it just won't connect. I have used a multimeter to verify that the wiring is correct and circuit continuity is present. I also made sure that the I2C connectivity is enabled in settings. I used an Arduino Uno to test the module and it successfully connected, indicating that the Raspberry Pi 5 is the problem. It is apparently a very common issue, with multiple online threads available. I have tried using both Raspberry Pi OS and Ubuntu 24.04 and the issue is present in both systems. The i2cdetect -y 1 command keeps detecting nothing. I have also ran the command with SDA and SCL connected and it ran on a slower timer, suggesting that the GPIO pins are working correctly.