raspberrypi/linux #6260: USB MIDI unplug + VL805 reset

pietervandermeer, 2024-07-08:
I've been testing for a few days and can conclude that the VL805 does not like USB MIDI devices getting unplugged. It may reset the entire USB bus. This means that mass storage devices are unmounted and then possibly remounted. If they contain your rootfs.. well then you can reboot. I've narrowed the issue down to systems with a VL805 on board. The Pi3 does not suffer from USB resets nor does the CM4 + CM4IO (I've unplugged 100's of times to confirm). The Pi4B is the easiest way to reproduce the problem. Since VL805 firmware is part of the Raspberry Pi firmware, I'm posting my issue here. I've also asked at https://forums.raspberrypi.com but there's no answer. 

**To reproduce**
The issue is easy to reproduce.
* Get a Pi4B, and run the latest Raspberry OS lite Bookworm (64 bit) from SD.
* Log in.
* Run `journalctl -fk`
* Plug in a USB stick. You'll now see the partitions getting recognized, and getting mounted.
* Plug in a USB MIDI keyboard of your choice (I've tried 3 models.. all have the same problem)... and then unplug. 
* Repeat 1-50 times. It helps to quickly insert and unplug.
* At some the USB stick gets recognized anew. 

Here you can clearly see that usb 2-2 (the usb stick) gets disconnected.. but I'm not touching it. I'm only touching the Akai usb plug. It happens already after a few unplugs.

pi@raspberrypi:~ $ journalctl -fk
jul 08 08:23:27 raspberrypi kernel: scsi host0: usb-storage 2-2:1.0
jul 08 08:23:28 raspberrypi kernel: scsi 0:0:0:0: Direct-Access     Kingston DataTraveler 3.0 1100 PQ: 0 ANSI: 6
jul 08 08:23:28 raspberrypi kernel: sd 0:0:0:0: [sda] 60463659 512-byte logical blocks: (31.0 GB/28.8 GiB)
jul 08 08:23:28 raspberrypi kernel: sd 0:0:0:0: [sda] Write Protect is off
jul 08 08:23:28 raspberrypi kernel: sd 0:0:0:0: [sda] Mode Sense: 43 00 00 00
jul 08 08:23:28 raspberrypi kernel: sd 0:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
jul 08 08:23:28 raspberrypi kernel:  sda: sda1
jul 08 08:23:28 raspberrypi kernel: sd 0:0:0:0: [sda] Attached SCSI removable disk
jul 08 08:23:28 raspberrypi kernel: sd 0:0:0:0: Attached scsi generic sg0 type 0
jul 08 08:23:29 raspberrypi kernel: exFAT-fs (sda1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
jul 08 08:24:06 raspberrypi kernel: usb 1-1.3: new full-speed USB device number 3 using xhci_hcd
jul 08 08:24:06 raspberrypi kernel: usb 1-1.3: New USB device found, idVendor=09e8, idProduct=0072, bcdDevice= 2.10
jul 08 08:24:06 raspberrypi kernel: usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
jul 08 08:24:06 raspberrypi kernel: usb 1-1.3: Product: Akai MPK25
jul 08 08:24:06 raspberrypi kernel: usb 1-1.3: Manufacturer: Akai
jul 08 08:24:06 raspberrypi kernel: usbcore: registered new interface driver snd-usb-audio
jul 08 08:24:07 raspberrypi kernel: usb 1-1.3: USB disconnect, device number 3
jul 08 08:24:08 raspberrypi kernel: usb 1-1.3: new full-speed USB device number 4 using xhci_hcd
jul 08 08:24:08 raspberrypi kernel: usb 1-1.3: New USB device found, idVendor=09e8, idProduct=0072, bcdDevice= 2.10
jul 08 08:24:08 raspberrypi kernel: usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
jul 08 08:24:08 raspberrypi kernel: usb 1-1.3: Product: Akai MPK25
jul 08 08:24:08 raspberrypi kernel: usb 1-1.3: Manufacturer: Akai
jul 08 08:24:08 raspberrypi kernel: usb 1-1.3: USB disconnect, device number 4
jul 08 08:24:09 raspberrypi kernel: usb 2-2: USB disconnect, device number 2
jul 08 08:24:09 raspberrypi kernel: sd 0:0:0:0: [sda] Synchronizing SCSI cache
jul 08 08:24:09 raspberrypi kernel: sd 0:0:0:0: [sda] Synchronize Cache(10) failed: Result: hostbyte=0x01 driverbyte=DRIVER_OK
jul 08 08:24:09 raspberrypi kernel: usb 2-2: new SuperSpeed USB device number 3 using xhci_hcd
jul 08 08:24:09 raspberrypi kernel: usb 2-2: New USB device found, idVendor=0951, idProduct=1666, bcdDevice=11.00
jul 08 08:24:09 raspberrypi kernel: usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
jul 08 08:24:09 raspberrypi kernel: usb 2-2: Product: DataTraveler 3.0
jul 08 08:24:09 raspberrypi kernel: usb 2-2: Manufacturer: Kingston
jul 08 08:24:09 raspberrypi kernel: usb 2-2: SerialNumber: 049226C3130421A019118XD1
jul 08 08:24:09 raspberrypi kernel: usb-storage 2-2:1.0: USB Mass Storage device detected
jul 08 08:24:09 raspberrypi kernel: scsi host0: usb-storage 2-2:1.0
jul 08 08:24:09 raspberrypi kernel: usb 1-1.3: new full-speed USB device number 5 using xhci_hcd
jul 08 08:24:09 raspberrypi kernel: usb 1-1.3: New USB device found, idVendor=09e8, idProduct=0072, bcdDevice= 2.10
jul 08 08:24:09 raspberrypi kernel: usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
jul 08 08:24:09 raspberrypi kernel: usb 1-1.3: Product: Akai MPK25
jul 08 08:24:09 raspberrypi kernel: usb 1-1.3: Manufacturer: Akai
jul 08 08:24:09 raspberrypi kernel: usb 1-1.3: USB disconnect, device number 5
jul 08 08:24:10 raspberrypi kernel: usb 2-2: USB disconnect, device number 3
jul 08 08:24:10 raspberrypi kernel: usb 1-1.3: new full-speed USB device number 6 using xhci_hcd
jul 08 08:24:10 raspberrypi kernel: usb 1-1.3: New USB device found, idVendor=09e8, idProduct=0072, bcdDevice= 2.10
jul 08 08:24:10 raspberrypi kernel: usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
jul 08 08:24:10 raspberrypi kernel: usb 1-1.3: Product: Akai MPK25
jul 08 08:24:10 raspberrypi kernel: usb 1-1.3: Manufacturer: Akai
jul 08 08:24:11 raspberrypi kernel: usb 2-2: new SuperSpeed USB device number 4 using xhci_hcd
jul 08 08:24:11 raspberrypi kernel: usb 2-2: New USB device found, idVendor=0951, idProduct=1666, bcdDevice=11.00
jul 08 08:24:11 raspberrypi kernel: usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
jul 08 08:24:11 raspberrypi kernel: usb 2-2: Product: DataTraveler 3.0
jul 08 08:24:11 raspberrypi kernel: usb 2-2: Manufacturer: Kingston
jul 08 08:24:11 raspberrypi kernel: usb 2-2: SerialNumber: 049226C3130421A019118XD1
jul 08 08:24:11 raspberrypi kernel: usb-storage 2-2:1.0: USB Mass Storage device detected
jul 08 08:24:11 raspberrypi kernel: scsi host0: usb-storage 2-2:1.0
jul 08 08:24:11 raspberrypi kernel: usb 1-1.3: USB disconnect, device number 6
[... 1213 lines cut ...]

There are different, but related problems, as well. At one point an I2S based alsa device was unreachable. If that's the case, it may also be there's a dangerous bug in one of the kernel modules.. which means i'll have to post this in the kernel repo as well. 

Additionally, if you monitor incoming midi data with the standard alsa tool ```amidi```:

```amidi -d -phw:3,0,0```

then you'll see hundreds, even thousands of ```urb status -32``` messages flying around in ```journalctl -fk``` (or ```dmesg```)

even up to the point of the kernel message buffer getting flooded over and over. this does _not_ sound healthy. 

```
...
jul 08 08:33:38 raspberrypi kernel: usb 1-1.3: urb status -32
jul 08 08:33:38 raspberrypi kernel: usb 1-1.3: urb status -32
jul 08 08:33:38 raspberrypi systemd-journald[245]: /dev/kmsg buffer overrun, some messages lost.
jul 08 08:33:38 raspberrypi kernel: usb 1-1.3: urb status -32
jul 08 08:33:38 raspberrypi kernel: usb 1-1.3: urb status -32
jul 08 08:33:38 raspberrypi kernel: usb 1-1.3: urb status -32
jul 08 08:33:38 raspberrypi kernel: usb 1-1.3: urb status -32
jul 08 08:33:38 raspberrypi systemd-journald[245]: /dev/kmsg buffer overrun, some messages lost.
...
```

needless to say, this is a bog standard pi4b and a bog standard raspberry os lite bookworm.

popcornmix, 2024-07-08:
> @6by9 so should I post this on the raspberry/linux repo instead?

I've transferred it. It's more likely to a linux issue that a firmware issue (the firmware does nothing related to usb once kernel has started).

pelwell, 2024-07-08:
Unless it's a VL805 firmware problem.

P33M, 2024-07-08:
It looks like the device in question is a full-size keyboard. Does it back-power VBus?

An easy (but relatively inaccurate) way to tell is if you plug the keyboard into a powered Pi 4, let it enumerate, then unplug the USB-C port on the Pi. If you still have a red LED, then the keyboard is misbehaving.

pietervandermeer, 2024-07-08:
The answer is no: the usb status -32 flood doesn't destabilize the system, but of course it fills up the system logs.

pelwell, 2024-07-08:
You may be able to use an unpowered hub as an "insulator"

pietervandermeer, 2024-07-09:
@pelwell That's pragmatic, but otoh.. The VL805 itself is already a hub. And the Pi3B's hub doesn't suffer from this problem.

pelwell, 2024-07-09:
The fact that some devices are tolerant of bad behaviour doesn't make the behaviour less bad.

6by9, 2024-07-09:
The USB specification states that the USB peripheral must not back power the USB host, so the Akai isn't compliant to the USB spec.
Having found a [review of the MPK25](https://www.soundonsound.com/reviews/akai-mpk25) it says it isn't supplied with a mains adapter as standard.

On all Pis all the USB connectors connect the VBUS lines together. Any one USB peripheral that does dubious things to that supply rail is likely to affect all USB devices connected directly to the Pi.
Short VBUS of one USB to ground and any bus powered devices will disconnect as they have no power. The USB over current protection in the Pi should stop the Pi from rebooting. 

As for the 3B not having the same issue, you're likely down to exactly how quickly the protection circuits recover from being reverse powered to having to supply power to all the USB devices.

pietervandermeer, 2024-07-10:
Indeed, this backpowering behavior is against USB specifications. It's best to not let this issue go philosophical. I'll accept that some device manufacturers need to do their homework, but also that host manufacturers, if they have the budget (and the users), need to develop defensively. I'll leave it at that.

For the Linux kernel it's more important to understand why it prints a thousand times the same messages for something as common as a broken pipe..