raspberrypi/linux #6261: only 39bit of virtual address space available

theofficialgman, 2024-07-08:
### Describe the bug

using 39bits of virtual address space causes issues primarily on emulation of x64 software (such as wine with box64) and it would be better to build the kernel with 48bits of virtual address space

this is a common usecase as evident by the 16K+ installs of `Wine (x64)` via pi-apps https://pi-apps.io/wiki/getting-started/apps-list/#wine-(x64) and ~44K installs of `Box64` via pi-apps https://pi-apps.io/wiki/getting-started/apps-list/#box64

### Steps to reproduce the behaviour

boot linux. observe that it uses the (default) 39bits address space on 4K pagesize and 47bits address space on 16K pagesize

### Device (s)

Raspberry Pi Zero 2 W, Raspberry Pi 3 Mod. A+, Raspberry Pi 3 Mod. B, Raspberry Pi 3 Mod. B+, Raspberry Pi 4 Mod. B, Raspberry Pi 400, Raspberry Pi 5, Raspberry Pi CM3, Raspberry Pi CM3 Lite, Raspberry Pi CM3+, Raspberry Pi CM3+ Lite, Raspberry Pi CM4, Raspberry Pi CM4 Lite

### System

all ARM64 kernel releases (both 4K and 16K pagesizes)

### Additional context

Solution:
the following args (though all are not explicitly necessary since they are dependencies of one another) achieve 48bits of virtual address space on ARM64

```
# enable 48bits virtual address space size
CONFIG_ARM64_VA_BITS_48=y
# set default virtual address space size to 48bits (default when CONFIG_ARM64_VA_BITS_48=y)
CONFIG_ARM64_VA_BITS=48
```

pelwell, 2024-07-09:
Duplicate of https://github.com/raspberrypi/linux/issues/4375?

theofficialgman, 2024-07-09:
> Duplicate of #4375?

as correctly described here https://github.com/raspberrypi/linux/issues/4375#issuecomment-862982033 , the page table size is a function of the virtual address space, not the other way around. so the other issue is asking for the same thing but (the title is) worded the wrong way.

theofficialgman, 2024-07-10:
52bits (with a runtime fallback to 48bits on hardware that doesn't support it) is the default on all arm64 kernels as of kernel 6.9

theofficialgman, 2024-07-22:
> this is a common usecase as evident by the 16K+ installs of `Wine (x64)` via pi-apps https://pi-apps.io/wiki/getting-started/apps-list/#wine-(x64) and ~44K installs of `Box64` via pi-apps https://pi-apps.io/wiki/getting-started/apps-list/#box64

the particular wine issue has been resolved via a custom patch https://github.com/Pi-Apps-Coders/files/commit/a8609ce0d57439f5ec082097d6158e28ba45da2b

closed 2024-07-23 by theofficialgman