raspberrypi/linux #6262: RTC `wakealarm` is improperly deleted from `sysfs`

seamusdemora, 2024-07-09:
### Describe the bug

A future time value stored in `/sys/class/rtc/rtc0/wakealarm` is deleted improperly; e.g. following a `halt` and re-start of the Raspberry Pi. 

### Steps to reproduce the behaviour

1.    Set the `wakealarm` (in sysfs) to a value in the future, let's call it Tf, and let's assume Tf is 24 hours beyond the time at which `wakealarm` is set; i.e. `date '+%s' -d '+ 24 hours' | sudo tee /sys/class/rtc/rtc0/wakealarm`; 

    IOW: Tf = date '+%s' -d '+ 24 hours'
    Let's assume the value of Tf = `1720585180`

2.    Shortly (e.g. 30 minutes) after setting `wakealarm` as in Step 1, issue a `halt` from the command line. 

    NOTE - I do not believe this is material to the bug report, but mention it "just in case" :  I have the `gpio-poweroff` overlay set in the `config.txt` file; i.e. `gpio-poweroff` "fires" after the `halt` is issued in Step 2.  In my "setup", I have `gpio-poweroff` connected to a "power switch" (GPIO 26 drives a small NPN transistor which in turn is connected to the coil of a latching relay, which causes the relay to remove power from the RPi.  

3.     Shortly (e.g. 30 minutes) after issuing the `halt` in Step 2, power is "re-connected" to the RPi, and it boots. 

4.     After booting, check the value stored in `wakealarm`. You will find there is nothing stored in `wakealarm`.  

5.     This is not a case of the `wakealarm` value still being retained in the DS3231 chip register; i.e. this is **NOT** a case of `/sys/class/rtc/rtc0/wakealarm` = NULL, but the chip register still contains the previously programmed value. I've confirmed that the alarm simply **does not go off / no #INTerrupt**; i.e. it *appears* the Tf value has been removed from the DS3231 register. 

### Device (s)

Raspberry Pi Zero W / WH

### System

`cat /etc/rpi-issue` ==>
Raspberry Pi reference 2022-01-28
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, f01430c9d8f67a4b9719cc00e74a2079d3834d5d, stage2

`vcgencmd version` ==>
Mar 17 2023 10:52:42
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start) 

`uname -a` ==>
Linux raspberrypi0w 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux 

### Additional context

What I expected to happen:

In Step 4 (Steps to reproduce behavior), I expected the `wakealarm` setting (Tf = `1720585180`) to be available by reading `wakealarm` (via `cat wakealarm`). I initially assumed this was just a case that the driver was not updating the `wakealarm` value from the DS3231 register.  However, further experiments revealed this to **not** be the case. Apparently the driver "zeros out" the DS3231's register at some time after the `halt` or subsequent boot. This of course results in no alarm/#INTerrupt.

pelwell, 2024-07-09:
This behaviour is not going to be unique to the Raspberry Pi kernel. You should email linux-rtc@vger.kernel.org.

If you think gpio-poweroff may be playing a part, try removing it.

seamusdemora, 2024-07-09:
Thanks - Edited bug report.

I've emailed kernel.org; do you agree that this is something that should be fixed?

pelwell, 2024-07-09:
I don't have an opinion on it, having not played with wakealarms enough to know what the behaviour should be.

seamusdemora, 2024-07-29:
@pelwell 

FWIW, I posted twice, but had zero luck raising anyone on the mail list at[linux-rtc@vger.kernel.org](mailto:linux-rtc@vger.kernel.org).  It seems this site is used for submitting patches - no discussion at all. Just FYI.

6by9, 2024-07-29:
Isn't the driver doing exactly what it says it's doing in the comment at https://github.com/raspberrypi/linux/blob/rpi-6.6.y/drivers/rtc/rtc-ds1307.c#L1799-L1804
```
		 * Using IRQ or defined as wakeup-source?
		 * Disable the square wave and both alarms.
...
			regs[0] &= ~(DS1337_BIT_A2IE | DS1337_BIT_A1IE);
```
The sysfs `show` hook doesn't display disabled alarms as https://github.com/raspberrypi/linux/blob/rpi-6.6.y/drivers/rtc/sysfs.c#L134-L136
```
	/* Don't show disabled alarms.  For uniformity, RTC alarms are
	 * conceptually one-shot, even though some common RTCs (on PCs)
	 * don't actually work that way.
```
and [ds1337_read_alarm](https://github.com/raspberrypi/linux/blob/rpi-6.6.y/drivers/rtc/rtc-ds1307.c#L441) won't have set `enabled` due to the interrupt being disabled.

Are there any other RTC drivers that do retain alarm status over a reboot? I don't know enough about the subsystem to say which is the expected behaviour.

Whenever emailing lists about a specific driver, it's always worth including the driver's maintainers as reported by [scripts/get_maintainer.pl](https://github.com/torvalds/linux/blob/master/scripts/get_maintainer.pl) (or a quick search in the [raw MAINTAINERS file](https://github.com/torvalds/linux/blob/master/MAINTAINERS#L19232))

seamusdemora, 2024-08-01:
> Isn't the driver doing exactly what it says it's doing in the comment at https://github.com/raspberrypi/linux/blob/rpi-6.6.y/drivers/rtc/rtc-ds1307.c#L1799-L1804
>
> ```
> 		 * Using IRQ or defined as [...]

I've read this comment four times now over two days, and its meaning remains unclear to me ... "IRQ OR wakeup-source..." What does that mean? The pin (on a DS3231) is labeled `#INT/SQW` - meaning that it either puts out a 1 PPS SQuare Wave - OR - it is pulled low (from a pulled-up state) at the `alarm time`. That being the case, what does the author mean when he says, *"IRQ or defined as wakeup-source"*? 

That said, I have difficulty in making the connection you propose; i.e. that the comment implies that the alarm time should be/is reset to zero during a reboot. In the first place: I can't see why anyone would _want_ to do that. For example, I might have an alarm set for 3 days from today; why would I want to reset it each time I have to `reboot` (or `halt`)? 

WRT your question: 
> Are there any other RTC drivers that do retain alarm status over a reboot? 

I can't answer that as this is the only RTC driver I've ever used. 

WRT your other suggestion about contacting the maintainer directly, I'll try that... although I'd ***think*** that he's subscribed to the mail list, and would have replied to one of my requests. Frankly - my opinion of *some of* the people who work on the kernel is *"lower than whale shit"*, but I do try to give every kernel maintainer the benefit of the doubt :)

pelwell, 2024-08-01:
That's quite enough time spent on a question for upstream.

closed 2024-08-01 by pelwell

seamusdemora, 2024-08-20:
I'd like to thank you for all the help. It's a small thanks - the best I can do.

seamusdemora, 2024-09-13:
I'm posting here b/c that seems *better* than opening a *new* issue on the same subject.

It's taken some time to get a reply, but A few days ago I got an email from the RTC kernel driver maintainer in which he said:

>... i was wrong. The bug is present since cb49a5e9ee4fb0ab6a84a29c4de6c1bd112d38df, 15 years ago... 

So it seems likely now that a patch will be implemented to correct the behavior I described above. I know you guys don't give a damn, but I thought I'd ask just in case: Would you like to have an *option* (*parameter*) to retain the existing behavior?

pelwell, 2024-09-13:
> Would you like to have an option (parameter) to retain the existing behavior?

It's good that the maintainer has identified this as a bug. This implies that other RTCs behave differently, i.e. they retain the alarm state, so I don't there's any good reason to allow the existing behaviour to be preserved.

seamusdemora, 2024-09-14:
OK - as you wish. Would you like to have the patch - or would you prefer to let it drift down from upstream?

pelwell, 2024-09-14:
Once it is merged upstream we can back-port it.