raspberrypi/linux #6264: Visual DRM plane artifacts on vertical edges.

dividuum, 2024-07-11:
### Describe the bug

While playing around in my software, I managed to come across a DRM plane layout that results in artifacts on vertical plane edges. Look at this screenshot:

[image]

I'm playing three videos across two displays (separated by the white line in the screenshot). As a result my software splits the middle video into two planes (labeled '2' and '3'). To reduce overdraw, my software also removes the non-visible part of the plane '3' that's hidden behind the plane '4'.

The video is 560x360 pixel, DRM "YU12". The relevant output in `/sys/kernel/debug/dri/1/state` for plane '3' is:

```
    crtc-pos=360x848+0+124
    src-pos=65.000000x320.000000+280.000000+0.000000
```

For plane '4':

```
    crtc-pos=1560x1080+360+0
    src-pos=560.000000x320.000000+0.000000+0.000000
```

So plane '4' touches plane '3' at x=360. That edge has odd bleeding of pixels going on:

[image]

### Steps to reproduce the behaviour

Not sure yet. That specific configuration doesn't work. Resizing the middle video (and thus plane '2' and '3') likely fixes the issue. Tested only on a Pi5 for now.

### Device (s)

Raspberry Pi 5

### System

My custom info-beamer OS. Using 6.6.33 kernel from [https://github.com/raspberrypi/rpi-firmware](rpi-firmware):

```
Linux info-beamer-xxxxxxxxx 6.6.33-v8+ #1770 SMP PREEMPT Wed Jun 12 17:47:55 BST 2024 aarch64 GNU/Linux                                                                                                                                                                                                                                                                                                                                  
2024/06/05 16:41:49 
Copyright (c) 2012 Broadcom
version 6fe0b091 (release) (embedded)
```

popcornmix, 2024-07-24:
Might be best to provide minimal test code that shows the issue.
Can you show the issue with a single display and two video planes?

6by9, 2024-07-24:
(Sorry, read and then forgot about this issue).

Can you also provide the contents of /sys/kernel/debug/dri/1/hvs_dlists, so that we can analyse what the HVS has been asked to compose?

dividuum, 2024-07-24:
> Can you show the issue with a single display and two video planes?

Not sure. I've not observed it with just a single display so far, but that might just be a coincidence. The dual display layout was just something I randomly put together and other layouts even across both displays don't show this glitch. 

```
HVS chan 0:
dlist: 98: 0x5b000008
dlist: 99: 0x007c0168
dlist: 100: 0x4205fff0
dlist: 101: 0x034f0617
dlist: 102: 0x013f0117
dlist: 103: 0x808f011c
dlist: 104: 0x00160c0a
dlist: 105: 0x7cf80000
dlist: 106: 0x00000240
dlist: 107: 0x001b100a
dlist: 108: 0x7cfad000
dlist: 109: 0x00000120
dlist: 110: 0x001e140a
dlist: 111: 0x7cfb8400
dlist: 112: 0x00000120
dlist: 113: 0x00000000
dlist: 114: 0x4016f970
dlist: 115: 0x40304d70
dlist: 116: 0xc00071e3
dlist: 117: 0x402df260
dlist: 118: 0x40609a60
dlist: 119: 0x0000e3c6
dlist: 120: 0x00000020
dlist: 121: 0x00000020
dlist: 122: 0x00000020
dlist: 123: 0x00000020
dlist: 124: 0x80000000
dlist: 125: 0x5b000008
dlist: 126: 0x01130034
dlist: 127: 0x4205fff0
dlist: 128: 0x024005e7
dlist: 129: 0x013f022f
dlist: 130: 0x809f013f
dlist: 131: 0x0042480a
dlist: 132: 0x7cf00000
dlist: 133: 0x00000240
dlist: 134: 0x00474c0a
dlist: 135: 0x7cf2d000
dlist: 136: 0x00000120
dlist: 137: 0x004a500a
dlist: 138: 0x7cf38400
dlist: 139: 0x00000120
dlist: 140: 0x00000012
dlist: 141: 0x402f6870
dlist: 142: 0x4046fc70
dlist: 143: 0x40007700
dlist: 144: 0x405ed060
dlist: 145: 0x408df860
dlist: 146: 0x0000ee00
dlist: 147: 0x00000020
dlist: 148: 0x00000020
dlist: 149: 0x00000020
dlist: 150: 0x00000020
dlist: 151: 0x80000000
dlist: 152: 0x80000000
HVS chan 1:
dlist: 208: 0x5b000008
dlist: 209: 0x007c0000
dlist: 210: 0x4205fff0
dlist: 211: 0x034f0167
dlist: 212: 0x013f0040
dlist: 213: 0x806900d1
dlist: 214: 0x0000180a
dlist: 215: 0x7cf80118
dlist: 216: 0x00000240
dlist: 217: 0x00051c0a
dlist: 218: 0x7cfad08c
dlist: 219: 0x00000120
dlist: 220: 0x0008200a
dlist: 221: 0x7cfb848c
dlist: 222: 0x00000120
dlist: 223: 0x00000035
dlist: 224: 0x40171c70
dlist: 225: 0x40304d70
dlist: 226: 0x800046a2
dlist: 227: 0x402e3860
dlist: 228: 0x40609a60
dlist: 229: 0x80008d44
dlist: 230: 0x00000020
dlist: 231: 0x00000020
dlist: 232: 0x00000020
dlist: 233: 0x00000020
dlist: 234: 0x80000000
dlist: 235: 0x5b000008
dlist: 236: 0x00000168
dlist: 237: 0x4205fff0
dlist: 238: 0x04370617
dlist: 239: 0x013f022f
dlist: 240: 0x806600ca
dlist: 241: 0x004d540a
dlist: 242: 0x7ce80000
dlist: 243: 0x00000240
dlist: 244: 0x0052580a
dlist: 245: 0x7cead000
dlist: 246: 0x00000120
dlist: 247: 0x00555c0a
dlist: 248: 0x7ceb8400
dlist: 249: 0x00000120
dlist: 250: 0x00000039
dlist: 251: 0x402df270
dlist: 252: 0x4025ed70
dlist: 253: 0x8000579b
dlist: 254: 0x405be461
dlist: 255: 0x404bda60
dlist: 256: 0x8000af36
dlist: 257: 0x00000020
dlist: 258: 0x00000020
dlist: 259: 0x00000020
dlist: 260: 0x00000020
dlist: 261: 0x80000000
dlist: 262: 0x80000000
HVS chan 2 disabled
```

dividuum, 2024-07-25:
I won't be able to provide a condensed test case within the next two weeks, but I can try to put one together after that. Don't put too much energy into this in the mean time.

dividuum, 2024-08-07:
Ok. Got something to easily reproduce this:

 * Latest Raspberry Pi OS (I've used the 32bit userland version for my test case)
 * Download [this tgz](https://usercontent.infobeamer.com/issue-6264.tgz) containing the 32bit info-beamer binary and stripped down code to trigger this issue.
 * Might need libao4 (`apt-get install libao4`)
 * Stop any wayland or X11, so DRM/KMS is free to use.
 * Connect two FullHD capable displays.
 * From a console, run `./run-testcase`

The issue also happens with 64bit. If you want to run that instead, download the 64bit binary of info-beamer [from here](https://info-beamer.com/download/player) and replace the included `info-beamer` binary with the one from that download.

Let me know if that's helpful or if there's anything else I might provide.

```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 6.6.31+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux
pi@raspberrypi:~ $ vcgencmd version
2024/06/05 16:41:49 
Copyright (c) 2012 Broadcom
version 6fe0b091 (release) (embedded)
```