raspberrypi/linux #6266: RPi 3B+ cabbot boot after commit e6c1e86

iucoen, 2024-07-13:
### Describe the bug

After updating to commit e6c1e86 my system no longer boots. I use Gentoo and OpenRC gets stuck at trying to mount the root file system. The commit in question seems to be relevant.

### Steps to reproduce the behaviour

- Install kernel at commit e6c1e86
- Boot process gets stuck at trying to mount the root filesystem

### Device (s)

Raspberry Pi 3 Mod. B+

### System

Gentoo aarch64 (musl + OpenRC)

P33M, 2024-07-13:
That commit should be a no-op on a 3B+. It doesn't support CQHCI?

Please post your full kernel .config and the console output from boot.

iucoen, 2024-07-13:
I'm using bcm2711_defconfig, which has CONFIG_MMC_CQHCI=y. Full MMC related configs:

```
CONFIG_MMC=y
CONFIG_PWRSEQ_EMMC=y
CONFIG_MMC_BLOCK=y
CONFIG_MMC_BLOCK_MINORS=32
# CONFIG_MMC_TEST is not set
# MMC/SD/SDIO Host Controller Drivers
CONFIG_MMC_BCM2835_MMC=y
CONFIG_MMC_BCM2835_DMA=y
CONFIG_MMC_BCM2835_PIO_DMA_BARRIER=2
CONFIG_MMC_BCM2835_SDHOST=y
# CONFIG_MMC_DEBUG is not set
# CONFIG_MMC_ARMMMCI is not set
CONFIG_MMC_SDHCI=y
CONFIG_MMC_SDHCI_IO_ACCESSORS=y
# CONFIG_MMC_SDHCI_PCI is not set
CONFIG_MMC_SDHCI_PLTFM=y
# CONFIG_MMC_SDHCI_OF_ARASAN is not set
# CONFIG_MMC_SDHCI_OF_AT91 is not set
CONFIG_MMC_SDHCI_OF_DWCMSHC=m
# CONFIG_MMC_SDHCI_CADENCE is not set
# CONFIG_MMC_SDHCI_F_SDH30 is not set
# CONFIG_MMC_SDHCI_MILBEAUT is not set
CONFIG_MMC_SDHCI_IPROC=y
# CONFIG_MMC_TIFM_SD is not set
CONFIG_MMC_SPI=m
# CONFIG_MMC_CB710 is not set
# CONFIG_MMC_VIA_SDMMC is not set
# CONFIG_MMC_DW is not set
# CONFIG_MMC_VUB300 is not set
# CONFIG_MMC_USHC is not set
# CONFIG_MMC_USDHI6ROL0 is not set
CONFIG_MMC_CQHCI=y
CONFIG_MMC_HSQ=y
# CONFIG_MMC_TOSHIBA_PCI is not set
# CONFIG_MMC_BCM2835 is not set
# CONFIG_MMC_MTK is not set
CONFIG_MMC_SDHCI_BRCMSTB=y
# CONFIG_MMC_SDHCI_XENON is not set
```

iucoen, 2024-07-13:
Console output will be a bit of a challenge for me since I don't have serial port hooked up right now.

pelwell, 2024-07-14:
There's another near-identical report here (now closed to avoid duplication): https://github.com/raspberrypi/linux/issues/6267

fishrockz, 2024-07-14:
[image]

As nether of us have a serial port set up here is a pic of the stuck condition. I can pull out the full boot log if you need it.

closed 2024-07-17 by iucoen