raspberrypi/linux #6270: USB3 is broken on latest kernel

solskogen, 2024-07-16:
### Describe the bug

On Arch Linux ARM USB 3 is broken. I cannot boot from it, and even if I boot from a SD card I cannot attach even the disk without the system freezing. The sources for the package (with kernel config) is here: https://github.com/archlinuxarm/PKGBUILDs/tree/master/core/linux-rpi-16k. Rolling back to 6.6.37-2, fixes the problem. while 6.6.39-1 introduces it. 

### Steps to reproduce the behaviour

Install the 6.6.39-1  (or 6.6.39-2 ) kernel on Arch Linux ARM.  6.6.39-2 corresponds with this repository commit: 1abc413af44652d6a76d5b5c2afe90788595008e

### Device (s)

Raspberry Pi 5

### System

Arch Linux ARM. 6.6.39-2 

### Additional context

USB2 works.

timg236, 2024-07-16:
Saying 'USB 3' doesn't work is non-specific and not very helpful when reporting issues against a different repository to the build that you are running.
Since USB3.0 seems to work fine with many USB3 devices on RPi OS rpi-update latest (Linux pi5 6.6.39-v8-16k+) you'll need to add more details about your particular setup, try on RPi OS to create a specific configuration that people could try to reproduce.

solskogen, 2024-07-16:
I've updated the issue a bit. 
This is the outout of lsusb:
`Bus 004 Device 002: ID 174c:55aa ASMedia Technology Inc. ASM1051E SATA 6Gb/s bridge, ASM1053E SATA 6Gb/s bridge, ASM1153 SATA 3Gb/s bridge, ASM1153E SATA 6Gb/s bridge
`

I've now also tried Raspberry Pi OS and ran rpi-update (without any spesific arguments), and it also have the same problem. The last line when booting is:
`Timed out for waiting the udev queue being empty`

solskogen, 2024-07-16:
The branches `rpi-6.9.y` and `rpi-6.10.y` does not have this problem.

P33M, 2024-07-16:
Confirmed on Pi 5 / 6.6.39 - something goes very very wrong.

It may seem like a hard freeze but things progress extremely slowly - if I have dmesg --follow running, I get:
```
[   31.901502] usb 4-1: new SuperSpeed USB device number 2 using xhci-hcd
[   31.925750] usb 4-1: New USB device found, idVendor=152d, idProduct=1561, bcd       Device= 2.04
[   31.925756] usb 4-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   31.925760] usb 4-1: Product: SABRENT
[   31.925763] usb 4-1: Manufacturer: SABRENT
[   31.925765] usb 4-1: SerialNumber: DB9876543214E
[   31.929673] scsi host0: uas
[   31.931306] scsi 0:0:0:0: Direct-Access     SABRENT                   0204 PQ       : 0 ANSI: 6
[   31.934944] sd 0:0:0:0: Attached scsi generic sg0 type 0
[   35.096483] sd 0:0:0:0: [sda] 500118192 512-byte logical blocks: (256 GB/238        GiB)
[   35.096491] sd 0:0:0:0: [sda] 4096-byte physical blocks
[   35.096694] sd 0:0:0:0: [sda] Write Protect is off
[   35.096697] sd 0:0:0:0: [sda] Mode Sense: 53 00 00 08
[   35.097058] sd 0:0:0:0: [sda] Write cache: enabled, read cache: enabled, does       n't support DPO or FUA
[   36.189450] ------------[ cut here ]------------
[   36.189455] Firmware transaction timeout
[   36.189466] WARNING: CPU: 1 PID: 1015 at drivers/firmware/raspberrypi.c:69 rp       i_firmware_property_list+0x228/0x268
[   36.189477] Modules linked in: sg rfcomm snd_seq_dummy snd_hrtimer snd_seq sn       d_seq_device algif_hash algif_skcipher af_alg bnep binfmt_misc vc4 aes_ce_blk ae       s_ce_cipher ghash_ce spidev brcmfmac_wcc gf128mul sha2_ce hci_uart snd_soc_hdmi_       codec drm_display_helper sha256_arm64 btbcm sha1_ce cec bluetooth drm_dma_helper        drm_kms_helper raspberrypi_hwmon ecdh_generic ecc snd_soc_core libaes brcmfmac        snd_compress snd_pcm_dmaengine brcmutil spi_bcm2835 i2c_brcmstb rpivid_hevc(C) c       fg80211 gpio_keys pisp_be v4l2_mem2mem videobuf2_dma_contig snd_pcm videobuf2_me       mops snd_timer snd v3d videobuf2_v4l2 videodev rfkill gpu_sched pwm_fan videobuf       2_common rp1_adc raspberrypi_gpiomem mc drm_shmem_helper nvmem_rmem uio_pdrv_gen       irq uio drm i2c_dev fuse drm_panel_orientation_quirks backlight dm_mod ip_tables        x_tables ipv6
[   36.189543] CPU: 1 PID: 1015 Comm: wayfire Tainted: G         C         6.6.3       9-v8-16k+ #1
[   36.189547] Hardware name: Raspberry Pi 5 Model B Rev 1.0 (DT)
[   36.189549] pstate: 60400009 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[   36.189551] pc : rpi_firmware_property_list+0x228/0x268
[   36.189554] lr : rpi_firmware_property_list+0x228/0x268
[   36.189557] sp : ffffc000809c3900
[   36.189558] x29: ffffc000809c3900 x28: ffff80010044eac0 x27: ffffd0008581dfe0
[   36.189562] x26: ffffd0008581def8 x25: ffffc000804f8008 x24: 0000000000004000
[   36.189565] x23: ffff80010a409740 x22: 00000000ffffff92 x21: 0000000000000018
[   36.189568] x20: ffff80010044ea80 x19: ffffc000804f8000 x18: 0000000000000000
[   36.189572] x17: 0000000000000000 x16: 0000000000000000 x15: 00007fff0288a628
[   36.189575] x14: 0000000000000000 x13: 74756f656d697420 x12: 6e6f69746361736e
[   36.189578] x11: fffffffffffe0000 x10: ffffd000856fa068 x9 : ffffd00084109184
[   36.189582] x8 : 00000000ffffefff x7 : ffffd000856f6708 x6 : 0000000000003960
[   36.189586] x5 : ffff8001ffaced48 x4 : 00000000fffff264 x3 : ffffb0017a800000
[   36.189589] x2 : 0000000000000000 x1 : 0000000000000000 x0 : ffff800105565c40
[   36.189592] Call trace:
[   36.189594]  rpi_firmware_property_list+0x228/0x268
[   36.189596]  rpi_firmware_property+0x78/0x118
[   36.189599]  raspberrypi_fw_get_rate+0x4c/0x88
[   36.189602]  clk_recalc+0x40/0xb8
[   36.189607]  __clk_recalc_rates+0x44/0xc0
[   36.189611]  clk_set_rate_range_nolock.part.0+0x1f0/0x2e8
[   36.189614]  clk_set_min_rate+0x4c/0x108
[   36.189616]  v3d_clock_up_get+0x6c/0xa0 [v3d]
[   36.189628]  v3d_job_init.constprop.0+0xe0/0x258 [v3d]
[   36.189636]  v3d_submit_cl_ioctl+0x9c/0x780 [v3d]
[   36.189643]  drm_ioctl_kernel+0xd4/0x188 [drm]
[   36.189689]  drm_ioctl+0x21c/0x4d8 [drm]
[   36.189726]  __arm64_sys_ioctl+0xb0/0x100
[   36.189730]  invoke_syscall+0x50/0x128
[   36.189735]  el0_svc_common.constprop.0+0x48/0xf8
[   36.189738]  do_el0_svc+0x28/0x40
[   36.189741]  el0_svc+0x40/0xf8
[   36.189744]  el0t_64_sync_handler+0x13c/0x158
[   36.189747]  el0t_64_sync+0x190/0x198
[   36.189749] ---[ end trace 0000000000000000 ]---
[   37.213447] raspberrypi-clk soc:firmware:clocks: Failed to change fw-clk-v3d        frequency: -110
[   37.853453] ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg        failed w/status -110
[   37.853458] brcmfmac: brcmf_dump_obss: dump_obss error (-110)
[   40.413454] ieee80211 phy0: brcmf_set_mpc: fail to set mpc
[   56.125441] rcu: INFO: rcu_preempt detected stalls on CPUs/tasks:
[   56.125446] rcu:     0-...0: (1 ticks this GP) idle=2d64/1/0x4000000000000002        softirq=7204/7204 fqs=1532
[   56.125451] rcu:     (detected by 1, t=5255 jiffies, g=2589, q=54 ncpus=4)
[   56.125454] Task dump for CPU 0:
[   56.125455] task:swapper/0       state:R  running task     stack:0     pid:0            ppid:0      flags:0x0000000a
[   56.125461] Call trace:
[   56.125462]  __switch_to+0xe8/0x168
[   56.125469]  cpu_number+0x0/0x8
[  119.145441] rcu: INFO: rcu_preempt detected stalls on CPUs/tasks:
[  119.145448] rcu:     0-...0: (1 ticks this GP) idle=2d64/1/0x4000000000000002        softirq=7204/7204 fqs=4683
[  119.145453] rcu:     (detected by 2, t=21010 jiffies, g=2589, q=795 ncpus=4)
[  119.145456] Task dump for CPU 0:
[  119.145457] task:swapper/0       state:R  running task     stack:0     pid:0            ppid:0      flags:0x0000000a
[  119.145462] Call trace:
[  119.145465]  __switch_to+0xe8/0x168
[  119.145472]  cpu_number+0x0/0x8

```
It took about 2 minutes to print out that stack trace.

HiassofT, 2024-07-16:
This reads like it might be this 6.6.39 regression:
https://bugzilla.kernel.org/show_bug.cgi?id=219039
https://lore.kernel.org/r/CA+3zgmvct7BWib9A7O1ykUf=0nZpdbdpXBdPWOCqfPuyCT3fug@mail.gmail.com

That was fixed with 6.6.40 upstream stable yesterday by reverting commit 9a24eb8010c2dc6a2eba56e3eb9fc07d14ffe00a "usb: xhci: prevent potential failure in handle_tx_event() for Transfer events without TRB"

see also https://cdn.kernel.org/pub/linux/kernel/v6.x/ChangeLog-6.6.40

Might be worth testing with that commit reverted in the rpi-6.6.y tree

P33M, 2024-07-16:
Reverting that commit does appear to fix the hang.

P33M, 2024-07-16:
6.6.39 -> 6,6,40 contains just the revert, so to avoid churn pulling in from upstream is going to be better.

solskogen, 2024-07-17:
Confirmed

closed 2024-07-17 by solskogen

pelwell, 2024-07-17:
Thanks, @HiassofT!