raspberrypi/linux #6281: Raspberry Pi 5 HDMI interlaced does not work properly

zoli0726, 2024-07-24:
### Describe the bug

Since Pi 5 DSI does not support interlaced anymore I went towards the route of using an HDMI->VGA dongle to produce 15khz signal for my CRT tv.

240p works very well, however 480i and 576i looks to be missing one of the fields, means the signal looks like  progressive 240p with slight flickering and scanlines.
I checked whether the dongle might be an issue, but on my PC with the same modeline it produces proper 480i or 576i.

I messed around in 

`static void vc5_hdmi_set_timings(struct vc4_hdmi *vc4_hdmi,
				 struct drm_connector_state *state,
				 const struct drm_display_mode *mode)`

especiall with 

`	HDMI_WRITE(HDMI_VERTA0, verta);
	HDMI_WRITE(HDMI_VERTA1, verta);

	HDMI_WRITE(HDMI_VERTB0, vertb_even);
	HDMI_WRITE(HDMI_VERTB1, vertb);`

and looks like some of these registers do not have any effect, writing them to 0 changes absolutely nothing about the image. (if I remember correctly, HDMI_VERTB1) , I'm wondering whether it might be, that these are not backwards compatible with Pi4, because the code itself looks like was inherited from Pi4. 
	

### Steps to reproduce the behaviour

Connected to HDMI-A-2
I loaded a custom edid using cmdline.txt.
Set the resolution to the interlace mode the edid supports. Tested it with Raspberry OS and Batocera 40 beta build.

extended cmdline.txt: 

drm.edid_firmware=HDMI-A-2:edid-hdmi-2.bin

extended config.txt: 

hdmi_edid_file=1
hdmi_edid_filename:0=edid-hdmi-1.bin
hdmi_edid_filename:1=edid-hdmi-2.bin

### Device (s)

Raspberry Pi 5

### System

Raspberry Pi OS with desktop
Release date: July 4th 2024
System: 64-bit
Kernel version: 6.6
Debian version: 12 (bookworm)

### Additional context

I can upload my edid file somewhere if needed.

popcornmix, 2024-07-24:
> hdmi_edid_file=1
hdmi_edid_filename:0=edid-hdmi-1.bin
hdmi_edid_filename:1=edid-hdmi-2.bin

remove these - config.txt settings apply to the deprecated firmware driver.

Can you confirm if you are booting to console or desktop? Note that wayland doesn't support interlaced modes.
Through what means are you choosing the interlaced hdmi mode?

Can you post output of `raspinfo | pastebinit` so we can see what is going on.

6by9, 2024-07-24:
Link to https://forums.raspberrypi.com/viewtopic.php?p=2239265 as your original post.

DSI has never supported interlaced output. I don't think the MIPI specification even allows it.
Perhaps you meant DPI, which also hasn't officially supported interlaced output, but some have hacked it to do so.

Interlacing is configured in both the Pixel Valve (vc4_crtc.c) and HDMI (vc4_hdmi.c). The HDMI block does have an autodetection mode to nominally pick up the PV timings, but that has never been used (it is also present on Pi0-3).

I'll fire up our HDMI analyser to see if it will capture the two fields separately and confirm what it puts in each.

6by9, 2024-07-24:
OK, this does warrant more investigation.

On Pi4, selecting 1080i60 (`kmstest -r 74250000,1920/88/44/148/+,1080/4/10/31/+,i`) is interpreted by the analyser as interlaced 1080i60.
Pi5 with the same command is detected as progressive, 1920x540@60p.

6by9, 2024-07-25:
By generating a test pattern of alternating black and white lines, the analyser has confirmed that we are getting the odd lines in one field and the even lines in the other. So it is only the interlaced signalling that it is getting wrong.

Reverting "drm/vc4: crtc: Support odd horizontal timings on BCM2712" seems to make the analyser happy that it's interlaced.
That's largely a PV related change (only controls filtering in HDMI), so presumably it's related to how the signalling from PV to HDMI works in the hardware.
We can probably derive a hybrid where interlacing disables the odd horizontal timings bit, but that does mean the standard PAL 720x576i and NTSC 720x480i timings will be rejected (720/12/63/69 and 720/19/62/57).

(It's also confirmed that one of my monitors here doesn't handle interlaced properly as it is flicking between a black and white image in a way that is likely to produce an epileptic fit! It's a reasonably reliable make and claims 1080i60 in the EDID! My other monitor is quite happy and trying to show the alternating lines but gives some aliasing)

6by9, 2024-07-25:
PR #6282 seems to resolve the issue, at the expense of breaking odd horizontal timings for interlaced modes.

Having looked in more detail there is already special handling for 720x576i and 720x480i modes as they are double clocked modes, so they do work.

zoli0726, 2024-07-25:
Thank you for your help!
I can confirm, that it does work on my end as well!

closed 2024-08-20 by 6by9