raspberrypi/linux #6285: PR 6280 breaks SPI communication and wm5102 detection on RPi5

HiassofT, 2024-07-26:
### Describe the bug

With current rpi-update kernel the WM5102 soundchip on the Cirrus Logic Audio Card is no longer detected, dmesg shows `arizona spi0.0: Unknown device ID: 0` which is typical for a SPI comminication issue. I suspect other SPI devices might be affected, too.

Bisecting lead me to the two "Save bandwidth with the TMOD_TO/RO" commits 6014649 and cd9084c, with those two reverted the chip is detected fine.

Just reverting the second TMOD_RO commit does not help and reverting only the first TMOD_TO one isn't easy because of shared context.

Checking with my scope it looks like the initial address (RPi output) phase is terminated too early, instead of the expected 48 clock cycles (1 bit r/w, 31bit address, 16 bit padding) I only see 21 cycles. In the following 16 (RPi input) clock cycles the WM5102 doesn't yet drive MISO as it's still waiting for the missing address cycles and so the RPi reads a device id of 0 (instead of 0x5102).
[image]

With the two TMOD commits reverted the SPI communication works fine:
[image]

### Steps to reproduce the behaviour

Use a Cirrus Logic Audio Card on a RPi5, just enabling the `cirrus-wm5102` should also the SPI issue (detecting the chip ID by reading register 0 is the first thing the arizona-spi driver does).

### Device (s)

Raspberry Pi 5

### System

RPi OS bookworm, 64bit, kernel 
```
Linux raspberrypi 6.6.42-v8-16k+ #1787 SMP PREEMPT Thu Jul 25 17:45:51 BST 2024 aarch64 GNU/Linux
```

### Logs

https://paste.debian.net/hidden/4840050a/

### Additional context

ping @pellwell

pelwell, 2024-07-29:
#6286 should get it working again, but I don't have one of those devices (or many other SPI devices at all).

HiassofT, 2024-07-31:
Thanks a lot for the quick fix, testing looks good here so I'm happy to close the issue

closed 2024-07-31 by HiassofT